crankyoldgit/IRremoteESP8266 #1770: Another variant for LG swing command to add. (request)

crackn, 2022-03-05:
Hi, @crankyoldgit 

I would like to know if its possible to add another command to turn on and off the swing function on LG (older model)

this is the code that is working for turn on and turn off swing: { "Protocol":"LG","Bits": 28,"Data": 0x8810001 }

I noted that in LG_ir.h all the swing variants are 0x8813xxx and none of that works on this older model of LG.

thanks in adv.

crankyoldgit, 2022-03-05:
Can you please give us your brand/model numbers for your remote and A/C unit?

crackn, 2022-03-05:
Remote is 6711A20083V
Air is TS-H122ERM1

if I use IRsend with { "Protocol":"LG","Bits": 28,"Data": 0x8810001 } it works but I am trying to send commands with IRHVAC and SwingV option but I cant make it work.

crankyoldgit, 2022-03-05:
The brand is LG too?

crackn, 2022-03-05:
LG yes

crankyoldgit, 2022-03-05:
Sorry, another question. Is that the only swing command your remote produces?

crackn, 2022-03-05:
Yes, 0x8810001 for both swing On and Swing Off (toggle)

crankyoldgit, 2022-03-07:
@crackn Can you please download and test the following branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1770

You will need to set the `protocol` to `LG`, & the `model` to `LG6711A20083V` or `5` in order for it to work correctly. I'm not 100% sure I've got sending correct for it. :-/

Please also test the remote with IRrecvDumpV2 or V3, it should detect as:
```
Model: 5 (LG6711A20083V), Swing(V): Toggle
```
When it gets your swing toggle message code, otherwise it should be Model: 1 / GE6711AR2853M

crackn, 2022-03-07:
ok, I will test it and return to you asap. thanks

btw, I was doing some tests and noted that some comands are identified as diferent controllers. so there are some bugs in the LG detection code, i will do more tests to give you feedback about this ok.

crankyoldgit, 2022-03-07:
> btw, I was doing some tests and noted that some comands are identified as diferent controllers. so there are some bugs in the LG detection code, i will do more tests to give you feedback about this ok.

Model Detection is based on a per message basis. Some messages are unique to a model, so it's detected as the correct model. And some messages are common to multiple models, so it reports them as a less specific (base) model.
I.e. sometimes there is no way to tell models apart based on an individual single message.

crackn, 2022-03-07:
here are the results:

the swing function works great. 

*** LG SWING-V toggle command (remote LG6711A20083V) OK!
Protocol  : LG
Code      : 0x8810001 (28 Bits)
Mesg Desc.: Model: 5 (LG6711A20083V), Swing(V): Toggle
uint32_t address = 0x88;
uint32_t command = 0x1000;
uint64_t data = 0x8810001;

and below are some other functions not implemented.

*** LG SWING-H toggle command (remote LG6711A20083V)
    obs. remote have the button but my hardware isn't supported.
Protocol  : LG
Code      : 0x8813004 (28 Bits)
Mesg Desc.: Model: 1 (GE6711AR2853M)
uint32_t address = 0x88;
uint32_t command = 0x1300;
uint64_t data = 0x8813004;

*** LG JET-COOL ON command (remote LG6711A20083V)
    obs. there is NO OFF command, it just return to what was on remote memory.
Protocol  : LG
Code      : 0x8810089 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: On, Mode: 0 (Cool), Temp: 15C, Fan: 8 (UNKNOWN)
uint32_t address = 0x88;
uint32_t command = 0x1008;
uint64_t data = 0x8810089;

*** LG AUTO-CLEAN ON command (remote LG6711A20083V)
Protocol  : LG
Code      : 0x88C00B7 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: Off
uint32_t address = 0x88;
uint32_t command = 0xC00B;
uint64_t data = 0x88C00B7;

*** LG AUTO-CLEAN OFF command (remote LG6711A20083V)
Protocol  : LG
Code      : 0x88C00C8 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: Off
uint32_t address = 0x88;
uint32_t command = 0xC00C;
uint64_t data = 0x88C00C8;

*** LG PLASMA-ION ON command (remote LG6711A20083V)
Protocol  : LG
Code      : 0x88C000C (28 Bits)
Mesg Desc.: Model: 1 (GE6711AR2853M), Power: Off
uint32_t address = 0x88;
uint32_t command = 0xC000;
uint64_t data = 0x88C000C;

*** LG PLASMA-ION OFF command (remote LG6711A20083V)
Protocol  : LG
Code      : 0x88C0084 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: Off
uint32_t address = 0x88;
uint32_t command = 0xC008;
uint64_t data = 0x88C0084;

it would be great and easy to implement if all functions belonged to the same mode but I understand your point. 
if you need any other tests please let me know.

thanks.

crankyoldgit, 2022-03-08:
Please create a new issue for the extra features you're now suggesting.

> *** LG SWING-V toggle command (remote LG6711A20083V) OK!
> obs. one minor thing I noted is when I send the command from remote, AC only beep once and when I
> send the command by the lib, the AC beeps twice. I don't know why.

That is to be expected. When using the interface that you're using, it will try to set the A/C to **all** the desired settings each time to ensure the device is completely in the desired operational mode.
i.e. The mode/temp/fan settings **and** the swing settings. In order to do so, it must send a "normal" settings message, followed by the desired "swing" setting.

crackn, 2022-03-08:
Ok, no problem, I will do that. thanks for the fast solution.

PR #1771 (2022-03-09): LG: Add Swing Toggle support for Model `LG6711A20083V`
* Detect swingv toggle messages and set appropriate model.
* Allow sending of swingv toggle messages.
* Update supported models info
* Various collateral updates.
* Minor code cleanup.
* Supporting unit tests.

* Confirmed working on real device by requester.

Fixes #1770
commit 3a438198 (2022-03-09): LG: Add Swing Toggle support for Model `LG6711A20083V` (#1771)

* Detect swingv toggle messages and set appropriate model.
* Allow sending of swingv toggle messages.
* Update supported models info
* Various collateral updates.
* Minor code cleanup.
* Supporting unit tests.

* Confirmed working on real device by requester.

Fixes #1770
closed 2022-03-09 by crackn