crankyoldgit/IRremoteESP8266 #1775: generic IRsend() method, which just emits given HEX/Dec value thorught IR transmitter.

rohitsg, 2022-03-12:
There should be a **generic IRsend() method,** which just emits given HEX/Dec value through IR transmitter (LED). The receiver I have is Symphony Air cooler, and i want to emit D81 ( this is for ON/OFF) to on/off the cooler.
**How I can emit these HEX values (other hex value for turning on/off - anti-mosquitto, timer, swing, cool functionality etc) from esp8266 board.?**

NiKiZe, 2022-03-12:
There is no way to just emit a "HEX value" without knowing the protocol.
You need to know the frequency (which most of the time is standard at 38Khz, but not always)
Then you need to know several pulse width values, what is a 1 and what is a 0. After this you can convert the D81 value to a number of pulse widths that are then sent out with the given frequency.

So no, there can not be a "generic IRsend() method", it already is as generic as it can be.
If you had taken the time and read thru and fill in the issue template when you created this issue, that would have been clear.

Try to go back and read the FAQ,.
To support your device whatever it is, you need to provide the details that was asked for in the issue template. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/.github/ISSUE_TEMPLATE/problem-report.md

crankyoldgit, 2022-03-12:
@NiKiZe is correct.

@rohitsg Have you tried [`IRsend::sendSymphony()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a1f1d5a30660ab0061f64d559d4916d4e)?
Where/how are you getting your "hex" codes?
Have you tried [`IRrecvDumpV3`](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) in the examples to obtain your "hex" codes? It will tell you the code, the nr. of bits, and the protocol it uses.

For a "**generic IRsend() method**", you've got the following options:
* [`IRsend::sendRaw()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a2b9b84f828918f933bd1764d113b53f8) Example usage: In [`IRsendDemo.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino)
https://github.com/crankyoldgit/IRremoteESP8266/blob/8f7510986e81c49423f21189f97ad19f6f31f1ff/examples/IRsendDemo/IRsendDemo.ino#L39-L46
https://github.com/crankyoldgit/IRremoteESP8266/blob/8f7510986e81c49423f21189f97ad19f6f31f1ff/examples/IRsendDemo/IRsendDemo.ino#L69
* [`IRsend::sendGeneric()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a5215fd797dfd490816f31bb99b38c273) Example usage:
https://github.com/crankyoldgit/IRremoteESP8266/blob/8f7510986e81c49423f21189f97ad19f6f31f1ff/src/ir_Symphony.cpp#L52-L56

rohitsg, 2022-03-14:
@crankyoldgit 
- **Where/how are you getting your "hex" codes?** -> connected IR receiver to esp8266 and pressed cooler remote buttons, and get all values from serial output.

- **Have you tried [IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3)** -> Yes tried, as shown in below code.

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// An IR detector/demodulator is connected to GPIO pin 5 (D1 on a NodeMCU board).
// Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.

const uint16_t kRecvPin = 5;
unsigned long key_value = 0;
IRrecv irrecv(kRecvPin);

decode_results results;

// NOTE - rawData and data, these are received **by running IRrecvDumpV3.ino** code and pressed ON/OFF button
// of symphony air cooler
uint16_t rawData[215] = {1338, 368,  1314, 370,  468, 1214,  1286, 396,  1312, 370,  470, 1212,  468, 1212,  470, 1212,  444, 1238,  470, 1214,  468, 1214,  1284, 7494,  1336, 368,  1312, 370,  468, 1190,  1336, 370,  1312, 372,  468, 1212,  442, 1240,  470, 1212,  470, 1212,  470, 1212,  470, 1212,  1286, 7494,  1336, 370,  1312, 370,  472, 1210,  1314, 370,  1312, 370,  468, 1214,  444, 1240,  442, 1216,  492, 1214,  470, 1212,  468, 1190,  1336, 7490,  1312, 372,  1312, 368,  472, 1212,  1292, 390,  1314, 370,  468, 1214,  468, 1214,  468, 1212,  470, 1212,  470, 1212,  470, 1214,  1312, 7488,  1314, 370,  1312, 370,  470, 1212,  1312, 370,  1312, 370,  470, 1214,  442, 1238,  442, 1240,  470, 1212,  470, 1212,  470, 1214,  1312, 7490,  1314, 368,  1288, 394,  470, 1214,  1312, 370,  1312, 368,  444, 1240,  442, 1240,  470, 1214,  444, 1238,  468, 1216,  470, 1212,  1312, 7492,  1312, 370,  1286, 396,  470, 1214,  1314, 370,  1314, 368,  444, 1240,  468, 1214,  442, 1238,  470, 1212,  444, 1238,  464, 1218,  1286, 7492,  1336, 372,  1286, 396,  468, 1214,  1312, 370,  1314, 368,  472, 1212,  470, 1212,  468, 1214,  470, 1210,  470, 1212,  474, 1210,  1314, 7466,  1338, 368,  1314, 370,  472, 1212,  1314, 368,  1316, 366,  472, 1210,  470, 1212,  472, 1210,  474, 1210,  470, 1210,  470, 1212,  1316};  // SYMPHONY D81
uint64_t data = 0xD81;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
  irsend.begin();

  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
}

void loop() {
  delay(2000);
  irsend.sendRaw(rawData, 215, 38);
  
  if (irrecv.decode(&results)) {
   // **HERE IT SHOULD PRINT, D81 HEX VALUE, BUT GETTING DIFF VALUES ON EACH TIME, SERIAL OUTPUT SHOWN AT END**
    serialPrintUint64(results.value, HEX);
    Serial.println();
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}

```
/* IR Reciepver codes, Symphony air cooler - touch 32 ltr, all remote keycodes in HEX and decimal number
   D81- 3457 - ON OFF
  D82- 3458 - FAN
  D84 - 3460 - COOL
  D88 - 3464 - SWING
  D90 - 3472 - TIMER
  DC3 - 3523 - MOSQUITO
*/

// Serial OUTPUT
/*
79A116F1
8B425CCE
E3541989
B7DF3171
5C3E906B
CCEF91BF
F65E3E5D
F8B3A04C
CB7E0E8D
673C0613
 */

rohitsg, 2022-03-14:
**IRrecvDumpV3 - Serial monitor output,**

`Timestamp : 000060.494
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xD81 (12 Bits)
uint16_t rawData[119] = {1288, 396,  1286, 398,  416, 1242,  1310, 396,  1260, 422,  416, 1266,  418, 1264,  416, 1266,  416, 1266,  468, 1214,  416, 1244,  1308, 7490,  1284, 422,  1260, 422,  416, 1266,  1258, 424,  1286, 396,  442, 1240,  444, 1238,  444, 1240,  416, 1266,  416, 1266,  416, 1266,  1284, 7518,  1284, 398,  1258, 424,  442, 1238,  1288, 396,  1260, 422,  442, 1240,  416, 1266,  440, 1242,  444, 1238,  418, 1264,  416, 1272,  1254, 7542,  1258, 424,  1258, 422,  444, 1240,  1284, 398,  1286, 398,  442, 1240,  418, 1264,  418, 1264,  442, 1240,  418, 1264,  418, 1266,  1260, 7542,  1284, 400,  1284, 398,  416, 1266,  1258, 424,  1286, 396,  418, 1240,  466, 1240,  442, 1218,  466, 1240,  416, 1266,  442, 1242,  1258};  // SYMPHONY D81
uint64_t data = 0xD81;
// ON button

Timestamp : 000064.210
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xD82 (12 Bits)
uint16_t rawData[95] = {1258, 426,  1256, 426,  470, 1212,  1256, 426,  1314, 368,  416, 1266,  416, 1266,  412, 1268,  416, 1266,  414, 1268,  1284, 396,  416, 8386,  1256, 426,  1256, 426,  440, 1242,  1256, 426,  1256, 426,  440, 1242,  412, 1272,  438, 1242,  414, 1268,  416, 1268,  1254, 428,  412, 8390,  1284, 398,  1256, 426,  440, 1242,  1256, 428,  1312, 370,  442, 1238,  412, 1272,  442, 1240,  412, 1270,  412, 1270,  1256, 426,  414, 8388,  1258, 426,  1284, 396,  414, 1268,  1254, 430,  1258, 424,  412, 1268,  440, 1242,  412, 1268,  414, 1270,  414, 1268,  1258, 426,  414};  // SYMPHONY D82
uint64_t data = 0xD82;
// FAN button

Timestamp : 000066.819
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xD84 (12 Bits)
uint16_t rawData[95] = {1254, 428,  1254, 430,  406, 1272,  1254, 456,  1228, 428,  410, 1272,  412, 1272,  410, 1300,  412, 1244,  1256, 426,  412, 1270,  416, 8412,  1230, 428,  1254, 456,  358, 1322,  1230, 452,  1228, 454,  384, 1298,  358, 1300,  410, 1270,  410, 1274,  1226, 482,  358, 1298,  410, 8390,  1234, 450,  1224, 482,  386, 1274,  1250, 428,  1252, 456,  386, 1298,  358, 1298,  384, 1324,  384, 1274,  1250, 432,  410, 1324,  358, 8392,  1254, 452,  1200, 484,  382, 1300,  1228, 454,  1256, 426,  360, 1322,  386, 1270,  412, 1272,  406, 1300,  1228, 452,  382, 1276,  380};  // SYMPHONY D84
uint64_t data = 0xD84;
// Cool button

Timestamp : 000072.714
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xD88 (12 Bits)
uint16_t rawData[119] = {1314, 370,  1314, 346,  492, 1212,  1314, 344,  1312, 394,  472, 1210,  470, 1212,  468, 1190,  1310, 396,  468, 1190,  492, 1212,  468, 7956,  1310, 396,  1312, 370,  470, 1212,  1314, 368,  1314, 370,  470, 1212,  472, 1210,  470, 1214,  1314, 368,  470, 1212,  470, 1212,  472, 7952,  1310, 396,  1314, 368,  472, 1210,  1312, 370,  1286, 396,  470, 1212,  470, 1210,  470, 1214,  1316, 368,  468, 1190,  494, 1210,  470, 7954,  1338, 370,  1312, 370,  472, 1212,  1312, 370,  1312, 346,  492, 1212,  470, 1212,  470, 1190,  1334, 370,  468, 1212,  470, 1212,  472, 7954,  1338, 368,  1312, 370,  472, 1210,  1294, 366,  1334, 348,  492, 1212,  470, 1212,  470, 1214,  1314, 370,  466, 1214,  470, 1212,  472};  // SYMPHONY D88
uint64_t data = 0xD88;
//D88 Swing button

Timestamp : 000074.668
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xD90 (12 Bits)
uint16_t rawData[119] = {1318, 366,  1316, 344,  494, 1210,  1316, 366,  1316, 366,  470, 1210,  470, 1188,  1338, 342,  496, 1212,  444, 1238,  472, 1210,  472, 8310,  1338, 366,  1316, 366,  472, 1212,  1314, 368,  1316, 366,  472, 1210,  470, 1212,  1314, 368,  470, 1212,  472, 1186,  496, 1210,  472, 8308,  1336, 370,  1314, 344,  494, 1212,  1314, 344,  1338, 366,  470, 1212,  470, 1212,  1316, 368,  470, 1212,  470, 1212,  470, 1212,  472, 8306,  1338, 368,  1316, 368,  472, 1188,  1338, 368,  1316, 366,  472, 1212,  448, 1234,  1318, 366,  472, 1210,  444, 1238,  470, 1212,  472, 8308,  1340, 366,  1314, 370,  470, 1212,  1318, 366,  1316, 366,  472, 1210,  470, 1190,  1340, 366,  472, 1210,  472, 1210,  472, 1212,  472};  // SYMPHONY D90
uint64_t data = 0xD90;
// Timer button

Timestamp : 000075.975
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : SYMPHONY
Code      : 0xDC3 (12 Bits)
uint16_t rawData[119] = {1286, 398,  1256, 426,  414, 1268,  1260, 424,  1256, 426,  1260, 424,  416, 1266,  442, 1240,  414, 1268,  414, 1268,  1284, 398,  1258, 7186,  1258, 424,  1258, 424,  442, 1240,  1258, 424,  1258, 424,  1286, 398,  414, 1268,  414, 1270,  444, 1238,  416, 1268,  1258, 424,  1258, 7186,  1258, 424,  1258, 424,  416, 1266,  1260, 424,  1286, 396,  1260, 422,  416, 1268,  414, 1268,  416, 1266,  416, 1266,  1258, 424,  1258, 7164,  1284, 424,  1258, 424,  414, 1244,  1304, 400,  1258, 426,  1258, 424,  414, 1268,  440, 1240,  416, 1266,  442, 1240,  1260, 424,  1258, 7186,  1286, 398,  1258, 424,  416, 1266,  1258, 400,  1362, 322,  1280, 426,  414, 1266,  416, 1268,  414, 1266,  416, 1268,  1260, 424,  1258};  // SYMPHONY DC3
uint64_t data = 0xDC3;
// anti-mosquitto button`

crankyoldgit, 2022-03-14:
> ```
> irsend.sendRaw(rawData, 215, 38);
>   
>   if (irrecv.decode(&results)) {
>    // **HERE IT SHOULD PRINT, D81 HEX VALUE, BUT GETTING DIFF VALUES ON EACH TIME, SERIAL OUTPUT SHOWN AT END**
>     serialPrintUint64(results.value, HEX);
> ```

I'm not sure you know what you're doing.
a) Why are you trying to send an IR message while you are trying to receive/capture one? That is probably recipe for disaster.
b) You are not checking the protocol of the decode result, so you're probably receiving noise.
Either:
```c++
   if (irrecv.decode(&results)) {
    if (results.decode_type == decode_type_t::SYMPHONY) {  // Is the captured message of the protocol type I'm looking for.
      serialPrintUint64(results.value, HEX);
      Serial.println();
      irrecv.resume();  // Receive the next value
    }
  }
```
or reduce the "UNKNOWN" type messages
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#occasionally-im-getting-seemingly-random--unwanted-unknown-values-is-there-a-way-to-get-rid-of-them

> Protocol : SYMPHONY
Code : 0xD81 (12 Bits)
uint16_t rawData[119] = {1288, 396, 1286, 398, 416, 1242, 1310, 396, 1260, 422, 416, 1266, 418, 1264, 416, 1266, 416, 1266, 468, 1214, 416, 1244, 1308, 7490, 1284, 422, 1260, 422, 416, 1266, 1258, 424, 1286, 396, 442, 1240, 444, 1238, 444, 1240, 416, 1266, 416, 1266, 416, 1266, 1284, 7518, 1284, 398, 1258, 424, 442, 1238, 1288, 396, 1260, 422, 442, 1240, 416, 1266, 440, 1242, 444, 1238, 418, 1264, 416, 1272, 1254, 7542, 1258, 424, 1258, 422, 444, 1240, 1284, 398, 1286, 398, 442, 1240, 418, 1264, 418, 1264, 442, 1240, 418, 1264, 418, 1266, 1260, 7542, 1284, 400, 1284, 398, 416, 1266, 1258, 424, 1286, 396, 418, 1240, 466, 1240, 442, 1218, 466, 1240, 416, 1266, 442, 1242, 1258}; // SYMPHONY D81
uint64_t data = 0xD81;
// ON button

As for you captures with IRrecvDumpV3 etc, you can reproduce those signals via:
```c++
irsend.sendSymphony(0xD81);
```

rohitsg, 2022-03-14:
@crankyoldgit , irsend.sendSymphony(0xD81); works fine. Thanks for your support.

closed 2022-03-14 by crankyoldgit