crankyoldgit/IRremoteESP8266 #1779: Need help on how to set the AC model .

gladson1990, 2022-03-28:
Hi, this is just to seek help or some sort of an idea on how to set the AC model. I am kind of intermediate to Arduino.
The scenario is like I have a code where I set different ac models based on the user input. 
The Eg code which I have is, 
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Airwell.h>
#include <ir_Amcor.h>
#include <ir_Argo.h>
#include <ir_Carrier.h>
#include <ir_Coolix.h>
const uint16_t kIrLed = 25; // ESP8266 GPIO pin to use. Recommended: 4 (D2).
uint8_t cool_mode;
IRSamsungAc ac(kIrLed);
WiFiClientSecure wiFiClient;

void set_ac_model(String ac_models)
{

  if (ac_models == "Airwell")
  {
    
    IRAirwellAc ac(kIrLed);
    cool_mode = kAirwellCool;
  }
  else if (ac_models == "Amcor")
  {
    IRAmcorAc ac(kIrLed);
    cool_mode = kAmcorCool;
  }
  else if (ac_models == "Argo")
  {
    IRArgoAC ac(kIrLed);
    cool_mode = kArgoCool;
  }
  else if  (ac_models == "Carrier")
  {
    IRCarrierAc64 ac(kIrLed);
    cool_mode = kCarrierAc64Cool;
  }
  else if  (ac_models == "Coolix")
  {
    IRCoolixAC ac(kIrLed);
    Serial.print("Setting ac as ");
    Serial.println(ac_models);
    cool_mode = kCoolixCool;
  }}

void setup()
{
  Serial.begin(115200);

}

void loop(){
// Assume we get the ac model from the user via webserver.
Serial.println("*********");
Serial.println(ac.getMode());
ac_model = "Midea";
set_ac_model(ac_model);
ac.begin();
ac.setTemp(25);
ac.send();
Serial.println(ac.getMode());

delay(3000);
  ac_model = "Carrier";
    set_ac_model(ac_model);
ac.begin();
ac.setTemp(25);
ac.send();
Serial.println(ac.getMode());

delay(3000);

ac_model = "Mitsubishi";
    set_ac_model(ac_model);
ac.begin();
ac.setTemp(25);
ac.send();
Serial.println(ac.getMode());

delay(3000);

ac_model = "Argo";
    set_ac_model(ac_model);
ac.setTemp(25);
ac.begin();
ac.send();
Serial.println(ac.getMode());

delay(3000);

ac_model = "Haier";
    set_ac_model(ac_model);
ac.begin();
ac.setTemp(25);
ac.send();

delay(3000);
}

Based on this i have written it, but the problem i face is, even if I initiate a new object inside the set_ac_model function still I get the IRSamsungAc ac(kIrLed); object wherever I call the function.

is there a way to implement this kind of logic in my code ?
Kindly help. 
TIA.

crankyoldgit, 2022-03-28:
You have a fundamental problem/issue with your code. Namely variable [scoping](https://en.wikipedia.org/wiki/Scope_(computer_science)) [in C](https://www.google.com/search?q=variable+scope+in+c).
In short, for example, your `IRAirwellAc ac(kIrLed)` won't survive/exist out side of the surrounding `{` `}`'s. Only the only `ac` that will be accessible is the globally declared one. i.e. `IRSamsungAc ac(kIrLed)`.

You need some more programming knowledge and skill to understand why your program as written won't work. That teaching/knowledge is [beyond the scope](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me) of this project. You need to learn that yourself.

You might want to try using the [`IRac class`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html) instead.

crankyoldgit, 2022-04-04:
No response in a week, and it's really out of scope; So marking this closed.

closed 2022-04-04 by crankyoldgit

gladson1990, 2022-04-05:
Sorry.. forgot to close it. I had used [IRac class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html)] which seems to fix the problem or the solution  i was looking for. 
Thank you so much and appreciate your help and concern.