crankyoldgit/IRremoteESP8266 #1782: SOME MitsubishiHeavy AC ARE NOT WORKING

abdulhannan97, 2022-03-31:
HI 
I am Abdul i am trying to control Mitsubishi Heavy ac and THIS  CODE I AM USEING

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>

const uint16_t kIrLed = 4;  // The ESP GPIO pin to use that controls the IR LED.
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

void setup() {
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send.
  // See state_t, opmode_t, fanspeed_t, swingv_t, & swingh_t in IRsend.h for
  // all the various options.
  ac.next.protocol = decode_type_t::MITSUBISHI_HEAVY_152;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
  ac.next.degrees = 25;  // 25 degrees.
  ac.next.fanspeed = stdAc::fanspeed_t::kMedium;  // Start the fan at medium.
  ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
  ac.next.beep = false;  // Turn off any beep from the A/C if we can.
  ac.next.econo = false;  // Turn off any economy modes if we can.
  ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  ac.next.turbo = false;  // Don't use any turbo/powerful/etc modes.
  ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
  ac.next.sleep = -1;  // Don't set any sleep time or modes.
  ac.next.clean = false;  // Turn off any Cleaning options if we can.
  ac.next.clock = -1;  // Don't set any current time if we can avoid it.
  ac.next.power = false;  // Initially start with the unit off.

  Serial.println("Try to turn on & off every supported A/C type ...");
}

void loop() {
  // For every protocol the library has ...
  for (int i = 1; i < kLastDecodeType; i++) {
    decode_type_t protocol = (decode_type_t)i;
    // If the protocol is supported by the IRac class ...
    if (ac.isProtocolSupported(protocol)) {
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol) + " is supported.");
      ac.next.protocol = protocol;  // Change the protocol used.
      ac.next.power = true;  // We want to turn on the A/C unit.
      Serial.println("Sending a message to turn ON the A/C unit.");
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(5000);  // Wait 5 seconds.
      ac.next.power = false;  // Now we want to turn the A/C off.
      Serial.println("Send a message to turn OFF the A/C unit.");
      ac.sendAc();  // Send the message.
      delay(1000);  // Wait 1 second.
    }
  }
  Serial.println("Starting from the begining again ...");
}

i am tried Mitsubishi_Heavy_88 and Mitsubishi_Heavy_152 protocol BUT ITS NOT WORKING

I DONT HAVE REMOTE NUMBER OR AC MODEL NUMBER BUT IN HAVE IR REMOTE  RAWDATA

RAWDATA FOR ON
Protocol  : UNKNOWN
Code      : 0xEAF2F7E (67 Bits)
uint16_t rawData[133] = {6010, 7380,  542, 1438,  544, 3422,  544, 1438,  542, 3422,  544, 1438,  544, 1438,  544, 1438,  542, 1440,  544, 1440,  542, 1442,  540, 1440,  544, 1438,  544, 1440,  544, 3420,  544, 3424,  542, 1438,  546, 3420,  544, 1438,  544, 1438,  544, 3422,  546, 1438,  542, 3422,  544, 1440,  544, 3422,  544, 1440,  542, 1438,  544, 1440,  542, 1440,  544, 3420,  544, 1438,  544, 3422,  542, 1440,  544, 3422,  544, 1440,  542, 3422,  544, 1438,  544, 3424,  542, 3422,  546, 3422,  542, 3422,  544, 3422,  544, 3420,  544, 3422,  546, 3418,  544, 3424,  544, 1438,  544, 1438,  544, 3422,  544, 1438,  544, 3422,  546, 3422,  542, 1440,  542, 3424,  542, 1438,  546, 3422,  544, 1434,  548, 3420,  544, 3422,  544, 3422,  546, 3422,  542, 1440,  544, 3420,  544, 1436,  546, 3422,  544, 7384,  544};  // UNKNOWN EAF2F7E

RAWDATA FOR OFF

Protocol  : UNKNOWN
Code      : 0x127C1D3A (67 Bits)
uint16_t rawData[133] = {6010, 7380,  542, 1438,  544, 3422,  544, 1440,  544, 3420,  544, 1440,  542, 1438,  544, 1438,  546, 1436,  544, 1440,  542, 1440,  542, 1440,  544, 1440,  544, 1436,  546, 3422,  542, 3422,  542, 1440,  544, 3422,  544, 1438,  542, 1440,  542, 3424,  544, 1438,  544, 3422,  544, 1438,  546, 1436,  546, 1438,  544, 1438,  544, 1438,  544, 1440,  542, 3422,  544, 1440,  544, 3422,  542, 1438,  546, 3422,  544, 1440,  544, 3420,  544, 1436,  546, 3420,  546, 3422,  544, 3420,  546, 3420,  542, 3422,  542, 3422,  546, 3418,  546, 3422,  542, 3422,  544, 1436,  544, 1440,  546, 3418,  546, 1466,  516, 3424,  542, 3422,  544, 1438,  546, 3420,  544, 1438,  544, 3422,  544, 3420,  548, 3418,  546, 3418,  546, 3422,  544, 3422,  542, 1438,  546, 3420,  544, 1438,  544, 3420,  546, 7388,  542};  // UNKNOWN 127C1D3A

I LIKE TO CONTROL EVERYTHING NOT ONLY ON AND OFF

CAN U HELP ME ON THIS
THANK YOU

crankyoldgit, 2022-04-04:
_Sorry, Seems my update to this has gone missing. I replied a few days ago.
I'll retype it as best as I can remember._

> i am tried Mitsubishi_Heavy_88 and Mitsubishi_Heavy_152 protocol BUT ITS NOT WORKING

It looks like your device isn't (yet) supported. If it was supported, the IRrecvDumpV2 & V3 programs would detect it.

This is an [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) item, namely [this](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_) & [this](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Can_you_reverse_engineer_my_AirCon_protocol_for_me).

Also, you need to check out:
[Adding support for a new IR protocol](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol) & [Adding-support-for-a-new-AC-protocol](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol)

> I DONT HAVE REMOTE NUMBER OR AC MODEL NUMBER BUT IN HAVE IR REMOTE RAWDATA

This seems highly unusual. Most A/Cs have a brand/model number on them somewhere. e.g. Where the serial number is.

Please check again.

crankyoldgit, 2022-04-14:
@abdulhannan97 Ping. Any more info or updates?

crankyoldgit, 2022-05-06:
@abdulhannan97 Ping. Any more info or updates? It's been a month since we last heard from you. If we don't get a response soon, we will consider it stale and mark it closed.

crankyoldgit, 2022-05-16:
Marking this stale after over a month. Closing.

closed 2022-05-16 by crankyoldgit