crankyoldgit/IRremoteESP8266 #1787: Bosch CL3000i-Set 26 E       Remote: RG10A(G2S)BGEF

NicoThien, 2022-04-15:
I have an air conditioner from Bosch (CL3000i-Set 26 E) at home. On the back of the remote control is the designation RG10A(G2S)BGEF.
[image]

I use Tasmota-IR 11.0.0.4 (full) with IRremoteESP8266 v2.8.2

Different protocols were detected: Coolix and Coolix48
Unfortunately, not all remote control commands match the protocol (Coolix). I can set 16°C with the remote control. Not with the library. The fan speeds are also different.

There is raw data that may not quite fit the Coolix protocol. When temperature, mode or/and fan changes, the IRrecvDumpV2 tool determines 299 characters. This results in 18 bytes. In the first 6 bytes, every second is the inverted value of the previous one. The second 6 bytes are a repeat of the first 6. The last 6 bytes are not inverted. The last of these 6 bytes is the checksum (from bytes 13-17).

Timer settings are recognized as Coolix48. Only bytes 1 and 3 (7 & 9) are inverted. Here, too, 299 characters (18 bytes) are recognized.

Only 199 characters (2x 6 bytes) are transferred for the SwingV, Turbo, LED/Beep and Clean functions.

[Table](https://docs.google.com/spreadsheets/d/1Q77OnqRaLiAuWTHiTXmYRCSGOEnhUH7jF0tvh3HcW2A/edit?usp=sharing) still incomplete

crankyoldgit, 2022-04-15:
Can you please include all of the text from some of the captures of the remote using IRrecvDumpV2 or V3?
Especially for messages that decode incorrectly (and a few for where/when it does).

You might want to look at #1694 (#1697) as well. (FYI/paging @jimmys01)

It looks like we will have to work out some way of telling these Bosch devices from the Coolix ones.

See also: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

NicoThien, 2022-04-15:
Mode: Cool; Fan: 100% ; Temp: 16°C
```
Protocol  : COOLIX
Code      : 0xB23F00 (24 Bits)
Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 1 (Max), Temp: 17C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[299] = {4380, 4400,  528, 1646,  504, 570,  504, 1646,  504, 1646,  504, 572,  502, 570,  504, 1646,  504, 570,  504, 572,  502, 1646,  504, 570,  502, 570,  502, 1648,  502, 1646,  502, 570,  502, 1646,  504, 572,  502, 572,  502, 1644,  504, 1646,  504, 1646,  504, 1646,  502, 1648,  500, 1646,  504, 1646,  504, 1646,  504, 572,  502, 570,  504, 570,  504, 570,  504, 570,  504, 570,  506, 570,  502, 572,  502, 570,  502, 572,  502, 572,  502, 572,  502, 572,  502, 572,  500, 1648,  502, 1644,  502, 1646,  504, 1646,  502, 1646,  504, 1646,  504, 1644,  504, 1646,  504, 5234,  4360, 4422,  504, 1646,  502, 596,  478, 1670,  478, 1646,  504, 570,  504, 572,  500, 1646,  502, 572,  502, 572,  502, 1644,  506, 570,  502, 570,  504, 1644,  506, 1644,  502, 574,  502, 1644,  504, 570,  504, 570,  504, 1644,  504, 1646,  504, 1644,  506, 1644,  504, 1646,  504, 1646,  504, 1644,  504, 1646,  502, 570,  504, 570,  504, 570,  504, 570,  502, 570,  504, 570,  502, 572,  502, 570,  504, 570,  504, 570,  504, 570,  502, 572,  502, 570,  506, 570,  504, 1646,  502, 1646,  504, 1646,  504, 1646,  504, 1646,  502, 1644,  504, 1644,  504, 1646,  502, 5236,  4360, 4424,  504, 1646,  504, 1646,  502, 572,  504, 1644,  504, 570,  504, 1646,  504, 570,  502, 1644,  504, 570,  504, 1644,  506, 1646,  502, 572,  502, 572,  502, 1646,  504, 570,  504, 570,  504, 570,  502, 572,  504, 570,  504, 570,  504, 570,  502, 572,  502, 570,  504, 570,  502, 570,  504, 572,  502, 572,  502, 1646,  504, 570,  504, 570,  504, 570,  502, 574,  502, 572,  502, 572,  502, 572,  502, 572,  502, 572,  502, 570,  504, 572,  502, 572,  502, 572,  502, 1646,  504, 572,  502, 570,  502, 1646,  504, 572,  504, 570,  504, 1644,  504};  // COOLIX B23F00
uint64_t data = 0xB23F00;
```

Mode: Cool; Fan: 20%; Temp: 18°C
```
Protocol  : COOLIX
Code      : 0xB2FF10 (24 Bits)
Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 7 (Fixed), Temp: 18C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[299] = {4354, 4398,  530, 1648,  502, 598,  476, 1646,  502, 1646,  502, 572,  502, 574,  500, 1646,  502, 572,  502, 574,  500, 1648,  502, 572,  502, 572,  502, 1646,  502, 1646,  504, 574,  500, 1646,  504, 1646,  502, 1672,  476, 1648,  500, 1648,  502, 1648,  502, 1646,  502, 1646,  502, 1646,  504, 572,  502, 572,  502, 572,  502, 598,  476, 572,  502, 572,  502, 598,  476, 574,  500, 574,  502, 572,  502, 572,  502, 1646,  502, 570,  504, 570,  502, 572,  502, 574,  500, 1648,  502, 1646,  502, 1646,  502, 572,  502, 1646,  502, 1648,  502, 1648,  500, 1646,  504, 5236,  4360, 4424,  504, 1672,  476, 572,  502, 1644,  504, 1646,  504, 572,  502, 570,  502, 1648,  502, 572,  502, 574,  500, 1648,  502, 598,  476, 574,  500, 1648,  502, 1644,  504, 572,  502, 1648,  500, 1646,  502, 1646,  502, 1646,  504, 1672,  476, 1646,  502, 1646,  504, 1646,  502, 1648,  502, 598,  476, 572,  500, 572,  502, 572,  502, 574,  500, 598,  476, 574,  500, 572,  502, 574,  500, 572,  502, 572,  502, 1648,  500, 572,  502, 572,  502, 570,  502, 572,  502, 1646,  502, 1646,  504, 1648,  500, 572,  502, 1646,  502, 1646,  502, 1648,  502, 1646,  502, 5216,  4380, 4424,  504, 1646,  502, 1622,  526, 572,  500, 1648,  502, 574,  500, 1646,  502, 572,  502, 1646,  504, 570,  502, 574,  502, 572,  502, 1648,  502, 574,  500, 1648,  502, 574,  502, 572,  500, 572,  502, 572,  500, 598,  476, 572,  502, 596,  476, 574,  502, 572,  502, 572,  502, 596,  476, 574,  502, 570,  502, 572,  502, 572,  502, 598,  476, 572,  502, 572,  502, 570,  504, 572,  502, 572,  502, 598,  476, 572,  500, 574,  502, 572,  502, 572,  502, 1648,  502, 1646,  502, 1648,  500, 572,  502, 1646,  504, 572,  502, 572,  502, 1648,  502};  // COOLIX B2FF10
uint64_t data = 0xB2FF10;
```

Mode: Cool; Fan: 40%; Temp: 18°C
```
Protocol  : COOLIX
Code      : 0xB29F10 (24 Bits)
Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 4 (Min), Temp: 18C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[299] = {4352, 4424,  504, 1646,  502, 572,  502, 1644,  504, 1648,  502, 570,  502, 572,  504, 1646,  502, 572,  502, 572,  502, 1648,  502, 570,  504, 572,  502, 1648,  502, 1646,  502, 572,  500, 1648,  502, 1646,  502, 574,  500, 598,  478, 1646,  502, 1648,  502, 1648,  502, 1646,  504, 1646,  502, 572,  502, 1646,  504, 1644,  504, 572,  502, 570,  502, 572,  502, 570,  502, 572,  504, 572,  500, 572,  502, 598,  476, 1646,  502, 574,  500, 572,  502, 572,  502, 596,  478, 1646,  502, 1646,  502, 1646,  502, 574,  500, 1646,  504, 1646,  502, 1646,  504, 1646,  502, 5236,  4360, 4422,  504, 1646,  502, 572,  504, 1646,  502, 1646,  504, 572,  502, 572,  502, 1646,  502, 572,  502, 572,  502, 1646,  502, 572,  502, 572,  502, 1648,  502, 1646,  502, 572,  502, 1646,  504, 1646,  502, 572,  502, 572,  502, 1646,  502, 1646,  502, 1648,  502, 1646,  504, 1646,  502, 572,  502, 1646,  504, 1646,  502, 572,  502, 570,  504, 570,  502, 572,  502, 570,  504, 570,  504, 570,  502, 572,  502, 1646,  504, 572,  502, 572,  502, 572,  502, 570,  502, 1648,  502, 1646,  502, 1646,  504, 572,  502, 1648,  502, 1646,  504, 1646,  502, 1646,  502, 5212,  4384, 4424,  504, 1646,  504, 1646,  502, 572,  502, 1646,  502, 570,  504, 1648,  502, 572,  502, 1646,  504, 570,  504, 596,  478, 1646,  504, 570,  502, 1646,  502, 572,  504, 572,   [... cut ...] 04, 1646,  502, 572,  504, 570,  502, 572,  502, 1644,  504, 1646,  504, 570,  504, 1644,  504, 1646,  504, 1646,  502, 1646,  504, 572,  502, 1646,  504, 570,  502, 1646,  502, 570,  504, 572,  502, 570,  504, 570,  504, 1646,  502, 570,  504, 1644,  504, 570,  504, 570,  504, 570,  502, 1646,  504, 572,  502, 1646,  502, 572,  502, 1644,  504, 572,  502, 1646,  504, 1646,  502, 572,  502, 1644,  506, 570,  504, 1646,  504, 572,  502, 1646,  502};  // COOLIX B9F52A
uint64_t data = 0xB9F52A;
```

NicoThien, 2022-04-15:
Many thanks for the quick response. If my [spreadsheet](https://docs.google.com/spreadsheets/d/1Q77OnqRaLiAuWTHiTXmYRCSGOEnhUH7jF0tvh3HcW2A/edit?usp=sharing) needs to be formatted differently, I can change that.

crankyoldgit, 2022-05-16:
Yes, it does. I've left comments on the spreadsheet. Sorry, I should have gotten back to you sooner.

crankyoldgit, 2022-06-04:
Chasing up on this.

NicoThien, 2022-06-07:
I changed the table.

crankyoldgit, 2022-06-10:
@NicoThien I've created a [bosch](https://github.com/crankyoldgit/IRremoteESP8266/tree/bosch) branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/bosch

It should enable _basic_ sending and decoding of the 144 bit protocol you've documented. I'll leave the 48 bit (short) messages till later for now.

This 18byte/144bit structure/message includes the inverted bytes, and the duplicated sections. i.e. It's not compacted like the `COOLIX` protocol.

It currently doesn't do any of the integrity checks yet either. e.g. Inverted byes, 2nd section the same as the first, and a checksum in the 3rd section. I'll add those later.

Can you please download and try it out, and let me know how it goes?

NicoThien, 2022-06-13:
The commands of the original remote control are recognized as BOSCH144 protocol. But the remote ESP8266 does not transmit. The IR LED does not turn on.
The curly brackets of the for loop in line 45 of "ir-Bosch.cpp" are missing. But that doesn't seem to be the only problem.

crankyoldgit, 2022-06-13:
> The commands of the original remote control are recognized as BOSCH144 protocol.

Excellent!

> But the remote ESP8266 does not transmit. The IR LED does not turn on.

It should work/send something. It does in my Unit Tests. What code & circuit are you using? Is your setup transmitting other protocols successfully?

> The curly brackets of the for loop in line 45 of "ir-Bosch.cpp" are missing. But that doesn't seem to be the only problem.

That code (`for` loop) is correct syntax. There is only one _single_ statement inside the `for` loop. When that is the case, you don't need to specify a [`statement block`](https://www.learncpp.com/cpp-tutorial/compound-statements-blocks/) using `{}`'s.

i.e.
```c++
    for (uint16_t offset = 0; offset < nbytes; offset += kSectionByteSize)
      // Section Header + Data + Footer
      sendGeneric(kBoschHdrMark, kBoschHdrSpace,
                  kBoschBitMark, kBoschOneSpace,
                  kBoschBitMark, kBoschZeroSpace,
                  kBoschBitMark, kBoschFooterSpace,
                  data + offset, kSectionByteSize,
                  kBoschFreq, true, 0, kDutyDefault);
    space(kDefaultMessageGap);  // Complete guess

```
is syntactically _**exactly**_ the same as:
 ```c++
    for (uint16_t offset = 0; offset < nbytes; offset += kSectionByteSize) {
      // Section Header + Data + Footer
      sendGeneric(kBoschHdrMark, kBoschHdrSpace,
                  kBoschBitMark, kBoschOneSpace,
                  kBoschBitMark, kBoschZeroSpace,
                  kBoschBitMark, kBoschFooterSpace,
                  data + offset, kSectionByteSize,
                  kBoschFreq, true, 0, kDutyDefault);
    }
    space(kDefaultMessageGap);  // Complete guess
```

crankyoldgit, 2022-06-13:
I think there is "more to it" if you want to get it working in Tasmota. Not sure. You may also not be using the branch of the library required. i.e. You may have a Tasmota problem, not _this_ library per se.

Please use a simple example (appropriately modified) such as: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino

NicoThien, 2022-06-27:
I want to use commands like `IRhvac {"Vendor":"Bosch144","Power":"on","Temp":23,"Mode":"Cool","FanSpeed":"low"}` in Tasmota. I have looked at many files and have now created my own (copy/paste and a few thoughts of my own ;-). It's not all complete yet. But the basic things work. My programming skills are limited. I'm a little unsure if I can help with that.
What do I have to do so that someone can see my changes?
changed:
- IRac.h
- IRac.cpp
- ir_Bosch.cpp

new:
- ir_Bosch.h

NicoThien, 2022-08-11:
@crankyoldgit many, many thanks for your support

closed 2022-08-11 by NicoThien