crankyoldgit/IRremoteESP8266 #1793: sendRaw not working

karimk123, 2022-04-28:
### Version/revision of the library used
  v2.8.1

### Describe the bug
i followed the tutorial from https://iotcircuithub.com/esp8266-programming-arduino/
and i instead replaced the led in the tutorial with an IR led and i also used this library to send a raw IR signal instead of just turning on the LED. The IR emitter is connected to GPIO 0.
But it's not working, its not even emitting anything because i even tried replacing it with an led and its still not turning on with irsend.sendRaw

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable in the report](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._
Snippet of the code i used:
```ino 
void ac_on() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  const short unsigned int irSignal[] = {4352, 4396, 512, 1628, 512, 560, 512, 1628, 568, 1576, 512, 556, 512, 560, 512, 1628, 512, 560, 512, 560, 508, 1628, 512, 560, 512, 560, 568, 1572, 568, 1572, 516, 556, 512, 1628, 512, 1632, 512, 556, 512, 1632, 512, 1628, 512, 1628, 572, 1572, 512, 1628, 512, 1628, 512, 560, 512, 1632, 512, 556, 512, 560, 512, 560, 564, 504, 512, 556, 512, 560, 512, 560, 512, 1628, 512, 560, 568, 1572, 512, 560, 512, 556, 512, 560, 512, 556, 512, 1632, 568, 500, 568, 1576, 512, 556, 512, 1632, 512, 1628, 512, 1628, 512, 1632, 512, 5208, 4348, 4396, 512, 1632, 512, 556, 512, 1632, 512, 1628, 512, 560, 512, 556, 572, 1572, 508, 560, 512, 560, 512, 1628, 512, 560, 512, 556, 568, 1576, 512, 1628, 568, 504, 508, 1632, 512, 1628, 512, 560, 512, 1628, 512, 1632, 512, 1628, 568, 1576, 512, 1628, 512, 1632, 564, 504, 512, 1628, 512, 560, 512, 560, 512, 556, 512, 560, 512, 556, 572, 500, 512, 560, 512, 1628, 512, 556, 516, 1628, 512, 560, 508, 560, 512, 560, 508, 560, 568, 1576, 564, 504, 512, 1632, 508, 560, 512, 1632, 508, 1632, 512, 1628, 512, 1632, 512};
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.
  Serial.println("on");
}
IRsend irsend(3);

void ac_off() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  const short unsigned int irSignal[] = {4408, 4340, 512, 1632, 564, 504, 568, 1576, 508, 1632, 512, 560, 508, 560, 512, 1628, 568, 504, 512, 556, 512, 1632, 512, 560, 508, 560, 512, 1628, 512, 1632, 512, 560, 508, 1632, 512, 556, 512, 1632, 564, 1576, 512, 1632, 564, 1576, 564, 504, 512, 1632, 512, 1628, 512, 1632, 564, 504, 512, 560, 512, 560, 508, 560, 512, 1628, 568, 504, 512, 560, 508, 1632, 568, 1572, 512, 1632, 508, 560, 568, 504, 564, 504, 512, 560, 512, 560, 508, 560, 512, 560, 568, 500, 512, 1632, 564, 1576, 564, 1576, 512, 1632, 508, 1632, 508, 5212, 4348, 4396, 512, 1632, 512, 556, 512, 1632, 564, 1576, 568, 504, 512, 556, 512, 1632, 508, 560, 512, 560, 512, 1628, 512, 560, 512, 556, 568, 1576, 508, 1632, 512, 560, 508, 1632, 564, 508, 564, 1576, 512, 1628, 568, 1576, 508, 1632, 512, 560, 564, 1576, 512, 1632, 508, 1632, 508, 560, 512, 560, 512, 556, 512, 560, 568, 1572, 512, 560, 512, 560, 568, 1572, 508, 1632, 568, 1576, 508, 560, 568, 504, 512, 556, 568, 504, 512, 556, 512, 560, 512, 560, 564, 504, 568, 1576, 568, 1572, 512, 1628, 568, 1572, 568, 1576, 568};
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.
  Serial.println("off");
}
```

#### Expected behaviour
  the expacted behaviour is to send an IR signal from GPIO 0 when its called.

### What brand/model IR demodulator are you using?
5mm IR led

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._
[image]

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2022-04-29:
See the following for a _better_ circuit for the ESP-01: https://github.com/crankyoldgit/IRremoteESP8266/wiki/ESP01-Send-&-Receive-Circuit#sender
Your circuit design for driving the IR LED only from a GPIO pin will only draw a tiny bit of current, as the GPIOs are severely limited in how much current they can supply. However, it should be enough for you to see/power a small red LED to indicate that your code is working.

From your code above:

> ```c++
> IRsend irsend(3);
> ```

You're not sending on `GPIO 0`, that would be `IRsend irsend(0);`
Given your provided circuit diagram and code, you'd be sending on `GPIO 3` a.k.a. `RX`. Which you are currently also using for Serial comms. e.g. `Serial.println("on");` .. You can't use it for both simultaneously.
See ["General: point 2"](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#general) in the Troubleshooting guide.
```
Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
```

See also our [`IRsendDemo.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) example code:
https://github.com/crankyoldgit/IRremoteESP8266/blob/8b3539667d1689d3a85d1c3841c947f37df5a76d/examples/IRsendDemo/IRsendDemo.ino#L23-L28

In short, you've got a coding configuration & circuit problem. Not a bug or problem with the library.

karimk123, 2022-04-29:
Hello, thanks for your response. I realized afterwards that the problem was that i had forgotten to put the pinMode line for the IR led. I apologize for this. And yes the snippet i provided had 3 instead of 0 because i was debugging and forgot to change it.

closed 2022-04-29 by karimk123

crankyoldgit, 2022-04-29:
> I realized afterwards that the problem was that i had forgotten to put the pinMode line for the IR led.

Yes. That is handled by [`IRsend::begin()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a386f026bf739b0718efde4cffa6ce129)

That's one of the reasons we ask for the complete code. ;-)

karimk123, 2022-10-27:
Hello, a few months later i came back to this. I tried the better circuit for the IR blaster that you provided but the IR signal still isn't strong enough to reach my AC for some reason. 
i used this to convert the TX pin to actual usable GPIO 3:
`pinMode(IR_LED_PIN, FUNCTION_3);
 pinMode(IR_LED_PIN, OUTPUT);`

Then just the normal `irsend.sendRaw` to send the signal. It actually sends the IR signal but its not reaching the AC which is like 2 meters from the LED. Am i missing something here?

crankyoldgit, 2022-10-28:
Have you disabled/closed the `Serial` interface?

You're probably not supplying enough current to the IR Led. If you're using the 3V3 line from the Arduino, the voltage regulator will unlikely be unable to prove enough current.

Have you gone through ALL the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal)? Especially the [sending/transmission](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal) part?

Try building the circuit & software on a more featured ESP8266 board (e.g. Wemos D1 Mini, Nodemcu etc) to work out where your problem is. i.e. Is your code the issue, or is it your circuit design, or how you're using the ESP-01 module.

karimk123, 2022-10-28:
How exactly do I "disable" the Serial interface? I thought not having the `Serial.begin` function was enough to disable it.
And I'm using the 5V from the arduino so i think it should be enough for the IR led right ? 
I can post the full code if that would help.

crankyoldgit, 2022-10-29:
You're missing:
In `setup()`:
```c++
  irsend.begin();  // This will *also* handle setting the GPIO as output etc.
```

in your headers:
```c++
#include <IRremoteESP8266.h>
```

You're `loop()` routine is only waiting a second between each message. That may not be very long for the device to noticably react. Try 10 seconds i.e. `delay(10000);`

How about some pics & a diagram of your circuit setup, and what components you're using.

karimk123, 2022-10-29:
So I made the changes you said, and the IR led is working (I know because I tried with a normal LED and i also checked with a phone camera) but its still not reaching the AC.
I'm using this circuit but with the IR led and the transistor from the second image:
[image]

[image]

I'm also using 5V from arduino instead of 3.3V and i removed the resistor on the IR led.

NiKiZe, 2022-10-29:
5V is required to have any kind of range.
Try to question your assumptions.
Is the LED the correct wavelength?

That it lights is a good step, but there is many other ways that IR needs to work properly.

karimk123, 2022-10-29:
> 5V is required to have any kind of range.
> 
> Try to question your assumptions.
> 
> Is the LED the correct wavelength?
> 
> 
> 
> That it lights is a good step, but there is many other ways that IR needs to work properly.

I mean I think it should be able to reach an AC that is 2 meters away right ? Its the same IR led in every generic remote, im not sure how to really check the wavelength.

karimk123, 2022-11-01:
After a bit more testing, turns out the problem is actually the IR led code that is hard coded. 
This is what i'm using to determine the raw code for the IR signal: https://gist.github.com/probonopd/5793692#file-sendandreceive-ino

It works with every remote, except, my AC remote changes the code every time even if its the same button.
It only works when the temperature is set to 23 degrees which is really odd. I'm still trying to figure out why that's happening.
I still don't fully understand how the NEC protocol and the IR hex codes work so please let me know if i'm missing something here.

NiKiZe, 2022-11-01:
Great that you have narrowed the issue.
Have you tried to grab the message sent by your original remote using dumpv2+ examples.

Using sendRaw to an AC which uses states is often a bad idea to start with. It is often much more complex.

karimk123, 2022-11-01:
I just tried with the ReceiveDump in the IRremote library (im not sure if that's dumpv2 or not) and i got a bigger raw result with 199 numbers but still didnt work on the AC.

karimk123, 2022-11-01:
I used the Example from the library because the normal IRecvDumpv2 didn't work with the latest version of IRremote and kept getting things like "NEC" and "SONY" undefined/

crankyoldgit, 2022-11-01:
The IRremote library examples were limited in capture size. Our versions are not, but they are required to run on a ESP chip.
If you want to use it with our library, I **strongly** recommend using _our_ IRrecvDumpV2 or V3 examples to do the capture.

karimk123, 2022-11-01:
> The IRremote library examples were limited in capture size. Our versions are not, but they are required to run on a ESP chip. If you want to use it with our library, I **strongly** recommend using _our_ IRrecvDumpV2 or V3 examples to do the capture.

Sure i'll try capturing with the ESP and see how it goes.