crankyoldgit/IRremoteESP8266 #1794: Support for Ogeneral AC.

gladson1990, 2022-04-29:
### Version/revision of the library used
Library   : v2.7.19

### Describe the bug
I used to run the CommonAcControl.ino file to figure out whether the AC is getting supported or not.
The AC is from the brand Ogeneral 

#### Expected behaviour
the Ac should turn on and off based on the command send 

#### Output of raw data from [IRrecvDumpV2.ino]IRrecvDump is now running and waiting for IR input on Pin 14
Timestamp : 000008.006
Library   : v2.7.19

Protocol  : FUJITSU_AC
Code      : 0x146300101002FD (56 Bits)
Mesg Desc.: Model: 1 (ARRAH2E), Id: 0, Power: Off, Mode: 0 (Auto), Temp: 16C, Fan: 0 (Auto), Clean: Off, Filter: Off, Swing: 0 (Off), Command: N/A, Timer: Off
uint16_t rawData[115] = {3230, 1660,  450, 398,  540, 370,  478, 1146,  450, 368,  452, 1174,  450, 368,  452, 368,  478, 346,  478, 1146,  452, 1178,  450, 368,  476, 342,  478, 340,  478, 1146,  450, 1178,  452, 374,  476, 338,  478, 338,  478, 340,  478, 414,  478, 340,  478, 340,  478, 340,  480, 346,  478, 338,  480, 338,  478, 340,  478, 340,  478, 1146,  452, 368,  452, 368,  478, 348,  478, 336,  478, 340,  478, 340,  478, 340,  480, 1144,  452, 368,  476, 342,  478, 346,  478, 338,  478, 1176,  450, 368,  474, 344,  478, 340,  480, 338,  480, 340,  478, 346,  478, 1144,  452, 366,  476, 1152,  452, 1178,  452, 1176,  452, 1172,  452, 1178,  450, 1178,  452};  // FUJITSU_AC
uint8_t state[7] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x02, 0xFD};

Timestamp : 000008.718
Library   : v2.7.19

Protocol  : FUJITSU_AC
Code      : 0x1463001010FE0930810101000000202D (128 Bits)
Mesg Desc.: Model: 1 (ARRAH2E), Id: 0, Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 1 (High), Clean: Off, Filter: Off, Swing: 0 (Off), Command: N/A, Timer: Off
uint16_t rawData[259] = {3230, 1662,  448, 424,  448, 368,  476, 1154,  448, 370,  450, 1176,  450, 370,  450, 370,  476, 348,  476, 1148,  450, 1180,  448, 370,  450, 368,  476, 342,  476, 1150,  450, 1180,  450, 376,  450, 366,  476, 342,  476, 342,  476, 394,  476, 340,  476, 342,  476, 342,  476, 348,  476, 338,  476, 342,  476, 340,  478, 342,  476, 1150,  448, 370,  450, 370,  476, 350,  476, 340,  476, 342,  476, 342,  476, 342,  476, 1148,  448, 370,  450, 370,  476, 348,  476, 340,  478, 1178,  450, 1180,  450, 1180,  450, 1180,  450, 1176,  450, 1180,  450, 1182,  450, 1172,  450, 368,  450, 368,  476, 1148,  448, 370,  450, 368,  476, 342,  476, 348,  476, 338,  476, 340,  476, 418,  476, 342,  476, 1152,  450, 1174,  450, 370,  450, 374,  476, 1148,  450, 368,  448, 368,  476, 342,  476, 340,  476, 340,  476, 342,  476, 1156,  450, 1172,  450, 368,  450, 368,  476, 340,  476, 340,  476, 340,  476, 390,  476, 348,  476, 1144,  450, 370,  450, 368,  476, 344,  476, 340,  476, 342,  478, 342,  478, 348,  476, 340,  476, 340,  476, 342,  478, 342,  476, 342,  476, 342,  476, 342,  476, 348,  476, 338,  476, 342,  476, 342,  476, 342,  476, 342,  476, 342,  476, 348,  476, 348,  476, 338,  476, 340,  476, 342,  476, 342,  476, 340,  476, 342,  476, 342,  476, 350,  476, 340,  476, 342,  476, 342,  476, 342,  476, 340,  476, 1150,  450, 370,  448, 376,  470, 1154,  450, 368,  448, 1178,  450, 1204,  448, 368,  450, 1174,  450, 370,  450, 374,  470};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x81, 0x01, 0x01, 0x00, 0x00, 0x00, 0x20, 0x2D};

crankyoldgit, 2022-04-29:
There has been some recent changes to Fujitsu support. Can you please try the current `master` branch (at or after https://github.com/crankyoldgit/IRremoteESP8266/commit/a2f79550ddfbd39e3b2c94f5827493bcaf8b0944) rather than v2.7.19, and try again.

gladson1990, 2022-04-29:
Okies ... I ll try that and let u know

crankyoldgit, 2022-05-06:
Any improvement?

gladson1990, 2022-05-06:
Yes. I could see that the latest code works, one thing which I observed is that, the IR led of 5mm is only been considered by this AC, i had tried with IR led of 3mm and it did not work. 
 Plus one more thing , I tried the IRrecv2 dump using this receiver https://www.hnhcart.com/products/tsop1738-ir-receiver?variant=33027031236696&currency=INR&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gclid=CjwKCAjw682TBhATEiwA9crl3wJSLz0f6xY_snNut4KMc--Se0Zm6HFbC24KsIA-xBVlZeH3GvrkLhoC-goQAvD_BwE and it did not work, but when i used https://thinkrobotics.in/products/tsop1738-ir-receiver?variant=39536368582742&currency=INR&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gclid=CjwKCAjw682TBhATEiwA9crl35rFORHhrLI5VIZWivEq81nST78vPhVWzJXHKZgYp788oC5Mqsz2eRoCz8QQAvD_BwE it worked. Is there any separate program or circuit for the first receiver to work ?

gladson1990, 2022-05-06:
I am closing this defect as it has fulfilled my requirement. Thanks

crankyoldgit, 2022-05-06:
> Yes. I could see that the latest code works, one thing which I observed is that, the IR led of 5mm is only been considered by this AC, i had tried with IR led of 3mm and it did not work.

Size of the IR led doesn't matter. What will matter is the brightness, power, strength, & frequency of the IR led. It will all depend on the circuit and the components etc.

> Is there any separate program or circuit for the first receiver to work ?

The first one is a 5V only one. The second is a 3V-5V one.
You'd need to connect the `Vcc` of the first one to `5V`, not `3V3` (3.3V).
Note: The `Signal` pin on the first one will produce a `5V` signal, an ESP8266 can handle input of `5V`, but an ESP32 can't. They can only handle 3V signals. You will probably kill or damage an ESP32 if you feed it a 5V signal.

crankyoldgit, 2022-05-06:
Can you please supply the Brand & Model number of **both** the A/C & the remote so we can add it to the list of supported devices?

closed 2022-05-06 by crankyoldgit

gladson1990, 2022-05-06:
[image]

crankyoldgit, 2022-05-06:
Thanks. What about the actual A/C unit?

gladson1990, 2022-05-06:
its my friends AC and cant figure out the AC model , I shall get u the details when I go to his home.