crankyoldgit/IRremoteESP8266 #1802: Need support for Daikin Remote BRC4M150W16

SaiSridhargithub, 2022-05-11:
I have a remote from Daikin BRC4M150W16 that is reporting unknown using IRRecvdumpV2. 

Raw Data examples are below:

Power On:

20:23:36.998 -> Timestamp : 002090.245
20:23:36.998 -> Library   : v2.8.2
20:23:36.998 -> 
20:23:36.998 -> Protocol  : UNKNOWN
20:23:36.998 -> Code      : 0x3BFB2888 (204 Bits)
20:23:36.998 -> uint16_t rawData[407] = {4852, 2298,  202, 1942,  202, 870,  200, 870,  202, 868,  202, 1942,  202, 870,  202, 870,  202, 870,  202, 870,  202, 1942,  202, 870,  202, 1942,  202, 1942,  202, 868,  202, 1942,  202, 1942,  202, 1912,  232, 1942,  202, 1942,  202, 868,  202, 1942,  202, 870,  202, 870,  202, 868,  202, 868,  204, 868,  202, 870,  202, 1942,  202, 868,  204, 868,  202, 1942,  202, 870,  202, 870,  200, 870,  202, 1942,  202, 868,  202, 870,  202, 870,  202, 870,  202, 870,  202, 870,  202, 870,  202, 870,  202, 868,  202, 868,  204, 868,  202, 870,  202, 870,  202, 870,  202, 1942,  202, 1942,  202, 1942,  202, 868,  202, 868,  204, 1942,  202, 870,  202, 29468,  4880, 2266,  256, 1888,  230, 844,  202, 868,  202, 870,  256, 1888,  230, 842,  228, 842,  230, 842,  256, 814,  232, 1914,  254, 816,  230, 1916,  256, 1888,  256, 814,  256, 1890,  254, 1888,  256, 1888,  256, 1888,  256, 1888,  256, 816,  230, 1912,  258, 814,  256, 816,  256, 816,  282, 790,  280, 790,  282, 790,  282, 1862,  258, 814,  280, 790,  282, 1862,  280, 792,  280, 790,  282, 788,  282, 790,  282, 790,  282, 790,  282, 790,  282, 788,  282, 790,  282, 1862,  282, 1862,  282, 790,  282, 788,  282, 1862,  282, 1864,  280, 1862,  282, 790,  280, 790,  282, 790,  282, 790,  282, 788,  284, 790,  282, 760,  310, 790,  282, 788,  282, 1862,  282, 788,  282, 790,  282, 788,  282, 788,  282, 1862,  284, 788,  282, 788,  284, 788,  282, 790,  282, 788,  284, 788,  282, 790,  282, 788,  284, 758,  312, 788,  284, 788,  282, 788,  284, 788,  282, 788,  284, 788,  282, 788,  284, 788,  284, 758,  314, 786,  284, 788,  284, 1860,  282, 790,  282, 790,  282, 1862,  282, 758,  312, 762,  310, 788,  284, 788,  282, 758,  314, 758,  314, 1862,  282, 760,  312, 1860,  284, 760,  310, 788,  282, 760,  312, 788,  284, 760,  312, 788,  282, 790,  282, 788,  284, 788,  282, 790,  282, 788,  282, 758,  314, 788,  282, 758,  314, 1862,  282, 758,  312, 788,  284, 758,  312, 758,  314, 788,  282, 758,  314, 758,  312, 788,  284, 788,  284, 758,  312, 760,  312, 762,  310, 790,  284, 788,  282, 758,  314, 758,  312, 788,  282, 760,  312, 760,  312, 760,  312, 788,  282, 758,  314, 788,  284, 788,  284, 758,  312, 788,  282, 790,  282, 1832,  310, 758,  316, 788,  284, 1862,  282, 1862,  282, 1832,  310, 788,  284};  // UNKNOWN 3BFB2888
20:23:37.230 -> 
20:23:37.230 -> 

AC ON, temp Up from 29 to 30 

Timestamp : 002171.691
20:24:58.432 -> Library   : v2.8.2
20:24:58.432 -> 
20:24:58.432 -> Protocol  : UNKNOWN
20:24:58.432 -> Code      : 0x2DDC3E71 (204 Bits)
20:24:58.465 -> uint16_t rawData[407] = {4852, 2296,  202, 1942,  200, 868,  204, 870,  202, 868,  204, 1942,  202, 870,  200, 870,  202, 868,  202, 842,  230, 1942,  202, 870,  200, 1942,  202, 1882,  262, 870,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  204, 868,  202, 1942,  202, 868,  202, 870,  202, 870,  202, 870,  202, 868,  204, 842,  230, 1942,  202, 870,  202, 870,  202, 1940,  202, 868,  204, 870,  200, 870,  202, 1942,  202, 868,  204, 868,  202, 868,  204, 870,  202, 870,  202, 870,  200, 868,  202, 870,  202, 870,  202, 868,  204, 868,  204, 870,  202, 870,  202, 868,  202, 1944,  200, 1942,  202, 1942,  202, 870,  202, 870,  202, 1940,  204, 870,  202, 29468,  4850, 2270,  228, 1942,  202, 870,  202, 870,  202, 868,  230, 1914,  204, 870,  228, 844,  202, 870,  202, 870,  202, 1942,  202, 870,  202, 1942,  202, 1942,  228, 842,  230, 1914,  230, 1914,  204, 1942,  230, 1914,  202, 1940,  232, 842,  202, 1944,  200, 870,  230, 842,  230, 842,  230, 840,  230, 840,  256, 754,  318, 1888,  256, 818,  228, 840,  282, 1864,  230, 844,  256, 814,  256, 814,  230, 840,  256, 816,  280, 790,  258, 814,  256, 816,  258, 814,  258, 1886,  256, 1888,  258, 752,  294, 842,  282, 1862,  256, 1888,  282, 1862,  282, 792,  254, 814,  282, 790,  282, 790,  282, 790,  282, 790,  282, 786,  284, 790,  282, 788,  282, 1862,  282, 790,  282, 790,  282, 788,  282, 790,  284, 1862,  282, 790,  282, 788,  284, 788,  282, 790,  282, 788,  282, 790,  282, 788,  284, 788,  284, 790,  282, 790,  282, 790,  282, 788,  282, 788,  284, 788,  284, 788,  284, 788,  282, 788,  284, 788,  282, 790,  282, 1860,  282, 790,  282, 1862,  282, 788,  284, 1860,  282, 788,  282, 788,  286, 758,  312, 790,  282, 788,  282, 788,  282, 1860,  284, 790,  282, 1860,  284, 788,  284, 758,  314, 788,  282, 758,  312, 788,  284, 788,  284, 758,  312, 788,  284, 788,  282, 790,  282, 788,  284, 788,  284, 788,  284, 788,  282, 1860,  284, 758,  314, 760,  312, 758,  312, 760,  312, 762,  310, 788,  282, 790,  282, 788,  282, 788,  284, 788,  284, 758,  314, 762,  310, 790,  282, 760,  312, 760,  310, 788,  284, 788,  282, 762,  310, 760,  312, 758,  314, 788,  282, 758,  314, 760,  312, 758,  314, 788,  284, 758,  314, 758,  314, 788,  282, 790,  282, 1834,  312, 1832,  314, 1830,  314, 1828,  314, 758,  314};  // UNKNOWN 2DDC3E71
20:24:58.664 -> 
20:24:58.664 -> 

AC ON temp down from 30 to 29

Timestamp : 002171.691
20:24:58.432 -> Library   : v2.8.2
20:24:58.432 -> 
20:24:58.432 -> Protocol  : UNKNOWN
20:24:58.432 -> Code      : 0x2DDC3E71 (204 Bits)
20:24:58.465 -> uint16_t rawData[407] = {4852, 2296,  202, 1942,  200, 868,  204, 870,  202, 868,  204, 1942,  202, 870,  200, 870,  202, 868,  202, 842,  230, 1942,  202, 870,  200, 1942,  202, 1882,  262, 870,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  204, 868,  202, 1942,  202, 868,  202, 870,  202, 870,  202, 870,  202, 868,  204, 842,  230, 1942,  202, 870,  202, 870,  202, 1940,  202, 868,  204, 870,  200, 870,  202, 1942,  202, 868,  204, 868,  202, 868,  204, 870,  202, 870,  202, 870,  200, 868,  202, 870,  202, 870,  202, 868,  204, 868,  204, 870,  202, 870,  202, 868,  202, 1944,  200, 1942,  202, 1942,  202, 870,  202, 870,  202, 1940,  204, 870,  202, 29468,  4850, 2270,  228, 1942,  202, 870,  202, 870,  202, 868,  230, 1914,  204, 870,  228, 844,  202, 870,  202, 870,  202, 1942,  202, 870,  202, 1942,  202, 1942,  228, 842,  230, 1914,  230, 1914,  204, 1942,  230, 1914,  202, 1940,  232, 842,  202, 1944,  200, 870,  230, 842,  230, 842,  230, 840,  230, 840,  256, 754,  318, 1888,  256, 818,  228, 840,  282, 1864,  230, 844,  256, 814,  256, 814,  230, 840,  256, 816,  280, 790,  258, 814,  256, 816,  258, 814,  258, 1886,  256, 1888,  258, 752,  294, 842,  282, 1862,  256, 1888,  282, 1862,  282, 792,  254, 814,  282, 790,  282, 790,  282, 790,  282, 790,  282, 786,  284, 790,  282, 788,  282, 1862,  282, 790,  282, 790,  282, 788,  282, 790,  284, 1862,  282, 790,  282, 788,  284, 788,  282, 790,  282, 788,  282, 790,  282, 788,  284, 788,  284, 790,  282, 790,  282, 790,  282, 788,  282, 788,  284, 788,  284, 788,  284, 788,  282, 788,  284, 788,  282, 790,  282, 1860,  282, 790,  282, 1862,  282, 788,  284, 1860,  282, 788,  282, 788,  286, 758,  312, 790,  282, 788,  282, 788,  282, 1860,  284, 790,  282, 1860,  284, 788,  284, 758,  314, 788,  282, 758,  312, 788,  284, 788,  284, 758,  312, 788,  284, 788,  282, 790,  282, 788,  284, 788,  284, 788,  284, 788,  282, 1860,  284, 758,  314, 760,  312, 758,  312, 760,  312, 762,  310, 788,  282, 790,  282, 788,  282, 788,  284, 788,  284, 758,  314, 762,  310, 790,  282, 760,  312, 760,  310, 788,  284, 788,  282, 762,  310, 760,  312, 758,  314, 788,  282, 758,  314, 760,  312, 758,  314, 788,  284, 758,  314, 758,  314, 788,  282, 790,  282, 1834,  312, 1832,  314, 1830,  314, 1828,  314, 758,  314};  // UNKNOWN 2DDC3E71
20:24:58.664 -> 
20:24:58.664 -> 

Mode Change to Auto:

Timestamp : 002171.691
20:24:58.432 -> Library   : v2.8.2
20:24:58.432 -> 
20:24:58.432 -> Protocol  : UNKNOWN
20:24:58.432 -> Code      : 0x2DDC3E71 (204 Bits)
20:24:58.465 -> uint16_t rawData[407] = {4852, 2296,  202, 1942,  200, 868,  204, 870,  202, 868,  204, 1942,  202, 870,  200, 870,  202, 868,  202, 842,  230, 1942,  202, 870,  200, 1942,  202, 1882,  262, 870,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  202, 1942,  204, 868,  202, 1942,  202, 868,  202, 870,  202, 870,  202, 870,  202, 868,  204, 842,  230, 1942,  202, 870,  202, 870,  202, 1940,  202, 868,  204, 870,  200, 870,  202, 1942,  202,  [... cut ...]
        0b110100100000000000000000000000000010000000000000000000000000000000000000000000000000000100000100011000110000000001001000000101111101101000010001 (LSB first)

Total Nr. of suspected bits: 200

Using the above would be very grateful if you could help adding this remote to the library. I am happy to provide you any other items of data to help with the addition. This AC remote seems to be a fairly new unit as they just installed the VRV (central unit) in the dwelling. 

Many thanks and Warm regards, Sridhar

crankyoldgit, 2022-05-12:
@SaiSridhargithub I've created a branch of the library (daikin200_basic https://github.com/crankyoldgit/IRremoteESP8266/tree/daikin200_basic) that seems to do basic decoding of this new 200 bit Daikin protocol.

Please download, compile, & test it. Let me know how it goes etc. If it's working then you're practically at this step in the wiki: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data

As we have plenty of other Daikin protocols, I'm fairly sure we've got the bit ordering correct, so you can probably skip that part.

The ball is now in your court. We are waiting on your confirmation (or not) of it working, and then your analysis of what all they bits/bytes do etc.

SaiSridhargithub, 2022-05-12:
Thankyou @crankyoldgit, we are in business. It works, i did IRRECVDUMPV3 with the new library and the Proptocol now is recognized. Happy to do the next steps. Please guide me on the next steps. Many thanks really appreciate it. 

20:15:43.568 -> Library   : v2.8.2
20:15:43.568 -> 
20:15:43.568 -> Protocol  : DAIKIN200
20:15:43.568 -> Code      : 0x11DA174804004E11DA174800630001000000100000000000BE (200 Bits)
20:15:43.601 -> uint16_t rawData[407] = {4930, 2218,  282, 1862,  280, 792,  256, 816,  256, 816,  282, 1864,  254, 816,  256, 816,  256, 816,  282, 790,  280, 1836,  308, 792,  256, 1886,  282, 1864,  254, 816,  256, 1888,  280, 1864,  256, 1888,  256, 1888,  380, 1766,  280, 764,  308, 1864,  280, 792,  258, 812,  280, 790,  280, 792,  282, 790,  282, 790,  282, 1862,  280, 790,  282, 790,  282, 1862,  282, 790,  282, 788,  280, 790,  282, 1860,  282, 790,  280, 790,  282, 790,  280, 792,  280, 790,  280, 790,  282, 792,  278, 792,  280, 792,  280, 790,  280, 790,  282, 790,  282, 788,  282, 760,  310, 1862,  282, 1862,  282, 1862,  280, 792,  280, 792,  280, 1864,  282, 790,  282, 29386,  4932, 2218,  280, 1832,  312, 790,  282, 758,  312, 790,  282, 1832,  312, 762,  310, 788,  282, 788,  282, 790,  282, 1860,  282, 760,  314, 1862,  280, 1834,  310, 790,  282, 1862,  282, 1862,  282, 1862,  282, 1860,  282, 1864,  280, 790,  282, 1834,  310, 790,  282, 790,  282, 790,  282, 762,  308, 790,  282, 790,  282, 1862,  284, 788,  282, 760,  312, 1862,  280, 760,  312, 790,  282, 788,  282, 790,  282, 790,  282, 788,  282, 760,  312, 790,  282, 764,  308, 1862,  282, 1860,  282, 788,  282, 790,  280, 790,  282, 1862,  282, 1862,  282, 758,  314, 760,  312, 790,  280, 762,  312, 756,  314, 760,  312, 758,  312, 790,  282, 760,  312, 1862,  282, 760,  310, 760,  312, 788,  282, 760,  312, 758,  314, 790,  282, 790,  282, 760,  314, 760,  312, 760,  312, 758,  312, 758,  312, 788,  282, 790,  280, 790,  282, 760,  312, 760,  312, 758,  312, 756,  314, 760,  314, 790,  258, 786,  308, 760,  312, 758,  314, 760,  312, 758,  312, 762,  312, 758,  314, 760,  312, 758,  312, 760,  312, 760,  310, 760,  312, 760,  290, 780,  312, 1834,  288, 784,  312, 790,  258, 784,  286, 782,  290, 782,  288, 786,  286, 784,  288, 814,  258, 784,  286, 784,  288, 784,  288, 814,  258, 784,  288, 782,  290, 782,  290, 782,  288, 784,  288, 784,  288, 786,  286, 784,  286, 814,  258, 782,  288, 784,  288, 780,  290, 784,  288, 784,  286, 784,  288, 784,  286, 786,  312, 756,  290, 784,  312, 760,  288, 784,  310, 758,  290, 786,  310, 790,  280, 760,  312, 760,  312, 760,  312, 762,  310, 760,  310, 762,  310, 758,  314, 764,  306, 1832,  312, 1832,  312, 1862,  282, 1862,  282, 1834,  310, 762,  310, 1862,  282};  // DAIKIN200
20:15:43.799 -> uint8_t state[25] = {0x11, 0xDA, 0x17, 0x48, 0x04, 0x00, 0x4E, 0x11, 0xDA, 0x17, 0x48, 0x00, 0x63, 0x00, 0x01, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0xBE};

20:15:43.799 -> 
20:15:43.799 -> 
20:15:48.184 -> Timestamp : 000106.345
20:15:48.184 -> Library   : v2.8.2
20:15:48.184 -> 
20:15:48.184 -> Protocol  : DAIKIN200
20:15:48.184 -> Code      : 0x11DA174804004E11DA174800630000000000100000000000BD (200 Bits)
20:15:48.184 -> uint16_t rawData[407] = {5036, 2218,  282, 1834,  310, 790,  282, 760,  310, 760,  312, 1862,  282, 790,  280, 790,  282, 790,  282, 788,  282, 1862,  282, 792,  280, 1862,  282, 1862,  282, 790,  282, 1862,  282, 1860,  282, 1832,  312, 1832,  312, 1834,  310, 790,  280, 1832,  312, 790,  282, 788,  282, 758,  312, 792,  280, 788,  282, 790,  282, 1862,  282, 790,  282, 788,  282, 1862,  282, 790,  282, 790,  282, 788,  282, 1862,  282, 790,  282, 760,  312, 788,  284, 758,  312, 760,  312, 760,  312, 790,  282, 790,  282, 788,  282, 762,  310, 788,  282, 760,  314, 788,  282, 790,  282, 1862,  282, 1832,  312, 1828,  314, 762,  310, 760,  310, 1834,  312, 788,  282, 29388,  4932, 2216,  284, 1860,  284, 788,  284, 760,  312, 758,  312, 1832,  312, 760,  314, 758,  314, 790,  280, 760,  312, 1862,  282, 760,  312, 1830,  312, 1836,  308, 790,  282, 1862,  282, 1862,  282, 1834,  310, 1860,  282, 1832,  312, 758,  314, 1862,  282, 758,  314, 760,  312, 790,  282, 758,  314, 756,  314, 758,  312, 1832,  312, 760,  312, 758,  314, 1862,  282, 760,  312, 788,  284, 788,  282, 760,  312, 758,  314, 760,  312, 760,  312, 758,  314, 758,  312, 1832,  314, 1830,  314, 758,  314, 760,  312, 760,  312, 1830,  314, 1860,  282, 760,  310, 760,  312, 760,  312, 758,  312, 760,  312, 760,  312, 758,  316, 758,  312, 758,  314, 758,  312, 758,  314, 760,  310, 758,  314, 788,  282, 758,  314, 758,  312, 760,  312, 758,  312, 762,  312, 760,  312, 758,  312, 758,  312, 760,  310, 790,  282, 760,  312, 758,  314, 788,  280, 762,  312, 760,  310, 760,  312, 760,  310, 758,  312, 760,  312, 758,  312, 760,  312, 758,  312, 758,  290, 814,  258, 784,  312, 760,  310, 760,  312, 758,  290, 782,  290, 784,  310, 758,  314, 1832,  290, 784,  312, 758,  314, 758,  312, 760,  312, 760,  288, 780,  290, 780,  290, 784,  288, 782,  314, 758,  290, 780,  290, 784,  288, 784,  288, 782,  288, 782,  290, 782,  290, 782,  290, 780,  290, 784,  288, 784,  288, 782,  288, 782,  288, 784,  288, 782,  290, 782,  290, 782,  290, 782,  290, 784,  288, 784,  290, 780,  290, 782,  288, 784,  288, 782,  290, 780,  290, 782,  290, 784,  288, 784,  288, 784,  288, 782,  290, 782,  290, 782,  290, 782,  288, 782,  288, 1856,  288, 784,  288, 1856,  288, 1854,  288, 1856,  288, 1856,  288, 782,  290, 1854,  290};  // DAIKIN200
20:15:48.383 -> uint8_t state[25] = {0x11, 0xDA, 0x17, 0x48, 0x04, 0x00, 0x4E, 0x11, 0xDA, 0x17, 0x48, 0x00, 0x63, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0xBD};

crankyoldgit, 2022-05-12:
Thanks for confirming the capture part works. I've updated it to do checksum verification as well, as it uses the same approach as other Daikin protocols.

> Happy to do the next steps. Please guide me on the next steps.

As per previous message, please read & follow these steps:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data

Once you've done the analysis, I'll try to code it up eventually for you.

SaiSridhargithub, 2022-05-12:
Thankyou. I read up and started to make progress on the analysis. At this point have deciphered most, except the timer control. While I work on the timer, will send you the first tranche of mode, temp, swing etc codes shortly through a spreadsheet. You can confirm  that would be what you need to with on the code enhancements. Thanks for your assistance

SaiSridhargithub, 2022-05-13:
Created a googlesheet for Daikin200 with what i was able to decipher till now. Mode, ON/OFF, Temp, Swing updated. Can you have a look and let me know if the explanation is understandable. Thankyou

https://docs.google.com/spreadsheets/d/1ogsHYiikPGuZY8B1e3l8S5JOAp2-VGXLxe1kVF2m_QI/edit?usp=sharing

SaiSridhargithub, 2022-05-13:
updated the same sheet with analysis of Timer Data, Please let me know if you need any other pieces of information. Many thanks

crankyoldgit, 2022-05-16:
Okay, I think I mostly understand your spreadsheet.

Can you please check what happens if the ON & the OFF timer are simultaneously set?

I'll try to get to coding this soon-ish.

SaiSridhargithub, 2022-05-16:
[image]
Thanks. Please see attached remote with the slider open. I did not get your question on the ON & OFF simultaneously set. I will try and answer this from what i observed. No signals are sent from the timer on, off, select up / down buttons when pressed. Signals are emitted only when the sequence of ON-Reserve, Off-Reserve, On-Cancel, Off-Cancel are pressed. The other timer buttons when pressed individually (like the up/down after On button, or up/down after Off button) only change the LCD display values in the corresponding ON or OFF locations in the LCD. On pressing the Timer ON and reserve, the ON values are transmitted and when the Timer OFF and reserve pressed the corresponding off value is sent across.  Rows 51 to 69 in the sheet are the case in point. 0x81, 0x82, 0x83 are the 1HR, 2HR, 3HR data values for ON reserve sent through byte 15. Similarly, 0x81, 0x82, 0xC8 are sent across for 1HR, 2HR and 72HR through byte 16 when Timer OFF and Reserve are pressed. Those were the values on the LCD at that instant in time when the ON-Reserve and Off-Reserve were pressed. For On-Cancel combination, 0x00 is sent through byte 15 and for Off-Cancel sequence, 0x00 is sent through byte 16. hope it makes sense. Happy to further answer if you have questions

crankyoldgit, 2022-05-17:
Sorry. I probably phrased it poorly. What I meant was try to set both the ON & the OFF timers to being on/in use.
So that they both appear in the message, if we can. I need to see if that is a valid message, and if so, how it is represented.

SaiSridhargithub, 2022-05-17:
Thanks thats what i attempted to do in the instance explained in my last post. Set 13hr for on and 15hr for off. The data is in the spreadsheer as well rows 76 to 80. hope it helps

SaiSridhargithub, 2022-05-29:
Hello, are you looking for information from me on this one. Or are you all set? Please let me know.

crankyoldgit, 2022-05-30:
I think you're right for now. I just have to find time to get to it.

SaiSridhargithub, 2022-05-30:
Thankyou

SaiSridhargithub, 2022-08-10:
Sorry to bother you, any update on this?

SaiSridhargithub, 2022-10-18:
Release v2.8.3 says "Add basic support for 200-bit Daikin protocol". Does this mean I can use it for sending commands like mode, temp to the AC unit. please let me know so that I can test the same. Thankyou

ivishaltejwani, 2022-11-07:
any update on this?

ivishaltejwani, 2022-11-08:
Just tested this with Daikin176 peotocol everything works with fan speed

SaiSridhargithub, 2022-11-08:
BRC4M150W16 uses the DAIKIN200 protocol (200 bit). how is that the DAIKIN176 works ?

ivishaltejwani, 2022-11-08:
> BRC4M150W16 uses the DAIKIN200 protocol (200 bit). how is that the DAIKIN176 works ?

I just tried it , it works you can check it too , the receiver of the ac is calibrated to different remotes in case the current one is not easily available I guess , you can try it and check

SaiSridhargithub, 2022-11-20:
> Release v2.8.3 says "Add basic support for 200-bit Daikin protocol". Does this mean I can use it for sending commands like mode, temp to the AC unit. please let me know so that I can test the same. Thankyou

Can you help with this answer. If the support for DAIKIN200 is available for on /off and basic functionality, i will test it. The issue is it involves travel to go to a home to physically test the indoor unit where the remote and unit are located. Please let me know and we can test and close this issue.

NiKiZe, 2023-12-06:
> I've actually been trying to use the library to control an ac which uses this protocol, i can see that the protocol is being recognized and states data is also displayed but unlike the other protocols i don't see a class implemented for this one any updates on this ? or if I'm missing out something that would be of help as well

Don't reuse an old issue like this, create a new discussion instead, and include any output that you get and explain what you expect. Also consider reading the FAQ which might answer your question.

jagannathmurali, 2024-02-21:
Is it ok to continue this thread but on same topic [BRC4M150W Remote]?

I have tested DAIKIN200 by varying temperature with data received through IRrecvDumpV3 and sent using the below testCode:
````
/**
   IR AC DAIKIN200 TEST
*/

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 2;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Data captured by IRrecvDumpV3.ino for turning OFF AC at 16C
uint16_t rawDataOFF[407] = {5006, 2178,  324, 1792,  354, 720,  350, 724,  348, 754,  330, 1788,  356, 718,  352, 718,  356, 748,  322, 750,  322, 1794,  350, 752,  320, 1796,  358, 1790,  356, 718,  352, 1792,  352, 1794,  352, 1796,  358, 1788,  358, 1788,  354, 718,  354, 1794,  352, 722,  350, 722,  348, 754,  330, 744,  328, 744,  328, 744,  326, 1790,  354, 718,  354, 750,  322, 1794,  350, 722,  350, 752,  330, 744,  328, 1788,  356, 716,  356, 748,  324, 748,  324, 748,  324, 752,  318, 754,  318, 754,  330, 742,  330, 744,  326, 746,  326, 746,  326, 746,  326, 748,  324, 748,  322, 1796,  350, 1794,  358, 1790,  356, 716,  354, 748,  326, 1790,  352, 720,  352, 29350,  5010, 2176,  326, 1790,  352, 750,  324, 720,  352, 720,  352, 1794,  358, 714,  358, 744,  328, 746,  326, 746,  326, 1790,  352, 720,  354, 1792,  350, 1796,  360, 712,  358, 1788,  356, 1790,  352, 1796,  350, 1794,  348, 1796,  358, 746,  326, 1790,  356, 716,  356, 748,  322, 750,  322, 750,  320, 754,  318, 756,  328, 1786,  356, 746,  326, 716,  356, 1790,  354, 720,  352, 750,  320, 752,  320, 754,  328, 744,  326, 746,  326, 746,  326, 746,  326, 716,  356, 1790,  352, 1794,  352, 750,  322, 722,  358, 1788,  358, 1788,  356, 1792,  352, 720,  352, 750,  320, 752,  322, 752,  328, 744,  330, 744,  328, 744,  326, 748,  322, 748,  324, 750,  322, 750,  322, 750,  322, 750,  318, 754,  318, 1796,  358, 744,  328, 746,  324, 748,  324, 750,  320, 752,  322, 748,  322, 752,  320, 752,  318, 754,  328, 744,  328, 744,  328, 744,  326, 746,  324, 748,  324, 752,  322, 750,  320, 752,  320, 752,  320, 724,  360, 1788,  354, 1790,  354, 1790,  354, 718,  352, 722,  350, 754,  320, 754,  330, 714,  356, 716,  358, 744,  328, 746,  326, 748,  326, 716,  354, 750,  322, 750,  322, 752,  320, 752,  320, 754,  330, 714,  356, 746,  328, 746,  326, 746,  324, 748,  324, 748,  322, 750,  322, 752,  320, 752,  320, 752,  330, 1788,  356, 716,  356, 718,  354, 720,  352, 750,  322, 750,  324, 750,  320, 752,  320, 752,  330, 744,  330, 744,  326, 744,  328, 746,  324, 750,  322, 748,  324, 750,  322, 750,  322, 750,  320, 724,  360, 742,  330, 714,  358, 744,  326, 748,  326, 746,  324, 748,  324, 748,  324, 750,  318, 1796,  348, 1798,  358, 714,  356, 1790,  356, 718,  352, 720,  352, 750,  322, 750,  322};  // DAIKIN200
uint8_t stateOFF[25] = {0x11, 0xDA, 0x17, 0x48, 0x04, 0x00, 0x4E, 0x11, 0xDA, 0x17, 0x48, 0x00, 0x73, 0x00, 0x20, 0x00, 0x00, 0x0E, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x0B};

// Data captured by IRrecvDumpV3.ino for turning ON AC at 16C
uint16_t rawDataON[407] = {5002, 2152,  352, 1794,  350, 752,  330, 744,  326, 716,  356, 1790,  354, 748,  322, 750,  322, 750,  320, 752,  320, 1796,  358, 716,  356, 1790,  352, 1794,  352, 720,  352, 1794,  360, 1786,  358, 1788,  358, 1792,  350, 1794,  348, 724,  348, 1798,  356, 716,  356, 746,  324, 748,  324, 718,  352, 720,  350, 722,  350, 1796,  360, 742,  326, 716,  356, 1788,  354, 718,  352, 750,  322, 750,  322, 1794,  352, 722,  358, 714,  356, 746,  328, 746,  326, 746,  324, 750,  322, 748,  324, 750,  322, 750,  320, 752,  330, 744,  328, 744,  326, 746,  326, 746,  324, 1792,  354, 1792,  350, 1796,  358, 712,  358, 744,  326, 1790,  354, 718,  352, 29348,  5008, 2148,  354, 1790,  352, 720,  352, 750,  322, 752,  318, 1796,  358, 716,  356, 746,  326, 746,  326, 746,  324, 1792,  352, 722,  350, 1796,  358, 1788,  356, 716,  356, 1790,  352, 1794,  350, 1794,  360, 1790,  352, 1794,  350, 718,  354, 1792,  352, 720,  350, 754,  328, 714,  358, 714,  356, 716,  356, 746,  326, 1790,  354, 720,  352, 720,  352, 1794,  358, 714,  358, 746,  326, 746,  326, 718,  354, 746,  322, 750,  322, 750,  322, 722,  348, 754,  330, 1786,  356, 1790,  354, 748,  324, 748,  324, 1792,  350, 1798,  358, 1788,  356, 716,  356, 716,  356, 718,  352, 720,  352, 750,  320, 752,  320, 752,  330, 712,  358, 746,  326, 1788,  356, 718,  352, 748,  322, 750,  322, 750,  322, 1794,  360, 714,  358, 714,  358, 716,  354, 748,  324, 750,  322, 718,  354, 750,  322, 722,  348, 752,  330, 742,  330, 714,  356, 746,  326, 746,  326, 718,  354, 748,  322, 750,  322, 720,  352, 750,  320, 754,  328, 1786,  358, 1790,  356, 1790,  354, 720,  352, 720,  352, 722,  350, 722,  360, 714,  356, 744,  328, 746,  326, 716,  356, 718,  350, 720,  354, 750,  322, 750,  322, 750,  318, 754,  330, 714,  358, 744,  326, 748,  326, 746,  324, 718,  354, 748,  322, 750,  320, 722,  350, 754,  328, 714,  358, 744,  326, 1790,  356, 716,  354, 720,  352, 720,  352, 750,  320, 722,  350, 722,  358, 714,  358, 746,  326, 748,  324, 748,  322, 750,  322, 720,  350, 750,  322, 720,  350, 756,  314, 724,  360, 712,  358, 714,  358, 744,  326, 718,  354, 748,  324, 750,  322, 718,  354, 750,  320, 724,  348, 724,  358, 744,  326, 744,  328, 1790,  354, 1792,  350, 722,  350, 722,  350, 752,  332, 744,  326};  // DAIKIN200
[... cut ...]
The provided code successfully controls the AC, both turning it on and off. 

For further validation, I analyzed the logs obtained from IRrecvDumpV3 while increasing the temperature with the AC both on and off. Here are the logs for when the AC is on [(ON Logs)](https://drive.google.com/file/d/1A5jg9wh6CgqrYEnyVrBxnrRKOHhOs9L3/view?usp=sharing) and when the AC is off [(OFF Logs)](https://drive.google.com/file/d/1yOMoAx8ib8xlpZKMGtFLHe2XgqmBdZaf/view?usp=sharing).

Kindly specify any additional details or validation requirements that are required for proceeding further.

crankyoldgit, 2024-02-21:
> Is it ok to continue this thread but on same topic [BRC4M150W Remote]?

No. Please create a new issue, and reference this one.
See the wiki for how to add a new AC protocol.