crankyoldgit/IRremoteESP8266 #1804: Need help: Here are some questions about  HAIER_AC176.

luolei69, 2022-05-14:
### Version/revision of the library used
v2.8.2

### Describe the bug

I have a remote from Haier KFR-26GW/83@UI-Ge that is reporting unknown using IRRecvdumpV2. BUT I can use HAIER_AC176 control my AC (Not all function is ok),There is my test:
Power: all ok
Model: all ok
Temp: all ok
Fan: ok ( have four Fan mode: Auto\Low\Medium\High)
Swing(V): The "Auto" is ok, else is not normal(PS: just have four Swing(V) mode)
Swing(H): don't have this function.
Quiet: don't have this function.
Turbo: all ok
Econo: on is ok, off not ok.
Light: no
Filter: don't have this function.
Clean: don't have this function.
Beep: don't have this function.

There is my RAW data:

Press Power on (Model: cool, Temp: 26, Fan:Low, Swing(V): Auto, Turbo: off, Econo: off, Light: on,.)

Timestamp : 000370.734
Library   : v2.8.2

Protocol  : UNKNOWN
Code      : 0xB6B57D85 (163 Bits)
uint16_t rawData[325] = {3078, 3002,  3058, 4338,  590, 1612,  588, 516,  584, 1588,  586, 542,  560, 540,  560, 1590,  584, 1618,  584, 542,  558, 1592,  584, 542,  582, 1594,  584, 542,  582, 1592,  560, 1642,  560, 516,  558, 544,  558, 542,  558, 542,  582, 518,  558, 542,  558, 516,  558, 542,  558, 542,  582, 520,  558, 542,  558, 542,  564, 510,  608, 492,  558, 542,  582, 520,  558, 542,  558, 544,  582, 492,  608, 1594,  558, 544,  558, 542,  582, 494,  608, 494,  558, 544,  558, 542,  558, 544,  558, 1616,  582, 1620,  560, 542,  558, 542,  582, 494,  558, 542,  558, 542,  558, 544,  556, 542,  582, 520,  580, 494,  582, 520,  582, 520,  558, 542,  558, 542,  582, 520,  582, 492,  608, 1592,  582, 520,  558, 544,  558, 516,  584, 518,  558, 542,  582, 520,  580, 520,  582, 520,  580, 492,  584, 516,  556, 544,  582, 518,  580, 520,  580, 520,  580, 494,  606, 494,  580, 520,  580, 520,  582, 520,  580, 520,  556, 518,  608, 492,  580, 520,  556, 544,  580, 520,  580, 520,  580, 494,  606, 494,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  606, 496,  578, 522,  580, 520,  580, 520,  580, 520,  580, 494,  606, 494,  580, 520,  580, 520,  580, 1594,  608, 494,  580, 1620,  580, 522,  580, 520,  580, 494,  578, 1620,  582, 520,  580, 1596,  582, 1620,  582, 1594,  608, 1594,  582, 522,  580, 1594,  582, 1620,  582, 520,  580, 1596,  582, 520,  580, 1620,  582, 494,  606, 496,  580, 522,  580, 522,  580, 520,  580, 520,  580, 496,  604, 496,  580, 520,  580, 1620,  582, 1594,  582, 522,  578, 522,  580, 522,  578, 522,  580, 496,  580, 522,  578, 522,  578, 522,  578, 524,  578, 522,  578, 496,  578, 522,  578, 522,  578, 546,  554, 522,  578, 522,  578, 520,  556, 546,  554, 546,  554, 546,  556, 544,  556, 546,  554, 520,  554, 544,  556, 1620,  582, 544,  556, 1594,  580, 546,  556, 1594,  580};  // UNKNOWN B6B57D85

Model: Dry, Temp: 26, Fan:Low, Swing(V): Auto, Turbo: off, Econo: off, Light: on,.

Timestamp : 000612.706
Library   : v2.8.2

Protocol  : UNKNOWN
Code      : 0x8C9E34DB (163 Bits)
uint16_t rawData[325] = {3074, 3000,  3054, 4338,  588, 1612,  588, 540,  560, 1588,  586, 540,  584, 518,  560, 1590,  608, 1594,  582, 542,  582, 1590,  560, 540,  582, 1618,  584, 516,  584, 1590,  560, 1640,  584, 490,  558, 540,  560, 540,  582, 518,  582, 518,  582, 518,  558, 516,  558, 542,  582, 518,  582, 518,  582, 518,  582, 518,  558, 516,  584, 516,  582, 518,  582, 518,  558, 542,  582, 518,  580, 492,  584, 1614,  558, 542,  582, 518,  580, 492,  608, 492,  582, 518,  558, 542,  558, 542,  580, 1592,  560, 1642,  582, 518,  558, 542,  582, 492,  582, 520,  580, 518,  582, 518,  580, 518,  582, 518,  582, 492,  582, 518,  582, 518,  582, 520,  556, 542,  582, 520,  556, 1616,  582, 518,  582, 518,  580, 518,  558, 516,  608, 492,  580, 518,  582, 518,  582, 520,  580, 518,  582, 492,  606, 492,  580, 518,  582, 518,  582, 518,  580, 520,  580, 492,  608, 492,  582, 518,  582, 518,  582, 518,  580, 520,  580, 492,  608, 492,  580, 518,  582, 518,  580, 520,  580, 520,  582, 492,  606, 494,  580, 520,  580, 518,  580, 520,  582, 520,  580, 494,  606, 492,  582, 518,  580, 520,  580, 518,  580, 520,  580, 494,  606, 492,  580, 520,  580, 520,  580, 1618,  582, 1592,  582, 520,  580, 520,  580, 520,  580, 1592,  582, 1618,  582, 1592,  582, 520,  580, 520,  580, 520,  580, 1592,  582, 520,  580, 1620,  582, 1592,  582, 520,  580, 1618,  582, 494,  580, 1620,  582, 520,  578, 520,  580, 494,  606, 494,  580, 520,  580, 522,  580, 520,  580, 520,  580, 494,  606, 1594,  582, 1618,  582, 494,  580, 520,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  580, 520,  580, 520,  580, 520,  580, 522,  578, 520,  580, 494,  580, 520,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  580, 520,  580, 520,  580, 520,  580, 522,  578, 520,  580, 1592,  582, 520,  580, 1592,  608, 494,  580, 1620,  582};  // UNKNOWN 8C9E34DB

Model: Fan, Temp: 26, Fan:Low, Swing(V): Auto, Turbo: off, Econo: off, Light: on,.

Timestamp : 000710.484
Library   : v2.8.2

Protocol  : UNKNOWN
Code      : 0xEB5ACABD (163 Bits)
uint16_t rawData[325] = {3074, 2998,  3052, 4336,  588, 1612,  586, 540,  560, 1588,  586, 540,  560, 540,  558, 1590,  584, 1616,  584, 542,  558, 1614,  560, 542,  582, 1618,  558, 542,  558, 1616,  560, 1642,  558, 516,  558, 542,  558, 542,  558, 542,  558, 542,  558, 542,  558, 516,  582, 518,  558, 542,  558, 542,  582, 518,  582, 518,  558, 516,  584, 516,  558, 542,  558, 542,  558, 542,  558, 542,  558, 516,  584, 1616,  558, 542,  580, 518,  558, 516,  606, 492,  580, 518,  582, 518,  558, 542,  556, 1616,  582, 1618,  560, 542,  556, 544,  556, 516,  582, 518,  556, 544,  556, 542,  558, 542,  558, 542,  556, 518,  580, 520,  556, 542,  558, 542,  558, 542,  580, 1592,  558, 1642,  558, 544,  556, 542,  558, 516,  556, 544,  580, 520,  580, 520,  556, 544,  556, 542,  580, 492,  558, 544,  556, 542,  580, 520,  580, 520,  580, 520,  558, 516,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  582, 518,  580, 520,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  580, 520,  580, 520,  580, 520,  580, 520,  580, 520,  556, 518,  580, 520,  580, 520,  580, 520,  580, 520,  580, 520,  580, 494,  580, 520,  580, 520,  580, 520,  580, 520,  580, 1592,  582, 1618,  582, 520,  580, 1594,  582, 520,  580, 1620,  582, 1592,  582, 1618,  580, 494,  606, 494,  578, 522,  578, 1620,  582, 520,  580, 1592,  582, 1620,  580, 494,  606, 1594,  580, 520,  578, 1592,  582, 520,  580, 520,  578, 522,  580, 520,  578, 520,  578, 496,  578, 520,  578, 520,  580, 520,  580, 1592,  608, 1592,  582, 520,  580, 522,  578, 496,  606, 494,  578, 522,  578, 522,  578, 522,  578, 522,  578, 496,  604, 496,  578, 522,  578, 522,  578, 522,  578, 522,  578, 496,  604, 496,  578, 544,  556, 544,  554, 546,  554, 544,  556, 520,  580, 520,  554, 546,  554, 546,  556, 1620,  580, 518,  554, 1620,  580, 546,  554, 1594,  580};  // UNKNOWN EB5ACABD

Model: Heat, Temp: 26, Fan:Low, Swing(V): Auto, Turbo: off, Econo: off, Light: on,.

Timestamp : 000816.119
Library   : v2.8.2

Protocol  : UNKNOWN
Code      : 0x9A0F1E9F (163 Bits)
uint16_t rawData[325] = {3076, 3000,  3054, 4338,  588, 1614,  588, 538,  562, 1588,  586, 540,  560, 540,  560, 1590,  586, 1616,  608, 518,  582, 1568,  606, 518,  582, 1618,  560, 540,  582, 1590,  560, 1640,  560, 514,  558, 542,  558, 542,  582, 518,  558, 542,  558, 542,  582, 492,  582, 516,  582, 518,  582, 518,  582, 518,  582, 518,  582, 490,  584, 516,  558, 542,  558, 542,  582, 518,  582, 518,  558, 1616,  582, 1618,  582, 518,  582, 492,  580, 518,  582, 518,  580, 520,  558, 540,  582, 518,  582, 1592,  584, 1616,  560, 516,  582, 516,  582, 518,  582, 518,  582, 518,  558, 542,  558, 516,  584, 514,  582, 518,  582, 518,  556, 542,  558, 542,  582, 518,  582, 1590,  560, 542,  582, 518,  582, 518,  582, 492,  582, 518,  582, 518,  582, 518,  582, 518,  582, 518,  582, 492,  582, 518,  582, 518,  580, 520,  580, 518,  582, 518,  558, 516,  582, 518,  582, 518,  580, 518,  582, 520,  556, 516,  608, 492,  580, 518,  582, 520,  580, 518,  582, 518,  580, 520,  580, 492,  580, 520,  580, 520,  580, 518,  580, 520,  580, 518,  580, 494,  580, 520,  580, 518,  580, 520,  580, 520,  580, 520,  580, 492,  580, 520,  580, 518,  580, 520,  580, 518,  580, 1592,  582, 1618,  582, 520,  580, 1592,  582, 1618,  582, 1592,  582, 1618,  582, 1592,  582, 520,  580, 520,  580, 520,  580, 1592,  582, 520,  580, 1618,  582, 1592,  582, 520,  580, 1618,  582, 520,  580, 1592,  582, 520,  580, 520,  580, 520,  580, 494,  580, 520,  578, 520,  580, 520,  580, 520,  578, 520 [... cut ...] 578, 520,  578, 522,  578, 494,  604, 494,  578, 520,  578, 1616,  580, 1592,  580, 520,  578, 520,  578, 520,  578, 520,  578, 494,  578, 522,  578, 520,  578, 522,  578, 520,  578, 522,  578, 494,  578, 522,  578, 520,  578, 522,  576, 520,  578, 522,  578, 496,  576, 522,  576, 522,  578, 522,  576, 522,  578, 522,  576, 496,  578, 544,  556, 1616,  580, 522,  576, 1592,  580, 544,  554, 1592,  580};  // UNKNOWN 86992933

If you need other data from me, I'll be happy to help.
Thank you again.

crankyoldgit, 2022-05-16:
@luolei69 I am amazed that `HAIER_AC176` is working for you at all. It is certainly not designed to work with your _shorter_ HAIER protocol.
I can't/won't change `HAIER_AC176` to work fully for your A/C, but I will work with you to add support for your new variant.
It looks like you've given me enough information to add basic support for it. That is, recognise it & converting it to a state _array_. No extra support for individual controls etc. However that should be fairly trivial if the 176 version is working for you mostly.

In the mean time, please read & follow: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2022-05-16:
@luolei69  I've created a branch of the library (haier_160 https://github.com/crankyoldgit/IRremoteESP8266/tree/haier_160)  & PR #1805 that seems to do basic decoding of this new 160bit Haier protocol.

Please download, compile, & test it. Let me know how it goes etc. If it's working then you're practically at this step in the wiki: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data

As we have plenty of other Haier protocols, I'm fairly sure we've got the bit ordering correct, so you can probably skip that part.
As you say the 176 bit protocols mostly works for you, you can probably use that as a template for your analysis.
i.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/3c6b057579ac89390d0255d1323b857a5b24c91e/src/ir_Haier.h#L193-L261
Except `Sum2` is in `byte[19]`.
The ball is now in your court. We are waiting on your confirmation (or not) of it working, and then your analysis of what all they bits/bytes do etc.

crankyoldgit, 2022-05-17:
Spreadsheet looks workable to me. I'll try to get to it soon.
Thanks for confirming the branch works.

crankyoldgit, 2022-05-17:
Have you got any collected data for the On Timer?
and in your spreadsheet. "Byte 19" is not `Fan3`, that entire byte is the checksum.

crankyoldgit, 2022-05-17:
> My AC don't have On Timer function, just have Off Timer.

Okay. Noted. But maybe it does, just your remote doesn't. ;-)

crankyoldgit, 2022-05-17:
Oh and for your spreadsheet. Please include some `state[]` lines. That's what I use in the tests.

crankyoldgit, 2022-05-17:
Okay, I've got a new branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/haier160_detailed) for you to play with. It has detailed support for your protocol.
Please indicate (with `state[]` and the description (yours & IRrecvDumpV2s etc) what is decoded incorrectly.
Hopefully, it should handle your vertical swings correctly too.

Looking forward to your response.

luolei69, 2022-05-17:
I have updated my sheet, please make sure it meets your requirements.
https://docs.google.com/spreadsheets/d/1waNbqAgeGx2TDkzSrG9_2FEG7BXRGjvFZZreg6pJ0_M/edit?usp=sharing

luolei69, 2022-05-17:
Sorry, I found out that when AC is in Heat mode, the SwingV will increase by one position, So I added a new row of data to the sheet. and renamed the previous "Lowest" to "LowLowest”. I think you can completely ignore the "LowLowest" and "HighHighest" data, in fact, which differ very little from "Lowest" and "Highest".

https://docs.google.com/spreadsheets/d/16U5oO9dLzXmwVViP54A1RyolxTRhISpYaiqKmReJ8PA/edit?usp=sharing

There is my test with your new branch “[haier160_detailed]“:
1. When I use IRMQTTServer control my AC, All command will open the "CLEAN" mode. So I can't do any send test. On my AC, “CLEAN” mode is a cleaning program, so it should be off most of the time.
2. I use IRrecvDumpV2 accept data, Whether I turn "health" on or off, "Health" in IRrecvDumpV2 always shows off, But IRrecvDumpV2 knew that I pressed the health button.
3.  When I press my remote's "CLEAN", "Light", "Auxiliary heat", The "Button" in IRrecvDumpV2 shows unknown. I noticed that there is no "Auxiliary heat" function in IRMQTTServer, so you can ignore this part, but when switching to Heat mode, please keep the "Auxiliary heat" function turned on.
ps：I found that in my RAW data, there is no state of “Light”, only the data of the “Light” button, so I think you can set both “Light on” and “Light off” as the data of this button in the code.

crankyoldgit, 2022-05-18:
> 1. When I use IRMQTTServer control my AC, All command will open the "CLEAN" mode. So I can't do any send test. On my AC, “CLEAN” mode is a cleaning program, so it should be off most of the time.

Clearly that is not desired. I'll try to see what is doing that. FYI, in IRMQTTServer, "filter" is what will control your A/C's health function.

> 2\. I use IRrecvDumpV2 accept data, Whether I turn "health" on or off, "Health" in IRrecvDumpV2 always shows off, But IRrecvDumpV2 knew that I pressed the health button.

Please provide the text output from IRrecvDumpV2/etc for both messages. I need the data.

> 3\. When I press my remote's "CLEAN", "Light", "Auxiliary heat", The "Button" in IRrecvDumpV2 shows unknown. I noticed that there is no "Auxiliary heat" function in IRMQTTServer, so you can ignore this part, but when switching to Heat mode, please keep the "Auxiliary heat" function turned on.

Again, I need **data** as well as descriptions to be able to help you. Please provide the text output from the dump prog.

>     ps：I found that in my RAW data, there is no state of “Light”, only the data of the “Light” button, so I think you can set both “Light on” and “Light off” as the data of this button in the code.

I'm not sure what you mean here. Can you explain it another way?
Some A/C's send a special "toggle" code to control settings like Light etc. So, it may not be a bit in the state that is sent with every message.

luolei69, 2022-05-18:
Sorry for not giving you detailed data before, I added new RAW data at the bottom of sheet01.
You can see that the data of "light on" and "light off" are exactly the same, so I don't think the light state is judged in the raw data.
https://docs.google.com/spreadsheets/d/1RNJ7esbArS5fy1lmiM-i1PekXSNojCMad4WuuyunsC8/edit?usp=sharing

Thanks again for your help, Best regards.

luolei69, 2022-05-20:
@crankyoldgit 
Hi, do I still need to provide some data?

crankyoldgit, 2022-05-22:
> Sorry for not giving you detailed data before, I added new RAW data at the bottom of sheet01. You can see that the data of "light on" and "light off" are exactly the same, so I don't think the light [...]

Yes. I concur with your assessment of the Light button. It looks like it's a toggle message only.
I'll try to spend some time again on this protocol now.

crankyoldgit, 2022-05-22:
I've updated the branch with the changes/problems you've pointed out. Hopefully I fixed most of the satisfactorily.

Please download & test, and let me know how it goes.

luolei69, 2022-05-22:
I tested it in IRrecvDumpV2, There are only two small problems here.

1. You may not have seen this issue I reported earlier.
> Sorry, I found out that when AC is in Heat mode, the SwingV will increase by one position, So I added a new row of data to the sheet. and renamed the previous "Lowest" to "LowLowest”. I think you [...]

2. In IRrecvDumpV2, and now there is no Health status. In my sheet, the Econo data (bety 30) represents the Health.

crankyoldgit, 2022-05-22:
> 2\. In IRrecvDumpV2, and now there is no Health status. In my sheet, the Econo data (bety 30) represents the Health.

How was I to know "Econo" meant "Health" in your spreadsheet/issue?

luolei69, 2022-05-22:
> > 2. In IRrecvDumpV2, and now there is no Health status. In my sheet, the Econo data (bety 30) represents the Health.
> 
> How was I to know "Econo" meant "Health" in your spreadsheet/issue?

Sorry, this is my question, I didn't make it clear in the spreadsheet.

luolei69, 2022-07-08:
Hi @crankyoldgit , I'm very sorry, I didn't mark it clearly in the spreadsheet before, I remarked it in the spreadsheet, Please help to complete HAIER_AC160.
https://docs.google.com/spreadsheets/d/1CTaIPGhBXbdVwxZhjvJ-vhTJmfE1F0z6EgXF-zlGvNg/edit?usp=sharing
“Health”=“Econo” , "Clean" = "Filter" .

crankyoldgit, 2022-08-11:
I've updated the branch. How is it now? Also created PR #1852

closed 2022-08-17 by crankyoldgit

crankyoldgit, 2022-09-16:
FYI, the changes mentioned above have now been included in the new [v2.8.3 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.3) of the library.

GeorgHelmut, 2026-08-27:
Dear all, I just decoded the HAIER176 protokoll. Here it is.

[HAIER176_analyse.odt](https://github.com/user-attachments/files/31524723/HAIER176_analyse.odt)

commit 48f2db23 (2022-08-17): HAIER_AC160: Experimental detail support. (#1852)

* This is basically a cut & paste of the `IRHaierAC176` class.
  - Without some features. e.g. SwingH
  - Modified values for SwingV
* Rough unit test coverage for new code.
  - Needs more real data to confirm etc.
* Add in support required for `IRac` class to work with this protocol.
* Add Light support
  - Report the button press code.
  - Make a guess at how the setting works.
    - i.e. The button code toggles the light on and off.
  - Adjust `toCommon()` to handle previous state.
  - Handle the light toggles appropriately.
  - Add a unit  [... cut]