crankyoldgit/IRremoteESP8266 #1812: I need Support for Remote AR-715 for Climate-Butler AC (looks the same as Trotec PAC 4600)

benjy3gg, 2022-05-22:
Hi, i want to create support for the Remote AR-715 for my new AC.
It's reporting UNKNOWN using IRRecvdumpV2.

```
Protocol  : UNKNOWN
00:01:16.218 -> Code      : 0xE6CA5369 (55 Bits)
00:01:16.218 -> uint16_t rawData[109] = {554, 3512,  558, 1488,  580, 496,  522, 494,  546, 558,  552, 494,  548, 496,  546, 500,  548, 558,  544, 504,  544, 502,  540, 506,  538, 570,  514, 530,  514, 532,  512, 534,  512, 592,  512, 536,  510, 532,  510, 536,  510, 598,  510, 536,  506, 536,  514, 534,  510, 594,  514, 534,  510, 536,  534, 510,  510, 598,  514, 534,  510, 536,  508, 536,  534, 572,  534, 500,  512, 1536,  512, 526,  510, 1588,  512, 1536,  510, 538,  512, 532,  510, 588,  510, 1536,  514, 532,  510, 536,  534, 570,  512, 534,  510, 536,  514, 528,  540, 568,  540, 506,  512, 524,  510, 1534,  516, 532,  536, 3396,  544};  // UNKNOWN E6CA5369 POWER OFF
```
```
Protocol  : UNKNOWN
00:02:09.240 -> Code      : 0xA008E5A4 (55 Bits)
00:02:09.240 -> uint16_t rawData[109] = {536, 3530,  542, 1532,  544, 502,  546, 500,  546, 560,  550, 496,  548, 488,  552, 1488,  548, 1560,  550, 498,  548, 496,  550, 496,  548, 560,  546, 498,  550, 500,  546, 496,  548, 558,  548, 498,  550, 496,  548, 498,  548, 560,  548, 496,  548, 496,  550, 496,  548, 558,  548, 498,  548, 498,  548, 496,  552, 558,  548, 496,  550, 496,  550, 494,  548, 558,  550, 488,  546, 1498,  548, 490,  550, 1550,  522, 1524,  524, 522,  548, 496,  548, 552,  548, 1498,  548, 496,  548, 498,  548, 562,  548, 496,  548, 498,  550, 496,  550, 548,  548, 1492,  548, 1488,  526, 1514,  550, 1496,  548, 3358,  580};  // UNKNOWN A008E5A4 POWER ON
```

Tried to follow How to add new IR Remote in the wiki and implemented the TestExample Stuff, but i still get UNKNOWN when i use the remote.

Help would be appreciated.

NiKiZe, 2022-05-22:
You mention remote AR-715
Do you have brand and also model of the AC unit?

benjy3gg, 2022-05-22:
@NiKiZe hey. sry i only posted it in the title.
The Brand is Climate-Butler and the model is RCS-SD43UWI.
But it looks the same as this one https://de.trotec.com/produkte-services/maschinen-homecomfort/klimatisierung/komfort-klimageraete-pac-serie/pac-4600/ and has the same remote

crankyoldgit, 2022-05-22:
> Tried to follow How to add new IR Remote in the wiki and implemented the TestExample Stuff, but i still get UNKNOWN when i use the remote.

Can you link to your code? We might be able to spot what is wrong.

benjy3gg, 2022-05-22:
https://github.com/crankyoldgit/IRremoteESP8266/compare/master...benjy3gg:master

crankyoldgit, 2022-05-22:
Thanks, at a cursory glance, the code looks like it should work. I'll have a tinker later

crankyoldgit, 2022-05-22:
Did you try setting `DEBUG` on and have a look at the output?
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/4adcc7bcdeb922e28de9ee8937f65ccd18be3269/src/IRremoteESP8266.h#L1382-L1383

benjy3gg, 2022-05-22:
no, will try that now. 
btw: if i send the captured u_int array with sendRaw the AC responds to the code.

benjy3gg, 2022-05-22:
01:33:11.653 -> Attempting TestExample decode
01:33:11.653 -> Matching SPACE 550 vs 3492 - 50. Matching: 2581 <= 550 <= 4303

crankyoldgit, 2022-05-22:
Try changing:
```c++
  if (!matchSpace(results->rawbuf[offset++], kTestExampleHdrSpace))
    return false;
```
to:
```c++
  if (!matchMark(results->rawbuf[offset++], kTestExampleBitMark))
    return false;
  if (!matchSpace(results->rawbuf[offset++], kTestExampleHdrSpace))
    return false;
```

benjy3gg, 2022-05-22:
ah some commands are recognized now!

benjy3gg, 2022-05-22:
looks like most of them are working. so what are the next steps now?

crankyoldgit, 2022-05-23:
Okay.
1. Stop using `TestExample` etc. Use `CLIMATEBUTLER`/`ClimateButler` etc instead for all occurrences.
2. Rename `ir_Climate.cpp` to `ir_ClimateButler.cpp`.
3. Change:
```c++
  for (uint16_t r = 0; r <= repeat; r++) {
    uint64_t send_data = data;
    space(kTestExampleHdrSpace);
    // Data Section #1
    // e.g. data = 0x830000005880F, nbits = 52
    sendData(kTestExampleBitMark, kTestExampleOneSpace, kTestExampleBitMark, kTestExampleZeroSpace, send_data, 52, true);
    send_data >>= 52;
    // Footer
    mark(kTestExampleBitMark);
    space(kTestExampleHdrSpace);
    space(kDefaultMessageGap);  // A 100% made up guess of the gap between messages.
  }
```
to:
```c++
  for (uint16_t r = 0; r <= repeat; r++) {
    // Header + Data
    sendGeneric(kClimateButlerBitMark, kClimateButlerHdrSpace,
                kClimateButlerBitMark, kClimateButlerOneSpace,
                kClimateButlerBitMark, kClimateButlerZeroSpace,
                kClimateButlerBitMark, kClimateButlerHdrSpace,
                data, nbits, kClimateButlerFreq, true, 0, kDutyDefault);
    // Footer
    mark(kClimateButlerBitMark);
    space(kDefaultMessageGap);  // A 100% made up guess of the gap between messages.
  }
```
**Note:** We will do something similar for the `decodeBlah()` routine later.
4. Change:
```c++
//   Brand: TestExample,  Model: TODO add device and remote
```
to:
```c++
//   Brand: ClimateButler,  Model: AR-715 remote
//   Brand: ClimateButler,  Model: RCS-SD43UWI A/C
```
5. When you're sure you've got it right (i.e. sending and decoding works), send/create a PR to merge it in to the library. 
6. Now that it works, start following: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
7. One of the first things is to capture an entire temperature range (lowest to highest) to try to work out the bit ordering. i.e. _LSBF_ or _MSBF_.

That will do you for now.

benjy3gg, 2022-05-23:
I changed my code according to your instructions. Decoding Stoll works, but if I now try to send a command with "sendClimateButler(0x.....)" Nothing gets send.
(There is a small led on my ir transmitter that flashes up when sending).

benjy3gg, 2022-05-23:
Okay, works now. Had some error on my code and it wasn't even sending the ir command.
Will open a PR later.

benjy3gg, 2022-05-24:
Started to analyse the IR Codes here:
https://docs.google.com/spreadsheets/d/1cy5-5tNJvlfgpQws0UVYX0jEZs4xoefBOWScEjmtoX0/edit#gid=0

crankyoldgit, 2022-05-25:
> Started to analyse the IR Codes here: https://docs.google.com/spreadsheets/d/1cy5-5tNJvlfgpQws0UVYX0jEZs4xoefBOWScEjmtoX0/edit#gid=0

@benjy3gg You forgot to make it publicly readable.

benjy3gg, 2022-05-26:
@crankyoldgit you should have access now

benjy3gg, 2022-05-28:
i got most of the code figured out. struggling with the checksum right now. help would be appreciated.

crankyoldgit, 2022-05-29:
Looking at the temperature range, it looks like we've got the correct bit-ordering. So, nothing to change. Please create a PR.

As for the checksum. It looks like the last nibble (4 bits) is the check value. A quick look at it makes me think it's an inversion of all the previous nibbles added together.
e,g.
`0x830000007914F`: `0x8 + 0x3 + 0x0 + 0x0 + 0x0 + 0x0 + 0x0 + 0x0 + 0x7 + 0x9 + 0x1 + 0x4 = 0x20` & invert the last nibble. ie. 0x2**0**: `0x0` -> `0xF` gets you the check value (or `0xF - 0x0 = 0xF` if you prefer)

benjy3gg, 2022-05-30:
@crankyoldgit checksum seems to be right :)
Also opened the PR:
https://github.com/crankyoldgit/IRremoteESP8266/pull/1814

benjy3gg, 2022-05-30:
@crankyoldgit also got another question:
https://docs.google.com/spreadsheets/d/1cy5-5tNJvlfgpQws0UVYX0jEZs4xoefBOWScEjmtoX0/edit#gid=339282704 in the "Analysis Timer" sheet, there is something strange (orange marked cells). All other cells correspond to the timer, but the marked ones have some sort of other formatting.

crankyoldgit, 2022-09-16:
FYI, the changes mentioned above have now been included in the new [v2.8.3 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.3) of the library.

sorryusernameisalreadytaken, 2024-08-02:
Wow, really cool foundings here. Good job!

The Remote AR-715 should be for the AC's (I guess there will be more, it's just branding thing):
- Remko RKL495
- Remko RKL494
- Remko RKL491
- Remko RKL490
- TROTEC PAC 4600
- Novamatic CL 990
- Novamatic CL 1590
- Rexair C15000N
- Fischer ClimaButler (Clima|Butler) RCS-SD43UAI
- Fischer ClimaButler (Clima|Butler) RCS-SD43UWI
- freecom RCS-SD43UAI
- freecom RCS-SD43UWI
- Prem-i-air 15000 BTU

I try to add my Remko RKL490 DC also to esphome:
https://github.com/esphome/feature-requests/issues/2826