crankyoldgit/IRremoteESP8266 #1818: Midea空调型号问题

zxnliu, 2022-06-04:
How do you do?

Does this library only control the domestic air conditioner? This model of air conditioner is also owned by Midea.
The air conditioner on my side is the commercial central air conditioner of Clavio. I tried to control it by using the library, but I failed.

Wanted to ask how to control it.

Greetings.

crankyoldgit, 2022-06-04:
Try capturing the IR messages from the original remote using the `IRRecvDumpV2` example codes.

zxnliu, 2022-06-05:
The capture was successful, but I couldn't send out the transmitting tube. I found the circuit online and checked whether the ESP32 I used had a specific pin for transmitting.
Or there is no transmitter circuit recommended.

Thank you

Greetings.

zxnliu, 2022-06-05:
Please forgive me if youdao translation is wrong

zxnliu, 2022-06-06:
Problem solved

zxnliu, 2022-06-06:
Thank you for your valuable advice