crankyoldgit/IRremoteESP8266 #1821: Energy Saver mode on Gree protocol

yeahme49, 2022-06-14:
I have a Soleus Air window a/c that uses the Gree protocol (remote even says Gree on it, remote model YX1F5F) and everything works great except one thing.

My a/c has an additional mode when cycling through the mode button, Energy Saver.

Is it possible to add this additional mode?

```
Protocol  : GREE
Code      : 0x0809605800200050 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 0 (Auto), Temp: 77F, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Manual, Swing(V): 0 (Last), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[139] = {8994, 4486,  654, 548,  656, 550,  656, 550,  656, 1652,  654, 548,  656, 550,  656, 548,  658, 548,  656, 1652,  656, 550,  656, 548,  658, 1650,  654, 552,  654, 550,  656, 550,  658, 548,  656, 548,  658, 548,  656, 550,  658, 548,  656, 550,  656, 1652,  656, 1650,  656, 548,  658, 548,  656, 550,  656, 548,  656, 1652,  656, 1650,  656, 548,  658, 1650,  656, 548,  656, 548,  656, 1652,  656, 550,  654, 19986,  654, 550,  656, 550,  654, 550,  656, 550,  656, 550,  654, 550,  654, 550,  656, 550,  656, 550,  654, 550,  654, 552,  654, 550,  654, 550,  654, 1652,  654, 550,  654, 552,  654, 550,  656, 552,  654, 552,  654, 550,  656, 550,  654, 552,  654, 550,  654, 552,  654, 550,  654, 552,  654, 552,  654, 550,  654, 1654,  654, 550,  654, 1654,  652, 552,  652};  // GREE
uint8_t state[8] = {0x08, 0x09, 0x60, 0x58, 0x00, 0x20, 0x00, 0x50};

Timestamp : 003721.230
Library   : v2.8.2

Protocol  : GREE
Code      : 0x0909605800200060 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 77F, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Manual, Swing(V): 0 (Last), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[139] = {8996, 4484,  656, 1650,  656, 548,  658, 548,  656, 1670,  636, 548,  656, 550,  656, 548,  656, 548,  656, 1652,  656, 550,  656, 550,  656, 1652,  654, 550,  656, 550,  656, 550,  656, 548,  656, 550,  656, 550,  658, 548,  656, 548,  658, 548,  658, 1670,  636, 1652,  654, 548,  656, 548,  656, 548,  658, 548,  658, 1652,  654, 1652,  654, 550,  656, 1652,  652, 550,  658, 548,  656, 1652,  656, 550,  654, 20004,  636, 548,  656, 550,  658, 548,  656, 548,  656, 550,  654, 550,  656, 548,  658, 548,  656, 550,  656, 550,  656, 550,  656, 550,  656, 548,  656, 1652,  656, 548,  656, 550,  656, 550,  656, 550,  656, 548,  656, 550,  656, 550,  658, 548,  656, 550,  656, 548,  658, 550,  656, 548,  656, 548,  656, 548,  658, 548,  656, 1652,  656, 1650,  656, 550,  656};  // GREE
uint8_t state[8] = {0x09, 0x09, 0x60, 0x58, 0x00, 0x20, 0x00, 0x60};

Timestamp : 003722.400
Library   : v2.8.2

Protocol  : GREE
Code      : 0x1D096058002000A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 5 (UNKNOWN), Temp: 77F, Fan: 1 (Low), Turbo: Off, Econo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Manual, Swing(V): 0 (Last), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[139] = {9020, 4484,  658, 1648,  658, 548,  658, 1670,  636, 1672,  634, 1672,  636, 548,  658, 548,  656, 548,  656, 1650,  656, 550,  656, 548,  656, 1652,  656, 548,  656, 548,  658, 548,  656, 548,  656, 550,  656, 550,  656, 550,  656, 548,  656, 548,  656, 1652,  656, 1650,  654, 550,  656, 550,  656, 550,  656, 548,  656, 1650,  656, 1650,  656, 550,  656, 1650,  656, 548,  656, 550,  656, 1650,  656, 550,  656, 19982,  656, 550,  656, 550,  656, 548,  656, 550,  656, 550,  656, 550,  656, 548,  656, 550,  656, 548,  656, 548,  656, 548,  658, 548,  656, 550,  654, 1652,  656, 548,  656, 548,  656, 550,  656, 548,  656, 550,  656, 550,  656, 550,  656, 548,  658, 548,  656, 548,  658, 548,  656, 550,  654, 550,  656, 550,  654, 550,  656, 1652,  656, 550,  656, 1652,  654};  // GREE
uint8_t state[8] = {0x1D, 0x09, 0x60, 0x58, 0x00, 0x20, 0x00, 0xA0};
```

This shows me pressing mode button to switch to Auto, then Cool, then Energy Saver. Let me know if there is anything else I would need to provide.

Thanks

crankyoldgit, 2022-06-15:
We can probably as it easy enough. What does that nice do other then save energy etc. Eg. Is it cold, hot, auto etc in addition to energy saver?

crankyoldgit, 2022-06-15:
@yeahme49 Can you please download and try out the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/YX1F5F / PR #1823 and let me know how it goes.

I think it should mostly work.

Can you please supply the exact model number of the A/C unit too?

yeahme49, 2022-06-15:
> t/IRremoteESP8266/tree/YX1F5F / PR #1823

I will test when I get home.

AC model is GM-WAC-12ESE-C.

The energy saver mode turns the fan off when the compressor is not running.

yeahme49, 2022-06-15:
Appears to be working. I can set that mode by setting model to 3 and econo to on. If I receive the command it shows econo on as well. Thanks!

commit f706f99f (2022-06-16): GREE: Add model support for `YX1FSF`/Soleus Air Windown A/C (#1823)

* Allow detecting/setting new model.
* Allow setting Econo(Energy Saver) operation mode.
* Update supported model info.
* Add unit tests to cover new changes.

Fixes #1821
closed 2022-06-16 by crankyoldgit