crankyoldgit/IRremoteESP8266 #1824: Midea (Coolix) AC timer support

universal-iot, 2022-06-19:
I have a Midea AC with RG56CMI-WP2 remote. 
I am successfully able to perform the following operations using **IRCoolixAC**:
- on()
- off()
- setMode()
- setFan()
- setTemp()
- setTurbo()
- setSwing()
- setLed()
- setSwingVStep()

I was not able to get the setSleep() to work, I have not tested the other methods yet.

I read in the issues that there is some [COOLIX48 support added for timer](https://github.com/crankyoldgit/IRremoteESP8266/pull/1697) and I can confirm that the remote that I have also uses COOLIX48 for timers (both ON and OFF). 

I also noted that when the OFF timer is set, all controls for changing the temp/mode/fan sent via the remote starts using COOLIX48.

I wanted to know how this can be done using this library as I am not sure how to use the COOLIX48 methods. If I hard code the value sent by the remote using sendCoolix48() I am able to get the AC timer set. But any other call to change the temp (e.g. setTemp()) removes the timer.

I used the [SmartIRRepeater ](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/SmartIRRepeater) sample to understand that the remote uses COOLIX48 when using timer.

universal-iot, 2022-06-19:
I have added the captures for both when the off timer is on and off. Hope this helps.

IRrecvDumpV3 captures:

----------------------------------------------------------------------------------------------------------------------------
Mode: Cool
Fan: Auto
Temp: 21C
Off Timer: 0.5 hours
--------
Protocol  : COOLIX48
Code      : 0xB24DA15E607F (48 Bits)
uint16_t rawData[199] = {4444, 4374,  616, 1588,  614, 592,  614, 1592,  614, 1590,  592, 616,  614, 592,  614, 1592,  614, 592,  614, 594,  614, 1570,  634, 594,  590, 616,  590, 1614,  614, 1568,  636, 594,  614, 1590,  616, 1590,  614, 592,  614, 1592,  612, 594,  614, 592,  614, 594,  612, 594,  590, 1614,  590, 616,  614, 1590,  590, 618,  614, 1590,  616, 1590,  614, 1590,  590, 1614,  614, 572,  636, 592,  590, 1614,  590, 1614,  614, 592,  616, 590,  614, 594,  614, 592,  614, 570,  612, 616,  592, 1614,  614, 1590,  614, 1592,  616, 1588,  614, 1590,  616, 1588,  616, 1590,  616, 4372,  4422, 4374,  614, 1592,  612, 592,  592, 1614,  614, 1592,  590, 616,  592, 616,  612, 1592,  612, 594,  590, 616,  614, 1590,  590, 594,  636, 592,  590, 1614,  590, 1614,  590, 616,  614, 1590,  614, 1590,  614, 570,  612, 1592,  634, 594,  614, 570,  638, 592,  612, 594,  612, 1592,  614, 592,  616, 1588,  614, 594,  614, 1590,  592, 1612,  614, 1590,  614, 1568,  636, 570,  614, 616,  612, 1592,  614, 1590,  614, 570,  636, 592,  614, 594,  614, 570,  636, 594,  612, 596,  614, 1592,  590, 1614,  612, 1590,  590, 1614,  592, 1612,  614, 1590,  614, 1568,  638};  // COOLIX48 B24DA15E607F
uint64_t data = 0xB24DA15E607F;
----------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------
Mode: Cool
Fan: Auto
Temp: 22C
Off Timer: 0.5 hours
--------
Protocol  : COOLIX48
Code      : 0xB24DA15E707F (48 Bits)
uint16_t rawData[199] = {4446, 4372,  616, 1588,  590, 616,  616, 1588,  614, 1590,  614, 592,  616, 590,  616, 1588,  616, 592,  616, 592,  614, 1590,  616, 590,  592, 616,  616, 1588,  616, 1588,  592, 616,  614, 1590,  590, 1612,  616, 592,  614, 1590,  618, 590,  618, 590,  642, 566,  590, 616,  616, 1590,  614, 592,  616, 1588,  616, 590,  618, 1586,  592, 1614,  616, 1590,  616, 1588,  616, 590,  616, 592,  616, 1588,  616, 1588,  616, 1588,  616, 590,  616, 590,  590, 616,  644, 564,  618, 588,  616, 1588,  616, 1586,  618, 1588,  614, 1590,  616, 1590,  590, 1614,  616, 1588,  616, 4350,  4420, 4398,  614, 1590,  614, 592,  616, 1590,  616, 1588,  616, 592,  616, 592,  590, 1614,  614, 592,  614, 570,  638, 1590,  616, 590,  616, 592,  590, 1592,  638, 1590,  590, 616,  590, 1614,  616, 1566,  636, 592,  616, 1590,  614, 592,  614, 592,  616, 592,  616, 590,  618, 1588,  616, 592,  618, 1588,  616, 590,  618, 1588,  616, 1588,  616, 1588,  592, 1614,  618, 568,  636, 592,  618, 1586,  590, 1614,  590, 1614,  616, 590,  616, 592,  616, 590,  616, 592,  590, 616,  616, 1588,  616, 1590,  614, 1590,  614, 1590,  614, 1590,  616, 1588,  618, 1588,  616};  // COOLIX48 B24DA15E707F
uint64_t data = 0xB24DA15E707F;
----------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------
Mode: Cool
Fan: Auto
Temp: 21C
Off Timer: 1 hours
--------
Protocol  : COOLIX48
Code      : 0xB24DA35C607F (48 Bits)
uint16_t rawData[199] = {4448, 4368,  594, 1612,  592, 616,  592, 1590,  642, 1584,  594, 614,  592, 594,  614, 1612,  592, 614,  592, 592,  614, 1614,  592, 600,  608, 614,  592, 1592,  640, 1586,  592, 614,  592, 1612,  592, 1612,  592, 614,  592, 1612,  592, 592,  614, 616,  618, 588,  592, 1614,  590, 1592,  614, 614,  592, 1612,  592, 616,  590, 1614,  590, 1614,  590, 1592,  614, 616,  592, 614,  594, 614,  590, 1590,  616, 1612,  592, 616,  592, 616,  590, 616,  592, 616,  592, 614,  592, 616,  618, 1586,  592, 1612,  592, 1612,  618, 1588,  590, 1612,  592, 1614,  590, 1614,  592, 4396,  4396, 4374,  614, 1614,  590, 594,  642, 1584,  620, 1586,  592, 616,  592, 616,  590, 1614,  590, 616,  592, 616,  590, 1614,  592, 616,  592, 616,  590, 1592,  612, 1614,  592, 616,  590, 1614,  592, 1614,  590, 616,  592, 1614,  592, 616,  592, 616,  616, 590,  590, 1614,  618, 1586,  592, 616,  620, 1586,  592, 614,  592, 1614,  618, 1586,  592, 1592,  614, 614,  592, 616,  592, 616,  590, 1592,  612, 1614,  592, 616,  618, 588,  592, 616,  616, 590,  592, 616,  590, 616,  618, 1564,  614, 1614,  592, 1588,  614, 1614,  616, 1588,  590, 1614,  590, 1614,  590};  // COOLIX48 B24DA35C607F
uint64_t data = 0xB24DA35C607F;
----------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------
Mode: Cool
Fan: Auto
Temp: 22C
Off Timer: 1 hours
--------
Protocol  : COOLIX48
Code      : 0xB24DA35C707F (48 Bits)
uint16_t rawData[199] = {4424, 4398,  590, 1614,  592, 614,  592, 1612,  592, 1612,  592, 616,  592, 614,  592, 1614,  592, 616,  592, 616,  590, 1614,  592, 592,  614, 616,  592, 1612,  592, 1614,  592, 616,  590, 1614,  592, 1612,  592, 616,  592, 1612,  592, 614,  590, 594,  614, 616,  590, 1590,  616, 1612,  592, 616,  590, 1614,  5 [... cut ...] 86,  616, 1566,  638, 1586,  618, 1586,  616, 1588,  614, 592,  614, 1588,  592, 616,  616, 568,  630, 598,  616, 590,  590, 616,  618, 590,  642, 542,  640, 1564,  638, 1588,  618, 1586,  618, 590,  618, 568,  636, 592,  616, 590,  616, 1588,  590, 616,  616, 590,  590, 616,  616, 1588,  616, 1588,  616, 1588,  618, 1566,  638};  // COOLIX B2BF70
uint64_t data = 0xB2BF70;
----------------------------------------------------------------------------------------------------------------------------

universal-iot, 2022-06-19:
I have managed to build a logic to get the timer values for the following settings:
Mode: Cool
Fan: Auto
Temp: 27C

Note: The below timer code only works for the above settings.

`uint64_t GetOffTimer(uint16_t minutes) {
  if(minutes <= 0) {
    return 0xB2BF90;
  }
  uint64_t timerValue = 0xB24DA15E90FF;
  if(minutes <= 480) {
      timerValue += ((minutes / 30) - 1) * 0x1fe0000;
  }
  else if (minutes <= 960){
      timerValue = 0xB24DA15E91FF;
      timerValue += (((minutes - 480) / 30) - 1) * 0x1fe0000;
  }
  else {
      timerValue = 0xB24DA15E92FF;
      timerValue += (((minutes - 960) / 30) - 1) * 0x1fe0000;
  }
  return timerValue;
}

uint64_t GetOnTimer(uint16_t minutes) {
  if(minutes <= 0) {
    return 0xB27BE0;
  }
  uint64_t timerValue = 0xB24DBF409381;
  timerValue += ((minutes / 30) - 1) * 2;
  return timerValue;
}`

NiKiZe, 2022-06-19:
This is captures from when you are using the original remote?

Timers is not something that this lib cares to much about, since you are much better of have the logic in something with more smarts than the AC.

crankyoldgit, 2022-06-19:
@NiKiZe Is correct. Adding timer support is at the bottom of the priority list, because as he pointed out, there are other/better ways to do it, and with finer control. e.g. Via home assistant/Google Assistant/Siri/Homekit etc. etc.

But we will get to it eventually.

universal-iot, 2022-06-19:
> This is captures from when you are using the original remote?
> 
> Timers is not something that this lib cares to much about, since you are much better of have the logic in something with more smarts than the AC.

Yes, the captures are using the original remote.

universal-iot, 2022-06-19:
Thanks a lot @NiKiZe and @crankyoldgit for indicating that timers are currently not supported for Coolix. I thought it was added along with support for Coolix48 protocol.

I am closing this issue as my plan is to use it with home assistant.

Thanks for creating such an amazing library. It was a breeze to get it working with my AC with the help of the examples provided. :)

closed 2022-06-19 by universal-iot

NiKiZe, 2022-06-19:
I'd put them in the, "not so tested and might need more work" ;)