crankyoldgit/IRremoteESP8266 #1825: MISSING command for Air conditioning MITSUBISHI MSZ-FD25. How send the correct IRcommand (HEX) I captured?

lumanga, 2022-06-19:
Dear Sirs,
I have Just captured my Air conditioning MITSUBISHI MSZ-FD25 IR remote command:
1) From Tasmota IRreceiver, the original AC Remote control, give this results (The same as received from Arduino IR-receiver pogram)
`tele/NodeMCU17/RESULT = {"IrReceived":{"Protocol":"MITSUBISHI_AC","Bits":144,"Data":"0x23CB2601002058080640370000401800006A","Repeat":0,"IRHVAC":{"Vendor":"MITSUBISHI_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Auto","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

2) Reading the  MITSUBISHI MSZ-FD25 IR remote command with "**RAW-ir-decoder" with arduino** I have obtained this informations
`Hh110001001101001101100100100000000000000000000100000110100001000001100000000000100100011000000000000000000000000100001000000000000000000010110011
Bytes:
00:  1100|0100 | 23 | 00100011 
01:  1101|0011 | CB | 11001011 
02:  0110|0100 | 26 | 00100110 
03:  1000|0000 | 01 | 00000001 
04:  0000|0000 | 00 | 00000000 
05:  0000|0100 | 20 | 00100000 
06:  0001|1010 | 58 | 01011000 
07:  0001|0000 | 08 | 00001000 
08:  0110|0000 | 06 | 00000110 
09:  0000|0010 | 40 | 01000000 
10:  0100|0110 | 62 | 01100010 
11:  0000|0000 | 00 | 00000000 
12:  0000|0000 | 00 | 00000000 
13:  0000|0001 | 80 | 10000000 
14:  0000|1000 | 10 | 00010000 
15:  0000|0000 | 00 | 00000000 
16:  0000|0000 | 00 | 00000000 
17:  1011|0011 | CD | 11001101 
18:  1100|0100 | 23 | 00100011 
19:  1101|0011 | CB | 11001011 
20:  0110|0100 | 26 | 00100110 
21:  1000|0000 | 01 | 00000001 
22:  0000|0000 | 00 | 00000000 
23:  0000|0100 | 20 | 00100000 
24:  0001|1010 | 58 | 01011000 
25:  0001|0000 | 08 | 00001000 
26:  0110|0000 | 06 | 00000110 
27:  0000|0010 | 40 | 01000000 
28:  0100|0110 | 62 | 01100010 
29:  0000|0000 | 00 | 00000000 
30:  0000|0000 | 00 | 00000000 
31:  0000|0001 | 80 | 10000000 
32:  0000|1000 | 10 | 00010000 
33:  0000|0000 | 00 | 00000000 
34:  0000|0000 | 00 | 00000000 
35:  1011|0011 | CD | 11001101 
23,CB,26,01,00,20,58,08,06,40,62,00,00,80,10,00,00,CD,23,CB,26,01,00,20,58,08,06,40,62,00,00,80,10,00,00,CD
Timings (in us): 
PAUSE SPACE:  17088
HEADER MARK:  3434
HEADER SPACE: 1764
BIT MARK:     362
ZERO SPACE:   444
ONE SPACE:    1322
Decoding known protocols...Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE COOL
I-See: ON
Clean: OFF
Plasma: OFF
Temperature: 24
FAN AUTO
VANE: AUTO1?
WIDE VANE: AREA
AREA MODE: AUTO
INSTALL POSITION: CENTER
################# End `

Then I tried to SEND the IR command to the IR-Led transmitter connected to another NodeMCU with Tasmota IR-firmware:
A) If I send the command to NodeMCU with tasmota, using the **IRvac** commands present into tasmota guide :
example: `IRhvac  {"Vendor":"MITSUBISHI_AC","Power":"On","Mode":"Cool","Temp":24....... }`
The "RAW-ir-decoder" Arduino, gives me the right response, and the AirConditioner WORKS, BUT **some commands are not available in the library and are missed:  VANE, AREA MODE, INSTALL POSITION**
I'm asking if I CAN and HOW i can send the complete and correct command in **HEX ?** (The hex is the code 23,CB,26,01,00,20,58,08,06,40,62,00,00,80,10,00,00,CD,23,CB,26,01,00,20,58,08,06,40,62,00,00,80,10,00,00,CD).

B) If I try to send RAW:
`IRSend raw,17100,3410,1770,350,444,1320,110001001101001101100100100000000000000000000100000110100001000001100000000000100100011000000000000000000000000100001000000000000000000010110011`

Tasmota tell me it's send correctly: `stat/NodeMCU16/RESULT = {"IRSend":"Done"}`
BUT Arduino IR receiver program,   **doesn't give me any feedback of received IR signal**

C) If I try to send the my Ir remote command HEX values:
`IRSend {"Protocol":"MITSUBISHI2","Bits":144,"Data":0x23CB26010020580806406000004018000093}  or IRSend {"Protocol":"MITSUBISHI_AC","Bits":144,"Data":0x23CB26010020580806406000004018000093}`
Tasmota tell me it's send correctly: `stat/NodeMCU16/RESULT = {"IRSend":"Done"}`
BUT Arduino IR receiver program, give me only Hh00000000000000000000000000000000000..... and all ZERO Hex value!

QUESTIONS:
- Is it possible to send the HEX or the RAW to obtain a correct AC Mitsubishi complete  message? 

- Here:
https://github.com/r45635/HVAC-IR-Control
there are best analysys and a list of all the HEX value for the command for my AC,
I'm able to compile my own Tasmota firmware, There is some way to add the commands not available?
Thanks in advance.

crankyoldgit, 2022-06-19:
> * Is it possible to send the HEX or the RAW to obtain a correct AC Mitsubishi complete  message?

Yes.

Note: We don't (directly) support Tasmota or any other 3rd party code. We only support our library & our code example. e.g. IRMQTTServer.

Tasmota IRHVAC uses the our [IRac API](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html). That is a very generic base-subset of the options supported by the library.
If you want to get full control, you have to use the relevant a/c class in the library. e.g. [IRMitsubishiAC](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRMitsubishiAC.html)

If you want to use the full capabilities, you'll probably need to write the code yourself. Tasmota etal can't have full support for each and every A/C remote/unit. It takes to much space, time, and effort to emulate every remote.

We have some example code for how to use the individual A/C classes. They are all fairly similar.

You can look at the source code (or the API documentation) for the class, or run the `IRrecvDumpV2` etc code from the examples, to see what is supported for your device.

crankyoldgit, 2022-06-20:
FYI, the code you'd want to do to send the "hex" code using our library is:
```c++
uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x58, 0x08, 0x06, 0x40, 0x37, 0x00, 0x00, 0x40, 0x18, 0x00, 0x00, 0x6A};
irsend.sendMitsubishiAC(state, length);
```

You'll need to add that to some code like: [IRsendDemo.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino)

See: [sendMitsubishiAC() documentation](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a3600527a82f9f22387c9f16ae51fb06f)

As for how to send it via Tasmota, please ask them.

crankyoldgit, 2022-06-24:
Have you read the Tasmota IR documentation? If not, you should. ie. https://tasmota.github.io/docs/Tasmota-IR/

> `IRSend {"Protocol":"MITSUBISHI2","Bits":144,"Data":0x23CB26010020580806406000004018000093} or IRSend {"Protocol":"MITSUBISHI_AC","Bits":144,"Data":0x23CB26010020580806406000004018000093}`

These won't work according to the documentation. It says the max bit size that Tasmota handles is 64 bits.

I suggest you follow their documentation for how to receive and send raw codes with Tasmota.

As I suspected, this is not something we can fix. It's a Tasmota issue. You need to talk to them.

closed 2022-06-24 by crankyoldgit