crankyoldgit/IRremoteESP8266 #1826: Adding support for Sanyo AC 154bit protocol

MinePlugins, 2022-06-19:
Hello, I attemp to decode the IR of my remote for the A/C with Sanyo RCS-4MHVPIS4EE remote,
I created a [Google Sheet](https://docs.google.com/spreadsheets/d/1GH3vCZx7HE-GGwTOEJhruUdv4eUDJE1y-XbemlOxT-c/edit?usp=sharing) like recommended and user Auto_analyse_raw_data.py:
`Found 307 timing entries.
Potential Mark Candidates:
[3360, 440]
Potential Space Candidates:
[1660, 1274, 434]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3360
kHdrSpace  = 1660
kBitMark   = 419
kOneSpace  = 1261
kZeroSpace = 420

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+00000010000000000010100000000001011101101000000101000100101101111100000011001101001101000001111011000000000000000000100000000000000000000000000011101110
  Bits: 152
  Hex:  0x02002801768144B7C0CD341EC00008000000EE (MSB first)
        0x77000000100003782CB303ED22816E80140040 (LSB first)
  Dec:  44615103636278984136397141307252716643287278 (MSB first)
        2653788678708221178456034549947135807186010176 (LSB first)
  Bin:  0b00000010000000000010100000000001011101101000000101000100101101111100000011001101001101000001111011000000000000000000100000000000000000000000000011101110 (MSB first)
        0b01110111000000000000000000000000000100000000000000000011011110000010110010110011000000111110110100100010100000010110111010000000000101000000000001000000 (LSB first)

Total Nr. of suspected bits: 152`

Can I have some help or explaination for this protocole

NiKiZe, 2022-06-19:
The faq outlines the steps.

Please post the raw data from Irdumpv2+ with the full temperature range from lowest to highest. This will get us to the first stage where the basic protocol can be implemented with the correct bit ordering.

After that comes the part of understanding each part. (Using decoded data)

(Please excuse the brevity of this post due to writing from phone)

MinePlugins, 2022-06-20:
Hi, no problem.
Here date from 16c to 30c

16c :
`uint16_t rawData[307] = {3294, 1726,  420, 330,  458, 462,  382, 452,  438, 330,  456, 458,  384, 454,  384, 1312,  422, 324,  512, 336,  454, 458,  384, 450,  438, 358,  436, 464,  424, 326,  458, 476,  372, 458,  430, 328, 
 458, 464,  382, 1308,  424, 326,  510, 1264,  424, 268,  520, 460,  380, 460,  436, 324,  462, 400,  436, 474,  372, 456,  430, 342,  452, 450,  388, 446,  442, 1262,  422, 266,  520, 1314,  372, 1306,  424, 1258,  370, 390,  448, 1314,  372, 1310,  426, 308,  522, 338,  454, 470,  370, 454,  438, 330,  456, 468,  370, 456,  384, 464,  384, 1306,  422, 328,  460, 472,  374, 448,  442, 1258,  426, 1256,  426, 268,  520, 464,  382, 460,  430, 328,  508, 1264,  426, 262,  572, 1262,  424, 228,  604, 1262,  372, 1312,  372, 1310,  426, 1256,  422, 1258,  424, 262,  524, 418,  428, 456,  382, 1308,  372, 456,  386, 456,  382, 464,  378, 1308,  424, 360,  436, 454,  430, 
344,  450, 1306,  372, 1310,  424, 326,  510, 338,  452, 456,  384, 456,  436, 328,  510, 1258,  372, 1310,  422, 338,  454, 466,  424, 328,  460, 1310,  372, 1312,  424, 1258,  450, 262,  496, 1310,  372, 1310,  426, 1260,  
424, 260,  526, 442,  442, 1264,  426, 268,  520, 458,  380, 450,  386, 462,  436, 320,  518, 1256,  372, 394,  446, 398,  494, 334,  506, 326,  510, 276,  518, 460,  430, 332,  508, 326,  510, 356,  440, 448,  444, 264,  572, 336,  456, 408,  434, 454,  438, 332,  506, 336,  454, 460,  384, 448,  444, 326,  510, 332,  456, 442,  400, 456,  384, 456,  386, 452,  388, 454,  440, 326,  512, 272,  518, 450,  440, 334,  454, 458,  440, 320,  516, 268,  570, 340,  452, 468,  424, 260,  574, 336,  456, 394,  444, 458,  438, 328,  508, 1258,  370, 1312,  372, 1312,  424, 314,  468, 1312,  450, 1232,  370, 1314,  420, 324,  514};  // UNKNOWN 584FBE80`

17c:
`uint16_t rawData[307] = {3324, 1696,  394, 328,  484, 456,  388, 448,  392, 448,  394, 446,  390, 450,  390, 1314,  394, 386,  428, 444,  394, 456,  418, 378,  430, 452,  390, 440,  394, 450,  392, 460,  384, 444,  394, 446, 
 420, 396,  418, 1312,  370, 392,  448, 1314,  398, 386,  426, 444,  424, 398,  418, 444,  392, 442,  426, 396,  420, 440,  424, 396,  418, 450,  390, 444,  392, 1314,  396, 328,  484, 1314,  370, 1314,  370, 1310,  396, 378,  432, 1312,  370, 1312,  398, 324,  488, 444,  392, 454,  390, 450,  420, 390,  424, 440,  396, 452,  414, 406,  412, 1310,  396, 388,  426, 1312,  400, 398,  418, 1312,  374, 1302,  394, 336,  480, 452,  388, 450,  420, 380,  426, 1312,  396, 392,  422, 1312,  370, 444,  400, 1310,  396, 1288,  394, 1284,  394, 1282,  396, 1288,  380, 368,  464, 448,  418, 386,  426, 1316,  370, 386,  452, 458,  386, 446,  420, 1280,  396, 402,  416, 456,  418, 370,  432, 1312,  370, 1314,  396, 326,  486, 446,  392, 458,  416, 382,  428, 456,  388, 1306,  396, 1290,  370, 446,  424, 390,  422, 446,  392, 1308,  368, 1316,  396, 1286,  396, 378,  432, 1312,  370, 1314,  394, 1288, 
 398, 386,  426, 390,  448, 1316,  396, 332,  482, 444,  394, 450,  390, 452,  390, 450,  390, 1308,  370, 400,  442, 450,  418, 394,  420, 442,  396, 448,  420, 402,  416, 442,  394, 446,  392, 454,  418, 382,  428, 454,  418, 376,  432, 450,  390, 462,  414, 380,  428, 446,  392, 458,  384, 450,  418, 396,  420, 442,  394, 450,  420, 390,  424, 446,  418, 388,  452, 396,  418, 448,  392, 456,  388, 438,  398, 450,  392, 460,  384, 444,  426, 370,  436, 456,  418, 378,  432, 454,  418, 372,  434, 448,  418, 398,  420, 444,  394, 1310,  370, 404,  466, 390,  424, 1310,  396, 1286,  370, 1312,  396, 1288,  396, 382,  428};  // UNKNOWN D6875F4B`

18c:
`uint16_t rawData[307] = {3324, 1700,  388, 398,  418, 454,  390, 388,  474, 384,  452, 410,  410, 446,  394, 1312,  394, 390,  424, 446,  422, 392,  420, 448,  416, 406,  414, 442,  424, 388,  424, 456,  410, 394,  444, 396, 
 420, 454,  388, 1310,  396, 332,  508, 1290,  396, 334,  480, 444,  418, 392,  446, 410,  412, 440,  396, 444,  418, 404,  440, 406,  440, 390,  422, 440,  420, 1286,  394, 406,  414, 1314,  396, 1280,  396, 1286,  396, 330,  484, 1314,  396, 1280,  394, 392,  448, 408,  412, 444,  394, 446,  392, 454,  388, 458,  410, 400,  416, 444,  394, 1308,  396, 342,  478, 388,  450, 1314,  392, 1288,  396, 1284,  396, 392,  420, 454,  412, 398,  442, 392,  422, 1312,  392, 392,  450, 1286,  394, 408,  412, 1312,  396, 1286,  370, 1308,  394, 1290,  396, 1282,  368, 444,  418, 396,  446, 404,  414, 1310,  396, 378,  456, 408,  436, 394,  420, 1312,  370, 444,  420, 344,  498, 390,  422, 1312,  394, 1288,  370, 442,  420, 414,  434, 380,  430, 442,  420, 402,  442, 1288,  396, 1284,  396, 386,  450, 410,  410, 1306,  398, 1288,  394, 1282,  396, 1286,  394, 346,  474, 1312,  398, 1282,  392, 1286,  370, 446,  420, 394,  422, 1306,  394, 398,  446, 406,  412, 452,  390, 440,  420, 390,  448, 1292,  396, 386,  426, 444,  418, 398,  442, 406,  414, 450,  390, 442,  418, 400,  442, 406,  414, 446,  392, 446,  416, 394,  446, 406,  412, 454,  388, 440,  420, 396,  444, 406,  414, 448,  390, 446,  418, 396,  444, 406,  414, 450,  390, 440,  420, 396,  444, 406,  414, 446,  392, 446,  416, 396,  446, 406,  414, 452,  388, 442,  418, 396,  444, 412,  440, 388,  446, 388,  448, 396,  444, 408,  410, 452,  390, 438,  420, 398,  444, 410,  440, 378,  428, 1314,  370, 1310,  396, 1284,  394, 1290,  398, 1278,  394, 398,  442};  // UNKNOWN 333CFEC4`

19c:
[... cut ...]
368, 482,  396, 432,  436, 1244,  370, 474,  368, 470,  370, 470,  408, 444,  358, 474,  368, 1310,  402, 440,  370, 470,  368, 474,  368, 480,  392, 446,  398, 448,  356, 468,  370, 472,  404, 448,  360, 470,  368, 472,  368, 470,  370, 482,  396, 436,  368, 472,  366, 470,  406, 446,  360, 468,  368, 472,  370, 458,  380, 480,  396, 438,  366, 456,  380, 472,  404, 450,  360, 468,  368, 476,  366, 480,  360, 482,  360, 472,  370, 468,  368, 472,  406, 446,  360, 468,  368, 474,  368, 468,  368, 472,  370, 484,  358, 480,  358, 470,  372, 1312,  370, 470,  368, 1312,  368, 414,  430, 1310,  406, 1274,  370, 472,  370};  // UNKNOWN 9CF61D56`

crankyoldgit, 2022-06-22:
Thanks for the temperature sequence. With that, I've determined it is in LSBF bit order.
Please download and test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/SanyoAc152Basic / PR #1828 
It should detect and send the protocol now. Fingers crossed. Please let us know how it goes.

As the bit ordering has changed, you'll need to recapture everything again and do your analysis.
e.g. You're now up to here: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data except that I've worked out the bit ordering for you already.

Btw. Your IR receiver module, circuit, or environment is not very good. The signals you captured above are fairly poor quality. ie. the length of pulses and gaps varies a LOT.

MinePlugins, 2022-06-22:
Hello,

I have updated the first sheets with new value and started to decode the logic

Thanks you :)

crankyoldgit, 2022-06-22:
> I have updated the first sheets with new value and started to decode the logic

So it's working?

Also, can you supply the make & model number of the actual AC unit please.

MinePlugins, 2022-06-23:
Yes ! It's working perfectly

The manufacturer is Sanyo and the model number is : sap-kmrv124ehe

MinePlugins, 2022-06-24:
What can I do now, I have attempted to reverse a little but I thinks I'm blocked on TIMER on remote.

I will attempt to implement this in cpp

MinePlugins, 2022-07-10:
Hello, what can I do now to help for decoding data ?

crankyoldgit, 2022-07-11:
Do you need the Timer functionality? Normally that's a low priority as you can use a computer/service to turn on/off a device when needed. etc.

MinePlugins, 2022-07-12:
Non I don't need timer functionality.

crankyoldgit, 2022-09-16:
FYI, the changes mentioned above have now been included in the new [v2.8.3 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.3) of the library.

martin3000, 2023-08-22:
I do not understand how to use it. There is a class IRSanyoAc with "send()" and a class IRsend with sendSanyoAc152(). How does this belong together? If I make IRSanyoAc.setTemp(26), how to do a sendSanyoAc152()?

crankyoldgit, 2023-08-22:
The `IRSanyoAc` class is for the 72 bit Sanyo A/C protocol (`SANYO_AC`). Not the 152 bit (`SANYO_AC152`) protocol.

There is no "detailed" support yet for the `SANYO_AC152` protocol.

In short, you can't use the `IRSanyoAc` class to generate `sendSanyoAc152()` messages.