crankyoldgit/IRremoteESP8266 #1829: Daikin SetTemp to half degrees

AntonWert, 2022-06-25:
Hello,
great library, works well with my Daikin.
Just one thing: with the origianl remote you can set the temperatures to half degrees. But the temperature parameter is only int. Is there a chance to add the half degrees as well?

crankyoldgit, 2022-06-25:
If you can decode and work out in the protocol how it does half degrees, we can add it.
Some brands/models do support half degrees.
Also, some brands/models _lie_ and say they do half degrees on the remote, but don't actually do half degrees. I can't remember if it was a Daikin one that did that or not.

AntonWert, 2022-06-25:
I used the IRrecvDumpV3 example and recorded 2 signals:

1st is 21,5°C

11:01:42.763 -> Protocol  : UNKNOWN
11:01:42.763 -> Code      : 0xDD44A786 (322 Bits)
11:01:42.808 -> uint16_t rawData[643] = {476, 394,  472, 420,  448, 416,  448, 394,  474, 420,  442, 25108,  3516, 1694,  474, 1260,  474, 420,  448, 416,  448, 394,  474, 1258,  474, 422,  446, 416,  448, 394,  474, 420,  444, 1246,  490, 422,  442, 1260,  474, 1264,  474, 420,  442, 1260,  452, 1286,  474, 1246,  486, 1260,  474, 1246,  490, 420,  444, 422,  446, 1258,  474, 420,  448, 420,  422, 442,  446, 396,  472, 420,  444, 420,  448, 420,  448, 420,  444, 420,  446, 422,  416, 446,  448, 1258,  472, 420,  448, 420,  422, 446,  444, 420,  448, 422,  442, 420,  448, 394,  448, 446,  442, 1260,  452, 442,  448, 1246,  486, 420,  448, 420,  422, 1286,  474, 420,  444, 1260,  452, 446,  442, 1248,  460, 1286,  452, 1284,  474, 1286,  448, 420,  422, 446,  444, 420,  448, 420,  416, 448,  446, 394,  448, 446,  444, 420,  448, 420,  422, 442,  448, 394,  448, 446,  444, 420,  448, 420,  416, 446,  448, 396,  448, 446,  442, 1246,  462, 1284,  452, 446,  444, 420,  448, 420,  418, 446,  448, 420,  422, 446,  444, 420,  422, 446,  442, 422,  448, 420,  416, 446,  448, 420,  422, 446,  444, 420,  448, 420,  418, 450,  444, 420,  418, 450,  418, 446,  420, 446,  418, 450,  418, 446,  422, 1268,  464, 420,  442, 452,  418, 446,  422, 446,  416, 446,  422, 446,  418, 450,  418, 446,  422, 446,  416, 452,  416, 446,  422, 446,  418, 446,  422, 446,  418, 450,  418, 446,  422, 446,  416, 446,  422, 446,  418, 450,  418, 446,  422, 446,  418, 446,  422, 448,  416, 450,  418, 446,  422, 446,  418, 446,  422, 446,  416, 446,  422, 446,  422, 442,  422, 446,  422, 446,  418, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 446,  416, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 446,  422, 446,  418, 446,  422, 446,  422, 1286,  448, 1290,  442, 446,  418, 446,  422, 1272,  460, 446,  422, 446,  418, 35544,  3484, 1708,  460, 1290,  448, 446,  418, 446,  422, 446,  418, 1290,  448, 446,  416, 446,  422, 446,  418, 450,  418, 1272,  460, 446,  422, 1284,  448, 1272,  460, 452,  416, 1284,  448, 1274,  460, 1290,  448, 1284,  448, 1284,  448, 450,  418, 448,  422, 1284,  448, 446,  418, 450,  418, 446,  422, 446,  418, 450,  418, 446,  422, 446,  418, 446,  422, 446,  422, 446,  416, 446,  422, 446,  418, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 448,  422, 446,  418, 446,  422, 1272,  460, 1272,  460, 1290,  448, 446,  418, 446,  422, 1286,  448, 1314,  418, 446,  422, 1290,  442, 446,  422, 1290,  444, 446,  422, 446,  416, 446,  422, 446,  422, 446,  416, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 446,  422, 446,  418, 446,  420, 448,  420, 1286,  448, 446,  416, 1290,  448, 446,  418, 446,  422, 446,  422, 446,  418, 446,  422, 446,  416, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 450,  418, 446,  422, 446,  418, 446,  422, 446,  422, 446,  416, 1316,  418, 1290,  448, 446,  416, 446,  422, 446,  418, 450,  418, 446,  422, 446,  416, 446,  422, 446,  422, 446,  416, 446,  422, 1272,  460, 1290,  444, 450,  416, 448,  420, 448,  416, 446,  422, 446,  416, 452,  416, 446,  422, 446,  418, 450,  418, 446,  422, 446,  416, 446,  422, 448,  416, 450,  418, 446,  422, 446,  418, 450,  416, 1290,  444, 446,  422, 1290,  442, 446,  422, 446,  418, 450,  418, 1290,  442, 1290,  444, 450,  418, 446,  422, 446,  418, 450,  418, 446,  422, 446,  416, 450,  418, 448,  416, 450,  418, 446,  422, 446,  418, 1276,  490, 390,  444, 450,  418, 446,  416, 452,  418, 446,  422, 446,  416, 452,  416, 1290,  474, 394,  474, 390,  478, 1254,  478, 390,  478};  // UNKNOWN DD44A786

2nd  is 22,0°C

11:01:48.850 -> Timestamp : 000022.580
11:01:48.850 -> Library   : v2.8.2
11:01:48.850 -> 
11:01:48.850 -> Protocol  : UNKNOWN
11:01:48.850 -> Code      : 0xFA41F15A (322 Bits)
11:01:48.897 -> uint16_t rawData[643] = {478, 390,  478, 390,  472, 422,  448, 416,  452, 392,  472, 25106,  3514, 1690,  478, 1260,  474, 394,  474, 390,  478, 416,  448, 1260,  474, 394,  472, 420,  448, 416,  448, 394,  474, 1260,  474, 420,  446, 1260,  474, 1260,  474, 394,  474, 1258,  474, 1260,  472, 1260,  478, 1256,  478, 1286,  446, 420,  448, 390,  474, 1288,  448, 416,  448, 394,  474, 390,  478, 416,  448, 422,  446, 390,  474, 394,  474, 420,  446, 416,  448, 394,  474, 390,  478, 1284,  448, 390,  472, 394,  474, 420,  448, 416,  448, 394,  474, 390,  476, 416,  448, 420,  448, 1260,  472, 420,  448, 1242,  490, 420,  448, 416,  446, 1260,  474, 394,  474, 1260,  474, 394,  474, 1258,  474, 1258,  474, 1262,  474, 1254,  478, 420,  448, 392,  472, 394,  472, 420,  448, 416,  448, 394,  474, 390,  474, 420,  448, 420,  448, 416,  446, 396,  472, 420,  448, 416,  448, 422,  446, 390,  474, 420,  448, 420,  448, 1242,  490, 1260,  474, 420,  446, 390,  474, 394,  474, 420,  448, 416,  446, 396,  472, 420,  448, 416,  448, 420,  448, 390,  472, 420,  448, 420,  448, 416,  448, 394,  474, 422,  442, 420,  448, 420,  448, 390,  474, 394,  474, 420,  422, 442,  448, 420,  448, 1256,  478, 420,  422, 442,  448, 420,  448, 420,  444, 420,  448, 420,  422, 442,  448, 394,  474, 420,  444, 420,  448, 420,  422, 442,  448, 420,  422, 446,  444, 420,  448, 420,  422, 442,  448, 420,  422, 446,  444, 420,  448, 416,  422, 446,  448, 420,  422, 442,  446, 424 [... cut ...] 46,  416, 446,  422, 446,  422, 446,  416, 446,  422, 446,  422, 446,  418, 446,  422, 446,  418, 446,  422, 446,  422, 1286,  448, 446,  422, 446,  418, 446,  422, 446,  416, 1290,  478, 390,  442, 450,  418, 1288,  478, 386,  448, 446,  422, 1286,  504, 364,  448};  // UNKNOWN FA41F15A

I would have expected to have more bytes the same, but they look totally different.

The remote used is an ARC466A67
Fan is set to auto. Mode cooling.

I think the dump works, if more sinals needed, no problem.

NiKiZe, 2022-06-25:
It says unknown, which means the protocol is not detected. Verify that your receiver is good. If you still get unknown after that, then please post a full temperature range (lowest to highest) Ang give all the raw data.

Please include model of unit as well. (All this is covered in the FAQ)

AntonWert, 2022-06-25:
Ok, fist of all some more information:
I tried this list of decoders, all with the UNKNOWN result:
30 kHz TSOP31230 
33 kHz TSOP31233 
36 kHz TSOP31236 
38 kHz TSOP31238 
40 kHz TSOP31240 
56 kHz TSOP31256 

Manufacturer DAIKIN
Multisplit
Model FTXM20R5V1B

The remote ist called ARC466A67

Now the full list from 18,0°C to 32,0°C in 0,5 steps
Fan was set to auto
Mode cooling
I used the TSOP31238  to capture the data

19:03:05.621 -> Timestamp : 000465.020
19:03:05.621 -> Library   : v2.8.2
19:03:05.621 -> 
19:03:05.621 -> Protocol  : UNKNOWN
19:03:05.621 -> Code      : 0xB9C999D2 (322 Bits)
19:03:05.667 -> uint16_t rawData[643] = {476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 25108,  3512, 1694,  472, 1260,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 394,  472, 394,  472, 394,  474, 394,  472, 1260,  476, 390,  478, 1256,  476, 1256,  476, 390,  478, 1256,  476, 1258,  476, 1256,  478, 1256,  476, 1256,  478, 394,  472, 390,  476, 1256,  478, 394,  472, 394,  474, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1262,  472, 394,  472, 394,  472, 394,  474, 394,  472, 394,  472, 394,  472, 394,  474, 1260,  472, 1260,  472, 394,  472, 1262,  472, 1260,  472, 1262,  472, 394,  478, 390,  476, 390,  476, 1256,  476, 1256,  478, 1256,  476, 1256,  478, 1260,  472, 394,  472, 394,  472, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 392,  472, 394,  472, 394,  472, 394,  472, 394,  476, 390,  476, 390,  476, 390,  478, 388,  478, 1260,  472, 1260,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  474, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1264,  474, 388,  478, 390,  476, 390,  478, 390,  476, 388,  478, 390,  478, 388,  476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  450, 416,  476, 390,  478, 388,  478, 390,  476, 390,  476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  476, 390,  476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 394,  472, 1256,  478, 394,  472, 394,  472, 394,  472, 394,  472, 35518,  3512, 1694,  474, 1260,  472, 394,  472, 394,  472, 394,  472, 1262,  472, 394,  474, 392,  474, 392,  452, 416,  476, 1262,  472, 394,  474, 1260,  472, 1262,  472, 394,  472, 1260,  472, 1260,  472, 1262,  472, 1260,  452, 1282,  476, 390,  476, 390,  476, 1262,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  476, 390,  476, 1256,  476, 394,  472, 394,  472, 1256,  476, 1258,  478, 1260,  472, 394,  472, 394,  472, 394,  472, 394,  474, 1260,  474, 392,  472, 394,  474, 1262,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 394,  474, 1260,  472, 394,  474, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  476, 390,  476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 1262,  472, 1260,  472, 394,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  474, 1260,  472, 1262,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  450, 416,  450, 420,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 394,  472, 1262,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 1262,  472, 394,  472, 394,  450, 420,  472, 394,  472, 394,  474, 392,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1262,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1262,  472, 394,  472};  // UNKNOWN B9C999D2
19:03:05.954 -> 
19:03:05.954 -> 
19:03:09.884 -> Timestamp : 000469.301
19:03:09.884 -> Library   : v2.8.2
19:03:09.884 -> 
19:03:09.884 -> Protocol  : UNKNOWN
19:03:09.884 -> Code      : 0x6D27EFF8 (322 Bits)
19:03:09.931 -> uint16_t rawData[643] = {472, 394,  472, 394,  472, 394,  472, 394,  474, 394,  472, 25110,  3514, 1690,  476, 1262,  472, 394,  472, 394,  472, 394,  472, 1256,  478, 394,  472, 394,  472, 394,  472, 394,  472, 1256,  478, 394,  472, 1260,  474, 1260,  472, 394,  472, 1256,  478, 1256,  476, 1258,  476, 1256,  478, 1256,  478, 388,  478, 390,  476, 1256,  478, 390,  476, 390,  478, 388,  478, 390,  476, 390,  478, 388,  476, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  474, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1256,  478, 1256,  478, 394,  472, 1260,  474, 1260,  472, 1262,  472, 390,  476, 390,  478, 390,  476, 1256,  478, 1258,  476, 1258,  476, 1256,  478, 1256,  476, 394,  474, 394,  472, 394,  472, 394,  472, 394,  474, 392,  472, 390,  478, 390,  476, 390,  478, 390,  478, 388,  476, 390,  478, 390,  476, 390,  476, 390,  478, 388,  478, 390,  478, 1256,  478, 1260,  472, 394,  472, 394,  472, 394,  474, 394,  472, 394,  472, 394,  47 [... cut ...]   472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1262,  472, 394,  472, 1262,  472, 394,  472, 394,  474, 394,  472, 1260,  474, 1260,  472, 394,  450, 416,  452, 416,  450, 416,  450, 416,  452, 420,  472, 394,  446, 420,  472, 394,  474, 394,  446, 420,  446, 1288,  472, 394,  472, 394,  472, 394,  472, 394,  472, 394,  472, 1260,  474, 1260,  472, 1262,  472, 1260,  474, 394,  472, 1256,  478, 394,  472};  // UNKNOWN BA514928
19:04:02.519 -> 
19:04:02.519 ->

AntonWert, 2022-06-29:
I'm supprised, and maybe you have an hint.
The AC does understand the commands I sent with your library. If I decode these Commands they ar recognized as DAIKIN
But the used commands by the remote are different. 

Is it possible that the AC does understand different protocols?

crankyoldgit, 2022-06-29:
> But the used commands by the remote are different.
> 
> Is it possible that the AC does understand different protocols?

Yes, We have seen some A/Cs accept multiple different protocols before. It might be to help owners who've lost their original remote. Dunno.

Our previous raw data samples of a `DAIKIN` protocol have `583` entries, your's has `643` so it's considerably larger.
Ping @sillyfrog who did a lot of the testing and code for `DAIKIN`

So, it's fairly clear your remote uses a different protocol. i.e. Not one the library supports. No reason it can't, it just needs a lot of work. That probably also explains the "it doesn't do half degrees" issue.

If you really want support, we/you will have to follow https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
Which is a lot of effort for half degrees when people typically don't need.

NiKiZe, 2022-06-29:
We should have enough to do step one already with the above raw dumps?

AntonWert, 2022-06-29:
Because the AC is also controlable with the DAIKIN App, I can confirm that at least the half degree steps are transmitted via IR. I can read the changes from the app.
Also after some investigation ant trying, I am quite shure that the AC is controlable in half steps.

Personally I think this could be an approtch by the manufacturer to be better than others and therefore I think some effort should be spent to get this solved also for future devices.

I puttet all my output in individual files (named without comma) and run them via your python script. The output is very simmelar if 2 steps are compared (example 18,0 and 18,5 degrees, Find it also in the list of files marked with *.out.txt

18,0 °C:
Found 643 timing entries.
Potential Mark Candidates:
[3512, 478]
Potential Space Candidates:
[35518, 25108, 1694, 1282, 420]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3512
kHdrSpace  = 1694
kBitMark   = 472
kOneSpace  = 1259
kZeroSpace = 393
kSpaceGap1 = 35518
kSpaceGap2 = 25108

Decoding protocol based on analysis so far:

kBitMark(UNEXPECTED)00000GAP(25108)
  Bits: 5
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b00000 (MSB first)
        0b00000 (LSB first)
kHdrMark+kHdrSpace+1000100001011011111001000000000001000000011011100011111000000000000000001100000000000000000000001000000000000000000000000000000000000000000000000000000001010000GAP(35518)
  Bits: 160
  Hex:  0x885BE400406E3E0000C000008000000000000050 (MSB first)
        0x0A0000000000000100000300007C76020027DA11 (LSB first)
  Dec:  778471974331943905232123735067560604558359003216 (MSB first)
        57089907708238395321559615459378624305866201617 (LSB first)
  Bin:  0b1000100001011011111001000000000001000000011011100011111000000000000000001100000000000000000000001000000000000000000000000000000000000000000000000000000001010000 (MSB first)
        0b0000101000000000000000000000000000000000000000000000000000000001000000000000000000000011000000000000000001111100011101100000001000000000001001111101101000010001 (LSB first)
kHdrMark+kHdrSpace+10001000010110111110010000000000000000001001110000100100000000000000010100000000000000000110000000000110000000000000000010100011000000000001000001000010
  Bits: 152
  Hex:  0x885BE400009C240005000060060000A3001042 (MSB first)
        0x420800C5000060060000A0002439000027DA11 (LSB first)
  Dec:  3040906149402779811781434919464982245112025154 (MSB first)
        1472546343248397986703471739034529953342806545 (LSB first)
  Bin:  0b10001000010110111110010000000000000000001001110000100100000000000000010100000000000000000110000000000110000000000000000010100011000000000001000001000010 (MSB first)
        0b01000010000010000000000011000101000000000000000001100000000001100000000000000000101000000000000000100100001110010000000000000000001001111101101000010001 (LSB first)

Total Nr. of suspected bits: 317

-----

18,5°C
Found 643 timing entries.
Potential Mark Candidates:
[**3514**, 478]
Potential Space Candidates:
[35518, **25110**, 1694, **1262**, 420]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = **3513**
kHdrSpace  = **1692**
[... 72 lines cut ...]
[260.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012056/260.txt.out.txt)
[265.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012057/265.txt)
[265.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012058/265.txt.out.txt)
[270.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012059/270.txt)
[270.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012060/270.txt.out.txt)
[275.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012061/275.txt)
[275.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012062/275.txt.out.txt)
[280.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012063/280.txt)
[280.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012064/280.txt.out.txt)
[285.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012065/285.txt)
[285.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012066/285.txt.out.txt)
[290.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012067/290.txt)
[290.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012068/290.txt.out.txt)
[295.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012069/295.txt)
[295.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012070/295.txt.out.txt)
[300.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012071/300.txt)
[300.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012072/300.txt.out.txt)
[305.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012073/305.txt)
[305.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012074/305.txt.out.txt)
[310.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012075/310.txt)
[310.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012076/310.txt.out.txt)
[315.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012077/315.txt)
[315.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012078/315.txt.out.txt)
[320.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012080/320.txt)
[320.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9012081/320.txt.out.txt)

crankyoldgit, 2022-06-29:
> Also after some investigation ant trying, I am quite shure that the AC is controlable in half steps.
> 
> Personally I think this could be an approtch by the manufacturer to be better than others and therefore I think some effort should be spent to get this solved also for future devices.

Thanks for the extra data.

I'll look at coding up basic (ie. hex) support for it soon.

AntonWert, 2022-07-03:
I did some more capturing:

Cooling 21°C Fan Auto
14:27:56.636 -> Timestamp : 000005.757
14:27:56.636 -> Library   : v2.8.2
14:27:56.636 -> 
14:27:56.636 -> Protocol  : UNKNOWN
14:27:56.636 -> Code      : 0x34EEF8FF (322 Bits)
14:27:56.683 -> uint16_t rawData[643] = {508, 390,  452, 414,  452, 414,  452, 414,  452, 414,  452, 25104,  3518, 1688,  478, 1256,  478, 414,  452, 414,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1280,  454, 414,  452, 1256,  478, 1254,  478, 414,  452, 1280,  452, 1282,  452, 1282,  452, 1282,  478, 1256,  480, 386,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  454, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1280,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 1280,  452, 1280,  452, 418,  450, 1280,  452, 1282,  452, 414,  452, 1282,  452, 414,  452, 416,  452, 1280,  454, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  454, 414,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  454, 414,  452, 414,  452, 414,  452, 414,  452, 414,  478, 388,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  454, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  478, 388,  452, 414,  478, 388,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 420,  448, 418,  450, 418,  448, 1286,  448, 1286,  448, 420,  448, 418,  448, 418,  448, 418,  448, 1284,  448, 1286,  448, 35512,  3518, 1688,  478, 1282,  452, 414,  452, 414,  452, 414,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1280,  452, 414,  452, 1256,  478, 1254,  478, 414,  452, 1280,  452, 1282,  452, 1282,  452, 1282,  452, 1280,  452, 414,  452, 414,  452, 1280,  452, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  452, 414,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 1286,  452, 414,  452, 414,  452, 1280,  452, 1282,  452, 1282,  452, 414,  452, 414,  452, 414,  452, 1256,  478, 414,  452, 1282,  452, 414,  452, 1256,  478, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  454, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1282,  452, 414,  452, 1256,  478, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 414,  452, 1286,  448, 1286,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 1286,  448, 1286,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 420,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 1260,  474, 418,  448, 1264,  470, 418,  448, 418,  448, 418,  448, 1258,  474, 1260,  474, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  452, 414,  452, 414,  452, 414,  452, 416,  452, 414,  448, 1264,  478, 384,  474, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 1260,  504, 362,  474, 418,  448, 1260,  478, 388,  474};  // UNKNOWN 34EEF8FF

Off (but set to Cooling 21°C Fan Auto)
14:28:00.621 -> Timestamp : 000009.738
14:28:00.621 -> Library   : v2.8.2
14:28:00.621 -> 
14:28:00.621 -> Protocol  : UNKNOWN
14:28:00.621 -> Code      : 0x7BDC8406 (322 Bits)
14:28:00.621 -> uint16_t rawData[643] = {484, 414,  478, 388,  454, 412,  454, 414,  452, 414,  452, 25104,  3518, 1688,  476, 1284,  452, 414,  454, 412,  454, 412,  452, 1254,  480, 412,  448, 418,  450, 416,  450, 418,  448, 1258,  476, 416,  450, 1284,  448, 1286,  452, 412,  454, 1254,  480, 1254,  478, 1254,  478, 1254,  480, 1254,  478, 414,  454, 414,  452, 1254,  480, 412,  454, 412,  452, 414,  452, 414,  454, 412,  454, 414,  452, 414,  454, 414,  452, 414,  454, 414,  452, 414,  452, 1280,  450, 418,  452, 414,  454, 414,  452, 414,  452, 414,  454, 412,  454, 414,  452, 1254,  480, 412,  454, 414,  454, 412,  448, 1284,  450, 1284,  448, 418,  450, 1258,  478, 1254,  480, 414,  452, 1280,  454, 412,  454, 414,  448, 1284,  448, 1286,  452, 414,  452, 414,  454, 414,  452, 414,  452, 414,  452, 414,  454, 412,  454, 414,  454, 412,  452, 414,  454, 412,  454, 412,  452, 414,  452, 414,  454, 412,  454, 414,  452, 414,  452, 1254,  480, 414,  452, 414,  452, 414,  454, 412,  454, 414,  454, 414,  452, 412,  454, 414,  452, 414,  454, 412,  454, 412,  452, 414,  454, 412,  454, 414,  448, 418,  448, 418,  450, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 418,  448, 1260,  474, 418,  448, 418,  448, 418,  450, 418,  448, 418,  448, 418,  450, 416,  450, 418,  448, 418,  448, 418,  450, 418,  450, 416,  448, 418,  454, 412,  454, 414,  452, 414,  454, 4 [... cut ...]
[dehumidify_mode_Fan_Auto.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9034783/dehumidify_mode_Fan_Auto.txt.out.txt)
[fan_mode_Fan_Auto.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9034784/fan_mode_Fan_Auto.txt.out.txt)
[heat_mode_250_Fan_auto.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9034785/heat_mode_250_Fan_auto.txt.out.txt)
[off_coling_210_fan_auto.txt.out.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9034786/off_coling_210_fan_auto.txt.out.txt)

Even more modes you find in the files atteacehd. I also run the python program

crankyoldgit, 2022-07-05:
@AntonWert I've written the code to do basic sending and decoding of your 312 bit / 39 byte Daikin A/C Protocol.
Please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Daikin312 / PR #1836 and let us know how it goes.
Based on the raw data you provided, it should work.
Assuming it does work, you will need to follow the steps in:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet

As we have plenty of other Daikin Protocols, we are fairly sure we have the correct bit ordering for the protocol, as they all use the same. i.e. You can skip: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#determining-the-bit-order-of-the-protocol
but you'll still need to work out the temperature locations & specifics.

Good luck with the analysis in advance, looking forward to hearing from you soon.
Note, if you look in the code, e.g. [ir_Daikin.h](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.h) & [ir_Daikin.cpp](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp) you will see plenty of common layouts for other Daikin protocols. Chances are there will be a lot of similarities, especially the checksum calculations (which is vital).

AntonWert, 2022-07-05:
I'm currently bussy during the week, I will test on weekend and come back to you. Thanks for the effort so far

AntonWert, 2022-07-05:
I already had a peak to look inside,  and based on the changes i found sendDaikin312

Do I have to call the function myselv? or should I define something like #define SEND_DAIKIN312?

NiKiZe, 2022-07-05:
Just run `IRrecvDumpV3` as you have so far, but using the code from the branch `SEND_DAIKIN312` is already enabled by default.

If everything works as it should you don't need to look at the raw data any more, and instead only look at the "code"

AntonWert, 2022-07-05:
Thank you. I was compleatly on the wrong path and thougt i should transmitt the data to the AC.

Now I got it.

Well, 1st result:
Cooling 22°C Auto Fan Speed, Swing on

21:43:48.481 -> Timestamp : 000005.589
21:43:48.481 -> Library   : v2.8.2
21:43:48.481 -> 
21:43:48.481 -> Protocol  : DAIKIN312
21:43:48.481 -> Code      : 0x11DA27000216DD0000020000010000000000000A11DA270000382C00AF0F0006600000C5000867 (312 Bits)
21:43:48.529 -> uint16_t rawData[643] = {486, 386,  482, 384,  482, 384,  482, 384,  482, 386,  480, 25100,  3522, 1682,  486, 1252,  482, 386,  482, 384,  482, 384,  482, 1252,  482, 384,  482, 384,  482, 384,  484, 382,  480, 1252,  482, 384,  482, 1252,  482, 1252,  482, 386,  482, 1252,  482, 1252,  482, 1252,  480, 1252,  482, 1252,  482, 384,  482, 386,  480, 1252,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  482, 384,  482, 384,  482, 386,  480, 386,  480, 386,  482, 384,  482, 1252,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  482, 384,  482, 386,  480, 1252,  482, 1252,  480, 386,  480, 1252,  480, 386,  480, 384,  482, 384,  482, 1252,  482, 384,  482, 1252,  480, 1252,  482, 1252,  480, 386,  480, 1252,  482, 1252,  482, 386,  480, 384,  482, 386,  480, 384,  482, 384,  482, 384,  480, 386,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  482, 384,  482, 386,  480, 386,  482, 384,  482, 1252,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  480, 386,  480, 386,  482, 384,  482, 384,  482, 384,  480, 386,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  480, 386,  480, 386,  482, 384,  482, 384,  482, 1252,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 384,  482, 386,  480, 386,  480, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  482, 386,  482, 384,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  480, 386,  482, 384,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  482, 386,  480, 386,  480, 386,  480, 386,  482, 384,  482, 1256,  482, 384,  482, 1252,  480, 386,  480, 386,  480, 386,  480, 384,  482, 35508,  3516, 1686,  482, 1252,  480, 386,  480, 386,  482, 384,  482, 1252,  480, 386,  482, 384,  482, 384,  482, 386,  480, 1252,  482, 384,  482, 1252,  486, 1248,  486, 380,  486, 1248,  486, 1248,  486, 1248,  486, 1248,  486, 1248,  486, 380,  486, 380,  486, 1248,  486, 380,  486, 380,  486, 382,  486, 380,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 386,  482, 384,  482, 386,  480, 1252,  482, 1252,  482, 1252,  482, 384,  482, 386,  480, 386,  482, 386,  480, 1252,  482, 1252,  482, 386,  480, 1252,  482, 384,  482, 386,  482, 384,  482, 384,  482, 384,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 1252,  480, 1252,  480, 1252,  482, 1252,  482, 384,  482, 1252,  482, 384,  482, 1252,  480, 1252,  482, 1252,  480, 1252,  482, 1250,  482, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  480, 386,  482, 386,  480, 384,  482, 386,  482, 384,  480, 1252,  482, 1252,  480, 386,  480, 386,  480, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  482, 1252,  480, 1256,  482, 386,  480, 386,  482, 384,  476, 390,  482, 386,  482, 384,  482, 384,  482, 386,  480, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 386,  482, 384,  482, 384,  478, 390,  476, 1258,  480, 384,  482, 1252,  482, 384,  482, 386,  480, 386,  476, 1256,  482, 1254,  480, 384,  482, 384,  482, 386,  480, 386,  482, 384,  482, 384,  482, 384,  482, 384,  482, 386,  480, 386,  480, 386,  480, 1252,  482, 386,  480, 386,  480, 386,  480, 386,  482, 1252,  482, 1252,  480, 1252,  482, 384,  482, 384,  482, 1252,  480, 1252,  482, 384,  482};  // DAIKIN312
21:43:48.815 -> uint8_t state[39] = {0x11, 0xDA, 0x27, 0x00, 0x02, 0x16, 0xDD, 0x00, 0x00, 0x02, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x38, 0x2C, 0x00, 0xAF, 0x0F, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC5, 0x00, 0x08, 0x67};

Now I should use the spreadsheet to decode the data as in the example above 
0x11DA27000216DD0000020000010000000000000A11DA270000382C00AF0F0006600000C5000867 (312 Bits)
correct?
As already sead, I will have time on the weekend

AntonWert, 2022-07-06:
I could not wait , and have tested a loot of settings in an XLS Shee

t

Following Results:
Most of the Items match with the spreadsheet of DAIKIN2
Some items I was able to even confirm, some have been moved, see the red markins in "new DAIKIN312" table

Resluts:
Byte 4 is always 0x02 (instead of 0x01)
Byte 9 is always 0x07 (reason: unknown)
Byte 10 is always 0x00 (reason: unknown)
Byte 12 defines in the lowest 2 Bits the LED brightnes: 0x01 is normal 0x03 is dimmed
Byte 14 if Bit 5 is set e.g. 0x10 the air clean mode is on 
Byte 17 is always 0x00 (reason: unknown)
Byte 18 is always 0x00 (reason: unknown) the air flow direction is not longer coded here
Byte 19 is the checksum, I can confirm the calculation also used in DAIKIN2
Byte 26 is the temperature, same formula, but the x2 witch was mentioned in DAIKIN2 is now used to incelemtn in half degrees, calculation is as described in DAIKIN2
Byte 28: I can confirm the upper 4 bits for fan contol speed the lower 4 bits define the swing left-rigt 0xx0=no swing 0x0F= swing
Byte 29: upper 4 bits unknown, the lower 4 bits define the swing up-down 0xx0=no swing 0x0F= swing
Byte 33: I can confirm the qutdoor quiet mode (Bit 5)
Byte 35 is always 0xC5 (reason: unknown)
Byte 36: I can confirm the eco mode (Bit 3)
Byte 37 is always 0x08 (reason: unknown)
Byte 38: checksum is as before

AntonWert, 2022-07-06:
[proto.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/9056771/proto.xlsx)

And here is my XLS

AntonWert, 2022-07-06:
Not really, but I could save as CSV
Or feel free to public it yourselv

AntonWert, 2022-07-08:
I created a PDF so everyone could at least read the data
[proto.pdf](https://github.com/crankyoldgit/IRremoteESP8266/files/9073773/proto.pdf)

AntonWert, 2022-07-10:
Is there something I can now provide?

crankyoldgit, 2022-07-11:
I will try to find time to look at this. I've been very busy of late with other things.
The advantage of Google Docs/sheets or similar is that it's a live-update document. No one needs to download the latest, it's always up to date, and available.

sillyfrog, 2022-07-11:
I have uploaded to Google Drive here:
https://docs.google.com/spreadsheets/d/1r3yulakLwdlwXeu_UoALFlzxVsdQYhfSfVmyr0-RBR8/edit?usp=sharing

@crankyoldgit @AntonWert if you make an edit / comment on the document I should be able to keep you as full editors and remove public access tomorrow.

I no longer have a Dakin AC unfortunately, however I'm looking to get one in a few months, so will be able to test further again then.

Good Luck!

AntonWert, 2022-07-23:
Im not shure if I did not mix up some Bits, but the to de fact that the new Daikin 312 seem to be simmelar to Daikin 2 I set up the stuct as a first go:

`
union Daikin312Protocol{
  struct{
    uint8_t pad[3];
    uint8_t raw[kDaikin2StateLength];  ///< The state of the IR remote.
  };
  struct {
    // Byte -3~4
	//		Byte 0: Always 0x11
	//		Byte 1: Always 0xDA
	//		Byte 2: Always 0x27
	//		Byte 3: Always 0x00
	//		Byte 4: Always 0x02
    uint64_t :64;

    // Byte 5~6
    uint64_t CurrentTime  :12;
    uint64_t              :3;
    uint64_t Power2       :1;
    // Byte 7
    uint64_t              :4;
    uint64_t Light_old    :2;	// maybe not longer used
    uint64_t Beep         :2;
    // Byte 8
    uint64_t FreshAir     :1;
    uint64_t              :2;
    uint64_t Mold         :1;
    uint64_t              :1;
    uint64_t Clean        :1;
    uint64_t              :1;
    uint64_t FreshAirHigh :1;
    // Byte 9~10
	//		Byte  9: Always 0x3D
	//		Byte 10: Always 0x00
    uint64_t              :16;
    // Byte 11
    uint64_t              :8;
    // Byte 12
    uint64_t              :1;
    uint64_t Light		  :2;
    uint64_t              :5;

    // Byte 13
    uint64_t          		:7;
    uint64_t EyeAuto  		:1;
    // Byte 14
    uint64_t          		:4;
    uint64_t AirCleanMode 	:1;
    uint64_t          		:3;
    // Byte 15~16
    uint64_t          		:16;
    // Byte 17: Always 0x00
    uint64_t          		:8;
    // Byte 18: Always 0x00
    uint64_t          		:8;
    // Byte 19
    uint64_t Sum1     :8;
    // Byte 20: Always 0x11
    uint64_t          :8;

    // Byte 21~24
 	//		Byte 21: Always 0xDA
	//		Byte 22: Always 0x27
	//		Byte 23: Always 0x00
	//		Byte 24: Always 0x00
   uint64_t          :32;
    // Byte 25
    uint64_t Power    :1;
    uint64_t OnTimer  :1;
    uint64_t OffTimer :1;
    uint64_t          :1;
    uint64_t Mode     :3;
    uint64_t          :1;
    // Byte 26
    uint64_t TempHalfDegree :7;
    uint64_t HumidOn  		:1;
    // Byte 27
    uint64_t Humidity :8;
    // Byte 28
    uint64_t SwingV   :4;
    uint64_t Fan      :4;

    // Byte 29
    uint64_t SwingH   :4;
    uint64_t          :4;
    // Byte 30~32
    /// @see https://github.com/crankyoldgit/IRremoteESP8266/pull/1264
    uint64_t OnTime     :12;
    uint64_t OffTime    :12;
    // Byte 33
    uint64_t Powerful   :1;
    uint64_t            :4;
    uint64_t Quiet      :1;
    uint64_t            :2;
    // Byte 34~35
 	//		Byte 35: Always 0xC5
   uint64_t            :16;
    // Byte 36
    uint64_t            :1;
    uint64_t Eye        :1;
    uint64_t Econo      :1;
    uint64_t            :1;
    uint64_t Purify     :1;
    uint64_t SleepTimer :1;
    uint64_t            :2;

    // Byte 37: Always 0x08
    uint8_t       :8;
    // Byte 38
    uint8_t Sum2  :8;
  };
};
`

AntonWert, 2022-08-07:
Anything new?
Anything I could support? Test?

crankyoldgit, 2022-08-16:
Sorry, not yet. I've been to busy with non-code things lately & higher priority stuff.

It's not forgotten.

AntonWert, 2022-09-10:
I did some tests the week ago. Using the Daikin app I can confirm that a half degree is possible. Also the same is possible with the origial IR remote.
So it would still be nice to have half degrees via the library

AntonWert, 2022-09-17:
Thanks @crankyoldgit  for the update. 
I'm currently struggeling to built my the software to sent this new Daikin312 protocol.
Therefore I defined
#define SEND_DAIKIN312 true
#define _IR_ENABLE_DEFAULT_ false

But what do I not see to get it running?

Do I still have to create the class:
IRDaikinESP ac(kIrLed);  // Set the GPIO to be used to sending the message
and sent with     ac.send(); ?

It will not compile.

Please help me out to figure out the right configuration, It would be great for me to do the tests and confirm here the implementation.

tonhuisman, 2022-09-17:
Are your compilation issues perhaps related to #1880 ? If it does, the fix for that is in #1881, and you _could_ manually apply that to your local copy (when using that).

AntonWert, 2022-09-17:
As I understand #1880 blocks all builts. My problem is, that in standard configuration the library is usable , but uses the standard function to sent data. I need to use the new 312-protocol version just figured out.

AntonWert, 2022-09-20:
Here is an instruction how to reproduce the issue with Arduino 1.8.19
I use the example TurnOnDaikinAC
I use the ESB-12E NodeMCU module
Of course I had removed the #if and #endif arround ac.send(); to trigger the actual transmission.

The example compiles without any issues.

Because it is not possible to use command line switches, I define 
#define SEND_DAIKIN312 true
#define _IR_ENABLE_DEFAULT_ false
at the first lines of IRremoteESP8266.h (not a good solution, I konw, so improvements are welcome) to use the new protocoll.

Now it does no longer compile:

C:\Users\xxx\AppData\Local\Temp\arduino_modified_sketch_221928\TurnOnDaikinAC.ino: In function 'void loop()':
TurnOnDaikinAC:65:6: error: 'class IRDaikinESP' has no member named 'send'
   65 |   ac.send();
      |      ^~~~
Bibliothek IRremoteESP8266 in Version 2.8.3 im Ordner: K:\Arduino\libraries\IRremoteESP8266  wird verwendet
exit status 1
'class IRDaikinESP' has no member named 'send'

AntonWert, 2022-09-26:
Just to confirm, i also tried platformio to test, but there ist the exact same error:

src\main.cpp:429:8: error: 'class IRDaikinESP' has no member named 'send'
  429 |     ac.send();
      |        ^~~~
Compiling .pio\build\d1_mini\lib1fd\IRremoteESP8266\ir_Arris.cpp.o
*** [.pio\build\d1_mini\src\main.cpp.o] Error 1

I used the built flags in platformio.ini
build_flags = -D_IR_ENABLE_DEFAULT_=false
              -DSEND_DAIKIN312=true

Still, it would be very nice, if someone could help me out

tonhuisman, 2022-09-26:
I guess you also need to add `-DSEND_DAIKIN=true` as Daikin 312 seems to be an extension to the original Daikin support.

AntonWert, 2022-09-26:
Thanks @tonhuisman , good hint. At least it compiles now.

But I am not shure, in IRsend.cpp in the function ool IRsend::send(const decode_type_t type, const uint8_t *state, const uint16_t nbytes) there is called 
#if SEND_DAIKIN
    case DAIKIN:
      sendDaikin(state, nbytes);
      break;
#endif  // SEND_DAIKIN

if SEND_DAIKIN is defined

and later

#if SEND_DAIKIN312
    case DAIKIN312:
      sendDaikin312(state, nbytes);
      break;
#endif  // SEND_DAIKIN312

so - as I see it - now both protocolls are sent. 
And this is clearly not, what I do want to have. Only the new version should be sent.

AntonWert, 2022-09-26:
Something is wrong, it does not work. I do not get half degrees. 
But the problem is deeper, I would have expect to have a setTemp function with some kind of half-degree parameter, but all setTemp are const uint8_t, so it will not work i think.

AntonWert, 2022-10-16:
Any hints for me here? How can I help to go ahead?

crankyoldgit, 2022-10-16:
Do you have a link to the code you've written?

AntonWert, 2022-11-20:
Any help i can expect?

AntonWert, 2022-12-10:
@crankyoldgit: Are you able to give me any hint please how to solve the issue to get it compiled? Can you give me just one red line I can follow to continue?
It will be great to continue on this improvement and I would be able to test and confirm if I get my problem solved.
Please give me help.

crankyoldgit, 2022-12-10:
> Any help i can expect?

I think your problem is you are using the wrong AC class for Daikin 312. DaikinESP is a different protocol

crankyoldgit, 2022-12-10:
> > Any help i can expect?
> 
> I think your problem is you are using the wrong AC class for Daikin 312. DaikinESP is a different protocol

Also, I just checked the code. There is NO detailed support for the 312 bit protocol.

AntonWert, 2022-12-11:
> I think your problem is you are using the wrong AC class for Daikin 312. DaikinESP is a different protocol

So something like daikin312 would be correct? (Event it is not existing yet, see below)

> Also, I just checked the code. There is NO detailed support for the 312 bit protocol.

OK, I see, some work to do.

As you have already implemented IRsend::sendDaikin312 in "ir_Daikin.cpp" I think this step can be called as done.

Would be the next step to add i "IRac.cpp" a class method IRac::daikin312 ? If this would be the right way to continue the support I can start on that, or would be that the wrong approach?

Would be that the final step, or is there more open?

NiKiZe, 2022-12-11:
The FAQ explains next step.
Collecting the data and map the bits in the message to its meaning. After that generating the AC class is often straight forward when the structure is known.

AntonWert, 2022-12-11:
Well, but this was done a long time ago.
See the post of [sillyfrog](https://github.com/sillyfrog) commented [on Jul 11](https://github.com/crankyoldgit/IRremoteESP8266/issues/1829#issuecomment-1180283932) •
I already made a proposal in Julay, see: [AntonWert](https://github.com/AntonWert) commented [on Jul 23](https://github.com/crankyoldgit/IRremoteESP8266/issues/1829#issuecomment-1193116583)

AntonWert, 2022-12-27:
Dears, 
I did it. I added the class for Daikin 312 myself.

I used this .ino file to witch was the former exaple, to confirm that the built is compilable: (please rename to ino, i was not able to upload the ino file)
[TurnOnDaikinAC.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10309807/TurnOnDaikinAC.txt)

I added the class based on Daikin2 into these files:
[ir_Daikin.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10309819/ir_Daikin.h.txt)
[ir_Daikin.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10309820/ir_Daikin.cpp.txt)

May anyone (or even better @crankyoldgit - may you confirm) that this was the right way?
If so, I would be happy and continue with testing.

Also I would be happy about any comments/help/advice....

Nawile, 2023-01-19:
> 

Thanks, i need to emulate an ARC466A67, but i'm actually not able to compile a firmware with DAIKIN312. Still didn't understand which parts of DAIKIN312 are already part of the stable build Tasmota 12.3.1. 
Is there any fork with already compiled builds?

Does anybody know if the "Bedarfssteuerung" is accessible via ir?

Nawile, 2023-01-23:
> Dears, I did it. I added the class for Daikin 312 myself.
>
> I used this .ino file to witch was the former exaple, to confirm that the built is compilable: (please rename to ino, i was not able to [...]

Hi,

got a build on my own with your changes, but it does not work. :(
(Visual Studio, PlatformIO -> tasmota-ir based on Tasmota 12.3.1.4)

Always getting: 
`15:51:24.776 MQT: stat/tasmota_ir_test/RESULT = {"IRHVAC":"Wrong Vendor (LG|COOLIX|DAIKIN|KELVINATOR|MITSUBISHI_AC|GREE|ARGO|TROTEC|TOSHIBA_AC|FUJITSU_AC|MIDEA|HAIER_AC|HITACHI_AC|HITACHI_AC1|HAIER_AC_YRW02|WHIRLPOOL_AC|SAMSUNG_AC|ELECTRA_AC|PANASONIC_AC|LG2|DAIKIN2|VESTEL_AC|TECO|TCL112AC|MITSUBISHI_HEAVY_88|MITSUBISHI_HEAVY_152|DAIKIN216|SHARP_AC|GOODWEATHER|DAIKIN160|NEOCLIMA|DAIKIN176|DAIKIN128|AMCOR|DAIKIN152|MITSUBISHI136|MITSUBISHI112|HITACHI_AC424|DAIKIN64|AIRWELL|DELONGHI_AC|CARRIER_AC64|HITACHI_AC344|CORONA_AC|SANYO_AC|VOLTAS|TRANSCOLD|TECHNIBEL_AC|MIRAGE|PANASONIC_AC32|ECOCLIM|TRUMA|HAIER_AC176|TEKNOPOINT|KELON|TROTEC_3550|SANYO_AC88|RHOSS|AIRTON|HITACHI_AC264|HITACHI_AC296|HAIER_AC160|BOSCH144)"}`

Sending: `{"IRHVAC{"Vendor":"DAIKIN312","Mode":"Heat","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Stop","SwingH":"Off","Quiet":"On","Turbo":"Off","Econo":"On","Eye":"On","Light":"Off","Filter":"On","Clean":"On","Beep":"On"}}`

I tried to debug, but didn't get it. I don't know why the vendor list ends with  "BOSCH144", but in "IRremoteESP8266.h" it ends with "DAIKIN312". I tried to shorten it there and in "IRtext.cpp", but without success. Any ideas?

My settings: ARC466A67 with Perfera (2MXM40N + FTXM20R + FTXM20R) and Pearl NX-4519-675

AntonWert, 2023-02-02:
> Thanks, i need to emulate an ARC466A67, but i'm actually not able to compile a firmware with DAIKIN312. Still didn't understand which parts of DAIKIN312 are already part of the stable build Tasmota [...]

With the Files I posted above, it was compailable for me using latest Arduino.

Nawile, 2023-02-08:
> With the Files I posted above, it was compailable for me using latest Arduino.

@AntonWert can you compile an  tasmota-ir version and upload it? 
So i can make sure, that the mistake is in my sources.
Thanks.

I still get the "wrong vendor" failure. Maybe my call is still wrong: `{"IRHVAC{"Vendor":"DAIKIN312","Mode":"Heat","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Stop","SwingH":"Off","Quiet":"On","Turbo":"Off","Econo":"On","Eye":"On","Light":"Off","Filter":"On","Clean":"On","Beep":"On"}}`

AntonWert, 2023-02-11:
First of all, I was able to test now my version with the real DAIKIN AC and got it to work (at least fo far for the functions I need). But I built with Arduino.

@Nawile : I do have Platform IO and downloaded the current source of tasmota. I was able to built the tasmota-ir as supplied from github.

```
[mylib.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/10714365/mylib.zip)

Environment    Status    Duration
-------------  --------  ------------
tasmota-ir     SUCCESS   00:00:18.309
```

I replaced the 2 files (latest version atteaced)  in the subfolder 
Tasmota-development\lib\lib_basic\IRremoteESP8266\IRremoteESP8266\src
and was able to built again.
But I do not know if the new IRDaikin312 class is used by tasmota

[ir_Daikin.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10714369/ir_Daikin.cpp.txt)
[ir_Daikin.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10714370/ir_Daikin.h.txt)

Nawile, 2023-02-16:
Thanks @AntonWert .

Getting the puzzle piece by piece.

I can compile and send DAIKIN312 via "IRSend". Maybe this was my Problem.
But also helpfully ist deleting the caches and editing "my_user_config.h".

@AntonWert: How can you send (encode) the parameter like `"Mode":"Heat","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Stop","SwingH":"Off","Quiet":"On","Turbo":"Off","Econo":"On","Eye":"On","Light":"Off","Filter":"On","Clean":"On","Beep":"On"`?

The "IRHVAC" Command still doesn't because in files "IRac.cpp" and "IRac.h" several entries aren't made. I wasn't able to made them, because i still don't understand the encode parameter to build correct classes.

AntonWert, 2023-02-16:
Just for clarification. I use the library (and did therefore the extention) on a WemosD1mini.
I run my own programm, providing me a simple WEB-GUI and an REST API.
I do not run tasmota.

For sending the data to the AC i do have my own function:

`void MyIRSendFunction(const bool isOn, const uint8_t acmode, const float temp) {
	IRDaikin312 ac(kIrLed);      // Set the GPIO to be used to sending the message
   if (isOn)
      ac.on();
    else
      ac.off();
    ac.setFan(kDaikinFanAuto);
    ac.setMode(acmode);
    ac.setTemp(temp);
    ac.setSwingVertical(kDaikin312SwingVOff);
    ac.setSwingHorizontal(kDaikin312SwingVOff);
    Serial.println(ac.toString());  // Display what we are going to send.
    ac.setEcono(true);
    ac.setClean(true);
    ac.setLight(2);
    ac.send();
}
`

These functions seem to work since I posted them.

@Nawile: with these circumstances, what do you mean with encode parameters? this is done by the IRDaikin312 class now

Nawile, 2023-02-16:
Ok, here we go. :)

I modified the  files "IRac.cpp" and "IRac.h" and got an runable image. I just used code from DAIKIN2. Now i need some more modification to add parameters for "eye" and others.
  
[IRac.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10757578/IRac.cpp.txt)
[IRac.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10757580/IRac.h.txt)

Now i got some issues and questions:

- The absolute Parameter ("up" or "down") for SwingV/H doesn't work. Just auto and stop. My fault or just not supported?

I didn't test all, but other things like "Beep off" still aren't working. 

@AntonWert, thanks. And it's pitty, that you aren't using Tasmota.

AntonWert, 2023-02-16:
Thanks for your feedback @Nawile .
Maybe I do have time tomorrow to check the Swing and other Items you mentioned.

Nawile, 2023-02-16:
Here are my actual versions:

I'm trying to support the eyes...

[ir_Daikin.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10758873/ir_Daikin.cpp.txt)
[IRsend.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10758874/IRsend.h.txt)
[IRac.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10758875/IRac.cpp.txt)
[IRac.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10758876/IRac.h.txt)

Nawile, 2023-02-17:
I've added them to `struct state_t`, to `IRDaikin312::toCommon` and in `IRac.cpp #if SEND_DAIKIN312`.

I'm also don't understand why the eye-state isn't shown in the console output. I guess it is using `IRDaikin312::toString`

AntonWert, 2023-02-17:
So first of all for the swing. I checked the manual and there it is stated that you only can turn on swing, or stop it at the current position. According the manual there is no option for a fixed positoning.
I confimed this with my remote and IR-Dump. 
The IR Remote only sends "kDaikin312SwingVAuto" and "kDaikin312SwingVOff" 
The other modes (copied from DAIKIN2) seem not to work.
Same for "kDaikin312SwingHAuto" and "kDaikin312SwingHOff" where the constants seem to be wrong, what is now corrected.

With the eye it seems to be difficult to test, but maye it is fixed now due to moving from byte 13 to 36 from my remote IR debuggung.
Sorry I can't test this because my AC is lookint to an wall and I cannot realy use the function.
Maybe you can give me feedback

AntonWert, 2023-02-17:
[ir_Daikin.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10767230/ir_Daikin.h.txt)
Here is the updated file

Nawile, 2023-02-17:
Yeah great. 
With EyeAuto it can be activated!
but not deactivated. :(
The old DAIKIN switch it off (home automation send with still DAIKIN).

I had to uncomment all the positions, becaus .cpp has to adapted...

AntonWert, 2023-02-17:
That is strange.
have you tried setEye and setEyeAuto in all 4 possible combinations?

Nawile, 2023-02-18:
> That is strange. have you tried setEye and setEyeAuto in all 4 possible combinations?

Yes, i think so.
EyeAuto on -> Eye works. 
Both off -> Eye still works.

My smart home device send with std Tasmota ir an DAIKIN-protocol and it stops.

I can test on tuesday again.

[11 more replies]