crankyoldgit/IRremoteESP8266 #1830: Need support for Delonghi PAC EX130 ECO REAL FEEL

Dominik28111, 2022-06-25:
At the moment I am trying to add support for my Delonghi PAC EX130 ECO REAL FEE. Below you can find the current data I already collected. Sending commands is working so far but not every command works I have to send some and randomly one is working then.

Output of `python auto_analyse_raw_data.py -f ../raw.txt -g -n TestExample`:
```
Found 71 timing entries.
Potential Mark Candidates:
[9226, 644]
Potential Space Candidates:
[35040, 4546, 2284, 1724, 572]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kTestExampleHdrMark   = 9214
kTestExampleHdrSpace  = 2284
kTestExampleBitMark   = 600
kTestExampleOneSpace  = 1694
kTestExampleZeroSpace = 546
kTestExampleSpaceGap1 = 35040
kTestExampleSpaceGap2 = 4546

Decoding protocol based on analysis so far:

kTestExampleHdrMark+UNEXPECTED->GAP(4546)kTestExampleBitMark(UNEXPECTED)11111000111111111100000100111110GAP(35040)
  Bits: 32
  Hex:  0xF8FFC13E (MSB first)
        0x7C83FF1F (LSB first)
  Dec:  4177510718 (MSB first)
        2089025311 (LSB first)
  Bin:  0b11111000111111111100000100111110 (MSB first)
        0b01111100100000111111111100011111 (LSB first)
kTestExampleHdrMark+kTestExampleHdrSpace+
Total Nr. of suspected bits: 32
```

The value of `kTestExampleSpaceGap2` is working.

```cpp
// Copyright 2020 David Conran (crankyoldgit)
/// @file
/// @brief Support for TestExample protocol

// Supports:
//   Brand: TestExample,  Model: TODO add device and remote

#include "IRrecv.h"
#include "IRsend.h"
#include "IRutils.h"

// WARNING: This probably isn't directly usable. It's a guide only.

// See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
// for details of how to include this in the library.
const uint16_t kTestExampleHdrMark = 9214;
const uint16_t kTestExampleBitMark = 600;
const uint16_t kTestExampleHdrSpace = 2284;
const uint16_t kTestExampleOneSpace = 1694;
const uint16_t kTestExampleZeroSpace = 546;
const uint16_t kTestExampleSpaceGap1 = 35040;
const uint16_t kTestExampleSpaceGap2 = 4546;
const uint16_t kTestExampleFreq = 38000;  // Hz. (Guessing the most common frequency.)
const uint16_t kTestExampleBits = 32;  // Move to IRremoteESP8266.h
const uint16_t kTestExampleOverhead = 7;

#if SEND_TESTEXAMPLE
// Function should be safe up to 64 bits.
/// Send a TestExample formatted message.
/// Status: ALPHA / Untested.
/// @param[in] data containing the IR command.
/// @param[in] nbits Nr. of bits to send. usually kTestExampleBits
/// @param[in] repeat Nr. of times the message is to be repeated.
void IRsend::sendTestExample(const uint64_t data, const uint16_t nbits, const uint16_t repeat) {
  enableIROut(kTestExampleFreq);
  for (uint16_t r = 0; r <= repeat; r++) {
    uint64_t send_data = data;
    // Header
    mark(kTestExampleHdrMark);
    // Gap
    space(kTestExampleSpaceGap);
    // Data Section #1
    // e.g. data = 0xF8FFC13E, nbits = 32
    sendData(kTestExampleBitMark, kTestExampleOneSpace, kTestExampleBitMark, kTestExampleZeroSpace, send_data, 32, true);
    send_data >>= 32;
    // Footer
    mark(kTestExampleBitMark);
    space(kTestExampleSpaceGap);
    // Header
    mark(kTestExampleHdrMark);
    space(kTestExampleHdrSpace);
    space(kDefaultMessageGap);  // A 100% made up guess of the gap between messages.
  }
}
#endif  // SEND_TESTEXAMPLE

#if DECODE_TESTEXAMPLE
// Function should be safe up to 64 bits.
/// Decode the supplied TestExample message.
/// Status: ALPHA / Untested.
/// @param[in,out] results Ptr to the data to decode & where to store the decode
/// @param[in] offset The starting index to use when attempting to decode the
///   raw data. Typically/Defaults to kStartOffset.
/// @param[in] nbits The number of data bits to expect.
/// @param[in] strict Flag indicating if we should perform strict matching.
/// @return A boolean. True if it can decode it, false if it can't.
bool IRrecv::decodeTestExample(decode_results *results, uint16_t offset, const uint16_t nbits, const bool strict) {
  if (results->rawlen < 2 * nbits + kTestExampleOverhead - offset)
    return false;  // Too short a message to match.
  if (strict && nbits != kTestExampleBits)
    return false;

  uint64_t data = 0;
  match_result_t data_result;

  // Header
  if (!matchMark(results->rawbuf[offset++], kTestExampleHdrMark))
    return false;

  // Gap
  if (!matchSpace(results->rawbuf[offset++], kTestExampleSpaceGap))
    return false;

  // Data Section #1
  // e.g. data_result.data = 0xF8FFC13E, nbits = 32
  data_result = matchData(&(results->rawbuf[offset]), 32,
                          kTestExampleBitMark, kTestExampleOneSpace,
                          kTestExampleBitMark, kTestExampleZeroSpace);
  offset += data_result.used;
  if (data_result.success == false) return false;  // Fail
  data <<= 32;  // Make room for the new bits of data.
  data |= data_result.data;

  // Footer
  if (!matchMark(results->rawbuf[offset++], kTestExampleBitMark))
    return false;
  if (!matchSpace(results->rawbuf[offset++], kTestExampleSpaceGap))
    return false;

  // Header
  if (!matchMark(results->rawbuf[offset++], kTestExampleHdrMark))
    return false;
  if (!matchSpace(results->rawbuf[offset++], kTestExampleHdrSpace))
    return false;

  // Success
  results->decode_type = decode_type_t::TESTEXAMPLE;
  results->bits = nbits;
  results->value = data;
  results->command = 0;
  results->address = 0;
  return true;
}
#endif  // DECODE_TESTEXAMPLE
```

Raw data:
[raw.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/8985248/raw.txt)

NiKiZe, 2022-06-25:
Your raw file indicates NEC protocol, could you also post the model of the remote?

Dominik28111, 2022-06-25:
> Your raw file indicates NEC protocol, could you also post the model of the remote?

What do you exactly mean? 

This is my normal remote:
[image]

NiKiZe, 2022-06-25:
It is nec protocol, see your own dumps.

Protocol  : NEC
Code      : 0xF8FFC13E (32 Bits)

It is a dumb protocol, there is nothing more to support here.
Each button on the remote sends one code. There is no state. Just keypresses.

crankyoldgit, 2022-06-26:
@NiKiZe is correct. This is not a remote that sends an entire state. It's like a dumb TV remote. i.e. It's sending the equivalent of "Temp Up one" not "set the temp to 23C".
There is nothing for us to add support for. Your message is saying "The user just press XYZ button". Normal A/C remotes send a single message that says "The mode is Cool, Fan is Auto, the vanes to High, The temp is 20C, the Power is On, the filter is Off, the Off timer is set to 240 mins".

To send your your "power button"'s message you need to use `sendNEC(0xF8FFC13E, 32, 1);` or `sendSherwood(0xF8FFC13E);`.
The messages are standard NEC messages with a single repeat message following it. (Per the NEC protocol, NEC has an odd repeat structure).

closed 2022-06-26 by crankyoldgit