crankyoldgit/IRremoteESP8266 #1834: Cannot Disable Sleep Function (Midea)

drschnalli, 2022-06-30:
Using Version v2.8.2 i tried sleep today ( i thought it was the timer function )... ive send a "1" to ir_server/ac/cmnd/sleep and i dont know how to disable... wipe settings and reboot the ESP does not help :( im using the ir_mqtt_server. Everytime i change the fanspeed, mode or anything the sleep mode turns on again (when i turn it off on the ac) can you help me ? the fan is going crazy now :D

crankyoldgit, 2022-06-30:
`-1` is Off. >= 0 is on or how many minutes, or time in minutes to turn off sleep mode.

crankyoldgit, 2022-06-30:
See: https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html#ad60fbe1488efe2d02307d81b090b3b72

drschnalli, 2022-07-01:
Thank you! You saved my live (and nerves) again... i was trying with false, 0,1 and so on... forgot to try out -1. 

youre the best! thanks for keeping this alive so much! Most ridiculus project here!

closed 2022-07-01 by drschnalli

crankyoldgit, 2022-07-01:
For future reference, this is something you could have done too (i.e. set this to `true`):
https://github.com/crankyoldgit/IRremoteESP8266/blob/c053aa58a8056909b19ec7706e2c0556dfa2a0c2/examples/IRMQTTServer/IRMQTTServer.h#L167-L171

Some settings are stored (via [RETAIN](https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/)) in your MQTT Server's database.