crankyoldgit/IRremoteESP8266 #1838: can't display mode"off" in the homeassistant UI card

water3033, 2022-07-09:
I have connected my gree AC to my homeassistant with the IRMQTTSever, and can control the gree AC   very well.  but  when I turn off the gree AC with remote control  or homeassistant , the thermostat card not change to mode "off", though the AC is power off.  how can i do?
[image]

homeassistant error log:

Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:621
Integration: MQTT (documentation, issues)
First occurred: 14:43:46 (14 occurrences)
Last logged: 15:32:27
Invalid modes mode: Off

my homeassistant configuration.yaml:

`climate:
  - platform: mqtt
    name: gree Aircon
    modes:
      - "off"
      - "auto"
      - "cool"
      - "heat"
      - "dry"
      - "fan_only"
    fan_modes:
      - "Auto"
      - "min"
      - "low"
      - "medium"
      - "high"
      - "max"
    swing_modes:
      - "off"
      - "Auto"
      - "highest"
      - "high"
      - "middle"
      - "low"
      - "lowest"
    power_command_topic: "GreeAC/ac/cmnd/power"
    mode_command_topic: "GreeAC/ac/cmnd/mode"
    mode_state_topic: "GreeAC/ac/stat/mode"
    temperature_command_topic: "GreeAC/ac/cmnd/temp"
    temperature_state_topic: "GreeAC/ac/stat/temp"
    fan_mode_command_topic: "GreeAC/ac/cmnd/fanspeed"
    fan_mode_state_topic: "GreeAC/ac/stat/fanspeed"
    swing_mode_command_topic: "GreeAC/ac/cmnd/swingv"
    swing_mode_state_topic: "GreeAC/ac/stat/swingv"
    current_temperature_topic: "zigbee2mqtt/temsensor2"
    current_temperature_template: "{{ value_json.temperature }}"
    min_temp: 16
    max_temp: 32
    temp_step: 1
    retain: false`

crankyoldgit, 2022-07-09:
See: https://www.home-assistant.io/integrations/climate.mqtt/
You've got a HomeAssistant problem, not a problem with this library. Please report the issue there.

But I think one of the problems is the version of HA you're using now uses a different format for this:
i.e. New:
```yaml
mqtt:
  climate:
    - name: gree Aircon
    - mode_command_topic: "GreeAC/ac/cmnd/mode"
etc
```
vs Old:
```yaml
climate:
  - platform: mqtt
    mode_command_topic: "GreeAC/ac/cmnd/mode"
etc
```

water3033, 2022-07-09:
thank you, but mode "cool"/"heat"/"dry" is shown correctly， I found the problem is caused by “Off” and “off”， in MQTT topic，the mode is "Off" which the fisrt letter is uppercase“O”， in configuration.yaml，the mode is “off” which the first letter is owercase "o".  if I change the "off" to "Off'" in the configuration.yaml,  the mode "off" will disappear. So I think if in the MQTT publish topic the mode  is "off".  it will be OK. but I don't kown how to modify the IRMQTTSever code.

water3033, 2022-07-09:
I have solved another similar problem， in MQTT topic，the fan_modes is "Auto" which the fisrt letter is uppercase“A”， in configuration.yaml，the mode is “auto” which the first letter is owercase "a". I modified  the "auto" to "Auto'" in the configuration.yaml, the error log in homeassisant not appear anymore.  But the mode"off" problem can't use this method to solve. I don't kown why.

before modified the "auto" to "Auto'" ,the error log in homeassistant is like this：
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:621
Integration: MQTT (documentation, issues)
First occurred: 10:43:46 (10 occurrences)
Last logged: 11:32:27
Invalid modes mode: Off
Invalid modes mode: Auto

crankyoldgit, 2022-07-11:
No idea off the top of my head. Have you talked to the HA crowd? What are the MQTT messages looking like??

crankyoldgit, 2022-08-17:
Chasing up on this. How did you go?

water3033, 2022-11-05:
I have solved the  problem with a clumsy method；In the HA configuration.yaml，add “mode_state_template:” `    mode_state_template: '{{ {"Off": "off", "off": "off", "auto": "auto", "cool": "cool", "heat": "heat", "dry": "dry", "fan_only": "fan_only"}[value] | default(''off'') }}'`

closed 2022-11-05 by water3033

1043717432, 2024-05-31:
> I have solved the problem with a clumsy method；In the HA configuration.yaml，add “mode_state_template:” ` mode_state_template: '{{ {"Off": "off", "off": "off", "auto": "auto", "cool": "cool", "heat": "heat", "dry": "dry", "fan_only": "fan_only"}[value] | default(''off'') }}'`

Hello, I also encountered this problem. How do I write the complete configuration format? Thank you!