crankyoldgit/IRremoteESP8266 #1840: Can't use the method in example of "CommonAcControl"

qifan2019, 2022-07-18:
### Version/revision of the library used

I used the 2.8.2 version and the chip was esp-01s

### Describe the bug

I wrote the "CommonAcControl.ino" in chips which was example code.  set the  "kIrLed" =0 or 3（RX）, linking the GPIOs  to the irsend module's "in" pin,  it can work 

properly, sending ir signal  intermittently.

But when I wrote my code also used the GPIO0 or GPIO3 as "kreLed", the problem came.

**The GPIOs would keep high potential**，of course, the irsend module would keep lighting. But  the "ac.sendAc()" is not work at all.   I was so puzzled.

I have tried not performing any serial I/O or using “Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY)”.

Those are useless
#### Example code used

My code:

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>
#include <string>
#include <WiFiManager.h>
#include <EEPROM.h>

// wifiname
const char *ssid = "qifan";
// wifipassword
const char *password = "123456789";

const char *mqttServer = "120.76.140.89";
const char *topic = "smartAcControl/esp8266";
const char *username = "esp8266";
const char *userpassword = "123456789";
const int mqtt_port = 1883;
////////////////////////////////////////

const uint16_t kIrLed = 0; // The ESP GPIO pin to use that controls the IR LED.
bool flag = true;
// if ac matched
bool res;
char *acControl = "acControl";
char *setProtocol = "setProtocol";
char *matchProtocol = "matchProtocol";
char *power = "power";
char *temperature = "temperature";
char *mode = "mode";

/////////////////initwifi
IRac ac(kIrLed);
WiFiClient wifiClient;
PubSubClient mqttClient(wifiClient);
// Create a A/C object using GPIO to sending messages with.

////////////////////////////

void setup()
{
    // digitalWrite(kSendSinal,LOW);
    Serial.begin(115200);
    delay(200);
    // set wifi workmode
[... 242 lines cut ...]
    ac.sendAc(); // Have the IRac class create and send a
    Serial.println("Sending a message to turn ON the A/C unit.");
}

void temperatureControl(int temp)
{
    ac.next.degrees = temp;
    ac.sendAc();
}

```
#### Expected behaviour
_What steps did you do and what should it have done? A clear and concise description of what you expected to happen._

I tried to test every function, but didn't find the reason

I guess the problem is about the function “loop” ，but i don’t know why and how to solve it

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

### Has this library/code previously worked as expected for you?
No. 

### Other useful information

crankyoldgit, 2022-07-19:
> I have tried not performing any serial I/O or using “Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY)”.

I am confused. Your code is littered with `Serial` commands. e.g.
```c++
    Serial.println("Sending a message to turn ON the A/C unit.");
```

Have you tried simplifying your code down to determin where the problem might be?

Have you tried it on a more capable board than the ESP-01 to debug your code? e.g. a WEMOS D1-mini or a NodeMCU board?
i.e. One with more GPIOs so you can debug a bit better.
What's the debug output on the serial?

Are you sure you're parsing your MQTT values correctly? Are you setting the parameters with the values you think you are? i.e. `integers` not `strings` etc.

crankyoldgit, 2022-07-19:
See also: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue

qifan2019, 2022-07-20:
> I am confused. Your code is littered with `Serial` commands. e.g.
>
> ```c++
>     Serial.println("Sending a message to turn ON the A/C unit.");
> ```
>
> Have you tried simplifying your code down to determin [...]

Thank you very much for your reply.

There are some supplemental Information

1、I had deleted all the “Serial”  commands，but it didn‘t solve problem. The code I gave is the version  that didn't delete “Serial”  commands

2、I have bought NodeMCU board based esp-12F，it is on delivery.

3、I have tried to delete or add code to debug， it is so interesting ，if i add codes：

```
 for (int i = 1; i < kLastDecodeType; i++) {
    decode_type_t protocol = (decode_type_t)i;
    // If the protocol is supported by the IRac class ...
    if (ac.isProtocolSupported(protocol)) {
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol) + " is supported.");
      ac.next.protocol = protocol;  // Change the protocol used.
      ac.next.power = true;  // We want to turn on the A/C unit.
      Serial.println("Sending a message to turn ON the A/C unit.");
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(5000);  // Wait 5 seconds.
      ac.next.power = false;  // Now we want to turn the A/C off.
      Serial.println("Send a message to turn OFF the A/C unit.");
      ac.sendAc();  // Send the message.
      delay(1000);  // Wait 1 second.
    }
  }
```
in my program，the GPIOs would work normally

4、I determine  that I parsing my MQTT values correctly，bacause the output on the serial is normally

5、So far I think the problem is with the function loop, but I don't know why and how to fix it

crankyoldgit, 2022-07-20:
> 4、I determine that I parsing my MQTT values correctly，bacause the output on the serial is normally

Um, you do know that:
```c++
String str = "2";
Serial.println(str);  // An arduino string
Serial.println("2");  // A c-style string.
Serial.println('2');  // An ascii uint8_t/byte.
Serial.println(2);  // An integer
int i = 2;
Serial.println(i);  // An integer.
```
all produce exactly the same output, right?! Yet they are all different types.
e.g. '2' == 0x32 == 50  and '2' != 2 != "2"

`Serial.println()` will convert types to give you a result most people want.

I suggest you put in some appropriate debugging code to indicate you're parsing the numbers from MQTT correctly.

You need to test and debug all of your assumptions.
eg.
```c++
if (my_parsed_protocol_number == EXPECTED_PROTOCOL_NUMBER) {  // e.g. 42 or something
  // Success. I got the correct protocol number I expected.
  Serial.println("DEBUG: Yay, I got the value of 42!");
}
```

or simplify your code to remove the parsing from MQTT, hardwire it, and slowly re-add bits back to see where it fails.

But, we are now heading into [this](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me) teritory.

> 3、I have tried to delete or add code to debug， it is so interesting ，if i add codes：

To me, this really indicates you've got a problem parsing the values to put into the `ac.next` structure.

crankyoldgit, 2022-08-16:
How's this going?

qifan2019, 2022-11-09:
Thank you for your support, I was looking for a job in the past few months, so I forgot about it, I will continue to troubleshoot the problem afterwards