crankyoldgit/IRremoteESP8266 #1848: Need Help: Detail Support For Hitachi PC-LH3B

bingooo, 2022-07-29:
Hello, I have a Hitachi PC-LH3B, and did some data collection base on #1063 .

[Google Sheet: HITACHI PC-LH3B](https://docs.google.com/spreadsheets/d/1aLBjY75B8-MAbgXZJMtWbgb5Ncry6fmqMH7YWDY48_M/edit?usp=sharing)

Is that possiable to add `Irac` support for this?

Thanks.

kmihaylov, 2023-04-06:
Hello. Thank you for your data. How do you use the library with HITACHI_AC3?

Something like ...?
```
begin();
stateReset();
setRaw();
send();
```

crankyoldgit, 2023-04-06:
@kmihaylov  `HITACHI_AC3` does not yet have detailed support. The code for it has not been added to the library.

kmihaylov, 2023-04-07:
@crankyoldgit thank you for your fast reply. I'm looking ```ir_Hitachi.cpp``` where the ```IRHitachiAc3``` class has ```begin()```, ```setRaw()```, etc. Isn't these enough to send some of the codes given by @bingooo ?

crankyoldgit, 2023-04-07:
Ah yes. It doesn't have detailed support, as in you can't tell the library to change the AC to Heat mode etc, or set the fan to Medium etc. You can however set the state code/array, as it does have enough for that.

So yes, you can set the state via `setRaw()` for that class, and use the `send()` method.
You should only need to call/use the `begin()` and `stateReset()` once. Heck, you don't even need to invoke `stateReset();` actually.

See the IRsendDemo.ino code for an example.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/c1a014a61fa739d7af88526b1a0ed26d779d8fb4/examples/IRsendDemo/IRsendDemo.ino#L47-L50
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/c1a014a61fa739d7af88526b1a0ed26d779d8fb4/examples/IRsendDemo/IRsendDemo.ino#L71-L72

The above shows you how to do it without using the `IRHitachiAc3` class. i.e. via `irsend.sendHitachiAc3(state_array);`
Currently the class is just scaffolding for when someone writes the rest of the code/decodes the protocol fully. There really is no point using the class till that's been added.