crankyoldgit/IRremoteESP8266 #1851: Question: How to detect at runtime what protocols are actually included in the build

tonhuisman, 2022-08-07:
While working on reducing the memory use and .bin size of ESPEasy, also see issue #1849, we are currently working on a way of excluding unused protocols (and that's already working as intended, but needs some fine-tuning, see https://github.com/letscontrolit/ESPEasy/pull/4181).

In ESPEasy, an end-user can select from a list of protocols for received IR codes (that also can be 'learned' automatically), but that list includes all theoretically available protocols, not the list of actually available protocols in the current build.
I have not found a way (yet) how to determine wether a protocol is available or not, so it can either be not included in the list (while parsing over the list as explained in #1181), or reducing that list to the selected/enabled protocols.

If needed I can start a PR to help solve this, f.e. by adding `#if DECODE_<protocol> || SEND_<protocol>` (or similar, as that's one way to change this) to the required spots in the code.

Not sure if this would be applicable to other parts of the code too.

NiKiZe, 2022-08-07:
I don't think there is any good way to do this.

Maybe we could have a send to gpio0 that returns failure (on non matched switch)

Another way could be to replace the name strings with nulls instead which could be used to skip them?

tonhuisman, 2022-08-07:
> replace the name strings with nulls instead which could be used to skip them?

Maybe that's the simplest way to solve that, I'm already testing the waters on what _I_ can do about this 😃 but it could coincide with the end-of-strings test 🤔, I'll try.

tonhuisman, 2022-08-07:
> replace the name strings with nulls instead which could be used to skip them?

Empty string won't work, as I expected, but using a question mark does, and then checking for the output length > 1 (shortest name is `LG`), giving this code for filling the vectors that will eventually populate the combobox:

```c++
  int size = static_cast<int>(decode_type_t::kLastDecodeType) + 1;
  //...
  int protocolCount = 0;
  for (int i = 0; i < size; i++) {
    const String protocol = typeToString(static_cast<decode_type_t>(i), false);
    if (protocol.length() > 1) {
      decodeTypeOptions.push_back(i);
      decodeTypes.push_back(protocol);
      protocolCount++;
    }
  }
```

crankyoldgit, 2022-08-08:
Currently the ways to detect which protocols are supported at runtime are:
[`bool IRac::isProtocolSupported(const [decode_type_t] protocol)`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html#ad9c2fc9d07db70704f78a2d5f7be5b1c) which should be a static method. Unfortunately, that only works for A/C protocols that have have _detailed_ support.

You could abuse/repurpose the [`IRsend::send()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a204eedc3ad182fb2f40c42ef58f78cfc) (there are [two](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#ac684c209ea8722f0a377070752df0040), one for `uint64_t`'s and one for `*uint8_t`'s). It/They will return `true` if it was able to send using it.

Note: Both the above are only valid for _sending_, not receiving/decoding.

Adding something similar to [`IRac::isProtocolSupported(protocol)`](https://github.com/crankyoldgit/IRremoteESP8266/blob/85d587fbbf804c25c15632849386091e712a6cae/src/IRac.cpp#L147-L333) to the [`IRsend`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html) & [`IRrecv`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html) classes is the best/better solution. It shouldn't be much effort.

jimmys01, 2022-08-08:
How about this:
from the file P035_data_struct.cpp from the ESPeasy project.

`String P035_data_struct::listProtocols() {
  String temp;
  for (uint32_t i = 0; i <= kLastDecodeType; i++) {
    if (IRsend::defaultBits((decode_type_t)i) > 0) {
      temp += typeToString((decode_type_t)i) + ' ';
    }
  }
  return temp;}`

`String P035_data_struct::listACProtocols() {
  String temp;
  for (uint32_t i = 0; i <= kLastDecodeType; i++) {
    if (hasACState((decode_type_t)i)) {
      temp += typeToString((decode_type_t)i) + ' ';
    }
  }
  return temp;
}
`

tonhuisman, 2022-08-08:
> `IRsend::defaultBits((decode_type_t)i)`

That `switch` then should have the `case` lines wrapped with #if SEND_<protocol>/#endif` to return 0, and smth. similar for `IRrecv`.

crankyoldgit, 2022-08-16:
@jimmys01 That won't work really well either. `FUJITSU` defaults to `0` bits if/when enabled due to the fact of legacy/no standard bit size.

PR #1853 (2022-08-17): Inhibit protocol names for not-included protocols
Resolves: #1851 

To solve the request of how to determine at runtime what protocols are actually compiled in, after disabling unneeded protocols, I chose to replace the protocol name by a question mark (empty string can't work in a `\0x0` separated and `\0x0\0x0` terminated string sequence). The shortest protocol name found so far is 'LG'.
To avoid having to do a double/double check, for both IR and AC, and both SEND and RECEIVE protocols, and usually both SEND and RECEIVE for a used protocol are expected to be included, checking for a valid n [... cut]
commit 7a8cf9c6 (2022-08-17): Inhibit protocol names for not-included protocols (#1853)

* Inhibit protocol names for not-included protocols
* Move Protocol string-selection to macro, in new file IRmacros.h
* Regenerate IRtext.h

Resolves #1851
closed 2022-08-17 by crankyoldgit