crankyoldgit/IRremoteESP8266 #1854: IRsendDemo compiling error with nikai protocol

lucki1000, 2022-08-13:
### Version/revision of the library used
IRremoteESP8266 in Version 2.8.2
### Describe the bug
Hi,
I captured with the IRrecvDumpV3 example one key from my remote control and want to output the captured signal, but while compiling I get this error: `compilation error: invalid conversion from 'uint64_t*' {aka 'long long unsigned int*'} to 'uint64_t' {aka 'long long unsigned int'} [-fpermissive]`
here is my code, maybe I misunderstand which parameters I have to pass to the `irsend.sendNikai(rawData, 259, 1);` function:
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint64_t rawData[259] = {
          4100, 3950,  604, 1946,  602, 1948,  602, 1948,  602, 1948,  600, 1948,  602, 1948,  
          604, 948,  600, 952,  600, 1948,  604, 1948,  600, 1948,  600, 1950,  602, 948,  602,
          988,  604, 934,  602, 948,  602, 948,  604, 948,  602, 1948,  602, 1948,  604, 948,
          602, 946,  604, 948,  604, 946,  604, 8470,  4100, 3946,  604, 1948,  604, 1946,  604, 1946,  
          604, 1946,  602, 1948,  602, 1948,  602, 946,  604, 948,  604, 1946,  602, 1950,  602, 1948,  
          602, 1948,  602, 948,  604, 986,  602, 936,  604, 948,  602, 948,  604, 948,  602, 1948,
          602, 1948,  604, 946,  604, 946,  604, 948,  604, 946,  602, 8468,  4104, 3944,  604, 1946,
          604, 1948,  602, 1948,  602, 1948,  604, 1948,  602, 1948,  602, 948,  602, 948,  600, 1948,
          600, 1952,  598, 1950,  600, 1950,  598, 952,  598, 1058,  574, 952,  598, 954,  598, 952,  598,
          954,  598, 1952,  596, 1954,  594, 956,  596, 956,  594, 956,  594, 958,  594, 8478,  4090, 3956,
          592, 1958,  592, 1958,  592, 1958,  592, 1958,  592, 1958,  590, 1960,  566, 986,  590, 962,  588,
          1962,  586, 1966,  564, 1986,  564, 1988,  564, 988,  562, 1040,  536, 990,  562, 990,  560, 1012,
          538, 1012,  538, 2012,  538, 2012,  538, 1014,  536, 1014,  536, 1014,  536, 1014,  536, 8564,  
          4030, 4018,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1020,  530,
          1020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1022,  530, 1074,  502, 1022,  530, 1022,  
          528, 1022,  530, 1022,  530, 2022,  528, 2020,  530, 1022,  530, 1022,  530, 1022,  528, 1022,  528
          };  // NIKAI 30FCF

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  Serial.println("NIKAI:");
  irsend.sendNikai(rawData, 259, 1);
  delay(2000);
}
```
#### To Reproduce
1. Flash IRrecvDumpV3 on a device.
2. capture signal with a TV Control who uses Nikai protocol
3. add captured signal to your send code & try to compile with this send command: ` irsend.sendNikai(rawData, length, repeats);`

#### Example code used
I use IRsendDemo as template.

#### Expected behaviour
It should compile, and I can test if I done it correctly. :)

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
My capture:
```
Timestamp : 000011.967
Library   : v2.8.2

Protocol  : NIKAI
Code      : 0x30FCF (24 Bits)
uint16_t rawData[259] = {4100, 3950,  604, 1946,  602, 1948,  602, 1948,  602, 1948,  600, 1948,  602, 1948,  604, 948,  600, 952,  600, 1948,  604, 1948,  600, 1948,  600, 1950,  602, 948,  602, 988,  604, 934,  602, 948,  602, 948,  604, 948,  602, 1948,  602, 1948,  604, 948,  602, 946,  604, 948,  604, 946,  604, 8470,  4100, 3946,  604, 1948,  604, 1946,  604, 1946,  604, 1946,  602, 1948,  602, 1948,  602, 946,  604, 948,  604, 1946,  602, 1950,  602, 1948,  602, 1948,  602, 948,  604, 986,  602, 936,  604, 948,  602, 948,  604, 948,  602, 1948,  602, 1948,  604, 946,  604, 946,  604, 948,  604, 946,  602, 8468,  4104, 3944,  604, 1946,  604, 1948,  602, 1948,  602, 1948,  604, 1948,  602, 1948,  602, 948,  602, 948,  600, 1948,  600, 1952,  598, 1950,  600, 1950,  598, 952,  598, 1058,  574, 952,  598, 954,  598, 952,  598, 954,  598, 1952,  596, 1954,  594, 956,  596, 956,  594, 956,  594, 958,  594, 8478,  4090, 3956,  592, 1958,  592, 1958,  592, 1958,  592, 1958,  592, 1958,  590, 1960,  566, 986,  590, 962,  588, 1962,  586, 1966,  564, 1986,  564, 1988,  564, 988,  562, 1040,  536, 990,  562, 990,  560, 1012,  538, 1012,  538, 2012,  538, 2012,  538, 1014,  536, 1014,  536, 1014,  536, 1014,  536, 8564,  4030, 4018,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1020,  530, 1020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1022,  530, 1074,  502, 1022,  530, 1022,  528, 1022,  530, 1022,  530, 2022,  528, 2020,  530, 1022,  530, 1022,  530, 1022,  528, 1022,  528};  // NIKAI 30FCF
uint64_t data = 0x30FCF;

```

### What brand/model IR demodulator are you using?
_Brand/Model code_ or _None_

My [sender](https://www.az-delivery.de/en/products/ir-sende-modul) & [reciever](https://www.az-delivery.de/en/products/ir-empfanger-modul)

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
There is no compiling section.

### Has this library/code previously worked as expected for you?
Not tested. 

### Other useful information
None

NiKiZe, 2022-08-13:
The code you probably want is 0x30FCF
If you want to use raw, then you need to use the function to send raw (but in this case the code is what you want)

crankyoldgit, 2022-08-13:
> here is my code, maybe I misunderstand which parameters I have to pass to the `irsend.sendNikai(rawData, 259, 1);` function:

Yes, you are misunderstanding the parameters.

Your current code:
```c++
void loop() {
  Serial.println("NIKAI:");
  irsend.sendNikai(rawData, 259, 1);
  delay(2000);
}
```
You're basically trying to send raw data with [`sendNikai()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a693e6616b81509cf27d1345c140acc96), to send raw data, you use [`sendRaw()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a2b9b84f828918f933bd1764d113b53f8).
[`sendNikai()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a693e6616b81509cf27d1345c140acc96) is used when you have the `NIKAI` code, which you have.
e.g. (from your text above)
> Protocol  : NIKAI
Code      : 0x30FCF (24 Bits)

You would send this using similar code that you're trying to currently use:
```c++
void loop() {
  Serial.println("NIKAI:");
  irsend.sendNikai(0x30FCF);
  delay(2000);
}
```

To send the `rawData[259]` from your code (first off, use `uint16_t` not `uint64_t` for the `rawData[]` type)
you would use:
```c++
uint16_t rawData[259] = {
          4100, 3950,  604, 1946,  602, 1948,  602, 1948,  602, 1948,  600, 1948,  602, 1948,  
          604, 948,  600, 952,  600, 1948,  604, 1948,  600, 1948,  600, 1950,  602, 948,  602,
          988,  604, 934,  602, 948,  602, 948,  604, 948,  602, 1948,  602, 1948,  604, 948,
          602, 946,  604, 948,  604, 946,  604, 8470,  4100, 3946,  604, 1948,  604, 1946,  604, 1946,  
          604, 1946,  602, 1948,  602, 1948,  602, 946,  604, 948,  604, 1946,  602, 1950,  602, 1948,  
          602, 1948,  602, 948,  604, 986,  602, 936,  604, 948,  602, 948,  604, 948,  602, 1948,
          602, 1948,  604, 946,  604, 946,  604, 948,  604, 946,  602, 8468,  4104, 3944,  604, 1946,
          604, 1948,  602, 1948,  602, 1948,  604, 1948,  602, 1948,  602, 948,  602, 948,  600, 1948,
          600, 1952,  598, 1950,  600, 1950,  598, 952,  598, 1058,  574, 952,  598, 954,  598, 952,  598,
          954,  598, 1952,  596, 1954,  594, 956,  596, 956,  594, 956,  594, 958,  594, 8478,  4090, 3956,
          592, 1958,  592, 1958,  592, 1958,  592, 1958,  592, 1958,  590, 1960,  566, 986,  590, 962,  588,
          1962,  586, 1966,  564, 1986,  564, 1988,  564, 988,  562, 1040,  536, 990,  562, 990,  560, 1012,
          538, 1012,  538, 2012,  538, 2012,  538, 1014,  536, 1014,  536, 1014,  536, 1014,  536, 8564,  
          4030, 4018,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1020,  530,
          1020,  530, 2020,  530, 2020,  530, 2020,  530, 2020,  530, 1022,  530, 1074,  502, 1022,  530, 1022,  
          528, 1022,  530, 1022,  530, 2022,  528, 2020,  530, 1022,  530, 1022,  530, 1022,  528, 1022,  528
          };  // NIKAI 30FCF
void loop() {
  Serial.println("NIKAI:");
  irsend.sendRaw(rawData, 259, 38000);  // Send the 259 item `rawData` array at 38,000Hz (38kHz)
  delay(2000);
}
```

Hope that helps.

lucki1000, 2022-08-13:
Thanks for the explanation, know I also understand my mistake @crankyoldgit :)

closed 2022-08-13 by lucki1000