crankyoldgit/IRremoteESP8266 #1856: Gree codes transmit not working

mjtrini, 2022-08-13:
I captured codes from a Lennox mini split air conditioner. The codes were identified as Gree however when sending the codes, the air conditioner does not respond. I used the command irsend.sendGree(0x4904605007C00030), for example to send a temperature command of 20C. The IR receiver on the ESP8266 test setup reads the reflected codes as unknown and 70bit.
Protocol  : UNKNOWN
Code      : 0x978C4EF6 (70 Bits)
uint16_t rawData[139] = {8978, 4550,  612, 1620,  588, 580,  590, 578,  612, 1616,  590, 580,  590, 580,  590, 1638,  590, 584,  590, 578,  612, 556,  606, 1622,  612, 556,  588, 582,  588, 578,  590, 578,  612, 556,  614, 554,  612, 556,  590, 578,  612, 556,  614, 556,  588, 1640,  612, 1616,  612, 558,  614, 556,  612, 556,  614, 556,  590, 578,  612, 1616,  612, 556,  612, 1616,  614, 572,  612, 556,  612, 1616,  614, 556,  614, 1038,  612, 1616,  612, 1614,  614, 1616,  612, 556,  612, 558,  612, 556,  614, 554,  614, 556,  614, 554,  614, 554,  614, 556,  612, 556,  614, 556,  614, 548,  612, 1616,  616, 1610,  614, 548,  614, 550,  616, 548,  616, 548,  614, 548,  614, 550,  614, 550,  614, 552,  612, 552,  612, 554,  612, 554,  614, 574,  584, 1644,  584, 1644,  604, 566,  602, 570,  578};  // UNKNOWN 978C4EF6

Below are the captured codes (via the remote control) from the temperature range 20 to 30C, fan auto, mode cool. Any help will be appreciated.

Protocol  : GREE
Code      : 0x4904605007C00030 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 20C, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 7 (UNKNOWN), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[279] = {9002, 4478,  654, 1652,  656, 552,  656, 548,  654, 1652,  652, 554,  652, 552,  650, 1656,  628, 578,  654, 550,  654, 552,  656, 1652,  628, 576,  654, 552,  656, 548,  654, 550,  630, 578,  654, 550,  658, 548,  654, 552,  654, 552,  656, 550,  654, 1654,  652, 1652,  656, 550,  652, 554,  652, 554,  652, 552,  654, 552,  654, 1652,  652, 552,  656, 1652,  654, 550,  654, 552,  652, 1654,  652, 552,  652, 19984,  652, 1654,  652, 1654,  650, 1656,  654, 552,  652, 552,  652, 554,  652, 554,  650, 556,  650, 554,  652, 554,  650, 556,  650, 554,  650, 556,  650, 556,  650, 1656,  650, 1658,  650, 556,  650, 554,  650, 558,  650, 556,  648, 556,  650, 556,  650, 558,  648, 558,  650, 558,  646, 558,  648, 558,  646, 558,  648, 1662,  646, 1660,  646, 560,  646, 560,  646, 40002,  8986, 4518,  624, 582,  622, 582,  622, 582,  622, 584,  622, 584,  622, 582,  622, 584,  620, 584,  620, 586,  620, 584,  620, 586,  620, 584,  618, 588,  616, 588,  618, 590,  616, 590,  616, 588,  616, 590,  616, 588,  618, 590,  594, 610,  616, 588,  616, 590,  616, 590,  616, 588,  616, 594,  612, 588,  616, 588,  594, 614,  616, 1690,  594, 612,  594, 1714,  616, 588,  616, 1690,  606, 600,  616, 20022,  592, 612,  592, 612,  594, 610,  616, 588,  618, 588,  616, 590,  592, 612,  618, 588,  616, 590,  616, 590,  616, 590,  616, 590,  594, 612,  616, 590,  594, 610,  616, 588,  594, 612,  616, 590,  616, 590,  594, 612,  616, 588,  618, 590,  594, 610,  592, 612,  616, 590,  616, 590,  592, 612,  616, 590,  616, 590,  616, 1690,  616, 590,  616, 1692,  594};  // GREE
uint8_t state[8] = {0x49, 0x04, 0x60, 0x50, 0x07, 0xC0, 0x00, 0x30};

Protocol  : GREE
Code      : 0x4905605007C00040 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 21C, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 7 (UNKNOWN), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[279] = {8998, 4478,  656, 1650,  658, 548,  652, 554,  654, 1652,  654, 550,  654, 552,  654, 1654,  652, 552,  656, 1652,  656, 550,  652, 1652,  656, 550,  678, 526,  656, 548,  656, 550,  654, 552,  654, 552,  654, 552,  654, 550,  654, 550,  654, 550,  660, 1648,  656, 1652,  654, 550,  654, 552,  654, 552,  656, 548,  654, 552,  654, 1652,  654, 552,  656, 1652,  656, 550,  658, 548,  656, 1650,  652, 552,  656, 19980,  656, 1652,  654, 1652,  654, 1652,  658, 546,  654, 552,  656, 548,  656, 550,  654, 550,  654, 552,  654, 550,  654, 552,  654, 550,  654, 552,  656, 550,  654, 1654,  654, 1654,  654, 552,  654, 550,  654, 550,  654, 550,  656, 550,  654, 552,  654, 552,  654, 554,  652, 554,  652, 554,  654, 552,  652, 552,  652, 552,  654, 552,  650, 1656,  654, 552,  652, 40000,  8988, 4488,  652, 554,  650, 556,  650, 554,  650, 556,  650, 554,  650, 558,  648, 556,  650, 554,  652, 554,  652, 552,  650, 556,  650, 556,  650, 556,  650, 556,  650, 556,  650, 556,  650, 554,  650, 554,  650, 556,  650, 554,  652, 556,  650, 556,  650, 558,  650, 556,  648, 558,  648, 556,  650, 558,  648, 558,  650, 558,  648, 1658,  648, 558,  648, 1660,  646, 560,  646, 1658,  646, 558,  646, 19990,  646, 558,  648, 560,  646, 560,  646, 560,  646, 560,  646, 580,  624, 582,  624, 582,  624, 582,  622, 582,  624, 582,  624, 582,  624, 582,  622, 584,  622, 582,  624, 582,  622, 582,  622, 584,  622, 584,  620, 584,  622, 584,  620, 584,  620, 584,  620, 586,  618, 586,  618, 590,  616, 588,  592, 614,  616, 588,  616, 1692,  594, 612,  618, 1692,  616};  // GREE
uint8_t state[8] = {0x49, 0x05, 0x60, 0x50, 0x07, 0xC0, 0x00, 0x40};

Protocol  : GREE
Code      : 0x4906605007C00050 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 22C, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 7 (UNKNOWN), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[279] = {8994, 4484,  654, 1650,  628, 576,  656, 552,  654, 1654,  652, 552,  650, 552,  654, 1654,  628, 576,  654, 552,  656, 1652,  630, 1676,  654, 552,  654, 550,  654, 552,  652, 554,  658, 548,  656, 548,  652, 554,  654, 552,  654, 550,  656, 550,  652, 1656,  654, 1652,  652, 552,  656, 550,  656, 550,  658, 548,  654, 550,  656, 1650,  658, 550,  654, 1654,  658, 548,  656, 550,  654, 1652,  654, 552,  656, 19980,  654, 1652,  652, 1654,  654, 1652,  658, 548,  658, 546,  652, 554,  654, 552,  656, 548,  654, 552,  656, 548,  654, 550,  656, 550,  654, 550,  656, 550,  656, 1652,  652, 1654,  654, 550,  656, 550,  654, 552,  652, 554,  652, 554,  652, 552,  654, 552,  654, 552,  652, 552,  654, 552,  652, 554,  652, 554,  652, 1654,  654, 552,  652, 1654,  652, 554,  652, 39996,  8992, 4486,  652, 554,  650, 554,  652, 556,  650, 556,  650, 554,  650, 556,  650, 554,  650, 556,  650, 556,  650, 556,  652, 554,  650, 554,  650, 556,  650, 556,  650, 554,  650, 556,  650, 556,  648, 558,  648, 556,  648, 556,  648, 558,  648, 558,  646, 558,  648, 556,  648, 556,  648, 556,  648, 558,  648, 556,  648, 558,  648, 1658,  648, 558,  646, 1660,  646, 560,  646, 1662,  646, 558,  646, 19990,  646, 560,  646, 580,  624, 582,  624, 582,  624, 580,  624, 580,  624, 580,  626, 580,  624, 582,  622, 582,  622, 582,  622, 582,  622, 584,  622, 582,  622, 582,  622, 584,  622, 584,  622, 584,  622, 584,  620, 584,  620, 586,  620, 586,  620, 586,  618, 588,  616, 590,  616, 588,  618, 588,  616, 590,  616, 588,  616, 1690,  618, 588,  616, 1690,  616};  // GREE
uint8_t state[8] = {0x49, 0x06, 0x60, 0x50, 0x07, 0xC0, 0x00, 0x50};

Protocol  : GREE
Code      : 0x4907605007C00060 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 23C, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 7 (UNKNOWN), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
uint16_t rawData[279] = {9050, 4454,  652, 1656,  650, 554,  628, 576,  654, 1654,  652, 552,  652, 552,  656, 1652,  652, 552,  654, 1652,  652, 1654,  654, 1654,  652, 552,  628, 576,  656, 550,  656, 552,  654, 550,  654, 552,  654, 552,  654, 552,  654, 550,  656, 550,  654, 1652,  658, 1650,  652, 554,  656, 550,  654, 550,  656, 548,  654, 552,  652, 1654,  652, 552,  652, 1656,  654, 552,  656, 548,  654, 1654,  654, 548,  656, 19984,  654, 1654,  678, 1628,  658, 1650,  654, 550,  656, 548,  654, 552,  656, 548,  656, 550,  654, 552,  654, 552,  658, 548,  656, 548,  682, 526,  654, 550,  656, 1650,  654, 1652,  654, 552,  654, 548,  654, 550,  656, 550,  656, 550,  654, 550,  654, 552,  652, 552,  656, 550,  654, 552,  652, 552,  656, 550,  652, 552,  652, 1656,  650, 1654,  652, 552,  654, 39992,  8992, 4486, [... cut ...] 0,  616, 1692,  616, 588,  616, 1690,  616, 590,  616, 20020,  616, 590,  616, 590,  616, 590,  616, 590,  616, 588,  618, 588,  616, 590,  616, 590,  616, 590,  616, 590,  616, 588,  592, 612,  616, 590,  616, 590,  616, 590,  616, 590,  616, 588,  616, 590,  616, 590,  614, 590,  618, 590,  616, 588,  616, 590,  616, 588,  616, 588,  616, 590,  616, 590,  616, 588,  616, 590,  616, 1692,  616, 590,  616, 1688,  616};  // GREE
uint8_t state[8] = {0x49, 0x0E, 0x60, 0x50, 0x07, 0xC0, 0x00, 0xD0};

crankyoldgit, 2022-08-14:
I think the problem here is the actual protocol being used is the `KELVINATOR` protocol. It is very similar (2x longer) than the `GREE` protocol and has a LOT of overlap with the first half. i.e. First half of `KELVINATOR` is almost exactly the same as the _entire_ `GREE` protocol
So we are getting a false detection .. I think. I still need to confirm that.

I am certain that your captured codes are not using the `GREE` protocol as we know it. The first part, as above, is very similar.

Try the [TurnOnKelvinatorAC.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnKelvinatorAC/TurnOnKelvinatorAC.ino) example code and let us know how it goes for you,

Meanwhile, I'll try to see if I can fix/correct the detection.

crankyoldgit, 2022-08-16:
Did the Kelvinator one work at all?

mjtrini, 2022-08-20:
Hi Sorry for the late reply, work has been crazy hectic. I didn't try the Kelvinator codes since the Gree example code works so well. Thanks for the assistance!

closed 2022-08-20 by mjtrini