crankyoldgit/IRremoteESP8266 #1859: Argo Ulisse 13 IR messages not decoding correctly

zpin, 2022-08-18:
### Version/revision of the library used
2.8.2/master

### Describe the bug
IR messages captured from an Argo Ulisse 13 remote are not detected as such. Also there seem to be messages of different length while the library only tries to match the longer ones.

#### To Reproduce
Use IRremoteESP8266: IRrecvDumpV2 to capture messages

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
```
10:18:33.818 -> Library   : v2.8.2
Manual ON
10:18:33.818 -> Protocol  : UNKNOWN
10:18:33.818 -> Code      : 0x5D1B22E5 (92 Bits)
10:18:33.818 -> uint16_t rawData[183] = {6418, 3168,  442, 836,  442, 834,  442, 2114,  442, 2114,  442, 836,  442, 2114,  442, 836,  440, 2114,  442, 2114,  442, 836,  442, 2112,  442, 834,  442, 2112,  442, 2112,  444, 2112,  442, 2114,  442, 836,  442, 836,  442, 834,  442, 834,  442, 2112,  442, 836,  442, 834,  442, 836,  440, 2114,  442, 836,  440, 2114,  442, 836,  442, 836,  440, 836,  442, 2112,  442, 2114,  442, 836,  442, 2114,  442, 2114,  442, 836,  442, 836,  442, 836,  440, 836,  442, 836,  442, 836,  442, 836,  442, 836,  440, 836,  442, 836,  440, 2114,  442, 2114,  442, 2114,  442, 836,  440, 838,  440, 836,  442, 836,  440, 838,  440, 836,  440, 836,  440, 836,  442, 2114,  440, 2116,  440, 2114,  442, 2114,  440, 2114,  440, 2114,  442, 2114,  442, 836,  440, 2114,  440, 836,  442, 2114,  442, 836,  440, 836,  442, 836,  442, 836,  440, 836,  440, 836,  440, 836,  440, 836,  440, 836,  440, 836,  440, 2114,  442, 836,  440, 2114,  442, 836,  440, 2114,  442, 836,  440, 2116,  440, 836,  440, 836,  440, 836,  440, 836,  442, 836,  440, 2114,  442};  // UNKNOWN 5D1B22E5

Manual OFF
10:18:40.641 -> Protocol  : UNKNOWN
10:18:40.641 -> Code      : 0x4162A6F0 (92 Bits)
10:18:40.641 -> uint16_t rawData[183] = {6384, 3206,  414, 864,  414, 864,  412, 2144,  412, 2142,  412, 864,  412, 2142,  414, 864,  412, 2144,  412, 2144,  410, 866,  412, 2142,  412, 866,  412, 2142,  412, 2144,  412, 2144,  412, 2170,  386, 866,  412, 864,  412, 866,  410, 866,  412, 2144,  412, 866,  412, 864,  412, 864,  412, 2144,  410, 866,  412, 2144,  412, 890,  386, 864,  412, 866,  412, 2144,  410, 2144,  412, 866,  410, 2172,  386, 2168,  386, 892,  360, 916,  386, 892,  386, 892,  386, 892,  386, 890,  386, 892,  384, 892,  386, 868,  408, 892,  386, 2170,  386, 2170,  384, 2170,  384, 892,  384, 892,  362, 916,  384, 892,  360, 916,  360, 916,  360, 916,  360, 918,  360, 2194,  360, 2194,  362, 2194,  386, 2170,  360, 2196,  360, 2196,  358, 2196,  360, 916,  384, 2172,  360, 918,  360, 2194,  360, 918,  358, 918,  362, 916,  360, 918,  360, 918,  360, 918,  360, 918,  360, 918,  360, 916,  360, 918,  360, 916,  360, 918,  360, 2196,  360, 918,  360, 2194,  362, 916,  360, 2196,  360, 918,  360, 918,  360, 918,  360, 2196,  360, 2196,  360, 918,  360};  // UNKNOWN 4162A6F0

Manual target temperature change 24->25
10:19:45.713 -> Protocol  : UNKNOWN
10:19:45.713 -> Code      : 0xD21627D (93 Bits)
10:19:45.713 -> uint16_t rawData[185] = {634, 402,  5320, 3258,  362, 916,  360, 916,  384, 2144,  412, 2170,  386, 892,  384, 2170,  360, 916,  360, 2194,  384, 2170,  360, 916,  362, 2194,  360, 916,  360, 2194,  362, 2194,  362, 2194,  360, 2168,  388, 916,  360, 916,  360, 916,  360, 916,  362, 2194,  360, 916,  362, 2194,  360, 916,  360, 2196,  360, 918,  360, 2194,  360, 918,  360, 916,  360, 918,  360, 2194,  360, 2194,  362, 916,  360, 2196,  360, 2194,  362, 916,  360, 916,  360, 918,  358, 918,  360, 918,  360, 916,  360, 918,  360, 918,  360, 916,  360, 918,  360, 2196,  358, 2196,  360, 2196,  360, 918,  358, 918,  360, 918,  360, 918,  360, 918,  360, 918,  360, 918,  360, 918,  358, 2196,  360, 2196,  360, 2196,  360, 918,  358, 918,  358, 918,  362, 916,  360, 2196,  358, 2196,  360, 918,  358, 2196,  358, 918,  360, 918,  358, 918,  358, 918,  358, 918,  360, 918,  360, 918,  358, 918,  358, 920,  358, 918,  358, 2198,  358, 918,  358, 2196,  358, 918,  358, 2196,  358, 1010,  268, 2198,  358, 920,  356, 2198,  358, 920,  358, 1010,  266, 2198,  358, 2198,  358};  // UNKNOWN D21627D

Manual target temperature change 25->24
10:20:14.030 -> Protocol  : UNKNOWN
10:20:14.030 -> Code      : 0xE923CB07 (92 Bits)
10:20:14.030 -> uint16_t rawData[183] = {6386, 3206,  414, 862,  416, 862,  414, 2140,  414, 2140,  414, 862,  416, 2140,  414, 864,  414, 2140,  414, 2142,  414, 862,  414, 2142,  414, 864,  414, 2142,  414, 2142,  414, 2142,  414, 2140,  414, 862,  414, 864,  414, 862,  414, 864,  414, 2142,  414, 864,  412, 862,  416, 862,  414, 2142,  414, 864,  412, 2142,  414, 864,  414, 866,  412, 864,  412, 2142,  414, 2142,  414, 864,  412, 2142,  412, 2142,  412, 864,  414, 862,  414, 864,  414, 864,  412, 864,  412, 864,  414, 888,  388, 874,  404, 864,  414, 890,  386, 2142,  414, 2142,  412, 2142,  412, 866,  412, 890,  388, 890,  386, 890,  388, 890,  388, 890,  386, 866,  412, 890,  388, 2168,  388, 2168,  388, 2168,  386, 890,  386, 890,  386, 890,  386, 890,  386, 2168,  388, 2168,  388, 890,  386, 2168,  386, 892,  386, 890,  386, 890,  386, 890,  388, 890,  386, 892,  386, 890,  386, 892,  386, 890,  386, 890,  386, 2170,  386, 892,  386, 2170,  386, 890,  386, 2170,  386, 890,  386, 2170,  386, 892,  386, 2170,  386, 890,  386, 892,  386, 890,  386, 2170,  386};  // UNKNOWN E923CB07

Manual target temperature change 24->25
10:42:12.339 -> Protocol  : UNKNOWN
10:42:12.339 -> Code      : 0x2D0000E3 (92 Bits)
10:42:12.339 -> uint16_t rawData[183] = {6418, 3168,  442, 834,  442, 834,  442, 2112,  444, 2112,  444, 832,  444, 2112,  444, 834,  442, 2112,  444, 2112,  442, 834,  444, 2112,  444, 834,  442, 2112,  442, 2112,  442, 2112,  444, 2112,  444, 834,  444, 834,  442, 834,  442, 834,  442, 2112,  444, 834,  444, 2112,  444, 834,  442, 2112,  442, 834,  442, 2112,  442, 836,  442, 834,  442, 2112,  444, 2112,  442, 2112,  442, 834,  442, 2112,  444, 2112,  442, 834,  442, 834,  442, 834,  444, 834,  442, 834,  442, 834,  442, 836,  442, 834,  442, 834,  442, 836,  442, 2112,  442, 2112,  444, 2112,  442, 836,  442, 834,  442, 834,  442, 834,  442, 836,  442, 836,  442, 834,  442, 836,  442, 2114,  442, 2112,  442, 2112,  442, 834,  442, 2112,  442, 2114,  442, 834,  442, 836,  442, 836,  442, 2112,  442, 2112,  442, 834,  442, 836,  442, 836,  442, 834,  442, 836,  442, 836,  442, 836,  442, 836,  442, 836,  442, 836,  442, 2114,  442, 836,  440, 2114,  442, 836,  442, 2114,  442, 2114,  442, 2114,  442, 836,  440, 2114,  442, 2114,  442, 836,  442, 836,  440, 2114,  442};  // UNKNOWN 2D0000E3

Silent temperature message 30C:
10:22:04.919 -> Protocol  : UNKNOWN
10:22:04.919 -> Code      : 0x936639B2 (34 Bits)
10:22:04.919 -> uint16_t rawData[67] = {6386, 3204,  414, 862,  416, 862,  414, 2140,  414, 2140,  414, 864,  414, 2140,  416, 862,  414, 2140,  416, 2140,  416, 862,  414, 2142,  414, 864,  414, 2142,  414, 2140,  414, 2142,  414, 2142,  414, 862,  414, 2142,  416, 862,  414, 864,  414, 2140,  414, 862,  414, 2142,  414, 2142,  414, 2142,  412, 2142,  414, 864,  412, 864,  414, 2142,  412, 2144,  412, 2142,  414, 864,  412};  // UNKNOWN 936639B2

Silent temperature message 29C:
10:23:09.423 -> Protocol  : UNKNOWN
10:23:09.423 -> Code      : 0x716DD4C0 (34 Bits)
10:23:09.423 -> uint16_t rawData[67] = {6420, 3168,  442, 834,  442, 834,  442, 2112,  444, 2112,  442, 834,  442, 2112,  444, 834,  442, 2112,  442, 2112,  442, 834,  444, 2112,  442, 834,  442, 2112,  442, 2114,  442, 2112,  442, 2112,  442, 836,  442, 2112,  444, 836,  442, 2112,  444, 834,  442, 836,  442, 2112,  442, 2114,  442, 2112,  442, 2112,  442, 834,  442, 2114,  442, 836,  442, 2112,  442, 2114,  442, 836,  442};  // UNKNOWN 716DD4C0

Silent temperature message 28C:
10:27:22.916 -> Protocol  : UNKNOWN
10:27:22.916 -> Code      : 0x6149090 (34 Bits)
10:27:22.916 -> uint16_t rawData[67] = {6418, 3168,  444, 834,  444, 834,  442, 2112,  444, 2112,  444, 834,  442, 2114,  442, 834,  442, 2112,  444, 2112,  444, 832,  442, 2114,  442, 834,  442, 2112,  442, 2114,  442, 2112,  444, 2112,  442, 834,  442, 2112,  444, 834,  442, 834,  442, 834,  442, 834,  442, 2114,  442, 2114,  442, 2112,  442, 2112,  442, 834,  444, 834,  442, 834,  442, 2112,  442, 2112,  442, 836,  442};  // UNKNOWN 6149090
```

### Circuit diagram and hardware used (if applicable)
https://www.wemos.cc/en/latest/d1/d1_mini.html
https://www.wemos.cc/en/latest/d1_mini_shield/ir.html

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

crankyoldgit, 2022-08-21:
@zpin Can you please download and try branch [Issue1859](https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1859) https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1859 and let me know how it goes.

It should decode the (very) short messages now. It doesn't do everything yet.
I've also tried to use more common code for your `sendSensorTemp()` too. So please re-test that too.

Also, can you determine the minimum and maximum for the sensor temp from the remote?
e.g. Put it in the fridge/freezer to get the minimum temp. Maybe a hairdryer for the maximum.

zpin, 2022-08-23:
Seems to work fine from what I can tell. It's hard to test exactly because there's no acknowledgement and the unit takes several minutes to react to a change. It also manages to decode the temperature messages but still doesn't recognize the normal messages.

Temperature range looks like [5...35] celsius.
Capture starting at 29, heating up (40+ C I'd say), then letting cool down:
[tmphigh.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9400704/tmphigh.txt)
Capture starting at <0 C, then letting warm up:
[tmplow.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/9400705/tmplow.txt)

crankyoldgit, 2022-08-23:
Looking at that data, it seems 35C is the max sensor temp recorded, and the lowest is 5C.

I'll look at adding the appropriate limits to `sendSensorTemp()` so we don't have anything go terribly wrong.

Thanks for confirming the sensor temps are being detected.

crankyoldgit, 2022-08-23:
I've added bounds checking to that funtion, so we don't get any unexpected results.
FWIW, mathematically the minimum sensor temp looks like it could be 4C. So I've allowed that.
Branch updated etc.

Now, what is the deal with the rest of the messages.

What messages are correctly decoded & which ones are not?

Also, can you please provide all the brand/model numbers of the remote and the A/C unit?

The A/C support for this protocol was user submitted code. So it doesn't have the same level of unit test coverage/support and confidence. (i.e. I can't verify that it works fully)

zpin, 2022-08-23:
It's a `AxAir Ulisse 13 DCI ECO` and the remote says `SAC WREM-2 V1`. It looks like it comes under different brands (Argo, AxAir, Krueger, AirBlue, ...), as `Ulisse 13 DCI ECO` but there's others that look exactly the same, e.g. `Technibel SCDF32 4000W`, `Climia CMK 4000`.

It might have been an incomplete capture before, it does show ARGO now but the sensor temp is wrong, it should use the `RoomTemp` from these messages instead. Also, these messages are not 32bit but somewhere around 90-96 (96 according to the size of the state struct but that didn't capture correctly before). I suspect the temperature messages might also be longer but maybe don't get recorded correctly because they are mostly empty.

Here's a few captures of normal remote functions:
```
ON:
13:48:02.392 -> Timestamp : 000090.093
13:48:02.392 -> Library   : v2.8.2
13:48:02.392 -> 
13:48:02.392 -> Protocol  : ARGO
13:48:02.392 -> Code      : 0xACF5D0E4 (32 Bits)
13:48:02.392 -> Mesg Desc.: Sensor Temp: 30C
13:48:02.392 -> uint16_t rawData[183] = {6418, 3168,  442, 836,  440, 836,  440, 2114,  440, 2114,  442, 836,  440, 2114,  440, 836,  442, 2114,  442, 2114,  440, 836,  440, 2114,  442, 836,  440, 2114,  442, 2114,  442, 2114,  442, 2114,  442, 836,  440, 836,  442, 836,  440, 836,  440, 2114,  440, 838,  440, 2114,  442, 2114,  440, 836,  440, 836,  442, 2114,  440, 836,  440, 836,  442, 2114,  440, 2116,  440, 2114,  440, 836,  440, 2114,  442, 2114,  440, 836,  440, 838,  440, 838,  440, 836,  440, 836,  440, 836,  440, 836,  442, 836,  438, 838,  440, 836,  440, 2114,  440, 2114,  440, 2114,  440, 838,  440, 838,  440, 836,  440, 838,  440, 838,  440, 838,  440, 836,  440, 838,  440, 2116,  440, 2114,  440, 2116,  440, 838,  440, 838,  438, 838,  440, 836,  440, 838,  440, 838,  438, 838,  438, 2116,  440, 2114,  440, 838,  440, 838,  438, 838,  440, 838,  440, 838,  438, 840,  438, 838,  440, 838,  440, 838,  440, 2116,  440, 838,  438, 2116,  440, 838,  440, 2116,  440, 838,  440, 838,  438, 838,  438, 838,  438, 2116,  440, 2116,  438, 2116,  440, 2116,  438};  // ARGO
13:48:02.491 -> uint8_t state[4] = {0xAC, 0xF5, 0xD0, 0xE4};
13:48:02.491 -> 
13:48:02.491 -> 

OFF:
13:48:04.312 -> Timestamp : 000092.033
13:48:04.312 -> Library   : v2.8.2
13:48:04.312 -> 
13:48:04.312 -> Protocol  : ARGO
13:48:04.312 -> Code      : 0xACF5D0E4 (32 Bits)
13:48:04.312 -> Mesg Desc.: Sensor Temp: 30C
13:48:04.345 -> uint16_t rawData[183] = {6418, 3170,  440, 838,  438, 838,  438, 2114,  440, 2114,  440, 866,  412, 2114,  440, 840,  438, 2114,  440, 2114,  440, 864,  412, 2114,  442, 864,  412, 2114,  442, 2114,  440, 2116,  440, 2114,  440, 866,  412, 866,  412, 866,  412, 866,  412, 2116,  440, 864,  412, 2116,  440, 2114,  440, 866,  412, 866,  412, 2116,  440, 866,  412, 866,  412, 2116,  440, 2114,  440, 2116,  440, 866,  412, 2114,  440, 2116,  440, 866,  412, 864,  412, 866,  412, 864,  412, 864,  412, 866,  410, 866,  410, 866,  412, 866,  410, 864,  412, 2116,  440, 2116,  440, 2116,  440, 866,  410, 866,  412, 866,  412, 866,  412, 866,  412, 866,  410, 866,  412, 866,  412, 2116,  440, 2116,  440, 2114,  440, 866,  412, 866,  412, 866,  410, 866,  412, 866,  410, 866,  412, 866,  412, 2116,  440, 2116,  438, 866,  412, 866,  412, 866,  412, 866,  412, 866,  412, 866,  412, 866,  412, 866,  412, 866,  410, 866,  412, 866,  412, 2116,  438, 866,  410, 2116,  440, 866,  410, 866,  410, 866,  412, 866,  410, 2116,  438, 868,  410, 2116,  438, 2116,  438};  // ARGO
13:48:04.412 -> uint8_t state[4] = {0xAC, 0xF5, 0xD0, 0xE4};
13:48:04.412 -> 
13:48:04.412 -> 

ON:
13:48:06.233 -> Timestamp : 000093.964
13:48:06.233 -> Library   : v2.8.2
13:48:06.266 -> 
13:48:06.266 -> Protocol  : ARGO
13:48:06.266 -> Code      : 0xACF5D0E4 (32 Bits)
13:48:06.266 -> Mesg Desc.: Sensor Temp: 30C
13:48:06.266 -> uint16_t rawData[183] = {6418, 3168,  440, 836,  440, 836,  442, 2114,  440, 2114,  442, 836,  440, 2114,  440, 836,  440, 2114,  442, 2114,  440, 836,  440, 2114,  440, 838,  440, 2114,  442, 2114,  442, 2114,  440, 2114,  440, 838,  440, 838,  440, 836,  440, 836,  440, 2114,  440, 836,  440, 2114,  440, 2116,  440, 838,  440, 838,  440, 2114,  440, 838,  440, 836,  440, 2116,  440, 2114,  440, 2116,  440, 838,  440, 2116,  440, 2114,  440, 838,  438, 838,  438, 838,  440, 838,  440, 838,  438, 838,  440, 838,  440, 838,  438, 840,  438, 838,  438, 2116,  440, 2116,  440, 2116,  440, 838,  438, 838,  438, 838,  438, 866,  412, 840,  438, 840,  438, 864,  412, 840,  438, 2116,  440, 2116,  440, 2116,  440, 840,  438, 840,  438, 866,  412, 838,  440, 864,  412, 866,  410, 866,  412, 2116,  440, 2116,  438, 866,  412, 866,  410, 866,  412, 866,  412, 866,  410, 866,  410, 866,  410, 866,  410, 866,  412, 2116,  438, 866,  412, 2116,  440, 866,  410, 2116,  440, 866,  412, 866,  412, 866,  412, 866,  410, 2116,  440, 2116,  440, 2116,  438, 2116,  438};  // ARGO
13:48:06.365 -> uint8_t state[4] = {0xAC, 0xF5, 0xD0, 0xE4};
13:48:06.365 -> 
13:48:06.365 -> 

Change target temperature from 23 to 24:
13:48:10.107 -> Timestamp : 000097.822
13:48:10.107 -> Library   : v2.8.2
13:48:10.107 -> 
13:48:10.107 -> Protocol  : ARGO
13:48:10.107 -> Code      : 0xACF510E5 (32 Bits)
13:48:10.107 -> Mesg Desc.: Sensor Temp: 6C
13:48:10.107 -> uint16_t rawData[183] = {6420, 3168,  442, 836,  440, 838,  440, 2114,  442, 2114,  442, 836,  440, 2114,  442, 836,  440, 2114,  440, 2114,  440, 836,  440, 2114,  440, 836,  442, 2114,  440, 2116,  440, 2114,  440, 2114,  440, [... cut ...]  438, 2116,  440, 2114,  440, 2116,  440, 840,  438, 840,  438, 838,  438, 840,  438, 864,  412, 840,  438, 840,  438, 2116,  440, 2116,  440, 866,  412, 866,  412, 866,  410, 866,  412, 866,  412, 864,  412, 866,  412, 866,  410, 866,  410, 2116,  440, 866,  412, 2116,  440, 866,  412, 2116,  438, 2116,  440, 866,  410, 866,  412, 866,  410, 2116,  438, 2116,  440, 866,  412, 866,  412};  // ARGO
13:48:10.207 -> uint8_t state[4] = {0xAC, 0xF5, 0x10, 0xE5};
13:48:10.207 -> 
13:48:10.207 -> 

```

mbronk, 2022-10-16:
Hi,
I'm currently in process of adding support for what seems to be next generation of this remote (`WREM-3` - it's capacitive btn./touch-based, see: https://argoclima.com/en/prodotti/argo-ulisse-eco/ ). 
While at it, I figured I could extract/reuse common code, but unfortunately as I don't own a `WREM-2` remote, so probably won't be able to refactor the code too much. 
From what I gathered, the new remote seems to have a slightly **different** AC protocol (message lengths, checksums calculated differently...), but seems to have quite a few similarities as well. 

For example, the new remote has 4 types of commands:
1. `[0b00]` Regular/operation  -  6-byte IR command (AC operation, like on/off, temp set...)
1. `[0b01]` iFeel Temperature report -  2-byte IR command (temp+3-bit checksum)
1. `[0b10]` Timer command - 9-byte
1. `[0b11]` Config command - 4-byte

Am I reading it correctly that for WREM-2, "regular" + "timer" is actually one jumbo command, and "config" does not exist at all? 

If someone who owns this remote could confirm it would help me decide how much is there to reuse. Thx in advance!

mbronk, 2022-10-16:
On another note... the [`IRArgoAC::sendSensorTemp()`](https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/ir_Argo.h#L134) method which sends the (shorter) iFeel report is only exposed from  `IRArgoAC` itself. I would _really_ like for this to be exposed through https://github.com/crankyoldgit/IRremoteESP8266/blob/9228143207ab7e2968a42d9a92e246237af7bb83/src/IRac.h#L74 ... which would allow for a **generic** interface, and straightforward use in downstream projects, like Tasmota IR, for example (e.g. somewhere [here](https://github.com/arendst/Tasmota/blob/5915ffdc48de3da82a879b5043a1261b64a1cab8/tasmota/tasmota_xdrv_driver/xdrv_05_irremote_full.ino#L482)).

----
Extending [`stdAc::state_t`](https://github.com/crankyoldgit/IRremoteESP8266/blob/9228143207ab7e2968a42d9a92e246237af7bb83/src/IRsend.h#L97-L117) to sth like 
```c++
struct state_t {
  // (...) existing fields
  float sensorTemp = -1.0;  // -1 for not present
                            // consider: std:optional<>, when project moves to c++17
};
``` 
seems straightforward enough (esp. since I see that lot of AC protocols have `iFeel` and `getSensorTemp()` / `setSensorTemp(...)`.

What I struggle with, is - given Argo sends separate commands (_e.g. 4B for iFeel report as per my post above_) - **how should the _generic i-face_ allow the user to choose what they want**? (it's an either-or choice for this protocol).
Naive implementation: (_if `sensorTemp` is set -> send it and disregard any other state param_) would work, but is not exactly scalable nor explicit. On the other hand, translating one `sendAc(...)` invocation into, potentially, a series of 1..4 commands does not seem to be too intuitive either :(

 @crankyoldgit do you have a preference or pointers here? Is something like this being done anywhere in the library already?

mbronk, 2022-10-22:
Sorry for the churn...
@crankyoldgit  - are you by any chance planning to merge `Issue1859` branch anytime soon?

I'm adding the WREM3 remote support (new Argo remote type) and, while the code would be mostly addon on top of what is there, there is a **common** part I'm touching as well. For now I've branched off of `Issue1859` to minimize merge conflicts, but can also base off of `master` if the former is not going to be merged soon.
I hope to have my PR ready this or next weekend, likely. Please advise

crankyoldgit, 2022-10-22:
Apologies. I had forgotten about that branh. No sure were I was with that. Created PR #1906 to merge it in.

crankyoldgit, 2022-10-22:
@mbronk it's now merged

mateuszdrab, 2024-08-28:
Hey guys

I don't mean hijack the issue; however, I also seem to be experiencing decoding issues with a `Technibel WREM2 v1` remote.
This issue can be reproduced on both Tasmota (tasmota-ir, version 14.2 and when using [IRrecvDumpV2.ino].(https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) pulled from `master`, Arduino IDE library version is `2.8.6`.

Symptom is that all IR communications received by Tasmota (and hence the library) are decoded as iFeel Temperature Reports instead of actual command messages. The provided reading for the sensor is erroneous, suggesting that the message bytes are not an actual iFeel report but an actual command which is misinterpreted by the library.

I tried this on 3 different builds, with 2 NodeMCUs and 1 pre-built (YTF) blaster with Tasmota on it.

Below is a capture from IRrecvDumpV2 with some basic operations carried out on the remote.

```
iFeel enabled
Temperature at remote 20
Target temp is 10
Cooling mode is on

# Turn off

Timestamp : 000036.015
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF58039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 20C
uint16_t rawData[183] = {6422, 3190,  408, 872,  428, 848,  428, 2126,  402, 2150,  408, 872,  404, 2152,  402, 874,  402, 2152,  402, 2154,  406, 872,  428, 2126,  404, 876,  404, 2150,  430, 2124,  432, 2122,  432, 2124,  404, 874,  404, 874,  426, 852,  428, 850,  412, 864,  406, 872,  404, 874,  400, 2154,  428, 2128,  402, 876,  402, 874,  400, 2154,  404, 2150,  406, 2150,  400, 878,  400, 876,  402, 874,  416, 2136,  404, 2152,  402, 876,  402, 874,  402, 874,  428, 852,  402, 874,  404, 872,  400, 876,  402, 876,  404, 872,  430, 846,  402, 2152,  406, 2150,  406, 2150,  404, 874,  432, 846,  426, 850,  428, 850,  400, 878,  398, 876,  404, 874,  404, 874,  400, 2152,  432, 2124,  428, 2128,  432, 846,  426, 852,  400, 2154,  404, 874,  426, 2128,  428, 2128,  404, 872,  430, 848,  426, 2128,  404, 872,  414, 2140,  432, 848,  400, 876,  426, 850,  404, 872,  402, 876,  400, 876,  430, 848,  404, 872,  404, 874,  402, 2154,  402, 876,  428, 2126,  406, 872,  426, 2128,  426, 852,  404, 872,  430, 2124,  404, 874,  430, 846,  426, 2128,  400};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x39};

# Turn on

Timestamp : 000038.077
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF58039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 20C
uint16_t rawData[183] = {6418, 3192,  406, 872,  432, 846,  402, 2154,  430, 2124,  432, 846,  404, 2150,  406, 870,  408, 2148,  406, 2148,  428, 850,  402, 2152,  430, 848,  404, 2150,  428, 2126,  404, 2150,  406, 2148,  404, 876,  428, 846,  430, 848,  428, 848,  428, 852,  428, 848,  402, 876,  426, 2128,  432, 2124,  406, 870,  404, 874,  430, 2126,  432, 2124,  404, 2150,  406, 872,  406, 870,  428, 848,  402, 2154,  430, 2126,  428, 850,  426, 848,  432, 846,  428, 852,  426, 850,  430, 848,  404, 874,  404, 874,  404, 874,  404, 872,  426, 2126,  404, 2152,  404, 2152,  430, 846,  434, 846,  404, 870,  408, 872,  430, 848,  404, 870,  402, 874,  432, 844,  406, 2150,  406, 2150,  432, 2120,  402, 878,  428, 848,  430, 2122,  406, 870,  406, 2150,  406, 2150,  406, 870,  430, 848,  400, 2154,  406, 870,  406, 2148,  406, 872,  432, 846,  404, 872,  428, 848,  404, 874,  404, 872,  402, 874,  402, 2154,  426, 852,  402, 2152,  406, 876,  424, 2128,  428, 850,  402, 2152,  406, 874,  400, 878,  402, 2150,  404, 2152,  428, 850,  404, 2150,  406};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x39};

# Target temp +1 (11)
Timestamp : 000049.843
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5C039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 28C
uint16_t rawData[183] = {6416, 3192,  436, 846,  404, 874,  404, 2148,  406, 2150,  406, 872,  430, 2122,  432, 846,  432, 2124,  432, 2124,  428, 850,  426, 2128,  402, 876,  428, 2124,  402, 2152,  408, 2150,  406, 2148,  430, 848,  400, 874,  432, 846,  434, 846,  426, 850,  432, 846,  428, 2126,  404, 2150,  404, 2152,  432, 846,  406, 872,  428, 2124,  432, 2124,  432, 2122,  406, 874,  404, 872,  428, 848,  402, 2152,  430, 2126,  430, 848,  404, 872,  404, 874,  404, 872,  430, 848,  430, 844,  406, 874,  402, 876,  400, 878,  430, 846,  404, 2150,  406, 2150,  428, 2126,  402, 874,  404, 874,  428, 850,  424, 852,  402, 876,  404, 870,  432, 846,  404, 874,  400, 2152,  430, 2126,  430, 2124,  404, 872,  428, 848,  428, 2126,  432, 846,  426, 2130,  402, 2152,  406, 872,  426, 850,  402, 2152,  430, 848,  402, 2152,  430, 846,  404, 874,  430, 844,  402, 878,  424, 850,  430, 848,  404, 874,  428, 2124,  428, 850,  428, 2126,  406, 870,  406, 2150,  402, 876,  428, 2124,  432, 848,  404, 872,  426, 2126,  430, 2124,  430, 2124,  406, 2150,  432};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0xC0, 0x39};

# Target temp +1 (12)
Timestamp : 000053.669
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5003A (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 4C
uint16_t rawData[183] = {6418, 3194,  404, 874,  404, 874,  404, 2148,  432, 2122,  434, 844,  428, 2126,  402, 876,  426, 2130,  404, 2152,  404, 872,  426, 2128,  404, 872,  406, 2148,  406, 2152,  428, 2128,  402, 2152,  402, 876,  404, 874,  404, 870,  404, 874,  404, 872,  400, 878,  398, 878,  426, 850,  400, 874,  406, 2150,  426, 852,  430, 2124,  430, 2124,  434, 2124,  404, 872,  404, 874,  406, 872,  402, 2152,  404, 2150,  432, 846,  430, 844,  432, 846,  430, 844,  432, [... cut ...] , 2126,  432, 2122,  404, 2152,  432, 846,  406, 2150,  404};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x39};
```

On Tasmota, all received messages look like this
```
21:16:13.773 MQT: tele/tasmota_087A7B/RESULT = {"IrReceived":{"Protocol":"ARGO","Bits":32,"Data":"0xACF58A2B","Repeat":0,"IRHVAC":{"Vendor":"ARGO","Model":"WREM2","Command":"IFeel Report","SensorTemp":21}}}
```

Happy to open a new issue if this seems irrelevant.

I'll try to test with older library versions and report back.

NiKiZe, 2024-08-29:
Is this all the output? These commands all seem to be identical, or close to it. (But temperature ranges seem a bit weird?)

Do you get better results if you disable iFeel?
Do we maybe cut of messages that have multiple parts?

crankyoldgit, 2024-08-29:
So, there appears to be an error in how we interpret the message/protocol.
Please, lets move this to a new issue.

At a quick glance, there are 90bits in the raw message, and we are claiming there is only 32. I'm guessing the code that determines if it is a short "sensor" message is faulty. i.e. It incorrectly thinks it is a sensor message.

mateuszdrab, 2024-08-29:
> So, there appears to be an error in how we interpret the message/protocol.
> Please, lets move this to a new issue.
>
> At a quick glance, there are 90bits in the raw message, and we are claiming there [...]

Thanks, will move into a new issue and capture more samples with iFeel off.