crankyoldgit/IRremoteESP8266 #1860: Logic problem in LG AKB74955603 state

CRCinAU, 2022-08-19:
### Version/revision of the library used
2.8.2

### Describe the bug
When I am trying to set the mode to 'heat' via MQTT, I set the property: `kLgAcHeat`

Code:
https://git.crc.id.au/netwiz/esp8266_code/src/branch/master/ClimateControl/src/ClimateControl.ino#L289

I can see the MQTT message being sent:
climate/ac/mode heat

The unit doesn't turn on.

If I set any other mode, ie auto, then heat again, the unit will correctly switch to heat mode.

After sending the `heat` mode, I get the following from `ac.toString()`:
`Model: 3 (AKB74955603), Power: Off`

If I send `auto`, I get:
`Model: 3 (AKB74955603), Power: On, Mode: 3 (Auto), Temp: 20C, Fan: 9 (Low)`

If I then send `heat` again, I get:
`Model: 3 (AKB74955603), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 9 (Low)`

I'm kind of lost with what's going on here - as the code works, but it seems the internal model as to the status of the AC doesn't update if the first mode is `heat`.

Any suggestions? Or have I made a stupid code problem that I just can't see?

crankyoldgit, 2022-08-19:
I'll look into it

crankyoldgit, 2022-08-19:
I've had a quick look.
If the power was "off" before hand. And I mean a real off (or the default state) e.g. `kLgAcOffCommand` or `stateReset()` and then you `setMode(kLgAcHeat);` it **won't** set power to on.
The `mode` bits for heat are the same in the off state.
The `IRLgAc` class is checking if the state is tne special `off` message for if it is off. With `heat`, it isn't changing the underly state.
`auto` does change the bits enough for it to not be the special `off` message.

If the class is in `off` mode, and you change it to `auto` a _sideeffect_ is making the library report differently. Because the raw state isn't the special `off` state.

So, setting `heat` doesn't make any change to the raw state, therefor it reports it as `off` because the state is still an `off` state.
That's just a quirk of the LG protocol's implementation.

Or to explain it another way.
When the AC or class is set to power off, it already has the `mode` bits set to `heat`. Thus setting it to `heat` mode does nothing at all, as it already was. Any other mode is not the special `off` command/state etc.

So, it sucks, but that's the way the protocol bits work. The  output of `toString()` doing the best it can.

CRCinAU, 2022-08-19:
Interesting.

In the cases of getting an auto / heat / cool / dry / fan_only, I set `ac.setPower(true)`... Is this not sufficient?

The whole `mode` block of code is here:
https://git.crc.id.au/netwiz/esp8266_code/src/branch/master/ClimateControl/src/ClimateControl.ino#L277

CRCinAU, 2022-08-19:
That code does seem to work better - I just removed the mode check with returns...

In other matters, this A/C has a function for 'cleaning' the air - which I think is just an ion generator type thing. It's buried in the remote menus, but is easily accessible via the app / wifi connection.

A long time ago, when you helped me with my old Coolix based unit, I put together an IR receiver with an ESP8266 and decoded a heap of things for the remote.

I put that ESP8266 in a safe place, so safe that I can't find it now.

Do you know what binary that would have been running so I could look at decoding these functions to pad out the command set a little more?

There's also a Jet mode as well as a energy restriction mode for cooling - which I think just limits the max energy draw of the unit.

crankyoldgit, 2022-08-19:
> EDIT: Wait, I'm a moron - that's _after_ the `return`, hence the rest doesn't even get processed.....
> 
> It took me a couple of reads of your post to figure out how that affected the code in place :)

Yeah. Don't worry, it took me a while to work out what was going on too.

crankyoldgit, 2022-08-19:
> Do you know what binary that would have been running so I could look at decoding these functions to pad out the command set a little more?

[IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) or V2 in the examples is what we recommend.

See also: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2022-08-19:
Please create a new issue when you've got the data & analysis for the new instructions.
I'm going to close this one as we've gotten to the bottom of it.

closed 2022-08-19 by crankyoldgit

CRCinAU, 2022-08-19:
> [IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) or V2 in the examples is what we recommend.

Awesome, thanks. I'll have to try to find it.... I only had 1 IR receiver diode, hence why I put it somewhere special :1st_place_medal:

Will do when I have something useful.

crankyoldgit, 2022-08-19:
Note: A IR reciever "diode" won't work with this library, you need a IR demodulater module. Hopefully that's what you meant.

CRCinAU, 2022-08-19:
Yeah - I think so - it was a black dark square with 3 legs on it..... It worked before, so likely, if I can actually find it, it'll work again :)

crankyoldgit, 2022-08-19:
3 legs == good