crankyoldgit/IRremoteESP8266 #1864: Problem sending commands to Sony Bravia TV

mjtrini, 2022-08-23:
Thanks again for all the hard work you have put into this awesome library. I recently captured codes from a Sony Bravia TV but when sending signals using irsend.sendSony38(code, 20, 4), the TV does not work. Also the reflected signals shows 20bit instead of the original 12bit. If I use irsend.sendSony38(code, 12, 4), the reflected signal is unknown 65 bits and the TV does not work.

Below are the captured codes. Can you provide any help?

Button: Power
Protocol  : SONY
Code      : 0xA90 (12 Bits)
uint16_t rawData[103] = {2392, 590,  1174, 614,  578, 612,  1174, 614,  578, 612,  1200, 588,  602, 590,  578, 614,  1198, 588,  628, 564,  602, 590,  578, 614,  604, 25684,  2392, 588,  1174, 612,  604, 588,  1198, 590,  578, 614,  1200, 588,  604, 590,  604, 588,  1202, 588,  604, 588,  604, 588,  602, 590,  630, 25666,  2394, 588,  1200, 586,  628, 564,  1224, 564,  630, 562,  1198, 588,  628, 564,  630, 562,  1226, 564,  630, 564,  628, 562,  604, 588,  630, 25666,  2416, 564,  1200, 590,  632, 560,  1224, 564,  630, 562,  1224, 564,  626, 564,  630, 564,  1198, 588,  630, 564,  630, 562,  634, 558,  632};  // SONY A90
uint32_t address = 0x1;
uint32_t command = 0x15;
uint64_t data = 0xA90;

Button: Volume Up
Protocol  : SONY
Code      : 0x490 (12 Bits)
uint16_t rawData[103] = {2434, 546,  648, 544,  1244, 544,  648, 544,  646, 546,  1242, 546,  650, 542,  648, 544,  1242, 546,  648, 544,  646, 546,  618, 574,  648, 26246,  2430, 552,  620, 572,  1216, 572,  618, 574,  644, 548,  1214, 574,  618, 574,  620, 572,  1242, 546,  644, 550,  616, 576,  616, 576,  616, 26284,  2404, 578,  616, 576,  1188, 598,  618, 574,  616, 576,  1188, 600,  592, 600,  592, 600,  1186, 602,  592, 600,  616, 576,  592, 600,  592, 26292,  2382, 598,  594, 598,  1188, 602,  590, 600,  590, 600,  1170, 620,  590, 602,  572, 620,  1188, 600,  572, 620,  590, 602,  572, 620,  572};  // SONY 490
uint32_t address = 0x1;
uint32_t command = 0x12;
uint64_t data = 0x490;

Button: Volume Down
Protocol  : SONY
Code      : 0xC90 (12 Bits)
uint16_t rawData[103] = {2440, 540,  1250, 538,  1248, 540,  652, 540,  680, 512,  1246, 542,  650, 542,  652, 538,  1244, 544,  648, 542,  650, 542,  652, 540,  652, 25638,  2438, 544,  1244, 544,  1244, 544,  654, 538,  648, 542,  1248, 542,  648, 544,  648, 544,  1244, 544,  648, 544,  646, 548,  644, 546,  646, 25650,  2440, 540,  1242, 546,  1242, 548,  644, 546,  646, 546,  1244, 544,  644, 548,  648, 544,  1244, 544,  644, 546,  646, 546,  622, 570,  622, 25676,  2434, 548,  1218, 570,  1218, 570,  620, 572,  620, 572,  1218, 570,  620, 572,  620, 572,  1216, 572,  620, 574,  620, 572,  620, 572,  618};  // SONY C90
uint32_t address = 0x1;
uint32_t command = 0x13;
uint64_t data = 0xC90;

Button: Channel Up
Protocol  : SONY
Code      : 0x90 (12 Bits)
uint16_t rawData[77] = {2384, 600,  592, 600,  592, 598,  594, 598,  594, 596,  1192, 598,  594, 598,  592, 600,  1190, 600,  592, 600,  590, 602,  572, 618,  594, 26904,  2362, 616,  576, 616,  574, 620,  574, 616,  574, 618,  1172, 618,  576, 616,  574, 616,  1172, 618,  574, 618,  574, 618,  574, 616,  576, 26914,  2366, 616,  576, 614,  578, 616,  576, 616,  576, 614,  1172, 618,  576, 614,  576, 618,  1172, 614,  578, 614,  576, 616,  578, 614,  578};  // SONY 90
uint32_t address = 0x1;
uint32_t command = 0x10;
uint64_t data = 0x90;

Button: Channel Down
Protocol  : SONY
Code      : 0x890 (12 Bits)
uint16_t rawData[103] = {2366, 614,  1172, 616,  576, 616,  576, 616,  574, 618,  1172, 616,  576, 616,  576, 616,  1174, 614,  576, 616,  578, 614,  578, 614,  576, 26316,  2366, 616,  1174, 614,  602, 590,  602, 590,  602, 590,  1200, 588,  578, 612,  602, 590,  1174, 614,  602, 588,  604, 590,  602, 588,  604, 26280,  2418, 562,  1202, 588,  630, 562,  602, 590,  628, 564,  1200, 588,  604, 588,  630, 562,  1224, 564,  628, 564,  630, 562,  632, 560,  632, 26270,  2368, 614,  1226, 562,  630, 562,  632, 558,  632, 560,  1226, 562,  630, 562,  632, 560,  1226, 564,  628, 562,  632, 560,  632, 562,  632};  // SONY 890
uint32_t address = 0x1;
uint32_t command = 0x11;
uint64_t data = 0x890;

Button: Input
Protocol  : SONY
Code      : 0xA50 (12 Bits)
uint16_t rawData[103] = {2438, 542,  1244, 544,  648, 544,  1244, 544,  648, 544,  650, 542,  1244, 544,  650, 542,  1244, 544,  646, 546,  646, 546,  648, 544,  644, 25662,  2408, 572,  1242, 546,  644, 546,  1242, 548,  644, 548,  644, 548,  1242, 546,  644, 548,  1240, 550,  618, 574,  620, 572,  642, 552,  620, 25676,  2408, 572,  1218, 564,  626, 574,  1214, 574,  620, 572,  618, 574,  1214, 574,  618, 576,  1214, 574,  618, 574,  618, 574,  618, 574,  618, 25680,  2408, 574,  1214, 572,  620, 574,  1192, 596,  618, 574,  616, 576,  1212, 576,  616, 576,  1214, 574,  594, 598,  616, 576,  618, 574,  592};  // SONY A50
uint32_t address = 0x1;
uint32_t command = 0x25;
uint64_t data = 0xA50;

crankyoldgit, 2022-08-23:
Have you tried:
```c++
irsend.sendSony(code, 12);
// e.g. irsend.sendSony(0xA90, 12);  // Send the 12-bit Sony code with 2 repeats.
```
Sony has a default number of repeats. i.e. From memory it sends a message a total of three times. (1 + 2 repeats). See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/IRsend.h#L272-L276
Your captures indicate 4 messages, 1 + 3 repeats).
So try:
```c++
irsend.sendSony(code, 12, 3);
```

Also, Sony (and `sendSony()` uses 40kHz typically. The `sendSony38()` is for the 38kHz variant.

Have you tried using `sendGeneric()` to send the raw captured code to see if TV responds to it?
Note: Try both 38kHz & 40kHz.
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/ir_Sony.cpp#L35-L36

mjtrini, 2022-08-23:
Thanks for the fast reply! Can you give an example of the sendGeneric() command? It seems complex to populate. I will test the other suggestions in the morning.

Regards,
Michael.

crankyoldgit, 2022-08-23:
Sorry, I meant [`sendRaw()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a2b9b84f828918f933bd1764d113b53f8)
See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino
e.g.
```c++
uint16_t rawData[103] = {2392, 590, 1174, 614, 578, 612, 1174, 614, 578, 612, 1200, 588, 602, 590, 578, 614, 1198, 588, 628, 564, 602, 590, 578, 614, 604, 25684, 2392, 588, 1174, 612, 604, 588, 1198, 590, 578, 614, 1200, 588, 604, 590, 604, 588, 1202, 588, 604, 588, 604, 588, 602, 590, 630, 25666, 2394, 588, 1200, 586, 628, 564, 1224, 564, 630, 562, 1198, 588, 628, 564, 630, 562, 1226, 564, 630, 564, 628, 562, 604, 588, 630, 25666, 2416, 564, 1200, 590, 632, 560, 1224, 564, 630, 562, 1224, 564, 626, 564, 630, 564, 1198, 588, 630, 564, 630, 562, 634, 558, 632}; // SONY A90

irsend.sendRaw(rawData, 103, 40);  // Send the raw data at 40kHz.
delay(10000);  // Wait ten seconds.
irsend.sendRaw(rawData, 103, 38);  // Send the raw data at 38kHz.
```

mjtrini, 2022-08-23:
Hi, good morning, I tried all suggestions above but the TV did not respond. I wonder if it is an issue with the 12 bit codes vs the 20bit as in the IRSend reference class link:
https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a02bb64503474a0841c51664cf4668d85

I am really hoping to get this Sony TV controlled.

Regards,
Michael.

mjtrini, 2022-08-23:
> Sorry, I meant [`sendRaw()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a2b9b84f828918f933bd1764d113b53f8) See: [...]

I have a quick question. In the IRSendDemo, the Sony command is shown as 
irsend.sendSony(0xa90, 12, 2);  // 12 bits & 2 repeats
The code is typed as 0xa90. Would using 0xA90 cause failure with an uppercase A?

Regards,
Michael.

crankyoldgit, 2022-08-24:
> The code is typed as 0xa90. Would using 0xA90 cause failure with an uppercase A?

No.

crankyoldgit, 2022-08-24:
> Hi, good morning, I tried all suggestions above but the TV did not respond. I wonder if it is an issue with the 12 bit codes vs the 20bit as in the IRSend reference class link: https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a02bb64503474a0841c51664cf4668d85

It is unlikely it is a problem with the `sendSony()` code. It is known to be working.
The `sendRaw()` example suggestions above should also work.
If it doesn't then it points to basic circuit, component or environment issues. See the [Troublehshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) for suggestions.

> I am really hoping to get this Sony TV controlled.

I'm sure you'll get there.

mjtrini, 2022-08-25:
Ok here's an update. None of the suggestions above worked at all for this TV. Now the interesting part is, I used the SmartIRRepeater sketch and it worked at both 38kHz and 40kHz! After the original Serial.printf(...) section, I added a 3 Serial prints :
Serial.println(protocol);Serial.println(results.value);Serial.println(size);

Is it possible to reverse engineer what protocol and results.value from the results i got? Would using irsend.send(protocol, results.value, size) work without explicitly calling the irsend.sendSony or irsend.sendSony38?

Below are the results:
**Power**
000308.049: A 12-bit SONY message was successfully retransmitted.
4
2704
12 

**Vol+**
000330.041: A 12-bit SONY message was successfully retransmitted.
4
1168
12 

**Vol-**
000331.899: A 12-bit SONY message was successfully retransmitted.
4
3216
12 

**Ch+**
000360.097: A 12-bit SONY message was successfully retransmitted.
4
144
12 

**Ch-**
000364.098: A 12-bit SONY message was successfully retransmitted.
4
2192
12 

**Input**
000393.863: A 12-bit SONY message was successfully retransmitted.
4
2640
12

crankyoldgit, 2022-08-25:
Yes, of course you can use `irsend.send()` if you want to.

All `irsend.send(protocol, results.value, size)` will do is call
`sendSony(2704, 12, 2);`  (for your "power" code) FYI: `2704` decimal is `0xA90` hex.
i.e. `sendSony(0xa90, 12, 2);`
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/IRsend.cpp#L1070-L1072

The fact that [SmartIRRepeater](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/SmartIRRepeater/SmartIRRepeater.ino) works proves the library **and** the particular hardware you ran it on is working as expected.
It _**logically**_ indicates that you're code that you're using is problem.
You haven't published your code here so we can't tell.
My _wild guesses_ are:
- you've forgotten to call [`irsend.begin();`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a386f026bf739b0718efde4cffa6ce129) (e.g. [here](https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/examples/IRsendDemo/IRsendDemo.ino#L53)) which enables the GPIO pin for output, before you sent anything; or
- You're passing the wrong values to `sendSony()`. i.e. Add some debug `Serial.println()` to _your_ code; or
- You're using the wrong GPIO in the `IRsend irsend(GPIO);` declaration.

If you trim out the other protocol/IR commands in [IRsendDemo](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) _(i.e. Remove the `sendNEC()`, `sendRaw()`, and `sendSamsungAC()` commands.)_ I'm sure you'll find that also works as does the SmartIRRepeater code.
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/examples/IRsendDemo/IRsendDemo.ino#L62-L72
i.e. Leaving only these lines in the `loop()` section:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/examples/IRsendDemo/IRsendDemo.ino#L65-L67

crankyoldgit, 2022-08-25:
Oh, and in case it wasn't obvious, protocol number `4` is the `SONY` protocol.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/IRremoteESP8266.h#L978-L988

mjtrini, 2022-08-26:
Hi, thanks again for your assistance and patience. The sketch i made actually works perfectly a several appliances (Coolix, Gree, NEC, Symphony, RAW, ToshibaAC, Airton). I have a webserver running on an esp8266 which presents all controls on a phone's browser. I can try to attach the sketch and supported header files but the coding is quite large. The only trouble to date is the Sony 12 bit protocol. Perhaps it is not possible to send 12 bit IR signals.

crankyoldgit, 2022-08-26:
>> Ok here's an update. None of the suggestions above worked at all for this TV. Now the interesting part is, I used the SmartIRRepeater sketch and it worked at both 38kHz and 40kHz!

>> Below are the results:
> **Power**
> 000308.049: A 12-bit SONY message was successfully retransmitted.
> 4
> 2704
> 12

> The only trouble to date is the Sony 12 bit protocol. Perhaps it is not possible to send 12 bit IR signals.

But you've clearly stated you are able to send the _"Sony 12 bit protocol."_ using `SmartIRRepeater`

That is all the SmartIRRepeater program is doing. It is receiving a Sony message, decoding it as valid, and then sending the values to `sendSony()`.

> I can try to attach the sketch and supported header files but the coding is quite large.

Then simplify your code down to the smallest example of non-working sony IR code.

> I have a webserver running on an esp8266 ...

Let me guess ... using the async webserver perhaps?
if so, see the FAQ:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#what-libraries-are-known-to-be-incompatible-with-this-library

mjtrini, 2022-08-26:
Hi!
Thanks for the fast reply. I will try a attach a chopped down version of my sketch. Yes I am using the async webserver! I have already used callback functions and the setup works fine (even with the occasional WDT reset). I used irsend.sendSony (0xA90, 12); in a callback function but it didn't work nor any variations based on your guidance.  I will take some time to read over your documentation and let you know of any success. If no success I will close off this issue and avoid all Sony products.

crankyoldgit, 2022-08-26:
The FAQ link I gave describes this EXACT issue:
> [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)
You can't send IR messages correctly from a callback in that library by default. ESPAsyncWebServer [does not allow or work well](https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember) with delay() or yield() calls which this library uses.
You can modify our library to work with their library by setting [ALLOW_DELAY_CALLS](https://github.com/crankyoldgit/IRremoteESP8266/blob/e7058c6aeace706bb8f4f2de2b3ae06a06cd4f03/src/IRremoteESP8266.h#L953-L959) to false which will remove all delay() calls from our library.
Note: That is far from optimal. You should instead move all IR send calls out of callback functions. IR messages can take 100's of milliseconds to send. Not allowing the ESPs RTOS to use those delay()s may cause other issues. e.g. WDT resets.

And the notes of the async webserver tell you not to do this:
https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember
> **Important things to remember**
> - This is fully asynchronous server and as such does not run on the loop thread.
> - **_You can not use yield or delay or any function that uses them inside the callbacks_**
> - The server is smart enough to know when to close the connection and free resources
> - You can not send more than one response to a single request

See also: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1187#issuecomment-1196566925

and finally:
https://github.com/crankyoldgit/IRremoteESP8266/blob/e7058c6aeace706bb8f4f2de2b3ae06a06cd4f03/src/IRremoteESP8266.h#L953-L959

In short, this is **not** a Sony device problem, **nor** is it an [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) problem, **nor** is it an [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) problem.
You are using _both_ libraries incorrectly in your code. (especially ESPAsyncWebServer)
Your code (the way your using the libraries) won't work for a number of other protocols either. See: Issue #1855

Async callbacks (like interrupt handlers) are to be as short & quick as possible. You're currently tying up them for 100s of milliseconds. That is not good.

mjtrini, 2022-08-26:
Hi! Total success! I put the below into the void loop of my sketch and it works perfectly!

if (sonyir==1){Serial.println("sony output");sonyir=0;
if (tnum==0) {irsend.sendSony (0xA90, 12, 3);}
else if (tnum==1) {irsend.sendSony (0x490, 12, 3);}
else if (tnum==2) {irsend.sendSony (0xC90, 12, 3);}
else if (tnum==3) {irsend.sendSony (0x90, 12, 3);}
else if (tnum==4) {irsend.sendSony (0x890, 12, 3);}
else if (tnum==5) {irsend.sendSony (0xA50, 12, 3);}
}

My lesson here is not to use a separate function for Sony devices but rather put the commands into the void loop. I want to thank you immensely and the other contributors. In my opinion you all are legends!

crankyoldgit, 2022-08-26:
I'm glad you now understand it and you've worked out how to code it up so it works for you.

I've created an(other) FAQ for this to hopefully save people time and effort.
See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Im_using_the_Async_webserver_library_and_some_IR_protocols_dont_seem_to_send_correctly_Why

For x-Ref with #1855 too

closed 2022-08-26 by crankyoldgit