crankyoldgit/IRremoteESP8266 #1865: Remote "miyako fan" which reads as a symphony protocol.

Meong26, 2022-09-03:
Hi everyone, 
I'm having a problem with decoding a fan remote with the brand "miyako" when decoded it reads as 12 bits symphony protocol, with the same reading. but when I code for IRserver and enter the decoded result. fan not working. that's not the case with the NEC protocol and the Samsung Protocol. is there a solution for this,
here I attach what is written on the serial monitor when decoding.

Timestamp : 000539.131
Library   : v2.8.2

Protocol : SYMPHONY
Code : 0xD81 (12 Bits)
uint16_t rawData[23] = {1282, 398, 1284, 398, 442, 1238, 1284, 398, 1284, 398, 442, 1238, 442, 1240, 442, 1238, 442, 1240, 440, 1240, 442, 1240 , 1282}; // SYMPHONY D81
uint64_t data = 0xD81;

thanks advance

NiKiZe, 2022-09-03:
Are you sending a Symphony 0xd81 message with irserver?

What is the exakt name and model number, both of the fan, and the remote?

crankyoldgit, 2022-09-03:
The [IRserver](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino) example code uses `sendNEC()` unless you modify the code, i.e. It only sends NEC messages without altering it.
It also expects the codes supplied in Decimal, not Hex. e.g. `0xD81` is `3457` in Decimal

You need to change:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/examples/IRServer/IRServer.ino#L89-L91
to
```c++
#if SEND_SYMPHONY
      irsend.sendSymphony(code);
#endif  // SEND_SYMPHONY
```

Please note FAQ: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

Meong26, 2022-09-03:
> Are you sending a Symphony 0xd81 message with irserver?
> 
> What is the exakt name and model number, both of the fan, and the remote?

no, I am sending decimals via IRserver. I don't know for sure the brand and type of the remote. only the brand of the fan is written.

Meong26, 2022-09-03:
> The [IRserver](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino) example code uses `sendNEC()` unless you modify the code, i.e. It only sends NEC messages [...]

yes I've done it changed IRsend.NEC to IRsend.Symphony but do I have to include the number of bits as well.. to
#if SEND_SYMPHONY
      irsend.sendSymphony(code, 12);
#endif  // SEND_SYMPHONY

or just like your example without including the number of bits..?

crankyoldgit, 2022-09-03:
`sendSymphony(code)` should default to the correct bit size of `12`. Adding the 12 shouldn't make a difference.

Meong26, 2022-09-04:
> `sendSymphony(code)` should default to the correct bit size of `12`. Adding the 12 shouldn't make a difference.

I have changed irsend.NEC to irsend.Symphony as in the example, the program runs perfectly, the Ir LED flashes, but nothing happens to the fan, I have entered the decimal value of the hexadecimal conversion given by IRrecvDumpV2. is the decimal entered in the code wrong... ?

crankyoldgit, 2022-09-04:
Try it in the [`IRsendDemo`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) example. That's as simple as we can make it.

It could be that your device/variant of the protocol does not use a repeat as default. e.g.:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ff045f149cbc2e31c8512169c1455578b8bf405e/src/IRremoteESP8266.h#L1343

The raw data you supplied indicates no repeat. so try:
```C++
irsend.sendSymphony(code, kSymphonyBits, kNoRepeat);
// or 
irsend.sendSymphony(code, 12, 0);  // Same thing
```

Meong26, 2022-09-05:
> Try it in the [`IRsendDemo`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) example. That's as simple as we can make it.
>
> It could be that your [...]

ahh yeah I've tried and it works. thanks dude, the only problem is the distance. IR nodeMcu can not be far a maximum of only 1 meter, I tried to replace it with IrLed from the existing remote, the result is the same. can the parameters affect the distance?

crankyoldgit, 2022-09-05:
Are you using a transistor, per the wiki?

Meong26, 2022-09-05:
> Are you using a transistor, per the wiki?

Yes, i use bc547

crankyoldgit, 2022-09-05:
Well, you've got a working signal, now you're in to the transmission/power territory. I'll suggest going through the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide).
It's up to your hardware and circuit design etc. That's out of scope for this library and my ability to help you.
It can be done. e.g. I control my TV & Stereo from over 4 meters away using basically the same circuit as the wiki suggests.

closed 2022-09-05 by crankyoldgit

Meong26, 2022-09-05:
> Well, you've got a working signal, now you're in to the transmission/power territory. I'll suggest going through the [Troubleshooting [...]

ok thank you very much for all the help and advice.. happy to discuss