crankyoldgit/IRremoteESP8266 #1880: v2.8.3 does not compile anymore with gcc 4.8.2 (used by Tasmota for ESP8266)

s-hadinger, 2022-09-16:
### Version/revision of the library used
v2.8.3

### Describe the bug
Tasmota uses gcc 4.8.2 for ESP8266 which does not support `__VA_OPT__` extension. 

Is there a way to rewrite the `COND(<define_to_test>[||/&&<more_define>...], <true_result>, <false_result>)` macro without `__VA_OPT__`?

Below is the first compilation error resported (there are many afterwards):
```
In file included from lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRtext.cpp:14:0:
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRmacros.h:22:40: error: pasting ")" and "THING" does not give a valid preprocessing token
 #define VA_TEST_P(a, ...) __VA_OPT__(NO)##THING
                                        ^
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRmacros.h:23:22: note: in expansion of macro 'VA_TEST_P'
 #define VA_TEST(...) VA_TEST_P(__VA_ARGS__)
                      ^
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRmacros.h:24:19: note: in expansion of macro 'VA_TEST'
 #define NEGATE(a) VA_TEST(a, a)
                   ^
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRmacros.h:25:51: note: in expansion of macro 'NEGATE'
 #define COND_P(cond, a, b) STUFF(a, cond)STUFF(b, NEGATE(cond))
                                                   ^
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRmacros.h:26:33: note: in expansion of macro 'COND_P'
 #define COND(cond, a, b) EXPAND(COND_P(cond, a, b))
                                 ^
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRtext.cpp:292:5: note: in expansion of macro 'COND'
     COND(DECODE_RC5 || SEND_RC5,
```

#### To Reproduce
Compilation error with gcc 4.8.2

#### Example code used

#### Expected behaviour

#### Output of raw data from [IRrecvDumpV2.ino]

### What brand/model IR demodulator are you using?

### Circuit diagram and hardware used (if applicable)

### I have followed the steps in the [Troubleshooting Guide]

### Has this library/code previously worked as expected for you?
Yes, 2.8.2 compiled fine

### Other useful information

Jason2866, 2022-09-16:
@tonhuisman Please have a look, i think EspEasy will fail to compile too.

tonhuisman, 2022-09-16:
Best solution would be to upgrade the gcc version, but I assume that's not an option, probably related to the Arduino framework version in use. (same problem at ESPEasy).

Simplest alternative is to add a test if `__VA_OPT__` is supported, and define the `COND()` macro (set) based on that, but that won't give the real solution, as I would have used it like that already.

I can add a different macro `PP_NARG()` (also from the [interwebz](http://jonjagger.blogspot.com/2010/11/c-macro-magic-ppnarg.html)) to check if `__VA_OPT__` can be used:
```
#if PP_NARG(__VA_OPT__(,)) != 2
#define COND(cond, a, b) a
#else
...// existing COND definition
#endif
```
This 'avoids' removing the not available protocols from the protocol-list, and also the check if a protocol is not available, but I _assume_ that is currently only used in ESPEasy, as I designed it for use there.

tonhuisman, 2022-09-16:
> I'm not aware of the protocol-list. What is it?

There is a function to fetch all supported protocols. After using the change I made, it will return `?` if a protocol is not available, so you can check if the length is > 1 and only show the enabled protocols (used in ESPEasy to reduce build size by excluding unneeded protocols in a Custom build)

tonhuisman, 2022-09-17:
I think I've fixed it (see linked PR), with the side-note/limitation that the list of protocols contains all names, even when not included, but _only_ for older compilers that don't support the `__VA_OPT__` macro.

NB: This feature was introduced to reduce code-size as much as possible.

s-hadinger, 2022-09-17:
Thanks a lot. I confirm it's compiling fine on both esp8266 and esp32

closed 2022-09-17 by s-hadinger

crankyoldgit, 2022-09-18:
v2.8.4 has been released which contains the fix.

PR #1881 (2022-09-18): [Bugfix] Handle gcc unsupported __VA_OPT___ macro (#1880)
Resolves #1880 

Add check to see if `__VA_OPT__` macro is supported, if not, applies `<true_result>` value for `COND()` macro by default.

This has the side-effect/limitation that all protocol names are included in the build, even if the protocol is not available in the code, but _only_ for compilers dat do _not_ support the `__VA_OPT__` macro, notably gcc 4.8.2 (and older).
commit c5dacad3 (2022-09-18): [Bugfix] Handle gcc unsupported __VA_OPT___ macro (#1881)

Add check to see if `__VA_OPT__` macro is supported, if not, applies `<true_result>` value for `COND()` macro by default.

This has the side-effect/limitation that all protocol names are included in the build, even if the protocol is not available in the code, but _only_ for compilers dat do _not_ support the `__VA_OPT__` macro, notably gcc 4.8.2 (and older).

Resolves #1880