crankyoldgit/IRremoteESP8266 #1890: Help decoding Carrier protocol

prbtt, 2022-09-26:
### Version/revision of the library used
Library   : v2.8.1

### Describe the bug
Help decoding Carrier AC. Remote is: Console CR 41014010

#### Output of raw data from [IRrecvDumpV2.ino]

_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

`
COOL LOW

18
[18:24:47]Timestamp : 000092.635
[18:24:47]Library   : v2.8.1
[18:24:47]
[18:24:47]Protocol  : UNKNOWN
[18:24:47]Code      : 0x1638003 (46 Bits)
[18:24:47]uint16_t rawData[91] = {9010, 4500,  548, 562,  546, 592,  544, 1698,  542, 592,  546, 1698,  544, 594,  544, 592,  546, 608,  544, 592,  572, 1668,  544, 1698,  544, 1696,  546, 592,  574, 564,  546, 1696,  546, 1714,  544, 594,  544, 594,  544, 592,  602, 536,  544, 592,  548, 590,  548, 590,  546, 608,  546, 592,  544, 592,  546, 594,  544, 592,  600, 538,  546, 592,  602, 536,  544, 610,  598, 538,  546, 592,  602, 536,  604, 534,  598, 538,  598, 538,  604, 534,  604, 556,  598, 1644,  598, 538,  544, 1698,  598, 1632,  544};  // UNKNOWN 1638003
[18:24:47]
[18:24:47]
19
[18:25:02]Timestamp : 000107.862
[18:25:02]Library   : v2.8.1
[18:25:02]
[18:25:02]Protocol  : UNKNOWN
[18:25:02]Code      : 0xEB34C3FC (46 Bits)
[18:25:02]uint16_t rawData[91] = {9010, 4494,  548, 562,  546, 592,  544, 1698,  544, 592,  548, 1694,  544, 592,  574, 564,  544, 610,  544, 594,  544, 1698,  542, 1698,  544, 1696,  546, 592,  544, 1696,  544, 592,  548, 606,  544, 592,  546, 592,  546, 592,  574, 564,  544, 578,  558, 592,  546, 592,  544, 610,  544, 592,  544, 592,  544, 592,  544, 592,  546, 592,  544, 594,  544, 592,  544, 610,  546, 592,  546, 590,  544, 594,  544, 592,  546, 592,  546, 592,  548, 588,  544, 616,  544, 1698,  544, 592,  544, 1698,  544, 580,  548};  // UNKNOWN EB34C3FC
[18:25:02]
[18:25:02]
20
[18:25:16]Timestamp : 000122.045
[18:25:16]Library   : v2.8.1
[18:25:16]
[18:25:16]Protocol  : UNKNOWN
[18:25:16]Code      : 0x25C096E9 (46 Bits)
[18:25:16]uint16_t rawData[91] = {9012, 4500,  548, 562,  544, 592,  544, 1698,  544, 592,  548, 1696,  544, 594,  544, 594,  544, 610,  544, 594,  542, 1698,  544, 1698,  544, 1696,  544, 592,  546, 1696,  546, 592,  544, 1714,  544, 594,  544, 592,  544, 594,  544, 592,  546, 590,  548, 592,  544, 592,  544, 610,  544, 594,  544, 594,  544, 592,  548, 590,  576, 562,  544, 594,  542, 594,  544, 610,  574, 564,  544, 594,  544, 594,  548, 588,  546, 592,  544, 594,  546, 592,  544, 614,  544, 1698,  542, 592,  546, 590,  544, 1686,  544};  // UNKNOWN 25C096E9
[18:25:16]
[18:25:16]

21
[18:25:24]Timestamp : 000129.819
[18:25:24]Library   : v2.8.1
[18:25:24]
[18:25:24]Protocol  : UNKNOWN
[18:25:24]Code      : 0xFB840412 (46 Bits)
[18:25:24]uint16_t rawData[91] = {9010, 4500,  548, 552,  556, 592,  574, 1670,  542, 592,  546, 1696,  544, 592,  544, 594,  544, 610,  544, 592,  544, 1696,  546, 1694,  546, 1696,  544, 592,  544, 1698,  544, 1698,  544, 612,  542, 592,  546, 592,  544, 592,  550, 588,  546, 592,  544, 594,  544, 592,  544, 612,  542, 594,  544, 592,  544, 592,  550, 588,  546, 592,  576, 562,  546, 592,  544, 608,  546, 594,  544, 592,  544, 594,  544, 592,  574, 562,  546, 592,  544, 594,  544, 616,  542, 1698,  544, 594,  544, 594,  544, 582,  544};  // UNKNOWN FB840412
[18:25:24]
[18:25:24]
22
[18:25:32]Timestamp : 000138.326
[18:25:32]Library   : v2.8.1
[18:25:32]
[18:25:32]Protocol  : UNKNOWN
[18:25:32]Code      : 0xD500C82D (46 Bits)
[18:25:32]uint16_t rawData[91] = {9010, 4498,  576, 534,  548, 588,  548, 1692,  546, 594,  544, 1698,  544, 594,  544, 592,  602, 552,  604, 536,  544, 1696,  544, 1698,  544, 1698,  542, 592,  546, 1698,  544, 1698,  544, 1714,  546, 592,  544, 592,  546, 592,  548, 588,  546, 592,  544, 594,  544, 594,  544, 610,  544, 592,  546, 592,  544, 592,  544, 594,  544, 592,  546, 592,  544, 594,  548, 606,  546, 592,  602, 536,  544, 594,  598, 538,  546, 594,  542, 592,  546, 594,  544, 614,  546, 592,  544, 1696,  546, 1696,  544, 1686,  544};  // UNKNOWN D500C82D
[18:25:32]
[18:25:32]
23
[18:25:41]Timestamp : 000146.490
[18:25:41]Library   : v2.8.1
[18:25:41]
[18:25:41]Protocol  : UNKNOWN
[18:25:41]Code      : 0xAC82095C (46 Bits)
[18:25:41]uint16_t rawData[91] = {9036, 4468,  574, 536,  602, 536,  600, 1644,  596, 540,  598, 1642,  600, 538,  624, 514,  598, 556,  600, 538,  598, 1644,  598, 1644,  598, 1644,  598, 1646,  596, 540,  598, 538,  600, 554,  598, 540,  598, 538,  598, 540,  596, 542,  596, 542,  596, 540,  598, 540,  598, 554,  598, 540,  598, 540,  596, 540,  598, 538,  596, 540,  598, 540,  598, 538,  598, 556,  598, 540,  598, 540,  596, 540,  598, 540,  598, 538,  598, 540,  596, 542,  596, 562,  596, 542,  598, 1644,  596, 1646,  596, 530,  596};  // UNKNOWN AC82095C
[18:25:41]
[18:25:41]
24
[18:25:49]Timestamp : 000154.916
[... 229 lines cut ...]
[18:30:30]uint16_t rawData[91] = {9032, 4478,  606, 502,  578, 560,  578, 1664,  578, 558,  578, 1664,  578, 560,  578, 558,  610, 544,  578, 560,  580, 1664,  576, 560,  578, 1664,  578, 1664,  576, 558,  578, 1664,  576, 1684,  576, 560,  578, 560,  580, 558,  580, 558,  578, 560,  600, 538,  578, 558,  578, 578,  578, 558,  576, 560,  602, 534,  578, 560,  580, 558,  576, 560,  578, 558,  580, 576,  576, 560,  578, 560,  576, 560,  604, 534,  602, 534,  578, 560,  578, 560,  578, 582,  578, 558,  580, 1664,  576, 558,  578, 1652,  578};  // UNKNOWN 72434281
[18:30:30]
[18:30:30]
27
[18:30:37]Timestamp : 000442.562
[18:30:37]Library   : v2.8.1
[18:30:37]
[18:30:37]Protocol  : UNKNOWN
[18:30:37]Code      : 0xB9E3A822 (46 Bits)
[18:30:37]uint16_t rawData[91] = {9032, 4478,  608, 500,  602, 538,  576, 1664,  578, 558,  634, 1608,  578, 560,  576, 560,  636, 518,  578, 558,  578, 1664,  576, 560,  578, 1662,  578, 1664,  578, 1664,  578, 558,  606, 548,  580, 558,  580, 558,  580, 556,  580, 560,  608, 528,  578, 558,  608, 530,  632, 522,  578, 558,  580, 558,  580, 558,  578, 560,  576, 560,  578, 558,  580, 558,  578, 578,  578, 558,  578, 560,  578, 560,  578, 558,  578, 560,  578, 560,  578, 560,  602, 556,  606, 532,  580, 1664,  576, 560,  604, 520,  580};  // UNKNOWN B9E3A822
[18:30:37]
[18:30:37]
28
[18:30:45]Timestamp : 000451.255
[18:30:45]Library   : v2.8.1
[18:30:45]
[18:30:45]Protocol  : UNKNOWN
[18:30:45]Code      : 0x56EAB14F (46 Bits)
[18:30:45]uint16_t rawData[91] = {9062, 4446,  614, 498,  608, 530,  578, 1664,  578, 558,  580, 1662,  580, 558,  632, 506,  602, 552,  578, 558,  610, 1634,  578, 558,  640, 1604,  578, 1662,  600, 1640,  602, 534,  602, 1658,  578, 558,  578, 558,  578, 558,  636, 502,  600, 536,  642, 496,  578, 560,  602, 552,  602, 534,  604, 534,  578, 558,  578, 560,  602, 534,  630, 506,  604, 534,  604, 550,  608, 530,  638, 498,  614, 524,  602, 534,  606, 534,  602, 534,  612, 526,  636, 524,  602, 536,  604, 532,  604, 1638,  604, 1626,  626};  // UNKNOWN 56EAB14F
[18:30:45]
[18:30:45]
`

### Other useful information
Im using IRrecvDumpV3

crankyoldgit, 2022-09-26:
Please upgrade to the latest version of the library (2.8.4), it may have added support, but unlikely.
Let us know how it goes with that. If it still doesn't detect it, we can add basic support for it fairly quickly.

prbtt, 2022-09-27:
Updated codes with Library   : v2.8.4 and IRrecvDumpV2

```
OFF

[17:19:34]Timestamp : 000179.569
[17:19:34]Library   : v2.8.4
[17:19:34]
[17:19:34]Protocol  : UNKNOWN
[17:19:34]Code      : 0xAA70AC93 (46 Bits)
[17:19:34]uint16_t rawData[91] = {9030, 4476,  610, 500,  638, 498,  636, 1604,  634, 504,  632, 506,  630, 504,  638, 500,  638, 514,  638, 500,  636, 1606,  632, 1608,  632, 1608,  632, 504,  634, 502,  640, 1602,  630, 1626,  632, 504,  638, 498,  638, 500,  636, 500,  636, 502,  636, 502,  636, 500,  636, 518,  632, 506,  658, 478,  630, 506,  662, 474,  636, 500,  636, 502,  634, 502,  634, 520,  632, 504,  638, 500,  632, 504,  638, 502,  628, 506,  634, 502,  634, 502,  634, 526,  634, 504,  632, 504,  634, 1608,  628, 1600,  630};  // UNKNOWN AA70AC93
[17:19:34]
[17:19:34]

COOL FAN LOW

30

[17:16:45]Protocol  : UNKNOWN
[17:16:45]Code      : 0xBFC872DF (46 Bits)
[17:16:45]uint16_t rawData[91] = {8982, 4520,  566, 542,  596, 542,  626, 1614,  626, 510,  596, 1646,  624, 512,  626, 512,  628, 526,  568, 570,  594, 1646,  622, 1618,  600, 1640,  602, 1640,  598, 1642,  626, 1614,  600, 1658,  602, 534,  598, 538,  602, 536,  596, 542,  602, 536,  596, 540,  598, 538,  600, 554,  624, 512,  600, 538,  596, 540,  568, 568,  626, 512,  600, 536,  626, 512,  594, 560,  600, 536,  602, 534,  568, 570,  600, 536,  600, 536,  624, 512,  570, 568,  626, 532,  592, 1650,  624, 1616,  602, 1640,  626, 1602,  626};  // UNKNOWN BFC872DF
[17:16:45]
[17:16:45]

29

[17:16:50]Timestamp : 000015.650
[17:16:50]Library   : v2.8.4
[17:16:50]
[17:16:50]Protocol  : UNKNOWN
[17:16:50]Code      : 0x5D9E88D6 (46 Bits)
[17:16:50]uint16_t rawData[91] = {9028, 4478,  572, 536,  600, 538,  598, 1642,  596, 538,  600, 1642,  598, 538,  598, 538,  600, 554,  600, 538,  600, 1642,  596, 1644,  598, 1644,  596, 1644,  598, 1644,  596, 1644,  596, 558,  598, 540,  596, 540,  598, 540,  596, 540,  598, 538,  596, 540,  598, 540,  596, 556,  598, 540,  598, 540,  598, 538,  598, 540,  596, 540,  598, 538,  598, 542,  596, 556,  596, 540,  600, 540,  596, 540,  598, 538,  600, 538,  594, 542,  598, 538,  596, 564,  594, 540,  596, 542,  598, 538,  600, 524,  598};  // UNKNOWN 5D9E88D6
[17:16:50]
[17:16:50]

28

[17:16:54]Timestamp : 000019.010
[17:16:54]Library   : v2.8.4
[17:16:54]
[17:16:54]Protocol  : UNKNOWN
[17:16:54]Code      : 0x6D457465 (46 Bits)
[17:16:54]uint16_t rawData[91] = {9030, 4470,  612, 498,  632, 506,  578, 1662,  604, 532,  612, 1630,  604, 534,  606, 530,  602, 552,  600, 536,  602, 1636,  604, 1636,  608, 1636,  602, 1638,  578, 1664,  602, 536,  604, 1654,  604, 534,  602, 534,  602, 534,  604, 532,  608, 530,  602, 534,  580, 558,  602, 552,  602, 534,  636, 502,  602, 534,  604, 534,  602, 534,  604, 534,  602, 534,  602, 552,  602, 534,  606, 532,  602, 534,  604, 530,  604, 536,  604, 532,  604, 534,  604, 556,  602, 534,  606, 532,  604, 534,  602, 1626,  604};  // UNKNOWN 6D457465
[17:16:54]
[17:16:54]

27

[17:16:56]Timestamp : 000020.814
[17:16:56]Library   : v2.8.4
[17:16:56]
[17:16:56]Protocol  : UNKNOWN
[17:16:56]Code      : 0x4D4F48C0 (46 Bits)
[17:16:56]uint16_t rawData[91] = {9056, 4450,  608, 502,  578, 560,  578, 1664,  576, 558,  578, 1666,  574, 560,  578, 560,  602, 552,  578, 560,  578, 1662,  578, 1664,  578, 1662,  578, 1664,  576, 1664,  578, 558,  610, 544,  578, 560,  576, 560,  600, 536,  610, 526,  578, 560,  576, 560,  600, 536,  578, 576,  578, 560,  578, 560,  600, 536,  602, 534,  600, 538,  578, 558,  600, 536,  576, 576,  578, 562,  600, 536,  576, 558,  602, 536,  576, 560,  578, 558,  634, 504,  578, 582,  576, 560,  576, 560,  580, 1662,  600, 524,  576};  // UNKNOWN 4D4F48C0
[17:16:56]
[17:16:56]

26

[17:16:57]Timestamp : 000022.691
[17:16:57]Library   : v2.8.4
[17:16:57]
[17:16:57]Protocol  : UNKNOWN
[17:16:57]Code      : 0xA57984B9 (46 Bits)
[17:16:58]uint16_t rawData[91] = {9030, 4470,  576, 534,  576, 562,  542, 1698,  544, 594,  542, 1698,  544, 592,  576, 560,  580, 574,  580, 556,  546, 1696,  576, 1666,  574, 1666,  544, 1696,  576, 560,  576, 1664,  578, 1682,  576, 560,  576, 560,  576, 560,  578, 558,  576, 560,  576, 560,  578, 558,  578, 578,  574, 562,  578, 560,  576, 560,  576, 560,  576, 562,  576, 560,  576, 560,  578, 576,  576, 560,  576, 560,  578, 560,  578, 558,  578, 560,  576, 560,  578, 560,  576, 582,  576, 562,  576, 560,  576, 1664,  576, 1654,  576};  // UNKNOWN A57984B9
[17:16:58]
[17:16:58]

25

[... 351 lines cut ...]
[17:25:52]

19

[17:25:54]Timestamp : 000558.877
[17:25:54]Library   : v2.8.4
[17:25:54]
[17:25:54]Protocol  : UNKNOWN
[17:25:54]Code      : 0x84691497 (46 Bits)
[17:25:54]uint16_t rawData[91] = {9030, 4470,  572, 536,  600, 538,  596, 1644,  598, 540,  596, 1644,  596, 540,  596, 540,  596, 558,  596, 528,  610, 1644,  596, 1644,  596, 542,  596, 540,  596, 1644,  596, 542,  594, 560,  596, 542,  598, 538,  598, 540,  596, 540,  598, 540,  594, 540,  598, 540,  598, 554,  598, 538,  596, 542,  594, 542,  596, 540,  596, 542,  596, 542,  596, 542,  594, 560,  596, 540,  596, 542,  594, 542,  596, 540,  596, 542,  594, 542,  596, 540,  598, 562,  596, 1646,  596, 540,  596, 1646,  596, 1634,  594};  // UNKNOWN 84691497
[17:25:54]
[17:25:54]

18

[17:25:56]Timestamp : 000561.111
[17:25:56]Library   : v2.8.4
[17:25:56]
[17:25:56]Protocol  : UNKNOWN
[17:25:56]Code      : 0x29117D76 (46 Bits)
[17:25:56]uint16_t rawData[91] = {9032, 4476,  570, 538,  596, 542,  596, 1644,  598, 540,  598, 1644,  596, 540,  596, 540,  598, 556,  600, 536,  598, 1644,  596, 1642,  598, 542,  596, 540,  596, 540,  598, 1644,  596, 1662,  596, 540,  596, 542,  596, 540,  598, 540,  596, 542,  596, 540,  598, 540,  596, 558,  596, 540,  596, 540,  596, 540,  594, 542,  598, 540,  594, 542,  596, 542,  596, 558,  596, 540,  596, 540,  594, 542,  596, 542,  596, 540,  596, 542,  596, 540,  596, 564,  596, 1644,  596, 1646,  594, 540,  598, 530,  596};  // UNKNOWN 29117D76
[17:25:56]
[17:25:56]

```

prbtt, 2022-11-01:
@crankyoldgit any updates on this?

Onepamopa, 2022-11-19:
If there's a way for us to "decode" the rawData array to help speed up adoption, I haven't been able to find it...

crankyoldgit, 2022-11-19:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#use-the-auto_analyse_raw_datapy-program

Onepamopa, 2022-11-19:
> https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#use-the-auto_analyse_raw_datapy-program

That's what I've done for Ikeda for both recv and send (no tests have been done because the AC unit is not here), however, everything still gets decoded as UNKNOWN. At the moment I'm trying with the other detected values for Potential Mark Candidates and Potential Space Candidates in various combinations, nothing so far.

crankyoldgit, 2022-11-19:
What is the output of the `auto_analyse_raw_data.py` program when you've used the raw data you captured? i.e. Paste it here.

Gsvitoria, 2023-11-05:
I have the same problem, I can't find which protocol is used