crankyoldgit/IRremoteESP8266 #1892: The TC112AC timer intervals decoded erroneously by IRrecvDumpV2

soosp, 2022-09-28:
### Version/revision of the library used
v2.8.4

### Describe the bug
The TC112AC timer intervals decoded erroneously by IRrecvDumpV2. Instead on half hours got 20 minutes and after 22 hours got incorrect values.

#### To Reproduce
1. Compile and upload IRrecvDumpV2
2. Set Off timer intervals by the remote contriller

#### Example code used
IRrecvDumpV2

#### Expected behaviour
Get proper intervals as set on the remote

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
There was sent every timer interval supported by remote controller, and the results pasted here.
```
0.5 h:
Timestamp : 002366.168
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002C03070003000080CE (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: 00:20
uint16_t rawData[227] = {3030, 1658,  500, 1062,  498, 1062,  498, 320,  496, 324,  492, 326,  502, 1060,  500, 318,  496, 330,  498, 1064,  498, 1064,  496, 324,  492, 1068,  492, 328,  500, 320,  498, 1064,  496, 1072,  498, 322,  496, 1066,  494, 1066,  494, 326,  490, 328,  500, 1062,  498, 322,  496, 332,  496, 1064,  496, 324,  492, 328,  500, 320,  496, 322,  494, 326,  492, 328,  498, 328,  500, 320,  496, 322,  494, 326,  500, 318,  498, 322,  496, 324,  492, 326,  500, 326,  500, 320,  496, 322,  496, 1066,  494, 1068,  494, 326,  500, 1060,  500, 318,  498, 330,  498, 1064,  498, 1064,  496, 324,  494, 326,  500, 318,  498, 322,  494, 324,  492, 334,  492, 1070,  492, 1070,  502, 1060,  500, 318,  498, 322,  494, 324,  492, 328,  498, 328,  500, 320,  498, 322,  494, 324,  492, 328,  498, 320,  496, 324,  492, 326,  500, 326,  502, 1060,  500, 1062,  500, 320,  498, 322,  494, 326,  502, 318,  498, 320,  496, 332,  496, 324,  494, 326,  502, 318,  498, 322,  496, 324,  492, 328,  500, 318,  498, 330,  498, 322,  494, 324,  492, 328,  500, 320,  498, 322,  494, 326,  492, 328,  498, 328,  500, 320,  498, 322,  494, 326,  500, 318,  498, 322,  496, 324,  492, 328,  500, 1068,  492, 328,  500, 1062,  500, 1062,  498, 1064,  498, 322,  494, 326,  492, 1070,  502, 1060,  468};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x2C, 0x03, 0x07, 0x00, 0x03, 0x00, 0x00, 0x80, 0xCE};

1.0 h:
Timestamp : 002366.168
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002C03070003000080CE (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: 00:20
uint16_t rawData[227] = {3030, 1658,  500, 1062,  498, 1062,  498, 320,  496, 324,  492, 326,  502, 1060,  500, 318,  496, 330,  498, 1064,  498, 1064,  496, 324,  492, 1068,  492, 328,  500, 320,  498, 1064,  496, 1072,  498, 322,  496, 1066,  494, 1066,  494, 326,  490, 328,  500, 1062,  498, 322,  496, 332,  496, 1064,  496, 324,  492, 328,  500, 320,  496, 322,  494, 326,  492, 328,  498, 328,  500, 320,  496, 322,  494, 326,  500, 318,  498, 322,  496, 324,  492, 326,  500, 326,  500, 320,  496, 322,  496, 1066,  494, 1068,  494, 326,  500, 1060,  500, 318,  498, 330,  498, 1064,  498, 1064,  496, 324,  494, 326,  500, 318,  498, 322,  494, 324,  492, 334,  492, 1070,  492, 1070,  502, 1060,  500, 318,  498, 322,  494, 324,  492, 328,  498, 328,  500, 320,  498, 322,  494, 324,  492, 328,  498, 320,  496, 324,  492, 326,  500, 326,  502, 1060,  500, 1062,  500, 320,  498, 322,  494, 326,  502, 318,  498, 320,  496, 332,  496, 324,  494, 326,  502, 318,  498, 322,  496, 324,  492, 328,  500, 318,  498, 330,  498, 322,  494, 324,  492, 328,  500, 320,  498, 322,  494, 326,  492, 328,  498, 328,  500, 320,  498, 322,  494, 326,  500, 318,  498, 322,  496, 324,  492, 328,  500, 1068,  492, 328,  500, 1062,  500, 1062,  498, 1064,  498, 322,  494, 326,  492, 1070,  502, 1060,  468};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x2C, 0x03, 0x07, 0x00, 0x03, 0x00, 0x00, 0x80, 0xCE};

1.5 h:
Timestamp : 002506.170
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002C03070009000080D4 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: 01:20
uint16_t rawData[227] = {3034, 1654,  492, 1068,  492, 1070,  492, 328,  500, 320,  496, 324,  494, 1066,  492, 328,  498, 326,  502, 1060,  500, 1036,  526, 320,  496, 1066,  496, 324,  492, 328,  500, 1062,  498, 1070,  492, 328,  498, 1062,  498, 1064,  496, 322,  494, 324,  492, 1072,  500, 318,  498, 328,  498, 1062,  498, 322,  496, 324,  490, 328,  498, 320,  496, 324,  492, 326,  500, 326,  492, 328,  498, 320,  496, 324,  492, 326,  500, 320,  496, 322,  494, 324,  492, 336,  492, 326,  500, 320,  496, 1064,  496, 1066,  496, 324,  492, 1068,  492, 328,  500, 326,  500, 1062,  498, 1062,  498, 322,  494, 324,  492, 328,  500, 320,  498, 322,  494, 332,  494, 1068,  494, 1068,  492, 1070,  492, 328,  500, 320,  496, 324,  494, 324,  502, 326,  490, 328,  500, 320,  496, 324,  492, 328,  500, 320,  498, 320,  494, 326,  492, 334,  492, 1070,  490, 328,  498, 320,  496, 1066,  496, 324,  492, 328,  500, 318,  498, 330,  498, 322,  494, 326,  492, 328,  500, 320,  498, 322,  494, 326,  490, 330,  498, 328,  500, 320,  496, 324,  494, 326,  500, 318,  498, 322,  496, 324,  492, 328,  500, 326,  500, 320,  498, 322,  494, 326,  492, 328,  500, 320,  496, 324,  494, 326,  500, 1068,  492, 326,  500, 320,  498, 1064,  496, 324,  494, 1068,  492, 328,  500, 1062,  500, 1062,  466};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x2C, 0x03, 0x07, 0x00, 0x09, 0x00, 0x00, 0x80, 0xD4};

2.0 h:
Timestamp : 002548.762
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002C0307000C000080D7 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: 02:00
uint16_t rawData[227] = {3038, 1652,  496, 1066,  496, 1066,  494, 326,  500, 318,  498, 320,  496, 1066,  494, 324,  492, 336,  492, 1070,  492, 1070,  500, 320,  496, 1064,  496, 324,  494, 326,  492, 1070,  500, 1068,  492, 326,  500, 1062,  498, 1062,  498, 322,  496, 324,  492, 1070,  490, 330,  498, 328,  498, 1062,  498, 322,  496, 324,  492, 328,  500, 320,  496, 322,  494, 326,  490, 336,  492, 328,  500, 320,  496, 324,  492, 328,  500, 318,  498, 322,  494, 324,  492, 334,  492, 326,  500, 320,  496, 1064,  496, 1066,  496, 324,  492, 1068,  492, 328,  500, 328,  498, 1062,  498, 1062,  498, 322,  494, 324,  496, 324,  500, 320,  496, 324,  494, 332,  494, 1068,  492, 1068,  492, 1070,  490, 330,  498, 320,  496, 324,  492, 326,  500, 326,  500, 318,  498, 322,  494, 324,  492, 328,  498, 320,  496, 324,  494, 326,  500, 326,  492, 328,  498, 322,  496, 1066,  494, 1068,  494, 326,  490, 328,  498, 320,  496, 330,  496, 324,  494, 326,  500, 318,  498, 322,  494, 324,  492, 328,  498, 320,  496, 330,  498, 322,  494, 326,  492, 328,  498, 320,  496, 324,  492, 326,  500, 318,  498, 330,  496, 322,  496, 324,  492, 328,  500, 320,  496, 322,  494, 326,  490, 328,  500, 1068,  502, 1060,  500, 1060,  500, 1062,  498, 322,  496, 1066,  494, 324,  492, 1070,  500, 1062,  468};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x2C, 0x03, 0x07, 0x00, 0x0C, 0x00, 0x00, 0x80, 0xD7};

2.5 h:
Timestamp : 002584.034
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002C0307000F000080DA (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: 02:20
uint16_t rawData[227] = {3038, 1650,  498, 1064,  496, 1064,  494, 326,  492, 328,  500, 320,  496, 1066,  494, 324,  492, 334,  492, 1070,  492, 1070,  502, 318,  498, 1062,  498, 322,  494, 326,  492, 1070,  500, 1068,  492, 328,  500, 1062,  498, 1064,  498, 322,  494, 324,  492, 1070,  502, 318,  498, 328,  500, 1062,  498, 322,  496, 32 [... cut ...]
### What brand/model IR demodulator are you using?
TSOP34838

### Circuit diagram and hardware used (if applicable)
TSOP34834 on D5 pin of a NodeMCU V2

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
The timer function was tested first time.

### Other useful information

NiKiZe, 2022-09-29:
Which model is the AC unit and remote?
Could it be that your device does it differently compared to what the initial protocol support was written for? If so we need to detect your remote and deal with it differently.

What you might need to do is grab the state and create a spreadsheet and then figure out exactly which bits needs to be used and in what way.
Also do set the timmer, wait a couple of minutes and then send something else, like changing temperature. The timer should probably have counted down.

soosp, 2022-09-29:
Thanks for the quick response!

> Which model is the AC unit and remote?

This is a CLair CLA-KE-WM12MONO 3.5 AC unit. I assume this is a rebranded TCL model. The details for the original TCL112AC protocol support was discovered using this model (see: #619), but without the timer code. The remote controller doesn't have any model number but the brand name. There are some marks on the PCB of it:

```
972-1(HS) V1.1
```

> What you might need to do is grab the state and create a spreadsheet and then figure out exactly which bits needs to be used and in what way.

You mean the hex value of byte 9 needed to convert into binary form?