crankyoldgit/IRremoteESP8266 #1894: Haier AC does not power on in correct mode

geertvda, 2022-09-28:
### Version/revision of the library used
v2.8.4

### Describe the bug
When sending a command to the airco via the MQTT server example using the detailed "Aircon" tab, the airco never goes in the desired "heat" mode, the unit shows a blue symbol which seems to indicate "cool".
When sending the hex code captured from the original remote, it works fine.

#### To Reproduce
Send any command via the detailed "aircon" page on the mqtt server. Same behaviour via the Tasmota integration.

#### Expected behaviour
The airco should go to the selected mode

### What brand/model IR demodulator are you using?
Wemos D1 mini with the IR shield.

### Circuit diagram and hardware used (if applicable)
https://www.wemos.cc/en/latest/d1_mini_shield/ir.html

### Has this library/code previously worked as expected for you?
Yes, works perfectly with an identical setup on a different brand of AC.

### Other useful information
The AC unit is Haier AS25TAEHRA-CLC, installed a week ago. Same behaviour with a more powerful unit installed at the same time.
Both remotes are YR-HE, as can be seen here: https://intech.cool/946-thickbox_default/pilot-bezprzewodowy-yr-he.jpg

The protocol is recognized as the 112-bit YRW02, and the power, mode, swing, fanspeed settings seem to decode correctly, but when sending via the library, the 8th byte is always "2" instead of "0" in the resulting hex code, no matter what other settings I choose.
When sending the hex code with the "2" replaced by "0" it seems to work fine.
For example, increasing temp from 17>18, mode heat with fan set to minimum:
physical remote:
0xA62A000**0**406000800000000000F0
generated by the library for the same settings:
0xA62A000**2**406000800000000005F7

From reading here, I understand that the third to last byte is "last button pressed", but that doesn't seem to matter for correct operation, but the 2 instead of zero seems to prevent going to the correct mode.
I'm trying to read into the code where I can change this myself, but I'm by no means a code expert, and it will take me time to figure out how to do it.

brgds,
geert

crankyoldgit, 2022-12-25:
> 0xA62A000**0**406000800000000000F0
> generated by the library for the same settings:
> 0xA62A000**2**406000800000000005F7

This equates to:
0xA6, 0x2A, 0x00 [Bytes 0 to 2]
and the odd byte in question is Byte 3, `0x02` or `0x00` by your collected data.

`0x02` is `0b00000010` in binary.
According to: 
https://github.com/crankyoldgit/IRremoteESP8266/blob/e6ec73c3b9f959b1c6ae939d15d25b2b47ae4416/src/ir_Haier.h#L220-L224

That's the `Health` bit/setting.
`Health` is set/on by default. I suggest you turn that bit/setting off. It should then fix your problem.
If you look at:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1508

You will see the "Health" setting is controlled by the "Filter" setting. So, set filter to`Off` and it should do what you want.
My guess is your A/C doesn't have the "Health" setting, or at least, it doesn't like it.

JVRMTS, 2024-05-04:
I am a beginner amateur and I have the same problem, I don't understand where and how to set the filter to off, could you help me?