crankyoldgit/IRremoteESP8266 #1896: GYKQ-52 TCl112AC remote is detected as various model depending on the mode

soosp, 2022-10-04:
### Version/revision of the library used
v2.8.4

### Describe the bug
GYKQ-52 TCl112AC remote (e.g.: https://www.aliexpress.com/item/1005004519422388.html) is detected as various model depending on the mode:
Cool mode: TAC09CHSD
Dry mode: TAC09CHSD
Fan mode: TAC09CHSD
Heat mode: GZ055BE1
Auto mode: GZ055BE1

It seems that the bit 7 of byte 12 is changing mode by mode and fools the `IRTcl112Ac::isTcl` function. It is annoying because some status element in not displayed by `IRTcl112Ac::toString` in Heat and Auto modes.

#### To Reproduce
1. Compile and upload IRrecvDumpV2
2. Set modes an see the output

#### Example code used
IRrecvDumpV2

#### Expected behaviour
The type of remote does not change

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
```
Timestamp : 001039.412
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB260100240808400000000089 (112 Bits)
Mesg Desc.: Model: 2 (GZ055BE1), Type: 1, Power: On, Mode: 8 (Auto), Temp: 23C, Fan: 0 (Auto), Swing(V): 0 (Auto), On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3038, 1652,  492, 1068,  496, 1066,  500, 320,  498, 320,  498, 322,  498, 1064,  500, 318,  500, 326,  492, 1070,  496, 1066,  498, 320,  498, 1064,  502, 318,  502, 318,  500, 1060,  494, 1076,  500, 320,  500, 1062,  492, 1070,  496, 324,  494, 324,  496, 1068,  496, 322,  496, 330,  500, 1062,  492, 328,  492, 328,  492, 328,  500, 318,  502, 318,  500, 320,  500, 328,  492, 328,  502, 318,  500, 320,  500, 318,  500, 320,  498, 322,  498, 322,  496, 328,  500, 320,  500, 320,  498, 1062,  492, 328,  502, 318,  502, 1060,  494, 326,  494, 332,  496, 324,  496, 324,  494, 326,  494, 1068,  498, 322,  498, 322,  496, 324,  496, 332,  498, 320,  498, 322,  496, 322,  496, 1066,  500, 320,  498, 320,  500, 320,  498, 328,  502, 318,  500, 320,  500, 320,  498, 322,  498, 322,  498, 322,  498, 1064,  500, 326,  492, 326,  492, 328,  502, 318,  502, 318,  500, 320,  500, 320,  500, 320,  498, 328,  502, 318,  500, 320,  500, 320,  498, 320,  498, 322,  498, 322,  498, 322,  496, 330,  500, 320,  500, 320,  498, 322,  498, 322,  496, 322,  496, 324,  496, 324,  496, 332,  498, 322,  498, 324,  496, 322,  496, 324,  496, 324,  494, 324,  494, 326,  492, 334,  496, 1066,  500, 320,  500, 320,  498, 1064,  502, 318,  502, 318,  500, 318,  500, 1064,  470};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x08, 0x08, 0x40, 0x00, 0x00, 0x00, 0x00, 0x89};

Timestamp : 001045.981
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002403080000000080C4 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 3 (Cool), Temp: 23C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3036, 1652,  502, 1060,  494, 1068,  496, 322,  496, 324,  496, 324,  496, 1066,  498, 322,  498, 328,  502, 1060,  494, 1068,  496, 322,  498, 1064,  500, 320,  498, 320,  498, 1064,  502, 1068,  498, 322,  496, 1064,  500, 1062,  494, 326,  494, 326,  492, 1070,  494, 324,  494, 332,  498, 1064,  500, 318,  500, 320,  500, 320,  498, 320,  498, 322,  498, 322,  496, 330,  500, 320,  498, 322,  498, 322,  498, 322,  496, 322,  498, 322,  496, 324,  496, 332,  498, 322,  496, 324,  496, 1066,  500, 320,  498, 320,  498, 1062,  492, 328,  502, 324,  494, 1068,  498, 1066,  500, 320,  500, 320,  500, 320,  498, 322,  498, 322,  496, 330,  500, 320,  498, 320,  498, 322,  498, 1064,  500, 318,  502, 318,  500, 320,  498, 328,  502, 318,  500, 318,  500, 320,  500, 320,  500, 320,  498, 320,  498, 322,  496, 330,  500, 320,  500, 320,  498, 322,  498, 322,  496, 322,  496, 324,  496, 324,  494, 332,  498, 322,  498, 322,  496, 322,  496, 324,  496, 324,  496, 324,  494, 326,  492, 334,  496, 324,  494, 324,  494, 326,  494, 326,  494, 326,  492, 328,  492, 328,  502, 324,  494, 326,  494, 326,  492, 328,  492, 328,  502, 318,  502, 318,  500, 320,  500, 1068,  496, 322,  496, 324,  494, 1068,  498, 322,  498, 322,  496, 324,  496, 1066,  500, 1064,  470};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x03, 0x08, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC4};

Timestamp : 001047.309
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002402080000000080C3 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 2 (Dry), Temp: 23C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3030, 1660,  494, 1068,  498, 1064,  502, 318,  500, 318,  500, 320,  498, 1064,  492, 328,  502, 326,  494, 1068,  496, 1066,  500, 320,  500, 1062,  492, 326,  492, 328,  502, 1060,  494, 1076,  500, 318,  500, 1062,  492, 1068,  498, 322,  496, 324,  496, 1066,  498, 322,  498, 328,  502, 1060,  494, 326,  492, 326,  492, 328,  502, 318,  502, 318,  500, 320,  500, 326,  492, 328,  490, 330,  500, 318,  500, 318,  500, 322,  498, 320,  498, 322,  498, 328,  500, 320,  500, 320,  498, 1062,  492, 328,  492, 328,  502, 1060,  494, 326,  494, 332,  496, 324,  494, 1066,  498, 322,  498, 322,  496, 322,  498, 322,  496, 324,  496, 332,  498, 322,  498, 322,  496, 322,  496, 1066,  498, 320,  500, 320,  498, 322,  498, 330,  500, 320,  500, 320,  498, 320,  498, 320,  498, 322,  498, 322,  496, 324,  496, 332,  498, 320,  498, 322,  498, 322,  496, 324,  496, 324,  496, 324,  494, 324,  494, 332,  496, 324,  496, 324,  494, 324,  494, 324,  494, 326,  492, 326,  492, 328,  492, 334,  494, 326,  494, 326,  494, 326,  492, 326,  492, 328,  502, 318,  500, 318,  502, 326,  492, 326,  492, 328,  490, 328,  502, 318,  500, 320,  500, 320,  498, 322,  498, 1072,  494, 1068,  498, 1064,  500, 318,  502, 318,  500, 320,  500, 320,  498, 1064,  490, 1072,  472};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x02, 0x08, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC3};

Timestamp : 001048.492
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB2601002407080500000080CD (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 7 (Fan), Temp: 23C, Fan: 5 (High), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3032, 1658,  496, 1066,  498, 1062,  492, 328,  492, 328,  502, 318,  500, 1062,  494, 326,  494, 334,  496, 1066,  500, 1062,  492, 328,  502, 1060,  494, 326,  494, 326,  492, 1068,  496, 1072,  492, 328,  492, 1070,  496, 1066,  500, 320,  498, 322,  498, 1064,  502, 318,  500, 326,  492, 1070,  496, 322,  496, 324,  494, 326,  494, 326,  494, 326,  492, 326,  492, 334,  494, 324,  494, 326,  494, 326,  492, 326,  492, 328,  490, 328,  502, 318,  502, 324,  494, 326,  492, 326,  492, 1072,  494, 324,  494, 326,  494, 1068,  498, 322,  496, 332,  498, 1062,  492, 1070,  496, 1066,  500, 320,  498, 322,  498, 322,  498, 322,  496, 330,  500, 320,  498, 322,  498, 322,  496, 1064,  500, 320,  500, 320,  498, 320,  498, 328,  502, 1060,  496, 324,  494, 1068,  498, 322,  496, 324,  496, 324,  496, 324,  494, 332,  498, 322,  496, 322,  496, 324,  496, 324,  494, 324,  494, 326,  492, 326,  494, 334,  494, 324,  494, 326,  494, 324,  494, 326,  492, 326,  492, 328,  502, 318,  502, 326,  494, 326,  492, 328,  490, 328,  502, 318,  502, 318,  500, 318,  500, 320,  498, 328,  492, 328,  502, 318,  500, 320,  500, 320,  500, 320,  498, 322,  498, 322,  498, 1072,  494, 1068,  496, 322,  498, 1064,  500, 1062,  492, 326,  492, 328,  492, 1070,  496, 1068,  466};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x07, 0x08, 0x05, 0x00, 0x00, 0x00, 0x80, 0xCD};

Timestamp : 001049.989
Library   : v2.8.4

Protocol  : TCL112AC
Code      : 0x23CB260100240108000000000042 (112 Bits)
Mesg Desc.: Model: 2 (GZ055BE1), Type: 1, Power: On, Mode: 1 (Heat), Temp: 23C, Fan: 0 (Auto), Swing(V): 0 (Auto), On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3034, 1654,  500, 1062,  492, 1070,  496, 324,  496, 324,  494, 326,  494, 1068,  496, 324,  496, 330,  500, 1062,  492, 1070,  496, 324,  494, 1068,  498, 322,  496, 322,  498, 1064,  500, 1070,  496, 324,  496, 1066,  500, 1062,  492, 326,  502, 318,  502, 1060,  494, 326,  494, 334,  496, 1066,  498, 320,  498, 322,  498, 322, [... cut ...]
### Circuit diagram and hardware used (if applicable)
TSOP34834 on D5 pin of a NodeMCU V2

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
Not tested before.

### Other useful information
It was tested with more remote controllers with same results.

NiKiZe, 2022-10-04:
Is that an original remote? Or just a clone that works with your AC?

soosp, 2022-10-04:
Yes, this remote model was shipped with both of my AC units. I tested them, and the results same as above.

crankyoldgit, 2022-10-04:
Please see #1486 for some history.

So, if [`_.isTcl`](https://github.com/crankyoldgit/IRremoteESP8266/blob/71cd956d1ed0a4f139ec214ea4a0de0afc7c4dda/src/ir_Tcl.h#L78) is not good enough to use as a model differentiator we need to come up with another way to tell the two protocols apart.

Can you please look over the `state[]` data and compare it with:
https://github.com/crankyoldgit/IRremoteESP8266/blob/71cd956d1ed0a4f139ec214ea4a0de0afc7c4dda/src/ir_Tcl.h#L28-L82
To see if there is some other bit, bits or pattern we can use to tell the two message formats apart.

#1486 #619 & [ir_Tcl_test.cpp](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/ir_Tcl_test.cpp) hopefully contain enough data to compare with. It's all the data we have.