crankyoldgit/IRremoteESP8266 #1903: Request support for Hisense AC(kelon168)

countrysideboy, 2022-10-18:
The Hisense AC model:hisense(KFR-33GWEF20A1).

The default IRMQTTServer example does not show the KELON168 Protocol until I modify the sourcecode of Irac.cpp and Irac.h.
I think KELON168 Protocol is under development, wihch cannot decode the power on status and the special swingV direction.

mp3-10, 2022-12-15:
Hello,
@countrysideboy I worked on decoding Kelon 168 bit for my Kelon/Hisense MINI APPLE PIE AC (I created #1745 issue) and after one year (yes, that took a long time for me... ) I can upload my results — probably all keys mapped from DG11R2-01. As I see, remote have very similar positions as Whirlpool AC.

As I see #1745 has been closed by @crankyoldgit and my skills are not enough to program in C++ so I need to found volunteer what can help me (and probably you) to move my Excel results into code :D (or we need to wait one year more, because I need to learn C++) 

My excel with decoded positions: https://1drv.ms/x/s!AqpZPk4W5RXIg-8136-zyOU_diZHjA?e=NKm3YS

NiKiZe, 2022-12-15:
Please share with Google drive or excel online so anyone can read it.

mp3-10, 2022-12-16:
@NiKiZe I shared using Excel Online with "everybody can view", I don't know why you can't open. 

Anyway Google Sheet version: https://docs.google.com/spreadsheets/d/1KERJSaXFS4T7RaMC3EUrNM2PA4QQRdzf/edit?usp=sharing&ouid=112270706018469707561&rtpof=true&sd=true (that will be too updated if I found something new)

leonardfactory, 2023-01-16:
Hello! Thank you for this detailed summary of the codes. I'd like to contribute to the Kelon168 / Hisense code, not an expert in C++ but I'm a programmer. I'd like to start but not sure about the path to follow. I've seen the Kelon168 code is embedded in the same file as Kelon (48 bits).

I'd like to ask @crankyoldgit if it's ok to add another `IRKelonAc168` class in the same file and adding its relative protocol, methods, etc.

Furthermore, has anybody got the checksum working? I've seen in the Sheet that the last checksum should be the sum of bytes 14-19, is it working or is it just an idea?

mp3-10, 2023-01-17:
@leonardfactory I tested - I made few using send RAW and sendKelon168(const unsigned char data[]) looks like working well ( setting temp, power off, power on, change mode, send iFell command etc) . On columns with 13/20 (IDEA) you can compare remote checksum (column 13 and 20) with generated by me

leonardfactory, 2023-01-17:
Thanks @mp3-10! After a bit of testing it seems the protocol is exactly the same as WhirlpoolAC, however the byte 18 differs (different remote model code, I suppose). 
I'll show my finding this evening in a PR, there are some bits I still need to check

mp3-10, 2023-01-17:
@leonardfactory I saw this also - sometimes (very often) remote has been decoded as Whirpool. But as I remember another was FAN settings (in Whirpool we have only 1-2-3 levels instead 5), but i'm not pretty sure (i don't remeber)

leonardfactory, 2023-01-24:
So, I've started an implementation in PR #1949, branching from Whirlpool since a lot of things are different (like you said). Currently it seems to be working pretty well, but there are some bits to check. Right now I'm waiting for a feedback about where to implement the code (Whirlpool merging vs Kelon168) given the comments on the PR.

ermes64, 2023-06-17:
sorry for my English
To calculate the checksum you must convert the pre-checksum values from hexadecimal to decimal and add them with xor and convert them again 
the preambes should not be added

mp3-10, 2023-07-27:
@ermes64 are you sure?

For now for bytes 2-12 i make MOD(2) for every byte position and look like working.

for bytes 14-19 checkum looks like sum of the values.

You can check this on my: https://docs.google.com/spreadsheets/d/1KERJSaXFS4T7RaMC3EUrNM2PA4QQRdzf/edit?usp=sharing&ouid=112270706018469707561&rtpof=true&sd=true

dgel27, 2024-08-26:
Hi @leonardfactory 
I want to compile your branch to use with tasmota.
Do i need add somthing to other files (irac.x...) to it working?
Thanks,