crankyoldgit/IRremoteESP8266 #1910: Support request for LUMENE IR REMOTE

pedlima, 2022-10-28:
This is a Projection Screens brand from France and I asked them to send me the details so that I could control it using Tasmota.

They were very nice and fast in answering with the details on how to do it for [this remote](https://www.avmat.ch/upload/rama/LU_FT_REMOTE-IR_2017_FR.pdf), which is the one I have.

So here are the technical details they sent me:

[image]

crankyoldgit, 2022-10-28:
That looks pretty simple. Have you tried to see if it is already supported by seeing if the library can decode it already/

pedlima, 2022-10-28:
> That looks pretty simple. Have you tried to see if it is already supported by seeing if the library can decode it already/

Thank you for your quick reply.

I have Tasmota installed (latest version) in a ESP-01M and when I pointed the remote (e.g. button MOVE UP) this is the result:

`11:06:02.739 MQT: tele/ProjectionScreen/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":480,"Hash":"0xCEBBB0DF","Repeat":0,"RawData":"+1290-380A-375A-405+1260-410+425-1235GhGhG-1240GhGhGh+1255fGhGhGhJfJfJfGiJfJfEfGhJfJfJfGhJfJfJfJfG-22235JfEfJfEfGiGhGhGiGhGhGhJfGhGhGiJfEfJfGiJfJfJfGhJ-415JfJfGiJfJfJfJfGkJfJfJfJfGiGiGhGiGhGhGhEfGi+420i+430hJfJfJfGiJfEfJfGiJlJfJfGiJfJfJfJfG-22250JfJfJfJfGhGhGhGiGhGhGiJfGiGhGhJfJfJfGiJfJfJfGiJfJfJfGiJfJfJfJfGoJfJfJfJfGiGhGiGiGiGiGhJfGiGhGiJfJfJfGiJfJfJfGiJfJfJfGiJfJfJfJfG-22260JfJfJfJfGiGhNhGiGhGiGiJfGiGhGiJlJfJfGiJfJfJfGiJlJfJfGiJfJfJfJfGoJfJfJfJfGiGiGhGiGiGiGiJfGiGiGiJfJfJfGiJfJfJfGiJfJfJfGiJfJfJfJfGoJfJfJfJfGiGiGiGiGiGiGiJfGiGiMiJlJfJfGiJfJfJfMiJlJfJfGiJfJfJfJfGoJfJfJfJfGiGiGiGiGiGiGiJfGiNhGiJlJl+1250fGiJfJfJfGiJlJfJfGiJfJfJ-435+1230fMoJfJfJfJfGiGiGiGiGiGiGiJfGiGiGiJlJfJfGiJfJfJfGiJlJrSr+400-1265SrSrSrSrT-22255JfJfJfJfGiGiGiGiGiGiGiJfGiMiGiJlJrSrTuSrSrSrTuS-440SrSrTuSrSnSrSrToJfJfJrSrTuTuTuT-1270TuTiMuSrTx+395uTuSwSrSrTuSrSrSrTuSwSrSrTuSrSrSrSrT-22280SrSrSrSrTeTuTuTxTuTuTuSrTuTuTuSwSrSrTuSrSrSrTuSwSrSrTuSrSrSrSrT-22275SrSrSrSrTuTuTuTxTuTeTuSrTuTuTuSwSrSrTuSw+1225rSrTuSrSrSrYuSrSrSwSrTzSrSrSrSrTuTuTuTxTuTuTuSrTuTuTuSwSrSrTuSrSrSrTuSwSwSrYuSwSwSw+1225wY","RawDataInfo":[959,959,0]}}`

But since you say it's simple, how can I use this to be able to communicate via Tasmota?

pedlima, 2022-10-29:
@crankyoldgit I don't know if this helps you, but the brand also sent these codes as well:

UP : 0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0030 02FB
DOWN : 0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0030 000F 0010 0030 0010 02F5
STOP : 0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0030 000F 0010 02F5
STEP UP : 0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 002F 0010 002F 0010 002F 0010 002F 0030 000F 0010 002F 0010 002F 0010 02F5
STEP DOWN : 0000 006D 0000 000C 0030 0010 0030 0010 0030 0010 0030 0010 0010 002F 0010 002F 0010 002F 0030 0010 0010 002F 0010 002F 0010 002F 0010 02FB

crankyoldgit, 2023-01-02:
FYI, the above codes decode as the `SYMPHONY` protocol.

e.g. The `UP` code above:
```
$ ./gc_decode -prontohex -rawdump "0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0010 0030 0030 02FB"
Code length 28
Code type      76 (SYMPHONY)
Code bits      12
Code value     0xf01
Code address   0x0
Code command   0x0
uint16_t rawbuf[24] = {
    1262, 394, 1262, 394, 1262, 394, 1262, 394, 
    420, 1262, 420, 1262, 420, 1262, 420, 1262, 
    420, 1262, 420, 1262, 420, 1262, 1262, 20066};
```

And `STEP DOWN` from above:
```
$ ./gc_decode -prontohex -rawdump "0000 006D 0000 000C 0030 0010 0030 0010 0030 0010 0030 0010 0010 002F 0010 002F 0010 002F 0030 0010 0010 002F 0010 002F 0010 002F 0010 02FB"
Code length 28
Code type      76 (SYMPHONY)
Code bits      12
Code value     0xf10
Code address   0x0
Code command   0x0
uint16_t rawbuf[24] = {
    1262, 420, 1262, 420, 1262, 420, 1262, 420, 
    420, 1236, 420, 1236, 420, 1236, 1262, 420, 
    420, 1236, 420, 1236, 420, 1236, 420, 20066};

```

And `STOP`:
```
$ ./gc_decode -prontohex -rawdump "0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0030 000F 0010 02F5"
Code length 28
Code type      76 (SYMPHONY)
Code bits      12
Code value     0xf02
Code address   0x0
Code command   0x0
uint16_t rawbuf[24] = {
    1262, 394, 1262, 394, 1262, 394, 1262, 394, 
    420, 1236, 420, 1236, 420, 1236, 420, 1236, 
    420, 1236, 420, 1236, 1262, 394, 420, 19908};
```

Can you please supply the Make & Model number of the screen & the remote so we can add it to the list of supported devices?

closed 2023-01-02 by crankyoldgit

pedlima, 2023-02-21:
@crankyoldgit I'm sorry, I thought I had answered this before and I clearly didn't.

I couldn't find the specific model for the screen since it is embeded in the ceil and as for the remote, it's the one on [this link](https://www.avmat.ch/upload/rama/LU_FT_REMOTE-IR_2017_FR.pdf).

Haven't yet tested this though with these new codes. But doing the irsend with those new "rawdump" codes, it should work, right?

crankyoldgit, 2023-02-23:
> And `STOP`:
>
> ```
> $ ./gc_decode -prontohex -rawdump "0000 006D 0000 000C 0030 000F 0030 000F 0030 000F 0030 000F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0010 002F 0030 000F 0010 02F5"
> Code [...]

You would send it like the following;
```c++
// Send the raw version.
uint16_t rawbuf[24] = {
     1262, 394, 1262, 394, 1262, 394, 1262, 394, 
     420, 1236, 420, 1236, 420, 1236, 420, 1236, 
     420, 1236, 420, 1236, 1262, 394, 420, 19908};
irsend.sendRaw(rawbuf, 24, 38000);
// send the protocol specific version
irsend.sendSymphony(0xf02);
```

See [`IRsendDemo.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) for more info.

pedlima, 2023-02-23:
I'm using Tasmota, how can I use this in the tasmota console?

crankyoldgit, 2023-02-23:
Sorry to be harsh, but: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#please-help-me-configure-homeassistantesphometasmotaespeasyetc-to-work-with-your-library-etc i.e. That's a tasmota problem/question, not ours.

See: https://tasmota.github.io/docs/Tasmota-IR/  , https://tasmota.github.io/docs/Tasmota-IR/#sending-ir-commands , & https://tasmota.github.io/docs/IRSend-RAW-Encoding/#new-ir-raw-compact-encoding

So, if you are using a build of Tasmota with it (Symphony) enabled, try:
```
IRSend {"Protocol":"SYMPHONY","Bits":12,"Data":0xF02}
```
But that is just a guess.