crankyoldgit/IRremoteESP8266 #1916: Ikeda AC

Onepamopa, 2022-11-18:
Not a bug report, just a question.

Has anyone tried getting Ikeda Air Conditioner codes to be read successfully ?
I don't see "Ikeda" anywhere in SRC, so I presume it's unsupported (yet).

How do we go about adding support for it? I've got an original working remote here.

crankyoldgit, 2022-11-18:
See the FAQ/Wiki: [https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol)

Onepamopa, 2022-11-18:
@crankyoldgit Do I have to #define DECODE_AC true in the sketch ?

I'm currently testing with TSOP6438TR but still seeing some differences in the on/off values, here's an example:

`
ON:
14:11:32.311 -> Timestamp : 000004.802
14:11:32.311 -> Library   : v2.8.4
14:11:32.311 -> 
14:11:32.311 -> Protocol  : NEC (Repeat)
14:11:32.311 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:32.311 -> uint16_t rawData[83] = {8936, 2044,  618, 474,  618, 1204,  618, 478,  620, 1204,  618, 474,  618, 1210,  640, 478,  618, 1204,  644, 1206,  618, 478,  618, 448,  644, 474,  618, 1208,  646, 474,  620, 474,  618, 478,  618, 1230,  618, 1210,  614, 480,  618, 474,  618, 480,  618, 1230,  618, 474,  618, 478,  618, 474,  644, 448,  644, 1210,  618, 474,  618, 474,  620, 1204,  618, 478,  618, 474,  618, 474,  618, 1236,  618, 474,  618, 474,  644, 452,  644, 448,  644, 478,  618, 474,  618};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:32.357 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:32.357 -> 
14:11:32.357 -> 

OFF:
14:11:36.236 -> Timestamp : 000008.756
14:11:36.236 -> Library   : v2.8.4
14:11:36.236 -> 
14:11:36.236 -> Protocol  : NEC (Repeat)
14:11:36.236 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:36.236 -> uint16_t rawData[83] = {8936, 2018,  640, 452,  618, 1232,  618, 478,  614, 1234,  618, 474,  618, 1236,  614, 478,  618, 1204,  644, 1206,  618, 478,  618, 474,  618, 478,  614, 478,  644, 448,  644, 452,  640, 478,  618, 1206,  644, 1210,  614, 454,  644, 448,  644, 478,  614, 1210,  644, 474,  618, 474,  618, 478,  618, 474,  644, 1182,  614, 478,  618, 500,  620, 1208,  618, 448,  644, 448,  644, 478,  618, 1204,  646, 478,  618, 474,  618, 1204,  618, 1210,  644, 1204,  618, 1204,  618};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:36.281 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:36.281 -> 

ON:
14:11:36.281 -> 
14:11:41.948 -> Timestamp : 000014.456
14:11:41.948 -> Library   : v2.8.4
14:11:41.948 -> 
14:11:41.948 -> Protocol  : NEC (Repeat)
14:11:41.948 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:41.948 -> uint16_t rawData[83] = {8936, 2044,  618, 474,  620, 1204,  620, 478,  618, 1204,  620, 474,  620, 1204,  644, 478,  618, 1204,  644, 1204,  618, 478,  618, 448,  644, 478,  618, 1204,  644, 474,  618, 478,  618, 474,  618, 1232,  618, 1204,  620, 478,  618, 474,  618, 474,  624, 1230,  618, 474,  618, 478,  618, 474,  644, 448,  648, 1204,  618, 474,  618, 478,  618, 1204,  618, 474,  618, 478,  618, 474,  618, 1236,  618, 474,  618, 474,  644, 452,  646, 448,  644, 478,  618, 474,  618};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:41.995 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:41.995 -> 
14:11:41.995 -> 

OFF:
14:11:44.685 -> Timestamp : 000017.208
14:11:44.685 -> Library   : v2.8.4
14:11:44.685 -> 
14:11:44.685 -> Protocol  : NEC (Repeat)
14:11:44.685 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:44.685 -> uint16_t rawData[83] = {8940, 2040,  620, 478,  614, 1210,  618, 474,  618, 1210,  614, 478,  618, 1204,  644, 478,  618, 1206,  644, 1204,  618, 478,  618, 448,  644, 478,  614, 478,  618, 474,  618, 478,  644, 474,  618, 1208,  644, 1178,  644, 448,  618, 478,  644, 474,  618, 1210,  640, 478,  618, 448,  644, 478,  618, 474,  644, 1184,  614, 478,  618, 500,  618, 1184,  644, 448,  618, 478,  640, 478,  618, 1204,  644, 478,  618, 474,  618, 1204,  618, 1208,  644, 1206,  618, 1180,  644};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:44.730 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:44.777 -> 
14:11:44.777 -> 

ON:
14:11:48.935 -> Timestamp : 000021.456
14:11:48.935 -> Library   : v2.8.4
14:11:48.935 -> 
14:11:48.935 -> Protocol  : NEC (Repeat)
14:11:48.935 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:48.935 -> uint16_t rawData[83] = {8942, 2018,  614, 478,  644, 1204,  618, 478,  614, 1210,  640, 452,  618, 1236,  614, 478,  618, 1232,  618, 1208,  614, 478,  618, 474,  644, 452,  614, 1236,  618, 474,  618, 478,  644, 474,  618, 1210,  640, 1184,  618, 474,  618, 480,  640, 478,  618, 1206,  644, 452,  644, 448,  618, 504,  614, 478,  618, 1206,  618, 478,  614, 478,  644, 1180,  618, 478,  614, 478,  644, 478,  614, 1210,  640, 454,  644, 448,  644, 478,  618, 474,  618, 478,  614, 478,  618};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:48.980 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:48.980 -> 
14:11:48.980 -> 

OFF:
14:11:50.808 -> Timestamp : 000023.301
14:11:50.808 -> Library   : v2.8.4
14:11:50.808 -> 
14:11:50.808 -> Protocol  : NEC (Repeat)
14:11:50.808 -> Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)
14:11:50.808 -> uint16_t rawData[83] = {8938, 2018,  640, 452,  618, 1230,  618, 478,  614, 1236,  618, 474,  618, 1204,  644, 478,  620, 1204,  644, 1206,  618, 478,  614, 478,  618, 474,  618, 478,  640, 454,  618, 474,  644, 478,  618, 1204,  644, 1206,  618, 452,  640, 454,  644, 474,  618, 1210,  640, 480,  618, 474,  618, 480,  618, 474,  644, 1178,  618, 478,  618, 500,  618, 1184,  640, 452,  620, 474,  644, 478,  618, 1206,  644, 474,  618, 478,  618, 1206,  618, 1204,  644, 1210,  614, 1208,  618};  // NEC (Repeat) FFFFFFFFFFFFFFFF
14:11:50.853 -> uint64_t data = 0xFFFFFFFFFFFFFFFF;
14:11:50.853 -> 

`

Onepamopa, 2022-11-18:
Any ideas on how to eliminate the inconsistencies so I can actually start gathering useful data ?

Onepamopa, 2022-11-18:
I think I'm getting somewhere, after adjusting the tolerance threshold to 7% (anything above that detects as NEC (Repeat)) - I'm seeing actual consistent codes and an UNKNOWN protocol:

ON:
17:12:55.587 -> Tolerance : 7%
17:12:55.587 -> Protocol  : UNKNOWN
17:12:55.587 -> Code      : 0x6C908182 (42 Bits)
17:12:55.633 -> No description
17:12:55.633 -> uint16_t rawData[83] = {8938, 2014,  618, 474,  644, 1210,  618, 448,  644, 1204,  644, 452,  618, 1230,  618, 478,  618, 1230,  618, 1204,  618, 478,  618, 474,  618, 478,  618, 1230,  618, 474,  620, 478,  644, 474,  618, 1208,  644, 1178,  618, 474,  618, 478,  644, 474,  618, 1204,  644, 452,  644, 448,  644, 1210,  618, 474,  618, 1206,  644, 452,  644, 474,  618, 1206,  618, 480,  618, 474,  618, 1204,  622, 1204,  618, 474,  618, 478,  618, 474,  644, 474,  622, 448,  644, 474,  618};  // UNKNOWN 6C908182
17:12:55.678 -> 
17:12:55.678 -> 

OFF:
17:12:57.779 -> Timestamp : 000006.115
17:12:57.779 -> Library   : v2.8.4
17:12:57.779 -> 
17:12:57.779 -> Tolerance : 7%
17:12:57.779 -> Protocol  : UNKNOWN
17:12:57.779 -> Code      : 0x5C713739 (42 Bits)
17:12:57.779 -> No description
17:12:57.779 -> uint16_t rawData[83] = {8938, 2018,  614, 478,  644, 1204,  618, 452,  640, 1208,  644, 448,  618, 1234,  614, 478,  618, 1230,  618, 1208,  618, 474,  618, 478,  614, 478,  618, 474,  644, 478,  618, 474,  618, 478,  614, 1208,  644, 1204,  618, 478,  618, 474,  618, 478,  614, 1208,  644, 474,  618, 478,  640, 1182,  618, 474,  618, 1236,  614, 478,  618, 474,  644, 1182,  614, 478,  618, 500,  618, 1184,  640, 1184,  618, 474,  644, 478,  618, 1178,  644, 1204,  618, 1208,  618, 1178,  644};  // UNKNOWN 5C713739
17:12:57.824 -> 
17:12:57.824 -> 

ON:
17:14:21.982 -> Timestamp : 000090.318
17:14:21.982 -> Library   : v2.8.4
17:14:21.982 -> 
17:14:21.982 -> Tolerance : 7%
17:14:21.982 -> Protocol  : UNKNOWN
17:14:21.982 -> Code      : 0x6C908182 (42 Bits)
17:14:21.982 -> No description
17:14:21.982 -> uint16_t rawData[83] = {8964, 2014,  618, 478,  618, 1204,  618, 478,  614, 1210,  618, 474,  618, 1208,  644, 474,  618, 1208,  644, 1204,  618, 448,  644, 452,  644, 474,  618, 1208,  640, 478,  618, 474,  618, 478,  618, 1204,  644, 1204,  618, 478,  618, 474,  618, 480,  618, 1230,  618, 474,  618, 478,  644, 1178,  618, 474,  618, 1236,  618, 474,  618, 504,  618, 1178,  618, 474,  618, 504,  618, 1204,  618, 1180,  618, 474,  644, 480,  618, 474,  618, 474,  618, 478,  644, 448,  644};  // UNKNOWN 6C908182
17:14:22.027 -> 
17:14:22.027 -> 

OFF:
17:14:24.124 -> Timestamp : 000092.445
17:14:24.124 -> Library   : v2.8.4
17:14:24.124 -> 
17:14:24.124 -> Tolerance : 7%
17:14:24.124 -> Protocol  : UNKNOWN
17:14:24.124 -> Code      : 0x5C713739 (42 Bits)
17:14:24.124 -> No description
17:14:24.124 -> uint16_t rawData[83] = {8934, 2018,  640, 480,  618, 1208,  640, 478,  614, 1214,  640, 452,  640, 1210,  644, 452,  614, 1234,  618, 1208,  614, 478,  614, 478,  644, 452,  614, 478,  644, 478,  614, 478,  614, 482,  614, 1210,  640, 1208,  618, 478,  614, 478,  618, 478,  614, 1234,  614, 482,  614, 478,  640, 1182,  618, 478,  614, 1234,  614, 482,  614, 504,  614, 1182,  618, 478,  614, 504,  618, 1208,  614, 1182,  614, 478,  644, 478,  614, 1210,  614, 1214,  614, 1208,  614, 1208,  614};  // UNKNOWN 5C713739

Onepamopa, 2022-11-18:
Here is a full set of codes:
- all modes
- temperatures
- fan speeds
- flap
- sleep (quiet?)
- FP (whatever that is)
- on timer 1-12h
- off timer 1-12h

[ikeda-ac-codes.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10043419/ikeda-ac-codes.txt)

Onepamopa, 2022-11-19:
Testing with auto_analyse_raw_data.py the defaut margin value of 200 produces basic code, when used - still recognized as "UNKNOWN". Reducing  the value to 50 or less - "data structures" start showing up with multiple SpaceGap's, however, the generated code doesn't utilize them. It only utilizes "SpaceGap" not SpaceGap1/2/3/4/n.

What do I do next?

crankyoldgit, 2022-11-19:
Please supply the input data & the output data when run with the normal `200` margin.

Onepamopa, 2022-11-19:
uint16_t rawData[83] = {8912, 2042,  640, 456,  610, 1214,  614, 478,  640, 1188,  640, 452,  614, 1240,  588, 504,  614, 1236,  614, 1214,  614, 478,  614, 482,  610, 484,  614, 1234,  614, 482,  588, 504,  640, 452,  640, 1214,  614, 1208,  614, 478,  614, 482,  640, 478,  614, 1208,  640, 456,  614, 478,  614, 508,  614, 478,  614, 1208,  614, 482,  614, 478,  640, 1182,  614, 482,  614, 478,  640, 456,  640, 1208,  640, 456,  610, 482,  640, 478,  614, 482,  614, 478,  614, 482,  588};  // UNKNOWN 67B84128

#### # python3 auto_analyse_raw_data.py -n IKEDA -f data.txt 
Found 83 timing entries.
Potential Mark Candidates:
[8912, 640]
Potential Space Candidates:
[2042, 1240, 508]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kIKEDAHdrMark   = 8912
kIKEDAHdrSpace  = 2042
kIKEDABitMark   = 619
kIKEDAOneSpace  = 1212
kIKEDAZeroSpace = 477

Decoding protocol based on analysis so far:

kIKEDAHdrMark+kIKEDAHdrSpace+0101010110001000110001000010010001000000
  Bits: 40
  Hex:  0x5588C42440 (MSB first)
        0x02242311AA (LSB first)
  Dec:  367366775872 (MSB first)
        9196212650 (LSB first)
  Bin:  0b0101010110001000110001000010010001000000 (MSB first)
        0b0000001000100100001000110001000110101010 (LSB first)

Total Nr. of suspected bits: 40

crankyoldgit, 2022-11-19:
Yep, that looks like it's examining and decoding the raw string correctly.
If you give it `-g` it should generate code that can be added to the library if you follow the rest of that wiki page.

Onepamopa, 2022-11-19:
I've already done all that, however, the protocol still doesn't get detected as IKEDA but as UNKNOWN

Onepamopa, 2022-11-19:
Here's what I've added to the library. IRrecvDumpV2 detects "UNKNOWN" or "NEC (Repeat)" depending on kTolerancePercentage

[library-patch.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/10046858/library-patch.zip)

crankyoldgit, 2022-11-19:
Did you follow all the suggested steps?
Do you have it hosted on Github somewhere that I can easily see the `diff`?

Onepamopa, 2022-11-19:
The .diff is within the zip archive (github doesn't allow .diff upload). If you are suspicious about a zip file - let me know and I'll upload it as .txt here.

crankyoldgit, 2022-11-19:
If you clone the repo and push your updates to your fork, it makes it easier to a) merge back in later, and b) for me to use other tools to help you and see what you've done.

The diff looks fine. Have you tried enabling `DEBUG` in `IRremoteESP8266.h` and checked the output when trying to decode it? i.e. If it is even attempting to decode it? You probably need to move the new clause in `decode()` to before the NEC protocol(s) so it gets detected before NEC etc. Or try disabling all the other protocols, and only enable this one.

Onepamopa, 2022-11-19:
Moving IKEDA decode function before SHARP (which probably was placed there for the same reason) did detect it as IKEDA (finally):

ON:
14:18:54.169 -> Using DECODE_AC
14:18:54.169 -> 
14:18:54.169 -> IRrecvDump is now running and waiting for IR input on Pin 14
14:18:55.183 -> UnknownThreshold Enabled
14:19:00.291 -> Timestamp : 000007.189
14:19:00.291 -> Library   : v2.8.4
14:19:00.291 -> 
14:19:00.291 -> Protocol  : IKEDA (Repeat)
14:19:00.291 -> Code      : 0x5588C42440 (40 Bits)
14:19:00.291 -> No description
14:19:00.291 -> uint16_t rawData[83] = {8938, 2040,  618, 478,  614, 1210,  614, 478,  618, 1206,  618, 480,  614, 1210,  640, 478,  618, 1206,  644, 1206,  618, 478,  614, 452,  644, 474,  618, 1210,  640, 480,  618, 474,  620, 478,  614, 1236,  614, 1210,  618, 474,  618, 478,  614, 478,  620, 1230,  620, 474,  618, 478,  614, 478,  644, 448,  646, 1204,  618, 480,  614, 480,  618, 1204,  620, 474,  620, 478,  614, 478,  618, 1232,  618, 474,  620, 474,  644, 452,  640, 452,  646, 474,  618, 478,  614};  // IKEDA (Repeat) 5588C42440
14:19:00.336 -> uint64_t data = 0x5588C42440;
14:19:00.336 -> 
14:19:00.336 -> 

OFF:
14:19:04.157 -> Timestamp : 000011.035
14:19:04.157 -> Library   : v2.8.4
14:19:04.157 -> 
14:19:04.157 -> Protocol  : IKEDA (Repeat)
14:19:04.157 -> Code      : 0x5580C4244F (40 Bits)
14:19:04.157 -> No description
14:19:04.157 -> uint16_t rawData[83] = {8938, 2018,  640, 452,  614, 1236,  614, 478,  618, 1232,  618, 478,  614, 1210,  640, 478,  618, 1206,  646, 1208,  614, 478,  614, 478,  618, 474,  618, 478,  640, 452,  618, 474,  644, 478,  614, 1210,  644, 1204,  620, 448,  644, 452,  640, 478,  618, 1204,  646, 478,  614, 478,  618, 474,  620, 478,  640, 1184,  618, 474,  618, 500,  620, 1184,  640, 452,  614, 478,  644, 474,  618, 1210,  640, 478,  618, 474,  618, 1206,  618, 1210,  640, 1210,  614, 1210,  618};  // IKEDA (Repeat) 5580C4244F
14:19:04.203 -> uint64_t data = 0x5580C4244F;
14:19:04.203 -> 
14:19:04.203 -> 

What are the next steps? How to break-down the received data structure, so I can generate & send commands?

crankyoldgit, 2022-11-19:
> What are the next steps? How to break-down the received data structure, so I can generate & send commands?

RTFM/Read The Fine Wiki.

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

Onepamopa, 2022-11-19:
By the way, sometimes it still gets detected as NEC instead of Ikeda.. what do I have to adjust to only match Ikeda?

crankyoldgit, 2022-11-19:
Hard to tell without raw data for the messages that are failing. It could just be your IR demodulator.
e.g. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator

Onepamopa, 2022-11-19:
Here are all codes detected as "IKEDA", all modes, fan speeds, temperatures, flap/fp/sleep(quiet), on-timers, off-timers.
So.. what do I do next? How to convert uint64_t data 0xXXXXXXXXXXXX to a format that would easily show the differences between all modes & How to construct the "Protocol" structure? I don't see that in the wiki.

[ikeda-ac-codes.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/10047189/ikeda-ac-codes.txt)

Onepamopa, 2022-11-19:
I've began adding the data from "ikeda-ac-codes.txt" to Google spreadsheet. Here is the link: https://docs.google.com/spreadsheets/d/1IOxfKN2PPuIayC1UFZqkHrk7LAqH0AwUjgUn6dIrnzg

I'll ping you here after I add all the data.

Onepamopa, 2022-11-19:
@crankyoldgit let me know if anything more is needed.

Onepamopa, 2022-11-23:
@crankyoldgit perhaps sponsoring the project in some way would lead to faster implementation? Let me know. I really need to be able to control this "Ikeda", as in - specify mode, fan, degrees, flap etc as soon as possible...

Onepamopa, 2022-11-23:
Converting MSB (which is the default, it seems) to LSB via:
  data_result = matchData(&(results->rawbuf[offset]), IKEDA_BITS,
                          kIKEDABitMark, kIKEDAOneSpace,
						  kIKEDABitMark, kIKEDAZeroSpace,
						  kUseDefTol, kMarkExcess, false, true);

Started showing the degrees, for example:
ON @ 27C, Heat, fan high:                                                    0x5   27  23 11 AA
OFF @ 27C, Heat, fan high:                                                   0xF5 27  23 01 AA

ON, 27C, Heat, Fan high, Sleep->On:                                    0x85 27  23 91 AA
OFF, 27C, Heat, Fan high, Sleep->On:                                   0xF5 27  23 01 AA

ON, 27C, Heat, Fan high, Flap->On:                                      0x15 27  33 11 AA
OFF, 27C, Heat, Fan high, Flap->On:                                     0x5   27  33 01 AA

ON, 27C, Heat, Fan high, FP->On:                                         0xD  27  2B 11 AA
OFF, 27C, Heat, Fan high, FP->On:                                        0xFD 27 2B  01 AA

ON, 27C, Heat, Fan high, Sleep->On+Flap->On:                  0x95  27  33 91 AA
OFF, 27C, Heat, Fan high, Sleep->On+Flap->On:                 0x5    27  33 01 AA

ON, 27C, Heat, Fan high, Sleep->On+FP->On:                     0x8D 27  2B 91 AA
OFF, 27C, Heat, Fan high, Sleep->On+FP->On:                    0xFD 27  2B 01 AA

ON, 27C, Heat, Fan high, Sleep->On+FP->On+Flap->On:   0x9D 27  3B 91 AA
OFF, 27C, Heat, Fan high, Sleep->On+FP->On+Flap->On:  0xD   27  3B 01 AA

ON @ 27C, Heat, fan high, Off timer->  1:                            0x25  27  23 31 AA
ON @ 27C, Heat, fan high, Off timer->  2:                            0x26  27  23 32 AA
ON @ 27C, Heat, fan high, Off timer->  3:                            0x27  27  23 33 AA
ON @ 27C, Heat, fan high, Off timer->  4:                            0x28  27  23 34 AA
ON @ 27C, Heat, fan high, Off timer->  5:                            0x29  27  23 35 AA
ON @ 27C, Heat, fan high, Off timer->  6:                            0x2A  27  23 36 AA
ON @ 27C, Heat, fan high, Off timer->  7:                            0x2B  27  23 37 AA
ON @ 27C, Heat, fan high, Off timer->  8:                            0x2C  27  23 38 AA
ON @ 27C, Heat, fan high, Off timer->  9:                            0x2D  27  23 39 AA
ON @ 27C, Heat, fan high, Off timer->  10:                          0x2E  27  23  3A AA
ON @ 27C, Heat, fan high, Off timer->  11:                          0x2F  27  23  3B AA
ON @ 27C, Heat, fan high, Off timer->  12:                          0x30  27  23  3C AA

ON @ 27C, Heat, fan high, On timer->  1:                            0x35  27  23 41 AA
ON @ 27C, Heat, fan high, On timer->  2:                            0x36  27  23 42 AA
ON @ 27C, Heat, fan high, On timer->  3:                            0x37  27  23 43 AA
ON @ 27C, Heat, fan high, On timer->  4:                            0x38  27  23 44 AA
ON @ 27C, Heat, fan high, On timer->  5:                            0x39  27  23 45 AA
ON @ 27C, Heat, fan high, On timer->  6:                            0x3A  27  23 46 AA
ON @ 27C, Heat, fan high, On timer->  7:                            0x3B  27  23 47 AA
ON @ 27C, Heat, fan high, On timer->  8:                            0x3C  27  23 48 AA
ON @ 27C, Heat, fan high, On timer->  9:                            0x3D  27  23 49 AA
ON @ 27C, Heat, fan high, On timer->10:                            0x3E  27  23 4A AA
ON @ 27C, Heat, fan high, On timer->11:                            0x3F  27  23 4B AA
ON @ 27C, Heat, fan high, On timer->12:                            0x40  27  23 4C AA

Hopefully this is helpful.

Onepamopa, 2022-11-25:
Some progress - turns out the checksum is the first 2 bits in LSB, the 2nd 2 bits are the temperature (in this case 17C)
0xF5 17 23 11 AA
0xF5 = 245 --> checksum?
	-------
0x17 = 23
0x23 = 35
0x11 = 17
0xAA = 170
	--------
	245

Onepamopa, 2022-11-26:
@crankyoldgit IKEDA fully decoded, all modes, timers, everything. Now... if I could only figure out how to implement into the library.... Give me a hand, would ya?

Onepamopa, 2022-11-26:
@crankyoldgit here's the initial support in my repo: https://github.com/crankyoldgit/IRremoteESP8266/compare/master...Onepamopa:IRremoteESP8266:master

Onepamopa, 2022-11-27:
@crankyoldgit I get you're a busy guy. At least let me know if Ikeda has similarities to another already-implemented protocol and I'll try to do the work for you. The checksum is in the front, and can be 1 or 2 bytes. That's what's hard for me at the moment. If I can figure out how to construct the protocol structure, I think I'll be able to do the rest.

Onepamopa, 2022-11-30:
Would anyone take a look at this?

crankyoldgit, 2022-12-01:
> @crankyoldgit I get you're a busy guy. At least let me know if Ikeda has similarities to another already-implemented protocol and I'll try to do the work for you. The checksum is in the front, and [...]

I'll try to get to working out a protocol structure for you next week. I'm flat out this week and on the weekend.

Onepamopa, 2022-12-01:
This is what I did yesterday, but ... can't be sure if it's correct:

```
union IkedaProtocol{
  uint8_t remote_state[IKEDA_BITS];  ///< The state in native IR code form
  struct {
    // Byte 0
		// checksum with first byte only 1 digit
		// 0x5 17 33 11 AA
		// 0101 00010111 00110011 00010001 10101010 // 36
		// if checksum is 1 digit (for example 0x2, 0x3, 0x4, 0x5, 0xD, etc =>> hex(100+hex2dec(DIGIT))
    uint8_t Checksum		:8;

    // Byte 1
    uint8_t Temperature		:8;		// DEC 16-30 C

    // Byte 2
    uint8_t FanSpeed		:4;		// DEC 0-7
		// Fan speed (auto=0, high=2, medium=4, low=6) OR
		// Fan speed flap (1=Flap_Fan_Auto, 3=Flap_Fan_High, 5=Flap_Fan_Medium, 7=Flap_Fan_Low)
    uint8_t OperatingMode	:4;		// DEC: 0-4, 9, 11
		// Operating mode "normal" (0=auto, 1=cool, 2=dry, 3=heat, 4=fan)
		// Operating mode "fp" (B(11)=heatFP, 9=coolFP) (Dry, Fan & Auto -> no FP & SLEEP modes)
		// Operating mode "sleep" (3=Heat_Sleep, 1=Cool_Sleep) + state[3][0] to 9
		// Combined "sleep" + "fp" (HEAT and COLD only): (B(11)=heatFP_SLEEP, 9=coolFP_SLEEP) + state[3][0] to 9

    // Byte 3
	uint8_t ACMode			:4;		// DEC: 0,1,3,4,9
		// 0=AC off; ON: 1=Normal, 3=Off-Timer, 4=On-Timer, 9=Sleep mode(with or without FP),
    uint8_t OnOffTimer		:4;		// DEC 1-12
		// on/off timer hours 1-9, 0xA=10, 0xB=11 0xC=12
    // Byte 4
    uint8_t unchanged		:8;  // value = 0xAA
  };
};
```

Onepamopa, 2022-12-16:
Turns out the chip inside the remote is NEC MYS LSB-R001

Onepamopa, 2022-12-16:
Here's a complete protocol structure down to the last bit, I'll try to implement the rest using "SanyoAc88" as a base:

```
/// Native representation of a Ikeda A/C message.
union IkedaProtocol{
  // const uint16_t kIkedaStateLength = 5; // 5 states * 8 bits
  // const uint16_t kIKEDABits = kIkedaStateLength * 8;
  // const uint16_t kIkedaMinRepeat = 2; // Sanyo88 (for now)
  uint8_t raw[kIkedaStateLength];  ///< The state in native IR code form
  struct {
    // Byte 0
    uint8_t Checksum      :8;

    // Byte 1
    uint8_t Temperature   :8;		// DEC 16-30 C -> @ HEX

    // Byte 2
    uint8_t               :1; // Empty 0x0
    uint8_t Fan           :2; // 3=Low, 2=Medium, 1=High, 0=Auto
    uint8_t Flap          :1; // 0=Off, 1=On
    uint8_t Fp            :1; // 0=Off, 1=On // FORCE RUN
    uint8_t FanOnly       :1; // 0=Off, 1=On (with mode Auto)
    uint8_t Mode          :2; // 0x11=Heat, 0x00=Auto(& Fan), 0x01=Cool, 0x10=Dry

    // Byte 3
    uint8_t Sleep         :1; // 0=Off, 1=On
    uint8_t OnT_Enable    :1; // 0=Off, 1=On
    uint8_t OffT_Enable   :1; // 0=Off, 1=On
	  uint8_t Power         :1; // 0=Off, 1=On
    uint8_t Timer         :4; // 1-12h
		// timer hours 1-9, 0xA=10, 0xB=11 0xC=12

    // Byte 4
    uint8_t               :8;  // Always 0xAA
  };
};
```

Onepamopa, 2022-12-16:
@crankyoldgit I've managed to get it to a point where it detects everything but the temperature.
For some reason the temperature just refuses to cooperate... The values are represented in hex, i.e. 0x16 is 16C, 0x30 is 30C
Also, haven't quite gotten the checksum part yet. I'll create a separate "fresh" branch and add everything there so you can help :) 

```
union IkedaProtocol{
  uint64_t remote_state;  ///< The state in native IR code form
  struct {
    // Byte 4
    uint8_t               :8;  // Always 0xAA

    // Byte 3
    uint8_t Timer         :4; // 1-12h
    uint8_t Power         :1; // 0=Off, 1=On
    uint8_t OffT_Enable   :1; // 0=Off, 1=On
    uint8_t OnT_Enable    :1; // 0=Off, 1=On
    uint8_t Sleep         :1; // 0=Off, 1=On
		// timer hours 1-9, 0xA=10, 0xB=11 0xC=12

    // Byte 2
    uint8_t Mode          :2; // 0b10001 (17) 0x11=Heat, 0b00 (0) 0x00=Auto(& Fan), 0b0001 (1) 0x01=Cool, 0b00010000 (16) 0x10=Dry
    uint8_t FanOnly       :1; // 0=Off, 1=On (with mode Auto)
    uint8_t Fp            :1; // 0=Off, 1=On // FORCE RUN (Turbo?)
    uint8_t Flap          :1; // 0=Off, 1=On
    uint8_t Fan           :2; // 3=Low, 2=Medium, 1=High, 0=Auto
    uint8_t               :1; // Empty 0x0

    // Byte 1
    //uint8_t               :3;
    uint8_t Temperature   :5;		// 0x16 = 16C . . . 0x30 = 30C
    uint8_t               :3;
    //uint8_t Temperature   :8;

    // Byte 0
    uint8_t Checksum      :8;
  };
};
```

Onepamopa, 2022-12-16:
@crankyoldgit Here's everything in one place. If you can give me a little help with why the temperature isn't getting detected properly.. And with the checksum :) 

https://github.com/crankyoldgit/IRremoteESP8266/compare/master...Onepamopa:IRremoteESP8266:master

Onepamopa, 2022-12-17:
@crankyoldgit if you take a look at https://docs.google.com/spreadsheets/d/1IOxfKN2PPuIayC1UFZqkHrk7LAqH0AwUjgUn6dIrnzg  in the "LSB" sheet - I've broken down the protocol.

The problem with temperature is that when it's 0x16 (16C on the remote) it actually shows 22C in the library (0x16 = 22).
As with the checksum - I can't figure out why sometimes it's just empty (i.e. the whole hex is 2 symbols shy - you can see @ the google docs)

closed 2022-12-18 by Onepamopa