crankyoldgit/IRremoteESP8266 #1922: If IR emitter has WiFi chip enabled, IRrecvDumpV2 receives many broken packets

Ks89, 2022-11-24:
### Version/revision of the library used

`version 2.8.4`

### Describe the bug

I'm using the `COOLIX` protocol to control a `Beko Air conditioner` (with remote `RG52A9/BGEF`).
My setup is a `ESP32S2 Devkit-C` sender with a `Grove IR emitter` and a `ESP32S2 Devkit-C` with `IRrecvDumpV2`.
Everything works as expected and I can read packets and control my AC successfully.

**However, when I enable the WiFi chip of the sender device, the receiver reads many broken packets.**
So, it works perfectly without WiFi, but when I connect to my WiFi network, the receiver starts to read many broken packets.

#### To Reproduce

Check this example `https://github.com/Ks89/ir-esp32-example`
It contains two projects:
- sender: is the project connected to the IR Emitter
- receiver: is the official IRrecvDumpV2 example

If you comment lines from 37 to 41 (https://github.com/Ks89/ir-esp32-example/blob/9559e02876779aa2ece39f40b3204452f6ffca6b/sender/sender.ino#L37) in this way:
```Cpp
//WiFi.begin(ssid, password);
//while (WiFi.status() != WL_CONNECTED) {
//    delay(500);
//    Serial.print(".");
//}
```
it works as expected with no broken packages, otherwise not.

#### Example code used

`https://github.com/Ks89/ir-esp32-example`

#### Expected behaviour

If a send a signal from the sender to the receiver I should be able to read valid packets also with WiFI enabled, instead of many broken data:
```
21:31:51.036 -> Timestamp : 000043.765
21:31:51.036 -> Library   : v2.8.4
21:31:51.036 -> 
21:31:51.036 -> Protocol  : COOLIX
21:31:51.036 -> Code      : 0xB2BF2C (24 Bits)
21:31:51.036 -> Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 5 (Auto), Temp: 20C, Zone Follow: Off, Sensor Temp: Off
21:31:51.036 -> uint16_t rawData[199] = {4700, 4500,  556, 1640,  556, 552,  562, 1650,  566, 1646,  560, 548,  564, 542,  560, 1652,  564, 550,  564, 544,  556, 1654,  562, 546,  558, 552,  562, 1650,  566, 1648,  568, 542,  562, 1662,  564, 1648,  558, 548,  564, 1654,  564, 1652,  564, 1646,  558, 1654,  562, 1650,  566, 1646,  560, 548,  564, 1648,  556, 552,  562, 546,  566, 544,  560, 550,  564, 544,  558, 562,  570, 542,  560, 548,  566, 1646,  560, 550,  562, 1650,  568, 1642,  564, 544,  556, 552,  560, 1652,  564, 1648,  558, 550,  562, 1654,  562, 544,  558, 550,  562, 1650,  566, 1668,  558, 5244,  4716, 4418,  568, 1652,  564, 542,  560, 1654,  562, 1682,  554, 532,  560, 546,  566, 1646,  562, 554,  558, 550,  562, 1648,  558, 552,  560, 546,  556, 1654,  562, 1650,  566, 542,  560, 1712,  564, 1646,  560, 550,  562, 1648,  568, 1650,  566, 1650,  566, 1644,  562, 1648,  558, 1656,  560, 548,  564, 1646,  560, 552,  560, 544,  560, 550,  562, 546,  558, 550,  562, 558,  566, 540,  572, 540,  564, 1652,  564, 544,  558, 1656,  560, 1650,  566, 546,  568, 540,  560, 1652,  564, 1648,  558, 550,  562, 1650,  556, 550,  562, 546,  558, 1660,  566, 1656,  560};  // COOLIX B2BF2C
21:31:51.132 -> uint64_t data = 0xB2BF2C;
21:31:51.132 -> 
21:31:51.132 -> 
21:31:52.319 -> Timestamp : 000045.061
21:31:52.319 -> Library   : v2.8.4
21:31:52.319 -> 
21:31:52.319 -> Protocol  : UNKNOWN
21:31:52.319 -> Code      : 0xC13D9FAE (99 Bits)
21:31:52.319 -> uint16_t rawData[197] = {4692, 4434,  560, 1656,  560, 546,  568, 1646,  560, 1654,  564, 544,  556, 552,  560, 1652,  564, 548,  564, 544,  558, 1654,  562, 546,  566, 544,  558, 1652,  564, 1648,  568, 542,  560, 1662,  564, 1648,  558, 552,  560, 1656,  570, 1638,  568, 2158,  366, 2070,  564, 1646,  560, 1658,  558, 550,  564, 1648,  568, 546,  556, 554,  560, 546,  566, 542,  560, 548,  566, 604,  560, 548,  564, 544,  558, 1652,  564, 544,  558, 1654,  562, 1656,  560, 544,  568, 540,  562, 1650,  566, 1646,  560, 548,  564, 1648,  558, 550,  562, 544,  558, 1654,  562, 1664,  562, 5246,  4694, 4444,  562, 1652,  566, 544,  558, 1654,  562, 1652,  564, 544,  558, 550,  564, 1650,  566, 542,  562, 546,  566, 1646,  560, 548,  556, 550,  562, 1650,  566, 1644,  560, 546,  578, 1654,  562, 1652,  554, 552,  562, 1650,  566, 1646,  560, 1652,  564, 1656,  560, 1648,  558, 2942,  560, 1654,  564, 544,  568, 546,  558, 554,  560, 546,  566, 544,  560, 614,  560, 548,  564, 542,  560, 1654,  562, 546,  558, 1656,  560, 1652,  564, 546,  558, 550,  562, 1650,  568, 1648,  568, 544,  558, 1654,  562, 546,  558, 550,  562, 1650,  566, 1658,  558};  // UNKNOWN C13D9FAE
21:31:52.449 -> 
21:31:52.449 -> 
21:31:53.629 -> Timestamp : 000046.353
21:31:53.629 -> Library   : v2.8.4
21:31:53.629 -> 
21:31:53.629 -> Protocol  : UNKNOWN
21:31:53.629 -> Code      : 0x9909D176 (99 Bits)
21:31:53.629 -> uint16_t rawData[197] = {4692, 4438,  566, 1646,  560, 548,  564, 1648,  558, 1654,  562, 548,  564, 542,  560, 1656,  560, 552,  562, 546,  566, 1646,  560, 548,  564, 544,  558, 1654,  562, 1652,  566, 542,  558, 1666,  562, 1650,  566, 544,  558, 1654,  562, 1652,  564, 1650,  558, 1654,  562, 1650,  566, 1648,  568, 542,  560, 1654,  564, 544,  556, 550,  562, 546,  558, 552,  560, 552,  570, 550,  564, 544,  558, 548,  566, 1650,  566, 542,  560, 1654,  562, 1656,  560, 544,  560, 550,  562, 1648,  558, 1652,  562, 546,  566, 1654,  562, 540,  562, 548,  564, 3998,  558, 5244,  4714, 4432,  562, 1650,  566, 540,  562, 1650,  556, 1656,  560, 548,  564, 542,  560, 1652,  566, 544,  566, 542,  560, 1654,  562, 546,  556, 552,  560, 1652,  566, 1646,  560, 546,  556, 1668,  560, 1654,  562, 546,  556, 1654,  562, 1654,  562, 1652,  564, 1644,  562, 1650,  566, 1648,  568, 542,  562, 1650,  566, 544,  558, 548,  566, 544,  558, 550,  562, 544,  558, 560,  562, 548,  566, 544,  568, 1644,  562, 546,  566, 1646,  560, 1652,  562, 546,  568, 540,  562, 1652,  564, 1646,  560, 546,  568, 1646,  560, 548,  564, 544,  558, 1652,  564, 1678,  558};  // UNKNOWN 9909D176
21:31:53.726 -> 
21:31:53.726 -> 
21:31:54.910 -> Timestamp : 000047.645
21:31:54.910 -> Library   : v2.8.4
21:31:54.910 -> 
21:31:54.910 -> Protocol  : COOLIX
21:31:54.910 -> Code      : 0xB2BF2C (24 Bits)
21:31:54.910 -> Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 5 (Auto), Temp: 20C, Zone Follow: Off, Sensor Temp: Off
[... 161 lines cut ...]
ac.setTemp(20);
ac.setFan(FAN_AUTO);
ac.setMode(HEAT_MODE);

// call send in a loop every second
ac.send();
```

### What brand/model IR demodulator are you using?

Emitter: https://www.seeedstudio.com/Grove-Infrared-Emitter.html

Receiver: https://www.seeedstudio.com/Grove-Infrared-Receiver.html?queryID=1622147f7b209c34a6cebe9444abb08e&objectID=2247&indexName=bazaar_retailer_products

### Circuit diagram and hardware used (if applicable)

Sender:
`ESP32S2 DevKit-C` with `Grove Infrared Emitter` connected to `GPIO 4`

Receiver:
`ESP32S2 DevKit-C` with `Grove Infrared Receiver` connected to `GPIO 14`

### Other useful information

Feel free to ask. I can reproduce this problem again and again.

crankyoldgit, 2022-11-25:
Something is screwing around with the timing system. i.e. The pulses produced are not as they should be.
This is probably due to the Wifi operating in the background while sending. In theory, it shouldn't interrupt it or cause a problem.

As I don't think I have an ESP32-S2 unit handy I'm going to have to get you to test a few things.
First. Let's start with adding a `ac.calibrate();` in your `init()` clause. The default timings for an ESP should be the same, but I never tested it on an S2 variant.

I doub't that will fix it but. I'll get back to you with some more ideas in a bit.

crankyoldgit, 2022-11-25:
Looking into this more, it seems the S2 being a single processor is more susceptable to causing delays in bit-banging (which is what we are doing to generate a PWM signal on the ESP chips). So it looks like this won't be properly fixed till we implement sending via the RMT hardware. I guess I'll need to find some time and up the priority on that.

Ks89, 2022-11-25:
Hi @crankyoldgit thanks for the info.
I tried to add `ac.calibrate();` in `setup()` before `ac.begin()` but it still not working. The same, also  with `ac.calibrate();` after `ac.begin()`.

> Looking into this more, it seems the S2 being a single processor is more susceptable to causing delays in bit-banging (which is what we are doing to generate a PWM signal on the ESP chips). So it [...]

Ok, yes this is a big problem, because ESP32 S2 is very common today (S3 is still quite rare, but it's already available on the market). I have many of them in my home.
Is the first ESP32 DevkitC supported?
If you want to work on this issue with high priority I can help you to do some tests on different ESP devices.
I have the first Esp32 devkitc (Az-delivery), ESP32S2 (official by Espressif) and I can find an ESPS3 DevkitC if required.

Ks89, 2022-11-26:
Tested on all 3 versions of ESP32 DevKits and these are the results:
- ESP32 DevKit-C (the first version with ESP32-WROOM-32): OK
- ESP32 S2 DevKit-C (ESP32-S2-SOLO): **not working as described above**
- ESP32 S3 DevKit-C (ESP32-S3-WROOM-1): OK

Onepamopa, 2022-12-15:
Is ESP-32S affected?

Ks89, 2022-12-15:
No idea. Is it a sort of mid version between ESP32 and ESP32S2?

crankyoldgit, 2022-12-15:
If it is dual core (2 CPUs) then you'll probably be fine

Onepamopa, 2022-12-16:
aren't all esp32's dual core ?

Onepamopa, 2022-12-19:
> > aren't all esp32's dual core ?
> 
> No. e.g. The ESP32-S2 is a single core, hence why we are have an issue on that platform when the CPU needs to multi-task.

Isn't ESP8266 also a single-core board? Why doesn't it have an issue?

NiKiZe, 2023-06-01:
Is there any chance that this maybe is something similar as #1995 where "the pin used was the issue"

PavelSheremetev, 2024-09-14:
This problem is not directly related to Tasmota. It stems from the [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) library which does not work with [RMT](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html) yet. On new chips (in my case esp32c6) sending signals works badly, for the air conditioner to receive a signal it has to be repeated 5 times.