crankyoldgit/IRremoteESP8266 #1937: LG AC AKB75415316

abdulhannan97, 2022-12-28:
Hi i am abdul
i am trying to control lg ac 

this is data what i get from remote

Timestamp : 000005.234
Library   : v2.8.2

Protocol  : LG2
Code      : 0x880094D (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 24C, Fan: 4 (Maximum)
uint16_t rawData[59] = {3144, 9658,  558, 1490,  550, 474,  530, 494,  530, 470,  560, 1484,  558, 468,  552, 474,  520, 504,  530, 470,  556, 468,  560, 460,  562, 468,  520, 504,  530, 494,  532, 468,  556, 468,  558, 1486,  554, 472,  520, 504,  530, 1518,  532, 468,  558, 1486,  558, 468,  552, 472,  522, 1524,  532, 1488,  558, 466,  556, 1490,  558};  // LG2 880094D
uint32_t address = 0x88;
uint32_t command = 0x94;
uint64_t data = 0x880094D;

Timestamp : 000008.215
Library   : v2.8.2

Protocol  : LG2
Code      : 0x88C0051 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: Off
uint16_t rawData[59] = {3146, 9654,  556, 1490,  526, 500,  522, 504,  530, 470,  558, 1488,  558, 468,  556, 468,  522, 504,  530, 1490,  558, 1490,  558, 466,  526, 500,  520, 506,  530, 470,  558, 466,  556, 466,  526, 500,  522, 508,  528, 468,  558, 466,  558, 470,  552, 1494,  532, 494,  530, 1490,  560, 466,  556, 470,  554, 472,  530, 1516,  530};  // LG2 88C0051
uint32_t address = 0x88;
uint32_t command = 0xC005;
uint64_t data = 0x88C0051;

and this is code what i am useing

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>

const uint16_t kIrLed = 4;  // The ESP GPIO pin to use that controls the IR LED.
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

void setup() {
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send.
  // See state_t, opmode_t, fanspeed_t, swingv_t, & swingh_t in IRsend.h for
  // all the various options.
  ac.next.protocol = decode_type_t::LG2;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
  ac.next.degrees = 25;  // 25 degrees.
  ac.next.fanspeed = stdAc::fanspeed_t::kMedium;  // Start the fan at medium.
  ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
  ac.next.beep = false;  // Turn off any beep from the A/C if we can.
  ac.next.econo = false;  // Turn off any economy modes if we can.
  ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  ac.next.turbo = false;  // Don't use any turbo/powerful/etc modes.
  ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
  ac.next.sleep = -1;  // Don't set any sleep time or modes.
  ac.next.clean = false;  // Turn off any Cleaning options if we can.
  ac.next.clock = -1;  // Don't set any current time if we can avoid it.
  ac.next.power = false;  // Initially start with the unit off.

  Serial.println("Try to turn on & off every supported A/C type ...");
}

void loop() {
  // For every protocol the library has ...
  for (int i = 1; i < kLastDecodeType; i++) {
    decode_type_t protocol = (decode_type_t)51;
    // If the protocol is supported by the IRac class ...
    if (ac.isProtocolSupported(protocol)) {
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol) + " is supported.");
      ac.next.protocol = protocol;  // Change the protocol used.
      ac.next.power = true;  // We want to turn on the A/C unit.
      Serial.println("Sending a message to turn ON the A/C unit.");
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(5000);  // Wait 5 seconds.
      ac.next.power = false;  // Now we want to turn the A/C off.
      Serial.println("Send a message to turn OFF the A/C unit.");
      ac.sendAc();  // Send the message.
      delay(1000);  // Wait 1 second.
    }
  }
  Serial.println("Starting from the begining again ...");
}

BUT LG AC DID NOT GET TURN ON or OFF

CAN YOU PLEASE HELP ME ON THIS

LG REMOTE MODEL NO IN BACKSIDE IS  :- AKB75415316

crankyoldgit, 2022-12-28:
The data you captured indicates:
> Mesg Desc.: Model: **2** (AKB75215403), ...

So try:
```c++
ac.next.model = 2; // Some A/Cs have different models. Try just the SECOND one.
```

abdulhannan97, 2022-12-28:
same not working...

change this line only

ac.next.model = 1;
replace with 
ac.next.model = 2;

crankyoldgit, 2022-12-28:
Have you ever had the library successfully send a signal yet? I.e. control a device from a distance?

Have you followed the trouble shooting guide yet?

What is the output on the serial interface?

abdulhannan97, 2022-12-31:
14:10:35.084 -> Protocol 51 / LG2 is supported.
14:10:35.084 -> Sending a message to turn ON the A/C unit.
14:10:40.176 -> Send a message to turn OFF the A/C unit.
14:10:41.296 -> Protocol 51 / LG2 is supported.
14:10:41.296 -> Sending a message to turn ON the A/C unit.
14:10:46.393 -> Send a message to turn OFF the A/C unit.
14:10:47.514 -> Protocol 51 / LG2 is supported.
14:10:47.514 -> Sending a message to turn ON the A/C unit.
14:10:52.607 -> Send a message to turn OFF the A/C unit.
14:10:53.727 -> Protocol 51 / LG2 is supported.
14:10:53.727 -> Sending a message to turn ON the A/C unit.

This is serial output what i am getting

on 2nd side i put receiver and this is data what i get in receiver

14:27:36.084 -> Protocol  : LG2
14:27:36.084 -> Code      : 0x88C0051 (28 Bits)
14:27:36.084 -> Mesg Desc.: Model: 2 (AKB75215403), Power: Off
14:27:36.084 -> uint16_t rawData[59] = {3268, 9864,  536, 1552,  516, 526,  548, 492,  546, 492,  550, 1542,  536, 504,  544, 492,  548, 492,  538, 1552,  548, 1540,  546, 494,  548, 492,  548, 492,  536, 504,  544, 492,  538, 504,  534, 530,  516, 492,  548, 492,  538, 502,  534, 502,  536, 1548,  550, 492,  538, 1548,  550, 492,  548, 492,  546, 498,  536, 1552,  534};  // LG2 88C0051
14:27:36.118 -> uint32_t address = 0x88;
14:27:36.118 -> uint32_t command = 0xC005;
14:27:36.118 -> uint64_t data = 0x88C0051;
14:27:36.152 -> 
14:27:36.152 -> 
14:27:39.888 -> Timestamp : 000023.714
14:27:39.888 -> Library   : v2.8.2
14:27:39.888 -> 
14:27:39.888 -> Protocol  : LG2
14:27:39.888 -> Code      : 0x880092B (28 Bits)
14:27:39.888 -> Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 24C, Fan: 2 (Medium)
14:27:39.888 -> uint16_t rawData[59] = {3266, 9862,  532, 1558,  548, 490,  504, 538,  534, 530,  516, 1544,  538, 530,  508, 502,  536, 502,  536, 502,  548, 490,  504, 534,  548, 492,  550, 490,  548, 490,  548, 492,  550, 490,  548, 1542,  550, 492,  538, 502,  544, 1542,  550, 490,  534, 508,  544, 1542,  536, 504,  546, 1542,  550, 492,  502, 1584,  550, 1544,  552};  // LG2 880092B
14:27:39.922 -> uint32_t address = 0x88;
14:27:39.922 -> uint32_t command = 0x92;
14:27:39.956 -> uint64_t data = 0x880092B;
14:27:39.956 -> 
14:27:39.956 ->

crankyoldgit, 2022-12-31:
So it seems:

Real remote OFF: `0x88C0051`
Library OFF: `0x88C0051`

So, the OFF signal should work.

Real remote ON:  `0x880094D`
Library ON: `0x880092B` (assuming I'm understanding your data correctly, IS THAT CORRECT?)

It differs in the last 8 bits.
e.g. `0x4D` is `0x4` for the fan speed and `0xD` for the checksum.
4 is Maximum fan, 2 is Medium fan.
So that explains the slightly different ON codes.

So, I'm initially at a loss as to why it isn't working.

Have you tried sending/replaying the captured codes via `sendRaw()` & `sendLG2()` directly yet?

if `sendRaw()` doesn't work for replaying a message, it seems more like a hardware/circut/environment issue.