crankyoldgit/IRremoteESP8266 #1943: Add support for CARRIER 42WM100CT

Petititi, 2023-01-04:
Hi,
I'm trying to control my AC device without luck. Here are the log of IRrecvDumpV2 :
```
IRrecvDump is now running and waiting for IR input on Pin 25
Timestamp : 000005.715
Library   : v2.8.4

Protocol  : UNKNOWN
Code      : 0xE366A1BC (86 Bits)
uint16_t rawData[171] = {5844, 1152,  1154, 462,  1152, 462,  418, 1194,  418, 1194,  420, 1194,  416, 1198,  418, 1196,  416, 1196,  418, 1194,  420, 1192,  420, 1194,  420, 1194,  1152, 460,  1152, 460,  420, 1192,  418, 1194,  1154, 458,  418, 1194,  420, 1192,  1154, 460,  1152, 458,  420, 1194,  418, 1194,  418, 1194,  1152, 460,  1154, 458,  418, 1196,  416, 1194,  418, 1194,  418, 1196,  420, 1194,  418, 1192,  420, 1194,  420, 1192,  418, 1196,  416, 1194,  416, 1196,  1152, 462,  416, 1194,  422, 1192,  1154, 458,  420, 1192,  418, 1194,  418, 1194,  418, 1196,  420, 1194,  418, 1192,  420, 1194,  418, 1192,  420, 1194,  418, 1194,  420, 1194,  418, 1194,  1154, 458,  418, 1194,  418, 1192,  1154, 458,  418, 1196,  416, 1194,  420, 1194,  416, 1192,  418, 1194,  1152, 462,  416, 1196,  1152, 458,  418, 1196,  1152, 458,  418, 1194,  1154, 460,  418, 1194,  418, 1196,  418, 1196,  418, 1194,  1154, 460,  420, 1192,  418, 1194,  420, 1194,  418, 1194,  418, 1196,  1152, 460,  418, 1194,  1154, 458,  1154, 460,  1152, 458,  1152};  // UNKNOWN E366A1BC

Timestamp : 000520.499
Library   : v2.8.4

Protocol  : UNKNOWN
Code      : 0x606EAF1C (86 Bits)
uint16_t rawData[171] = {5850, 1156,  1150, 460,  1152, 464,  412, 1200,  414, 1200,  416, 1196,  416, 1198,  414, 1198,  416, 1196,  416, 1198,  416, 1198,  414, 1200,  416, 1196,  1150, 462,  1150, 462,  416, 1200,  412, 1198,  1152, 460,  414, 1198,  416, 1196,  1152, 462,  1150, 462,  414, 1198,  414, 1198,  416, 1198,  1148, 462,  1150, 464,  414, 1198,  416, 1198,  412, 1200,  414, 1196,  416, 1198,  412, 1200,  412, 1198,  414, 1200,  414, 1198,  414, 1198,  414, 1200,  1152, 462,  414, 1196,  414, 1198,  1152, 462,  414, 1200,  416, 1196,  414, 1198,  414, 1200,  416, 1196,  414, 1200,  412, 1200,  414, 1198,  416, 1198,  414, 1198,  414, 1198,  414, 1200,  1148, 462,  412, 1198,  414, 1198,  1152, 462,  414, 1198,  416, 1196,  416, 1198,  1148, 464,  1150, 464,  412, 1198,  414, 1198,  414, 1202,  414, 1196,  414, 1200,  412, 1200,  416, 1198,  1148, 464,  412, 1200,  414, 1198,  414, 1200,  1146, 464,  414, 1200,  414, 1198,  1148, 464,  1152, 462,  410, 1202,  412, 1200,  412, 1200,  1148, 464,  1146, 466,  1148, 464,  1146};  // UNKNOWN 606EAF1C
```
Unfortunately, `auto_analyse_raw_data.py` doesn't understand the protocol:
```
Found 171 timing entries.
Potential Mark Candidates:
[5850, 1152, 416]
Potential Space Candidates:
[1202, 466]

Guessing encoding type:
Sorry, it looks like it is Mark encoded. I can't do that yet. Exiting.
```

Any clue of what happend?

NiKiZe, 2023-01-04:
Auto analyze only has support for the most common way to send data.

Can you collect more data with lowest to highest temperature setting?

Also do you have a model number of the remote?

crankyoldgit, 2023-01-05:
Looking at the raw data, it appears this protocol uses a constant bit-time transmission protool.

crankyoldgit, 2023-01-05:
I've got some half-arsed decoding working. But the data portion of the protocol appears to be 84 bits long.
We can handle unusual bit lengths <= 64 bits.
Over 64, we need it to be in nice 8-bit chunks to work without horrible hacks.
e.g. 80 or 88.

Out best bet is if **ALL** the raw messages start with the same 4 bits.
That means the `rawData` needs to have this pattern: (i.e Numbers roughly the same, not exactly. e.g With in +/- 200 of the corresponding array)
```c++
uint16_t rawData[171] = {5844, 1152,  1154, 462,  1152, 462,  418, 1194,  418, 1194, ...
```

Let me know how you go with your data collection.

Petititi, 2023-01-05:
Thanks a lot for your interest !
I dumped some frames in this file:
[IR dump.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/10354618/IR.dump.xlsx)
Tell me if this is enough and/or if I can do something else.

Petititi, 2023-01-05:
[IR dump.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/10354694/IR.dump.xlsx)

crankyoldgit, 2023-01-06:
I've cobbled together something that tries to match and send this odd-ball protocol.
Can you please download & test out branch [https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1943](https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1943) / PR #1945 

Let me know how it goes. It's very experimental, so don't rely on the data yet. Please capture a **full** temperature range as we have still to work out the bit ordering. (assuming it works enought to capture `state[]` data for it.

Petititi, 2023-01-06:
Ji !
Thanks a lot for your help. I captured those frames:
```
Timestamp : 000038.590
Library   : v2.8.4
Temp : 10
Protocol  : CARRIER_AC84
Code      : 0x0C00C08C004800488498D4 (84 Bits)
uint16_t rawData[171] = {5846, 1140,  1166, 466,  1148, 464,  412, 1202,  410, 1202,  412, 1200,  412, 1200,  412, 1202,  410, 1202,  412, 1198,  412, 1202,  410, 1202,  410, 1202,  1146, 464,  1148, 464,  412, 1200,  412, 1202,  412, 1216,  398, 1200,  412, 1200,  412, 1200,  1148, 464,  412, 1200,  412, 1200,  414, 1198,  1148, 462,  1150, 464,  412, 1200,  412, 1200,  414, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  414, 1200,  412, 1202,  412, 1202,  1144, 468,  412, 1212,  400, 1200,  1146, 466,  412, 1200,  414, 1198,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  412, 1200,  1148, 464,  410, 1200,  412, 1200,  1148, 464,  412, 1200,  414, 1202,  410, 1198,  1150, 462,  412, 1210,  402, 1200,  414, 1200,  412, 1202,  1146, 466,  410, 1202,  412, 1200,  1146, 466,  412, 1202,  412, 1200,  1148, 464,  1150, 462,  414, 1200,  412, 1200,  412, 1200,  1148, 462,  1150, 462,  412, 1198,  1148, 462,  414, 1202,  1146, 466,  410, 1198,  414, 1198,  1148};  // CARRIER_AC84
uint8_t state[11] = {0x0C, 0x00, 0xC0, 0x8C, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0xD4};

Timestamp : 000040.917
Library   : v2.8.4
Temp : 11
Protocol  : CARRIER_AC84
Code      : 0x0C00C88C00480048849834 (84 Bits)
uint16_t rawData[171] = {5856, 1132,  1150, 464,  1150, 462,  412, 1200,  416, 1200,  414, 1200,  412, 1202,  412, 1208,  406, 1200,  416, 1198,  412, 1198,  416, 1200,  414, 1196,  1150, 464,  1150, 462,  414, 1200,  414, 1206,  1142, 464,  412, 1202,  410, 1202,  412, 1200,  1150, 464,  414, 1198,  414, 1200,  414, 1200,  1150, 464,  1148, 462,  416, 1198,  412, 1202,  414, 1198,  412, 1200,  412, 1200,  416, 1196,  416, 1200,  412, 1202,  412, 1200,  414, 1198,  414, 1198,  1148, 464,  414, 1200,  414, 1198,  1150, 462,  416, 1200,  412, 1200,  412, 1200,  416, 1198,  414, 1198,  416, 1198,  412, 1200,  412, 1200,  414, 1198,  414, 1202,  412, 1200,  412, 1200,  1150, 462,  414, 1200,  412, 1200,  1150, 462,  412, 1200,  414, 1200,  414, 1200,  1148, 464,  416, 1200,  414, 1198,  414, 1198,  412, 1202,  1148, 464,  410, 1202,  412, 1206,  1142, 462,  414, 1202,  412, 1200,  1148, 464,  1148, 464,  414, 1198,  416, 1198,  414, 1200,  414, 1198,  412, 1200,  1148, 464,  1152, 462,  414, 1198,  1150, 462,  412, 1200,  414, 1200,  1148};  // CARRIER_AC84
uint8_t state[11] = {0x0C, 0x00, 0xC8, 0x8C, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0x34};

Timestamp : 000042.848
Library   : v2.8.4
Temp : 12
Protocol  : CARRIER_AC84
Code      : 0x0C00C48C004800488498B4 (84 Bits)
uint16_t rawData[171] = {5852, 1150,  1150, 466,  1148, 464,  414, 1198,  416, 1196,  416, 1198,  412, 1202,  412, 1200,  412, 1200,  416, 1198,  412, 1198,  414, 1202,  412, 1200,  1148, 466,  1150, 462,  414, 1200,  414, 1200,  414, 1198,  1152, 462,  414, 1198,  412, 1202,  1148, 464,  412, 1202,  412, 1200,  414, 1200,  1152, 460,  1148, 466,  414, 1198,  414, 1200,  414, 1200,  414, 1200,  416, 1198,  412, 1200,  412, 1200,  412, 1202,  414, 1196,  414, 1200,  414, 1200,  1150, 462,  412, 1200,  412, 1200,  1150, 464,  412, 1200,  414, 1200,  412, 1200,  412, 1200,  414, 1198,  416, 1200,  414, 1196,  414, 1198,  412, 1200,  416, 1198,  412, 1202,  414, 1200,  1150, 464,  412, 1202,  412, 1200,  1148, 464,  412, 1200,  414, 1202,  410, 1202,  1148, 462,  414, 1200,  414, 1200,  414, 1198,  414, 1200,  1150, 462,  412, 1200,  412, 1200,  1148, 464,  414, 1200,  414, 1200,  1150, 462,  1150, 464,  414, 1198,  416, 1200,  412, 1200,  1146, 464,  412, 1200,  1148, 464,  1150, 462,  416, 1198,  1150, 462,  416, 1198,  412, 1200,  1148};  // CARRIER_AC84
uint8_t state[11] = {0x0C, 0x00, 0xC4, 0x8C, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0xB4};

Timestamp : 000044.505
Library   : v2.8.4
Temp : 13
Protocol  : CARRIER_AC84
Code      : 0x0C00CC8C00480048849874 (84 Bits)
uint16_t rawData[171] = {5872, 1130,  1154, 462,  1150, 464,  412, 1200,  416, 1198,  414, 1200,  412, 1202,  414, 1198,  412, 1202,  414, 1200,  414, 1200,  410, 1200,  414, 1200,  1148, 464,  1148, 466,  412, 1200,  414, 1202,  1148, 464,  1148, 464,  414, 1200,  412, 1200,  1148, 466,  414, 1200,  412, 1200,  412, 1202,  1150, 464,  1148, 462,  412, 1200,  412, 1200,  412, 1202,  414, 1196,  414, 1200,  414, 1200,  414, 1198,  412, 1202,  414, 1200,  412, 1200,  414, 1200,  1148, 464,  414, 1198,  414, 1200,  1148, 468,  412, 1200,  414, 1200,  414, 1200,  416, 1198,  412, 1202,  414, 1198,  414, 1200,  412, 1202,  412, 1202,  412, 1200,  412, 1200,  412, 1200,  1148, 466,  412, 1200,  410, 1202,  1148, 462,  414, 1200,  410, 1200,  416, 1196,  1148, 466,  414, 1200,  414, 1200,  412, 1200,  414, 1200,  1148, 464,  414, 1200,  410, 1200,  1152, 464,  412, 1200,  414, 1200,  1150, 462,  1148, 466,  414, 1198,  414, 1200,  412, 1202,  412, 1200,  1150, 462,  1150, 464,  1148, 464,  414, 1198,  1150, 462,  412, 1202,  412, 1198,  1152};  // CARRIER_AC84
uint8_t state[11] = {0x0C, 0x00, 0xCC, 0x8C, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0x74};

Timestamp : 000046.167
Library   : v2.8.4
Temp : 14
Protocol  : CARRIER_AC84
Code      : 0x0C00C28C004800488498F4 (84 Bits)
[... cut ...] 1202,  1148, 464,  412, 1202,  412, 1200,  1148, 466,  1148, 464,  412, 1200,  414, 1200,  414, 1198,  1150, 464,  1148, 464,  412, 1200,  1148, 464,  412, 1202,  412, 1198,  1148, 466,  412, 1200,  1148};  // CARRIER_AC84
uint8_t state[11] = {0x0C, 0x00, 0xC0, 0xCC, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0xD2};
```

The signal seems quite noisy as I have to be very (very) close to the receptor to have a good response... Do you think we can have some tolerance (or increase it) around the bit-time ?

Petititi, 2023-01-06:
I found that there is a `const uint8_t tolerance` parameter. I'll try to play a bit with it (as well as slightly modify `kCarrierAc84XXX`)

crankyoldgit, 2023-01-07:
I'll look at changing those parameters. I need to change the bit ordering as the data you've collected indicates it is LSBF, not MSBF order.

Petititi, 2023-01-07:
Using your insights, I am now able to get data which seems correct, but I had to do something I don't fully understand: in `matchGenericConstBitTime` I had to change the bit shift from `result <<= 1;` to `result <<= 2;`. That way, I get something like that:
```
20°, à 9:43 =>
0x11, 0x00, 0x03, 0x32, 0x00, 0x12, 0x00, 0x12, 0x43, 0x09, 0x5C
                      0       2           12:00           12:00        9:43            ??
16°, à 9:47 =>
0x11, 0x00, 0x63, 0x31, 0x00, 0x12, 0x00, 0x12, 0x47, 0x09, 0x92
                      6       1           12:00           12:00        9:47            ??
```

crankyoldgit, 2023-01-07:
Please try the updated branch.

Petititi, 2023-01-07:
That's perfect 😍 !!!
For what I understand, we have a message like that:
- `0x03` => off ; `0x0C` => on
- `0xXY` => X: fan speed (0:auto, 1-3); Y: mode (2:only fan, 3:heat, 4:cold, 0:auto)
- `0xXY` => X: small digit of consign temp; Y: In binary `0bABCD`: A:timer on/off; B:Swing mode on/off; C & D always `1`?
- `0xXY` => X: always '3'? ; Y: big digit of consign temp
- `0xXX` => minute of the timer start
- `0xXX` => hour of the timer start
- `0xXX` => minute of the timer end
- `0xXX` => hour of the timer end
- `0xXX` => current minute
- `0xXX` => current hour
- `0xXX` => ??? A sort of CRC?

Petititi, 2023-01-07:
As a side note, I still have sometimes burst that are not recognized. I can try to play with different values but as I don't fully understand the meaning behind those, I'll probably not do something correct. For instance, this one isn't recognized: 
```
uint16_t rawData[171] = {5828, 1158,  1148, 464,  1148, 494,  384, 1200,  414, 1202,  412, 1202,  410, 1204,  408, 1204,  408, 1206,  408, 1228,  386, 1202,  408, 1232,  386, 1202,  1148, 464,  1146, 468,  412, 1200,  384, 1230,  412, 1200,  412, 1200,  410, 1204,  412, 1200,  412, 1202,  1146, 468,  410, 1202,  410, 1204,  1146, 466,  1146, 464,  410, 1230,  386, 1198,  414, 1200,  412, 1202,  410, 1202,  410, 1200,  410, 1204,  408, 1204,  410, 1202,  412, 1226,  386, 1202,  1144, 468,  408, 1204,  410, 1200,  1146, 468,  412, 1198,  410, 1206,  408, 1202,  410, 1228,  384, 1228,  386, 1202,  410, 1202,  412, 1202,  408, 1202,  410, 1202,  408, 1202,  410, 1204,  1146, 464,  412, 1202,  412, 1200,  1146, 468,  408, 1202,  412, 1200,  382, 1232,  1146, 488,  1122, 466,  410, 1228,  386, 1200,  1146, 492,  1118, 466,  412, 1202,  410, 1204,  1146, 464,  414, 1198,  410, 1204,  410, 1202,  1150, 464,  410, 1202,  412, 1202,  410, 1226,  386, 1200,  410, 1228,  1148, 440,  1148, 466,  1144, 468,  408, 1202,  1144, 470,  1144, 492,  1120};  // UNKNOWN 4CC7BE54
```
This one is correctly recognized (with the same meaning):
```
uint16_t rawData[171] = {5828, 1158,  1150, 466,  1148, 462,  414, 1200,  414, 1198,  414, 1198,  414, 1202,  414, 1200,  412, 1198,  414, 1200,  412, 1202,  414, 1200,  412, 1200,  1146, 468,  1146, 466,  412, 1198,  414, 1202,  412, 1200,  412, 1200,  414, 1200,  414, 1198,  416, 1198,  1150, 464,  414, 1198,  414, 1200,  1146, 466,  1148, 464,  416, 1198,  412, 1202,  414, 1200,  412, 1200,  412, 1198,  414, 1200,  412, 1202,  412, 1198,  414,
 1200,  414, 1198,  414, 1200,  1150, 462,  412, 1202,  412, 1200,  1148, 466,  410, 1202,  412, 1200,  412, 1202,  412, 1200,  414, 1198,  414, 1200,  412, 1200,  416, 1198,  414, 1198,  414, 1200,  416, 1198,  414, 1198,  1156, 458,  414, 1200,  414, 1198,  1150, 462,  414, 1198,  414, 1198,  414, 1198,  1150, 462,  1150, 462,  416, 1198,  416, 1198,  1150, 464,  1148, 464,  414, 1200,  414, 1200,  1150, 462,  412, 1200,  416, 1198,  414, 1198,  1150, 464,  414, 1198,  416, 1198,  414, 1200,  412, 1198,  418, 1196,  1150, 464,  1148, 464,  1150, 462,  414, 1198,  1150, 464,  1150, 462,  1152};  // CARRIER_AC84
uint8_t state[11] = {0x03, 0x00, 0x03, 0x32, 0x00, 0x12, 0x00, 0x12, 0x33, 0x11, 0xDC};
```
Do you have some tips of how to proceed to be more robust? Adjusting `kCarrierAc84Zero` and `kCarrierAc84One` is the best way or do I have an other option?

crankyoldgit, 2023-01-07:
I'll look into improving the matching.
You'll need to workout the checksum part/calculation as that will be critical to getting this supported.
I suggest you focus on working out how to calculate it before everything else.

Petititi, 2023-01-07:
I'll work on it, do you have any guess or other protocols which does a similar control bits?

crankyoldgit, 2023-01-07:
> I'll work on it, do you have any guess or other protocols which does a similar control bits?

See/read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#integrity-checks

Petititi, 2023-01-07:
Thanks a lot for your link, it help me a lot !
So that wasn't very easy, but I figured it out. Using the following code, I am able to compute the correct integrity check byte:
```
      u_int32_t CRC1 = 0, CRC2 = results->state[0];
      for (uint16_t i = 1; i < nbytes - 1; i++)
      {
        // 16 bit crc
        CRC1 = CRC1 + (results->state[i] & 0x0F);
        CRC2 = CRC2 + ((results->state[i]>>4) & 0x0F);
      }
      u_int8_t CRC = (CRC1<<4) + CRC2;
      output += F("uint8_t CRC = ");
      output += uint64ToString(CRC, 16);
      output += F(";\n");
```
What can I do then to help you?

Petititi, 2023-01-08:
I tried to send some datas without success. I'm not sure of my settings & hardware so the problem is probably on my side. Here is what I tried:
```
IRsend irSend(26/*PIN  NUMBER*/, true);
void setup() {
  irSend.begin();
  irSend.enableIROut(38);
}
void loop(){
  uint8_t stateTmp[11] = {0x0C, 0x00, 0xC0, 0x8C, 0x00, 0x48, 0x00, 0x48, 0x84, 0x98, 0xD4};
  irSend.sendCarrierAC84(stateTmp);
  sleep(1);
}
```

crankyoldgit, 2023-01-09:
Unless a lot of the bits line up the same, I wouldn't bother with inheritance.