crankyoldgit/IRremoteESP8266 #1950: TECO recieved but sended copy is not read in target device

whoim2, 2023-01-21:
### Version/revision of the library used
v2.8.4 from Library Manager in Arduino IDE 1.8.19

ESP8285 (ESP-M3) and recieve / send modules and IR-diodes with mosfet.
nonos-sdk-2.2.1 +100 (190703)
This problem only at TECO protocol, NEC and LG worked perfecly.
This is my universal IR center with http, mqtt and IR control: https://github.com/whoim2/esp-ir-center-and-relays

### Describe the bug
Readed string from IR remote / climat device: 25060053C, TECO, 35 bits.
Send this code:
`
struct ir_data { //for store IR full packet
  uint64_t value;
  decode_type_t decode_type;
  uint8_t bits;
};
...
ir_data _action_ir1;
...
//load values from memory and debug it
Serial.println("action1 started");
Serial.println(_action_ir1.value, HEX);
Serial.println(_action_ir1.decode_type, DEC);
Serial.println(_action_ir1.bits, DEC);
//send
if(_action_ir1.decode_type > 0 && _action_ir1.value != 0) {
    irsend.send(_action_ir1.decode_type, _action_ir1.value, _action_ir1.bits);
    Serial.println("action1 sended");
 }
`
Result from serial debug:
`action1 started
25060053C
55
35
action1 sended
`
Diodes flashed in camera view. Device ("Besshof" climat separated device) not view this code, not beep as original ir-remote.

#### To Reproduce
Read TECO data from IR-remote and send copy of this code.

NiKiZe, 2023-01-21:
Have this worked before?
What is the output from the dumpv2 application? (As described in the issue template) also have a read in the FAQ?

whoim2, 2023-01-21:
> 

I worked with nec, lg and also exported with some kind of protocol, not an exporter. The same code is executed using the console, by the value of the log line in littlefs (value, decode_type, bits) and by the trigger mqtt or http is called via send. It works right now with nec and lg, but does not work with teco, although the receiver always recognizes it perfectly, it turns out that this remote is teco.
Tomorrow I will try to master the dump.

whoim2, 2023-01-23:
Ok, im dump remote in 2-3 days

crankyoldgit, 2023-01-30:
How did you go?

whoim2, 2023-01-31:
> How did you go?

No time now, trouble with health. Need many time for this, sorry. Maybe, end of this week.