crankyoldgit/IRremoteESP8266 #1963: Add support for CARRIER_AC40 protocol

prbtt, 2023-03-01:
Using IRrecdumpv2, I captured these codes:
Can you add support for this protocol, please? I really have no idea how to start doing this...
```
16-30 fan max

Showing logs:
[14:02:12]E (218) esp_core_dump_flash: No core d partition found!
[14:02:12]E (218) esp_core_dump_flash: No core dump partition found!
[14:02:12]
[14:02:12]IRrecvDump is now running and waiting for IR input on Pin 14
[14:02:50]Timestamp : 000038.473
[14:02:50]Library   : v2.8.4
[14:02:50]
[14:02:50]Protocol  : CARRIER_AC40
[14:02:50]Code      : 0x1088800830 (40 Bits)
[14:02:50]uint16_t rawData[83] = {9086, 4514,  568, 526,  570, 524,  566, 528,  568, 1624,  568, 524,  572, 528,  542, 552,  570, 528,  544, 1642,  546, 550,  570, 526,  546, 548,  570, 1622,  546, 550,  544, 550,  544, 550,  570, 1622,  546, 548,  546, 552,  544, 552,  544, 550,  568, 524,  546, 552,  544, 550,  546, 550,  546, 548,  548, 550,  570, 524,  570, 1622,  544, 550,  568, 530,  544, 548,  570, 524,  544, 554,  568, 1620,  544, 1646,  570, 526,  544, 550,  546, 548,  568, 526,  568};  // CARRIER_AC40 1088800830
[14:02:50]uint64_t data = 0x1088800830;
[14:02:50]
[14:02:50]
[14:03:04]Timestamp : 000051.690
[14:03:04]Library   : v2.8.4
[14:03:04]
[14:03:04]Protocol  : CARRIER_AC40
[14:03:04]Code      : 0x1048800830 (40 Bits)
[14:03:04]uint16_t rawData[83] = {9084, 4510,  568, 526,  546, 548,  568, 526,  544, 1648,  542, 548,  546, 550,  566, 528,  570, 524,  548, 550,  544, 1642,  546, 550,  570, 526,  572, 1622,  546, 550,  546, 552,  546, 552,  544, 1644,  570, 528,  570, 526,  572, 526,  546, 552,  564, 528,  548, 546,  546, 552,  566, 530,  568, 526,  546, 550,  544, 550,  574, 1618,  546, 550,  548, 550,  570, 526,  546, 550,  548, 548,  548, 1648,  570, 1622,  546, 550,  566, 526,  572, 526,  544, 550,  544};  // CARRIER_AC40 1048800830
[14:03:04]uint64_t data = 0x1048800830;
[14:03:04]
[14:03:04]
[14:03:05]Timestamp : 000052.991
[14:03:05]Library   : v2.8.4
[14:03:05]
[14:03:05]Protocol  : CARRIER_AC40
[14:03:05]Code      : 0x10C8800830 (40 Bits)
[14:03:05]uint16_t rawData[83] = {9074, 4510,  568, 526,  570, 526,  544, 550,  568, 1622,  544, 550,  568, 524,  568, 526,  546, 548,  568, 1622,  544, 1646,  548, 550,  546, 548,  570, 1624,  542, 548,  568, 526,  546, 548,  572, 1620,  544, 548,  572, 526,  544, 548,  548, 546,  570, 526,  546, 548,  568, 526,  546, 548,  546, 550,  570, 524,  570, 528,  566, 1624,  546, 546,  570, 526,  568, 526,  546, 548,  572, 522,  546, 1644,  570, 1622,  570, 524,  544, 552,  544, 554,  546, 548,  544};  // CARRIER_AC40 10C8800830
[14:03:05]uint64_t data = 0x10C8800830;
[14:03:05]
[14:03:05]
[14:03:06]Timestamp : 000054.225
[14:03:06]Library   : v2.8.4
[14:03:06]
[14:03:06]Protocol  : CARRIER_AC40
[14:03:06]Code      : 0x1028800830 (40 Bits)
[14:03:06]uint16_t rawData[83] = {9086, 4510,  568, 526,  568, 526,  568, 524,  572, 1624,  544, 550,  546, 548,  544, 550,  568, 526,  546, 550,  570, 524,  548, 1646,  566, 526,  568, 1622,  566, 528,  544, 550,  544, 552,  570, 1622,  544, 552,  568, 526,  546, 550,  568, 526,  568, 526,  546, 548,  546, 546,  570, 528,  570, 524,  570, 524,  546, 548,  546, 1646,  568, 526,  542, 550,  568, 528,  544, 550,  568, 524,  572, 1620,  572, 1618,  568, 528,  570, 522,  550, 548,  568, 526,  546};  // CARRIER_AC40 1028800830
[14:03:06]uint64_t data = 0x1028800830;
[14:03:06]
[14:03:06]
[14:03:07]Timestamp : 000055.405
[14:03:07]Library   : v2.8.4
[14:03:07]
[14:03:07]Protocol  : CARRIER_AC40
[14:03:07]Code      : 0x10A8800830 (40 Bits)
[14:03:07]uint16_t rawData[83] = {9070, 4512,  546, 550,  568, 526,  572, 524,  544, 1646,  570, 528,  566, 528,  568, 526,  570, 528,  570, 1626,  568, 526,  568, 1626,  566, 528,  566, 1624,  566, 528,  568, 528,  570, 524,  572, 1620,  570, 526,  570, 522,  546, 548,  546, 550,  568, 526,  570, 526,  574, 520,  548, 548,  546, 548,  572, 524,  546, 550,  568, 1620,  544, 546,  572, 524,  566, 528,  572, 524,  568, 530,  542, 1648,  568, 1620,  548, 546,  548, 550,  544, 546,  572, 526,  568};  // CARRIER_AC40 10A8800830
[14:03:07]uint64_t data = 0x10A8800830;
[14:03:07]
[14:03:07]
[14:03:08]Timestamp : 000056.466
[14:03:08]Library   : v2.8.4
[14:03:08]
[14:03:08]Protocol  : CARRIER_AC40
[14:03:08]Code      : 0x1068800830 (40 Bits)
[14:03:08]uint16_t rawData[83] = {9072, 4510,  548, 548,  568, 526,  568, 526,  546, 1644,  544, 550,  568, 522,  574, 524,  568, 528,  548, 546,  570, 1622,  568, 1622,  566, 526,  570, 1622,  546, 546,  574, 524,  546, 548,  572, 1620,  546, 552,  568, 528,  546, 548,  570, 526,  568, 526,  572, 526,  544, 550,  544, 550,  570, 528,  568, 522,  572, 526,  572, 1622,  568, 524,  546, 550,  546, 548,  570, 526,  546, 550,  544, 1648,  544, 1648,  568, 526,  546, 552,  568, 528,  570, 526,  568};  // CARRIER_AC40 1068800830
[14:03:08]uint64_t data = 0x1068800830;
[14:03:08]
[14:03:08]
[14:03:09]Timestamp : 000057.609
[14:03:09]Library   : v2.8.4
[14:03:09]
[14:03:09]Protocol  : CARRIER_AC40
[14:03:09]Code      : 0x10E8800830 (40 Bits)
[14:03:09]uint16_t rawData[83] = {9084, 4512,  570, 526,  546, 552,  570, 524,  548, 1646,  546, 548,  570, 524,  570, 528,  568, 526,  572, 1622,  568, 1620,  548, 1646,  570, 528,  544, 1646,  544, 552,  542, 552,  570, 522,  548, 1646,  568, 528,  548, 550,  544, 550,  546, 550,  548, 550,  568, 526,  546, 548,  546, 550,  568, 526,  572, 524,  546, 548,  546, 1646,  548, 548,  546, 550,  544, 550,  546, 550,  570, 524,  548, 1644,  546, 1648,  566, 528,  568, 526,  570, 528,  546, 550,  570};  // CARRIER_AC40 10E8800830
[... 343 lines cut ...]
[14:06:28]
[14:06:28]Protocol  : CARRIER_AC40
[14:06:28]Code      : 0x10B8C00830 (40 Bits)
[14:06:28]uint16_t rawData[83] = {9092, 4510,  570, 528,  546, 548,  572, 524,  546, 1644,  570, 526,  568, 528,  550, 546,  548, 552,  546, 1648,  592, 500,  570, 1622,  546, 1646,  546, 1646,  546, 548,  568, 526,  550, 544,  546, 1648,  568, 1622,  546, 548,  548, 550,  546, 548,  568, 526,  546, 552,  544, 550,  548, 548,  546, 548,  546, 550,  546, 548,  548, 1644,  546, 550,  568, 524,  546, 550,  570, 528,  544, 550,  570, 1624,  568, 1626,  570, 528,  568, 526,  546, 552,  546, 550,  546};  // CARRIER_AC40 10B8C00830
[14:06:28]uint64_t data = 0x10B8C00830;
[14:06:28]
[14:06:28]
[14:06:29]Timestamp : 000257.480
[14:06:29]Library   : v2.8.4
[14:06:29]
[14:06:29]Protocol  : UNKNOWN
[14:06:29]Code      : 0xC73846E3 (42 Bits)
[14:06:29]uint16_t rawData[83] = {9088, 4508,  572, 528,  568, 526,  570, 524,  570, 1624,  566, 526,  568, 526,  546, 548,  568, 528,  546, 552,  568, 1624,  570, 1624,  568, 1622,  570, 1622,  546, 550,  546, 550,  570, 524,  570, 1622,  568, 1626,  542, 716,  380, 548,  574, 524,  544, 552,  542, 550,  546, 552,  546, 548,  546, 550,  548, 550,  568, 526,  574, 1618,  546, 548,  548, 546,  546, 552,  568, 528,  570, 528,  548, 1650,  544, 1646,  548, 550,  544, 550,  570, 526,  568, 526,  568};  // UNKNOWN C73846E3
[14:06:29]
[14:06:29]
[14:06:31]Timestamp : 000258.830
[14:06:31]Library   : v2.8.4
[14:06:31]
[14:06:31]Protocol  : CARRIER_AC40
[14:06:31]Code      : 0x10F8C00830 (40 Bits)
[14:06:31]uint16_t rawData[83] = {9056, 4542,  544, 550,  546, 552,  568, 526,  546, 1648,  568, 528,  568, 524,  570, 528,  566, 528,  570, 1620,  568, 1626,  548, 1646,  546, 1646,  570, 1622,  546, 550,  544, 550,  570, 524,  548, 1642,  574, 1620,  570, 524,  548, 550,  568, 526,  546, 550,  546, 550,  546, 550,  570, 522,  572, 526,  568, 530,  542, 548,  546, 1648,  572, 524,  570, 526,  570, 528,  544, 550,  548, 546,  548, 1644,  572, 1622,  546, 550,  570, 528,  548, 550,  546, 550,  548};  // CARRIER_AC40 10F8C00830
[14:06:31]uint64_t data = 0x10F8C00830;
[14:06:31]
[14:06:31]
```

NiKiZe, 2023-03-01:
As you can see the Protocol for you A/C is already supported,
What I think you are asking for is detailed support, for which you can read what is needed at https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#can-you-reverse-engineer-my-aircon-protocol-for-me

To avoid misunderstanding, can you clarify what you want to do and what you think is needed.

prbtt, 2023-03-01:
> As you can see the Protocol for you A/C is already supported, What I think you are asking for is detailed support, for which you can read what is needed at [...]

Yes, detailed support. So I can send a JSON with temp+mode+fan speed etc.

prbtt, 2023-03-02:
I read the docs and did some work. So far I know that bits [20-26] inform the temperature (16 to 30), bits [36-38] inform the fan speed (1,2,3) and bit[10] informs Power (on-off). I rounded the numbers to 500, 1600, 9000 and 4500. Bits [1-9], [11-19] [26-35] and [39-83] didn't change. Whats the next step?

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Temperature | Bits [20-26]
-- | --
16 | 1600,500,500,500,500,500,500
17 | 500,500,1600,500,500,500,500
18 | 1600,500,1600,500,500,500,500
19 | 500,500,500,500,1600,500,500
20 | 1600,500,500,500,1600,500,500
21 | 500,500,1600,500,1600,500,500
22 | 1600,500,1600,500,1600,500,500
23 | 500,500,500,500,500,500,1600
24 | 1600,500,500,500,500,500,1600
25 | 500,500,1600,500,500,500,1600
26 | 1600,500,1600,500,500,500,1600
27 | 500,500,500,500,1600,500,1600
28 | 1600,500,500,500,1600,500,1600
29 | 500,500,1600,500,1600,500,1600
30 | 1600,500,1600,500,1600,500,1600

</body>

</html>

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Fan | Bits [36-38]
-- | --
3 | 1600,500,500
2 | 500,500,1600
1 | 1600,500,1600

</body>

</html>

[image]

NiKiZe, 2023-03-02:
You need to work with the code/data hex value and convert to bits.
But you have a great start.

Maybe you can use this as a base https://docs.google.com/spreadsheets/d/1zzDEUQ52y7MZ7_xCU3pdjdqbRXOwZLsbTGvKWcicqCI/edit?usp=drivesdk

prbtt, 2023-03-02:
Not sure if I understand. How do I proceed with the timings to hex conversion?

NiKiZe, 2023-03-02:
> Not sure if I understand. How do I proceed with the timings to hex conversion?

You use the code/data values, not timings.

prbtt, 2023-03-02:
> > Not sure if I understand. How do I proceed with the timings to hex conversion?
> 
> You use the code/data values, not timings.

What would be the code/data values?

crankyoldgit, 2023-03-02:
Please read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

prbtt, 2023-03-07:
> Please read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

After reading it fully, here is what i've collected.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Settings   Desc. | state[1] with reversed bits
-- | --
Cool Mode, 16C | 0x10, 0x88, 0xC0, 0x08, 0x30
Cool Mode, 17C | 0x10, 0x48, 0xC0, 0x08, 0x30
Cool Mode, 18C | 0x10, 0xC8, 0xC0, 0x08, 0x30
Cool Mode, 19C | 0x10, 0x28, 0xC0, 0x08, 0x30
Cool Mode, 20C | 0x10, 0xA8, 0xC8, 0x08, 0x30

</body>

</html>
The order of state[1] controls the temperature and it looks to be inverted.
After inverting state[1], I can confirm that the byte increses by 1 like so:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Settings   Desc. | state[1] corrected
-- | --
Cool Mode, 16C | 0x10, 0x11, 0xC0, 0x08, 0x30
Cool Mode, 17C | 0x10, 0x12, 0xC0, 0x08, 0x30
Cool Mode, 18C | 0x10, 0x13, 0xC0, 0x08, 0x30
Cool Mode, 19C | 0x10, 0x14, 0xC0, 0x08, 0x30
Cool Mode, 20C | 0x10, 0x15, 0xC8, 0x08, 0x30

</body>

</html>

Furthermore, I've noticed that state[2] controls the fan speed.
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Settings   Desc. | state[2] changes
-- | --
Fan 1, 16C | 0x10,   0x11, 0xC0,   0x08, 0x30
Fan 2, 16C | 0x10,   0x11, 0x40,   0x08, 0x30
Fan 3, 16C | 0x10,   0x11, 0x80,   0x08, 0x30

</body>

</html>

Lastly, state[0] controls the power state (on/off)
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/paulo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

</span><span style='font-variant-ligatures: normal;font-variant-caps: normal;
orphans: 2;text-align:start;widows: 2;-webkit-text-stroke-width: 0px;
text-decoration-thickness: initial;text-decoration-style: initial;text-decoration-color: initial'>

<body link="#0563C1" vlink="#954F72">

28c fan   1 pon | 0x10, 0xB8, 0xC0, 0x08,   0x30
-- | --
28c   fan1 poff | 0x00, 0xB8, 0xC0, 0x08, 0x30

</body>

</html>

prbtt, 2023-03-28:
The next steps would be to turn this into code? Tried messing with it but I'm kinda lost. Is there any advice you could give me or just point in the right direction?

Solo157, 2024-04-18:
Add detailed support for CARRIER_AC protocol
Please, need CARRIER_AC for 32 bit
There is no structure and description for it