crankyoldgit/IRremoteESP8266 #1966: Decoding Benchmark DAC 3 remote protocol issue

iamelec, 2023-03-11:
Hi, I've been trying to capture my Benchmark DAC 3 DX's remote. However I'm receiving a message that the protocol is unknown. 
Unfortunately my knowledge regarding IR protocols is limited. Tasmota is currently running on my LoLin v3 NodeMcu ESP8266 chip.

I did manage to capture the remote control using LIRC on a Raspberry PI using the exact same IR receive sensor. Which gave me the following config file:
```
begin remote

  name  BenchmarkDAC3DXremote
  bits            8
  flags SPACE_ENC
  eps            30
  aeps          100

  header       2555  2627
  one          1167  2870
  zero         1167  1143
  ptrail       1166
  pre_data_bits   8
  pre_data       0x56
  gap          58220
  toggle_bit_mask 0x0
  frequency    38000

      begin codes
          VolumeUP                 0xE8
          VolumeDOWN               0xFC
          OFF                      0xF9
          ON                       0xF5
          Dim                      0x14
          Mute                     0x0C
          InputLEFT                0xF0
          InputRIGHT               0xED
          D1                       0xB1
          D2                       0x3C
          USB                      0x28
          D3                       0xA5
          ANALOG                   0xBD
          D4                       0xAC
      end codes

end remote
```

Benchmark kindly sent me the HEX codes for the remote. The codes are supported by Logitech Harmony and Pronto remote if that means anything.

Now when I "Scrutinize" the HEX codes using IrScrutinizer I get the same ish data as the LIRC config file. For each command it adds the pre data `0x56` + the begin code `E8` so for example `0x56e8`. I'm not sure what the purpose of IrScrutinizer is and I'm trying all options to make this work.

Now I can't find a way to send this code in Tasmota. I still don't know what protocol this remote uses exactly. Is there a way to find the IR protocol and add support for it in Tasmota?

IrScrutinizer gives me the following decode:
```
Decode:         SIM2: {D=106,F=23}, beg=0, end=35
IRP:            {38.0k,420,msb}<3,-3|3,-2772u>(2772u,-2772u,A:16,3,-47.49m)*{A=0x56e8}
Intro lenght:   0 Rep.lenght: 36 #reps.: 1 Ending lenght: 0 Frequency 38029
```

Benchmark's official HEX codes:
```
Command             Learn Freq Burst Len. M_L S_L M_C0 S_C0 M_C1 S_C1 M_C2 S_C2 M_C3 S_C3 M_C4 S_C4 M_C5 S_C5 M_C6 S_C6 M_C7 S_C7 M_D0 S_D0 M_D1 S_D1 M_D2 S_D2 M_D3 S_D3 M_D4 S_D4 M_D5 S_D5 M_T  S_T  M_P  S_P M_STP S_STP
POWER ON            0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 070E
POWER OFF           0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 002A 002E 006B 002E 070E
VOLUME UP           0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 002A 002E 002A 002E 070E
VOLUME DOWN         0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 002A 002E 070E
-20dB TOGGLE        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 002A 002E 070E
MUTE TOGGLE         0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 002A 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 070E
INPUT UP            0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 070E
INPUT DOWN          0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 002A 002E 070E
DAC INPUT A1/A2     0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 006B 002E 070E
DAC INPUT USB       0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 002A 002E 002A 002E 070E
DAC INPUT D1        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 006B 002E 070E
DAC INPUT D2        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 002A 002E 006B 002E 006B 002E 006B 002E 006B 002E 002A 002E 002A 002E 070E
DAC INPUT D3        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 002A 002E 006B 002E 002A 002E 006B 002E 070E
DAC INPUT D4/D5     0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 070E
DAC INPUT A2        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 002A 002E 002A 002E 002A 002E 006B 002E 070E
DAC INPUT D5        0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 006B 002E 002A 002E 002A 002E 002A 002E 006B 002E 002A 002E 002A 002E 070E
KEY18               0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 002A 002E 006B 002E 002A 002E 002A 002E 006B 002E 002A 002E 002A 002E 002A 002E 070E
KEY59               0000 006D 0000 0012 0063 0063 002E 002A 002E 006B 002E 002A 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 002A 002E 006B 002E 006B 002E 006B 002E 002A 002E 006B 002E 070E
```

For now I can send RAW remote commands that work on the DAC 3 DX. Though I can't seem to get the Volume control and ON command to work. In LIRC I had to repeat the command twice (so sending it two times in total). All other commands work by sending it once. I'm also using the same IR transmitter as I was using with my Raspberry PI using LIRC.
```
VolumeUP:   ?
VolumeDOWN: ?
OFF:        irsend 0,+2570-2625+1175-1140C-2865+1170dC-2860CdCgCeFdCgCgCeCeCgCdCdCg+1180
ON:         ?
Dim:        irsend 0,+2535-2665+1140-1175+1135-2900Cd+1160-2875C-1180G-2870Cf+1165-1150E-1200+1115dCdCfCmCjKlKlG
Mute:       irsend 0,+2540-2660+1140-1175C-2895C-1170CeCd+1165-2870CeCdCdCdCdCf+1145-2890IeCfCfC
InputLEFT:  irsend 0,+2565-2630+1170-1145C-2870CdD-2890Dc+1150-2885C-2865CdDfCiCiCiCdDcCdC-1140C
InputRIGHT: irsend 0,+2565-2630+1170-1145+1150-2890CdD-2895DcDf+1140gHcC-2870DgDg+1155-1160CiDgHc+1165iH
D1:         irsend 0,+2570-2625+1175-1140C-2865CdC-2860Cd+1170eCeGdCeCdCeGeGdGdG-1145GeC
D2:         irsend 0,+2570-2625+1175-1140C-2865CdC-2860Cd+1200-2840CeCd+1170dCdCeIeCfCeCdCdC
USB:        irsend 0,+2565-2630+1175-1140C-2865CdC-2860+1180dCfCfCdCdCdGfCdCfCdGdCdC
D3:         irsend 0,+2540-2660+1165-1145D-2890D-1170+1140-2895DfF-2870FiG-1175CiGjGhGjFdGhGjGhG
ANALOG:     irsend 0,+2570-2625+1170-1140+1175-2865CdEfCdCfCfCdCfEdCfCfCfCfCdCfC
D4:         irsend 0,+2570-2625+1175-1140C-2865CdC-2860+1170dCfCfCdCfCdGeGdCfCeCdG-1145G
```

I've tried to  add as much information as I have. If there's more information needed I'd be happy to add it to this post. 
Thanks

crankyoldgit, 2023-03-11:
Yep, that's a lot of info. But it would be best if you can supply it in our format(s) and using our tools.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

You can convert the pronto code versions (and see if we would decode them) by using the [`gc_decode` tool](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/gc_decode.cpp).

iamelec, 2023-03-11:
Is there maybe an easier way to send the codes in the right format for a non programmer like me?

Do I need to use Arduino to use the gc_decode tool? I've tried running it in Visual Studio Code in Windows but it says:
```
Untitled-1.cpp:10:10: fatal error: IRac.h: No such file or directory
   10 | #include "IRac.h"
```
Is there possibly a installation guide for this tool or an alternative tool that runs in Windows or in a browser?

NiKiZe, 2023-03-11:
This library is for the ESP8266 and/or ESP32,
You usually use the Arduino development environment, or  the platformio environment in vscode or other IDE.

So I do kind of wonder what your end goal here is. How would you use these codes?

iamelec, 2023-03-11:
If I understand your question correctly. What I want to do is to control my Benchmark DAC 3 using a Hue Tap Dial Dimmer via MQTT to Tasmota (installed on a ESP8266 chip). Tasmota uses the IRremoteESP8266 library. It will basically be an extended volume knob, which will be wireless and shouldn't have problems with IR "reach" (pointing the remote at the DAC). So turning the dial of the dimmer should send IR codes to the DAC for volume up or down.
The concept works so far for most buttons but I can't send volume commands yet to the DAC.

I will look further into the Arduino development environment or platformio in VScode. Thanks

NiKiZe, 2023-03-11:
Ok perfect, how do you usually upload tasmota to your ESP8266? Does that device also have a IR receiver?

iamelec, 2023-03-11:
There's a integrated USB input on the ESP8266 board. So I can directly connect it to my computer and flash the Tasmota firmware on there over the USB port. The ESP board does also have a IR receiver that I can put on there.

crankyoldgit, 2023-03-12:
That means you should be able to build & flash our IRrecvDumpV3 program.

The `gc_decode.cpp` program is a C++ program that should build under linux. No, it doesn't need arduino or the arduino IDE.

Your "Benchmark's official HEX codes" are PRONTO codes, I have no idea if Tasmota allows transmission of them. Our library does via the [`irsend.sendPronto()` function](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a0b349351e2ba19f87e6b01cde7e67c49).'
Your basically asking a Tasmota question, you'll have to ask them, not us (for how to send Prono codes).

iamelec, 2023-03-12:
Thanks for the suggestion. I've switched to ESPhome which uses this library as well and can send Pronto codes. Everything works now!

closed 2023-03-12 by iamelec