crankyoldgit/IRremoteESP8266 #1968: Wiki page ir-sending is missing 2 resistors

ArminJo, 2023-03-14:
Please update the schematic on the page https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending with two resitors to avoid [such ](https://forum.arduino.cc/t/esp8266-serial-not-working/1101509/4) problems.

crankyoldgit, 2023-03-14:
In the meantime, I've updated the wiki page to ensure that people don't treat those circuits as reference designs.

See: https://github.com/crankyoldgit/IRremoteESP8266/wiki#warning-warning-warning

ArminJo, 2023-03-14:
Oh sorry, the resistors required are 270 Ohm for the diode to VCC and 10 kOhm for the transistor base to D2.

ArminJo, 2023-03-14:
Here is a working schematic
[image]

NiKiZe, 2023-03-14:
I agree that there probably should be a resistor on the base, however with the 3.3v on ESP the resistor for diode is not required, and if it has a to high value it might not even work. 270Ohm is on the high end, but this also depends on the specification of the diode.

For the short pulses used with IR transmission the overdriving is early an issue, and might even be required to get proper signal.

ArminJo, 2023-03-14:
With 3.3V you get 2.6 V / 10 k = .26 mA base current. Times hfe of 100, which is very conservative, you get maximum 26 mA collector current. At 5 V  -1.2 for IR diode and transistor = 3.8 V and 270 Ohm you get 14 mA, which is not much, but good for programming errors, when D2 is always high.

NiKiZe, 2023-03-14:
You have 3.3V and a LED at 1.4-1.8V that should be driven at 100-120mA (burst). I would recommend 2 LEDs in series, and no resistor.

If it stays high due to errors it is good to test with more concervative values. But for working IR transmission you need these higher values, which is different from LEDs for visual indication which stays on for much longer.

jimmys01, 2023-03-15:
while you are at it, add a cap in GND and 5V

Jason2866, 2023-04-21:
A good description and circuit design is here https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/

ArminJo, 2023-04-22:
What is the exact reason for keeping the wrong (missing resistors) version, thus fooling all newbies?

closed 2023-04-22 by ArminJo

NiKiZe, 2023-04-22:
The above circuit uses a 2.5ohm resistor.
As already stated, this actually isn't necessarily wrong when dealing with IR diodes which only is on in bursts.

And I can say for sure that the 270ohm suggestion will not give you any good results.

This is an example, with warnings, you need to design your circuit with the values for you IR diode. Not random posts on the internet ;)