crankyoldgit/IRremoteESP8266 #1976: Adding Panasonic Ceiling Light Remote Data to IRremoteESP8266 Library

bennyyagiyama, 2023-04-10:
## Add Panasonic Ceiling Light remote data

### Description
I would like to request that the Panasonic Ceiling Light remote data be added to the IRremoteESP8266 library. When the "off -  Channel3" button  is pressed on the remote, the following data is obtained using IRrecvDumpV2:

```uint16_t rawData[83] = {3486, 1742,  432, 456,  406, 456,  406, 1336,  406, 1312,  430, 456,  406, 1334,  408, 456,  406, 456,  408, 456,  406, 1334,  408, 456,  408, 454,  408, 1334,  408, 456,  406, 1336,  406, 456,  408, 1334,  408, 456,  406, 456,  408, 1336,  406, 456,  408, 454,  406, 456,  406, 454,  408, 1332,  410, 1332,  408, 1334,  408, 1336,  406, 1334,  410, 1332,  410, 454,  406, 456,  406, 456,  406, 1332,  410, 1334,  408, 454,  406, 1336,  406, 1336,  406, 454,  410, 454,  408};  // UNKNOWN 1D41D404```

According to this [link](https://qiita.com/awawaInu/items/25c6e17fcc2e655d5d42), the data format is as follows:

|T [us] | Leader_high[T] | Leader_low[T] | signal[Byte] |
| --- | --- | --- | --- |
| 435 | 8 | 4 | 5 |

The remote has the following commands and corresponding channel values:

| Command | Channel 1 | Channel 2 | Channel 3 |
| --- | --- | --- | --- |
| on | 2c52092d24 | 2c5209353c | 2c52093d34 |
| off | 2c52092f26 | 2c5209373e | 2c52093f36 |
| full | 2c52092c25 | 2c5209343d | 2c52093c35 |
| night | 2c52092e27 | 2c5209363f | 2c52093e37 |
| high | 2c52092a23 | 2c5209323b | 2c52093a33 |
| low | 2c52092b22 | 2c5209333a | 2c52093b32 |
| warm | 2c523991a8 | 2c523995ac | 2c523999a0 |
| cool | 2c523990a9 | 2c523994ad | 2c523998a1 |

### Additional Information
I have confirmed that the data corresponds to the "off" command of channel3 for the Panasonic Ceiling Light. Please let me know if any further information is needed.
Thank you for your attention and I hope this data can be added to the library.

NiKiZe, 2023-04-10:
Can I ask that you collect more samples, can't promise when we get time to implement this. But data is good to have to verify.

bennyyagiyama, 2023-04-10:
I have collected data for all the buttons on my remote control. Please let me know if there is anything else I can help with. Thank you very much.

```// full ch1
uint16_t rawData[83] = {3494, 1736,  460, 404,  434, 426,  460, 1282,  434, 1308,  438, 426,  460, 1282,  460, 404,  436, 426,  436, 430,  432, 1306,  436, 430,  460, 402,  432, 1308,  460, 402,  434, 1306,  438, 426,  436, 1306,  436, 428,  436, 428,  458, 1282,  434, 428,  458, 404,  434, 426,  436, 426,  432, 430,  436, 428,  432, 1306,  434, 1308,  460, 400,  436, 1308,  434, 428,  432, 432,  436, 1302,  438, 426,  438, 1304,  436, 428,  458, 402,  434, 1308,  460, 402,  460, 404,  434};  // UNKNOWN CA21A230
// on ch1
uint16_t rawData[83] = {3492, 1736,  436, 426,  438, 426,  436, 1306,  436, 1304,  458, 404,  432, 1310,  460, 402,  464, 400,  434, 426,  436, 1306,  438, 428,  432, 428,  438, 1304,  436, 426,  434, 1308,  436, 426,  434, 1308,  434, 430,  434, 428,  438, 1306,  460, 404,  434, 430,  434, 426,  436, 426,  462, 1280,  436, 428,  436, 1304,  436, 1306,  434, 428,  436, 1304,  460, 402,  438, 428,  434, 430,  430, 430,  436, 1304,  464, 400,  434, 428,  460, 1282,  460, 404,  436, 426,  434};  // UNKNOWN F6B92168
// night ch1
uint16_t rawData[83] = {3514, 1714,  436, 426,  462, 404,  432, 1308,  436, 1306,  438, 426,  456, 1282,  436, 428,  438, 426,  460, 402,  434, 1308,  460, 404,  436, 430,  430, 1308,  436, 430,  434, 1306,  462, 402,  436, 1304,  438, 426,  438, 424,  438, 1304,  438, 426,  434, 428,  434, 428,  436, 428,  434, 426,  436, 1306,  436, 1304,  436, 1306,  436, 430,  434, 1308,  434, 426,  438, 426,  436, 1306,  438, 1304,  464, 1278,  438, 426,  460, 404,  460, 1282,  462, 400,  436, 426,  432};  // UNKNOWN A2768C92
// high ch1
uint16_t rawData[83] = {3490, 1736,  436, 428,  434, 426,  434, 1306,  460, 1282,  436, 428,  436, 1304,  436, 426,  438, 424,  436, 428,  436, 1304,  434, 430,  434, 430,  460, 1280,  462, 402,  434, 1308,  460, 402,  434, 1310,  434, 428,  434, 426,  436, 1304,  462, 402,  434, 428,  462, 400,  436, 428,  460, 400,  460, 1282,  432, 430,  434, 1306,  438, 426,  436, 1306,  436, 426,  432, 430,  434, 1306,  436, 1306,  434, 430,  434, 426,  436, 428,  434, 1306,  436, 426,  436, 426,  436};  // UNKNOWN 3D2149C8
// low ch1
uint16_t rawData[83] = {3518, 1710,  434, 428,  434, 430,  434, 1308,  436, 1304,  436, 426,  438, 1304,  434, 430,  436, 426,  438, 422,  438, 1304,  436, 428,  436, 426,  462, 1278,  438, 426,  438, 1304,  436, 426,  434, 1308,  438, 426,  434, 430,  432, 1308,  434, 428,  434, 428,  436, 426,  436, 426,  436, 1304,  434, 1310,  436, 426,  436, 1306,  438, 426,  436, 1306,  434, 430,  458, 402,  440, 426,  458, 1280,  436, 426,  438, 426,  434, 430,  462, 1278,  434, 430,  434, 428,  434};  // UNKNOWN 7F7FEDD0
// off ch1
uint16_t rawData[83] = {3492, 1734,  438, 426,  434, 430,  432, 1310,  436, 1306,  458, 402,  434, 1308,  438, 426,  436, 426,  436, 424,  462, 1280,  460, 404,  434, 428,  434, 1306,  436, 428,  432, 1308,  462, 402,  438, 1304,  436, 426,  460, 402,  436, 1306,  434, 428,  460, 402,  434, 426,  434, 428,  436, 1306,  436, 1306,  436, 1304,  438, 1306,  436, 428,  436, 1306,  434, 428,  436, 426,  460, 402,  434, 1306,  436, 1304,  438, 426,  436, 428,  438, 1304,  434, 428,  434, 428,  436};  // UNKNOWN 1B9C2A92
// full ch2
uint16_t rawData[83] = {3518, 1714,  436, 426,  460, 402,  436, 1306,  436, 1306,  436, 428,  436, 1304,  462, 400,  436, 426,  438, 426,  436, 1304,  438, 426,  436, 428,  460, 1280,  440, 426,  432, 1308,  438, 422,  436, 1304,  438, 428,  434, 428,  434, 1306,  438, 426,  434, 426,  436, 428,  434, 428,  434, 426,  436, 426,  438, 1304,  436, 454,  434, 1280,  464, 1278,  436, 426,  434, 428,  438, 1304,  436, 426,  458, 1286,  460, 1282,  458, 1282,  436, 1306,  462, 400,  434, 428,  436};  // UNKNOWN 5DF2C18E
// on ch2
uint16_t rawData[83] = {3488, 1736,  462, 402,  436, 426,  434, 1308,  436, 1306,  436, 428,  432, 1308,  436, 430,  432, 430,  434, 428,  460, 1282,  434, 430,  434, 426,  434, 1308,  438, 428,  434, 1304,  436, 430,  434, 1308,  438, 426,  434, 428,  432, 1308,  434, 428,  432, 434,  434, 428,  434, 426,  432, 1310,  436, 428,  436, 1308,  430, 432,  430, 1308,  438, 1304,  436, 428,  436, 426,  434, 430,  434, 428,  460, 1282,  434, 1308,  434, 1308,  436, 1306,  438, 426,  434, 430,  434};  // UNKNOWN D528FFCE
// night ch2
uint16_t rawData[83] = {3490, 1738,  436, 426,  436, 426,  436, 1306,  438, 1304,  434, 430,  436, 1302,  462, 404,  460, 402,  462, 400,  436, 1304,  438, 426,  436, 426,  434, 1308,  436, 428,  434, 1308,  436, 428,  464, 1276,  438, 428,  434, 428,  436, 1308,  434, 426,  460, 404,  436, 426,  436, 426,  438, 424,  460, 1282,  436, 1304,  440, 426,  434, 1304,  438, 1306,  436, 426,  436, 426,  438, 1304,  438, 1304,  462, 1282,  460, 1280,  436, 1304,  438, 1306,  438, 426,  460, 402,  438};  // UNKNOWN 50E4CAFC
// high ch2
uint16_t rawData[83] = {3520, 1710,  436, 428,  434, 426,  436, 1306,  462, 1282,  438, 426,  434, 1304,  436, 454,  414, 422,  438, 426,  436, 1304,  462, 402,  434, 428,  434, 1308,  432, 430,  434, 1306,  462, 402,  436, 1304,  436, 428,  436, 426,  436, 1304,  436, 426,  434, 428,  438, 426,  458, 404,  436, 426,  436, 1304,  438, 426,  436, 428,  436, 1306,  462, 1280,  436, 428,  436, 426,  436, 1306,  438, 1304,  436, 428,  434, 1308,  460, 1280,  460, 1280,  438, 426,  436, 426,  434};  // UNKNOWN DEAA2F2E
// low ch2
[... cut ...]
uint16_t rawData[83] = {3492, 1734,  438, 426,  434, 428,  434, 1308,  434, 1308,  434, 430,  436, 1304,  434, 430,  434, 428,  460, 400,  436, 1304,  438, 426,  436, 426,  434, 1306,  434, 430,  434, 1308,  434, 430,  460, 1282,  438, 426,  434, 428,  436, 1306,  436, 428,  460, 402,  458, 404,  436, 426,  456, 1284,  434, 1308,  438, 1304,  436, 1306,  438, 1304,  436, 1306,  434, 430,  436, 426,  438, 426,  436, 1306,  436, 1306,  436, 426,  434, 1308,  434, 1308,  434, 430,  434, 428,  434};  // UNKNOWN 1D41D404

crankyoldgit, 2023-04-18:
@bennyyagiyama Can you please download & try branch [Panasonic40](https://github.com/crankyoldgit/IRremoteESP8266/tree/Panasonic40) (https://github.com/crankyoldgit/IRremoteESP8266/tree/Panasonic40) & PR #1977 
It should now decode your remote as a 40 bit Panasonic message. Panasonic IR messages are typically 48 bits, so you will need to send the messages with a bit more specifics than usual.
e.g.
```c++
// send a 'night ch3' message.
irsend.sendPanasonic64(0x344A907CEC, kPanasonic40Bits);  // kPanasonic40Bits is 40.
```
Can you let me know how you go with that branch/code?

bennyyagiyama, 2023-05-02:
I apologize for the delayed response. Your prompt and thorough response has been an immense help to me. I am deeply grateful for your expertise and willingness to assist.

Using the Panasonic40 branch, I obtained the following result for the Ch1 FULL button:

Protocol : PANASONIC
Code : 0x344A9034A4 (40 Bits)
uint16_t rawData[83] = {3482, 1766, 408, 454, 406, 456, 406, 1336, 406, 1334, 408, 456, 406, 1336, 404, 456, 406, 456, 406, 458, 406, 1336, 406, 454, 406, 456, 408, 1334, 406, 458, 404, 1336, 406, 458, 406, 1336, 404, 456, 410, 454, 406, 1336, 406, 456, 404, 458, 404, 456, 406, 458, 406, 454, 406, 458, 404, 1336, 406, 1334, 406, 458, 406, 1312, 430, 456, 406, 454, 408, 1334, 406, 456, 406, 1336, 408, 454, 406, 456, 406, 1334, 408, 456, 406, 460, 402}; // PANASONIC 344A9034A4
uint32_t address = 0x34;
uint32_t command = 0x4A9034A4;
uint64_t data = 0x344A9034A4;

Only the code is listed below.
| Command | Channel 1 | Channel 2 | Channel 3 |
| --- | --- | --- | --- |
| on | 0x344A90B424 | 0x344A90AC3C | 0x344A90BC2C |
| full | 0x344A9034A4 | 0x344A902CBC | 0x344A903CAC |
| night | 0x344A9074E4 | 0x344A906CFC | 0x344A907CEC |
| high | 0x344A9054C4 | 0x344A904CDC | 0x344A905CCC |
| low | 0x344A90D444 | 0x344A90CC5C | 0x344A90DC4C |
| off | 0x344A90F464 | 0x344A90EC7C | 0x344A90FC6C |

(Warm button and cold button were not available on my IR remote.)

Also, I confirmed that the received code could be sent using the following code:
```
irsend.sendPanasonic64(code, kPanasonic40Bits);
```
Thank you very much. I look forward to it being incorporated into the main branch.

crankyoldgit, 2023-05-03:
Thanks for confirming it. It's now merged.

PR #1977 (2023-05-03): Add support for a 40bit variant of the standard Panasonic protocol
This seems to be a shorter version of the normal 48 bit protocol. Different manufacturer code, and slightly different checksum calc. Modified the exist code to support it rather than add a new protocol.

Fixes #1976
commit 575a81b2 (2023-05-03): Add support for a 40bit varient of the standard Panasonic protocol (#1977)

This seems to be a shorter version of the normal 48 bit protocol. Different manufacturer code, and slightly different checksum calc.
Modified the exist code to support it rather than add a new protocol.

Fixes #1976
closed 2023-05-03 by crankyoldgit