crankyoldgit/IRremoteESP8266 #1983: Calling IRrecv.enableIRIn() after calling IRrecv.enableIRIn() fails to attach timer handler

homy-newfs8, 2023-05-02:
### Version/revision of the library used
2.8.4

### Describe the bug
Calling IRrecv.enableIRIn() after calling IRrecv.enableIRIn() fails to attach timer handler, and receive timeout does not work.
Tested on ESP32 Arduino 2.0.8 based on ESP-IDF 4.4.4.

#### To Reproduce
Code below produces framework error message `timer_group: timer_isr_callback_add(236): register interrupt service failed`, 
and `IRrecv.decode()` does not finish until buffer overflow. I used compiler flags:  `-DDEBUG -D_IR_ENABLE_DEFAULT_=false -DDECODE_HASH=true`.

#### Example code used
```c++
#include <Arduino.h>
#include <IRrecv.h>

IRrecv recv{32, 256, 15};

void
setup() {
  Serial.begin(115200);
  delay(3000);

  recv.enableIRIn();
  recv.disableIRIn();
  recv.enableIRIn();
}

bool done;

void
loop() {
  decode_results results;
  if (!done && recv.decode(&results)) {
    Serial.printf("rawlen = %u, overflow = %u\n", results.rawlen, results.overflow);
    Serial.println("Decode done");
    done = true;
  }
  delay(1);
}
```
Output:
```
E (5039) timer_group: timer_isr_callback_add(236): register interrupt service failed
rawlen = 256, overflow = 1
Decode done
```

#### Expected behaviour
`IRrecv.enableIRIn` does not produce system error and `IRrecv.decode()` returns with proper length of data.

### What brand/model IR demodulator are you using?
M5Atom Lite with IR UNIT

### Other useful information
Call `timerDetachInterrupt()` in `disableIRIn()` seems fix this problem.

crankyoldgit, 2023-05-03:
@homy-newfs8 Thanks for reporting this. Yes, I do seem to miss clearing that Interrupt. Well spotted.

Can you please confirm that branch [ESP32_timerDetachInterrupt_bug](https://github.com/crankyoldgit/IRremoteESP8266/tree/ESP32_timerDetachInterrupt_bug) / PR #1984 addresses the problem for you?

homy-newfs8, 2023-05-03:
Tested the branch version. Above example code produces these messages:
```
E (3042) timer_group: timer_isr_callback_remove(246): HW TIMER NEVER INIT ERROR
E (3043) timer_group: timer_isr_callback_add(236): register interrupt service failed
```

It seems `timerDetachInterrupt()` must be called before `timerEnd()`.

crankyoldgit, 2023-05-04:
Thanks for that info. I've moved it before `timerEnd()` now. Can you download/pull etc that branch and try it again?

homy-newfs8, 2023-05-04:
Confirmd the problem fixed on the branch. Example code outputs no error messages and my personal project starts working well 😄 
Thank you 🙏

PR #1984 (2023-05-04): Fix a bug where we never detached the timer interrupt on ESP32s.
Kudos to @homy-newfs8 for finding, isolating, and providing a suggested fix for the bug.

Fixes #1983
commit d7ce2ca3 (2023-05-04): Fix a bug where we never detached the timer interrupt on ESP32s. (#1984)

Kudos to @homy-newfs8 for finding, isolating, and providing a fix for the bug.
Note: `timerEnd()` needs to happen after `timerDetachInterrupt()`

Fixes #1983
closed 2023-05-04 by crankyoldgit