crankyoldgit/IRremoteESP8266 #1998: TCL 112 Protocol HalfDegree Wrong Position

kickit2, 2023-05-29:
Version 2.8.5

### Describe the bug
IRL AC unit did not respond to half degree C commands for TCL 112 Protocol.  Compared to IR code from remote and found HalfDegree bit in wrong position.  Corrected on local copy of code and verified.  

Byte 12 of TCL protocol should be as follows
`	uint8_t                 :2;
    uint8_t HalfDegree		:1;
	uint8_t SwingH          :1;
    uint8_t                 :3;
    uint8_t isTcl           :1;`

crankyoldgit, 2023-05-31:
@kickit2 What is the make & model numbers for the a/c unit and the remote please?

kickit2, 2023-06-01:
Make and Model is a "Denali Aire", "1DAC10K" which seems to be just another reseller of 3rd party built equipment.  The remote is branded "Denali Aire" but without a model number.  Opening the remote revels a PCB with the following details 

> TCL Aircon GYKQ-58
> 3.60.011-2181001010
> V3.0
>2018.11.07

I suppose its possible this remote / unit uses a slightly modified protocol, but literally everything else matches and I've run in to no other issues (on things it supports anyways - this unit doesn't have heat for example)