crankyoldgit/IRremoteESP8266 #1999: Panasonic CS-E12QKEW SwingH not working

duczz, 2023-06-01:
Hey Guys, 

im using following Repo which integrate your IRremoteESP8266 repo. https://github.com/hristo-atanasov/Tasmota-IRHVAC

I found out, that the swing command is only working with hvac_model: 3, but with this hvac_model i cant start the climate (still be offline). it looks like the hvac_model: 3 is only for set settings, any chance to fix these?

My normal call (swingh not working, all other works):
13:24:09.685 MQT: stat/Klimaanlage_Buero_5A7A65/RESULT = {"IRHVAC":{"Vendor":"PANASONIC_AC","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Min","SwingV":"Middle","SwingH":"Right","Quiet":"Off","Turbo":"On","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}} 

Swing and all other works, but power not working (climate still be offline)
13:24:09.685 MQT: stat/Klimaanlage_Buero_5A7A65/RESULT = {"IRHVAC":{"Vendor":"PANASONIC_AC","Model":3,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Min","SwingV":"Middle","SwingH":"Right","Quiet":"Off","Turbo":"On","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}

NiKiZe, 2023-06-01:
Hi, it seems you ignored everything in the Issue template?
IRremoteESP8266 emulates the remote,  You should use the correct model for the remote you have.
In your case if SwingH is missing it is due to the fact that it is missing from the remote we emulate.

We will want to have the data collected by DumpV3 utility (as described in the issue template and FAQ) to verify what data you have from your original remote.

duczz, 2023-06-01:
Yes sorry for that.
i have no demodulator etc.

i have checked all hvac_models and swingh only working in model 3, all other settings works with model 2, all settings works with model 3 too, only the power does not work in model 3

i have only the tasmota results, if these is not enough i will close the ticket.

NiKiZe, 2023-06-01:
We do not want to introduce bugs due to guesses.
If model 3 works, then use that, do not switch between models for the same device.

Exactly which model of the remote do you have? Could you also post a picture?

NiKiZe, 2023-06-01:
So this makes me think you probably has something that is neither model 2 or model 3

duczz, 2023-06-01:
i thinked that too, but swingh works not with following model: -1,0,1,2,4, only with 3 (but than i cant power on the climate)
if you want to test some constellations, let me know to assist you to find a error.

i can try each setting with eatch model if this helps?

NiKiZe, 2023-06-01:
We need that the data is collected from the actual remote.

crankyoldgit, 2023-06-01:
See also: https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRPanasonicAc.html#ad60bf8a88d041f8e8ab3d728831ee8f3

We are going to need `state[]` data captured from the _real_ remote to see how this _different_ (KEW) model works.
i.e. Your devices isn't a known-supported device/model etc.
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d0a9196fd716879897b7bd739fef5dc6ee92551/src/ir_Panasonic.h#L7-L29

As @NiKiZe has correctly pointed out, we don't want to add "guesses" to the emulation.

PenthouseLA, 2023-06-01:
#Power on:
```16:59:16.220 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400393280310D000EE000008100009E","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":25,"FanSpeed":"Min","SwingV":"Highest","SwingH":"Auto","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
16:59:21.619 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":427,"Hash":"0xDD44A6E5","Repeat":0}}

swingh:
17:06:54.005 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620A000EE00000810000BE","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Left","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:06:55.582 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620B000EE00000810000BF","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Right","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:06:57.000 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620C000EE00000810000C0","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Right Max","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:06:58.605 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620D000EE00000810000C1","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Auto","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:07:00.001 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E004003924806206000EE00000810000BA","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Middle","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:07:01.314 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E004003924806209000EE00000810000BD","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Left Max","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:07:02.831 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620A000EE00000810000BE","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Left","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:07:04.598 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400392480620B000EE00000810000BF","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Right","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}```

swingv:
17:06:02.843 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E004003924806109000EE00000810000BC","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"Highest","SwingH":"Left Max","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:06:04.819 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E004003924806209000EE00000810000BD","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Left Max","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
17:06:06.838 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E004003924806309000EE00000810000BE","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"Middle","SwingH":"Left Max","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":" [... cut ...]
17:07:37.788 MQT: tele/Klimaanlage_Buero_5A7A65/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":216,"Hash":"0x0530044D","Repeat":0}}```

when i try to trigger swinghit now via following string (original Vendor string from top) it work not with model 0, ot works only with model 3

http://192.168.178.50/cm?cmnd=IRhvac{"Vendor":"PANASONIC_AC","Model":0,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"High","SwingV":"High","SwingH":"Left","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}

do you need more informations?

duczz, 2023-06-08:
works now brilliant with model 3, i need to push it only sometimes multiple

closed 2023-06-08 by duczz