crankyoldgit/IRremoteESP8266 #2000: Skip compiling module Y if both DECODE_Y and SEND_Y are not defined

dilyanpalauzov, 2023-06-03:
I want to compile the most current Tasmota (develop branch) with IR support only for Toshiba-Encode.   That is I set
```c
#define _IR_ENABLE_DEFAULT_ false
#define USE_IR_REMOTE
#define DECODE_HASH  true
#define SEND_GLOBALCACHE true
#define SEND_TOSHIBA_AC true
#undef USE_IR_SEND_NEC                        // Support IRsend NEC protocol
#undef USE_IR_SEND_RC5                        // Support IRsend Philips RC5 protocol
#undef USE_IR_SEND_RC6                        // Support IRsend Philips RC6 protocol
#undef USE_MODBUS_BRIDGE
```
I expect to have very small binary. Unfortunately as far as I see there are a lot of unused symbols compiled, as in
```
$ nm -C .pio/build/tasmota-vn/lib892/IRremoteESP8266/ir_Airwell.cpp.o |grep " T "
00000004 T IRAirwellAc::convertFan(stdAc::fanspeed_t)
00000008 T IRAirwellAc::stateReset()
00000004 T IRAirwellAc::convertMode(stdAc::opmode_t)
00000004 T IRAirwellAc::toCommonMode(unsigned char)
00000000 T IRAirwellAc::setPowerToggle(bool)
00000004 T IRAirwellAc::toCommonFanSpeed(unsigned char)
00000004 T IRAirwellAc::begin()
00000004 T IRAirwellAc::setFan(unsigned char)
00000000 T IRAirwellAc::setRaw(unsigned long long)
0000000c T IRAirwellAc::setMode(unsigned char)
00000004 T IRAirwellAc::setTemp(unsigned char)
0000000c T IRAirwellAc::IRAirwellAc(unsigned short, bool, bool)
0000000c T IRAirwellAc::IRAirwellAc(unsigned short, bool, bool)
00000000 T IRAirwellAc::getPowerToggle() const
00000000 T IRAirwellAc::getFan() const
00000000 T IRAirwellAc::getRaw() const
00000000 T IRAirwellAc::getMode() const
00000000 T IRAirwellAc::getTemp() const
0000001c T IRAirwellAc::toCommon(stdAc::state_t const*) const
00000048 T IRAirwellAc::toString() const
```
I assume these symbols are included in the final binary, as setting they are already in the .pio/build/tasmota-vn/lib892/libIRremoteESP8266.a file.

Please, alter the code, so that when both SEND_Y and DECODE_Y are `false`, for any protocol `Y`, that all the symbols for the Y protocol are not compiled.

NiKiZe, 2023-06-03:
Even if every file is built, the final binary should only have code that is actually used. So checking individual .o files will not give you the correct result.

And if you still have unused functions pulled in, the most likely reason is that you are building without optimizations, or with debug enabled.

Recheck your result for the final binary.

crankyoldgit, 2023-06-04:
> I want to compile the most current Tasmota (develop branch) with IR support only for Toshiba-Encode. That is I set
>
> ```c
> #define _IR_ENABLE_DEFAULT_ false
> #define USE_IR_REMOTE
> #define DECODE_HASH [...]

Where in your code are you putting this?
You had better be putting those first five (5) lines in the `IRremoteESP8266.h` file, otherwise they won't have the effect you want.
Realistically, you should be passing them to the compiler as `-D` (define) flags in the _build_ process.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d0a9196fd716879897b7bd739fef5dc6ee92551/examples/IRrecvDumpV2/platformio.ini#L64-L67
or
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d0a9196fd716879897b7bd739fef5dc6ee92551/src/IRremoteESP8266.h#L85-L102

Also, as @NiKiZe points out, at `link` time, items not needed are not included, and when the resultant binary is `strip`ed, it won't have any (debug) symbols in it.

Have you tested that your resultant binary is smaller or not when when you add or subtract protocols?

P.S. You seem to be having a "Tasmota" problem, not necessarily a problem with our Library. You should probably report your issues there initially. We realistically can't help you with down-stream programs & libraries.

dilyanpalauzov, 2023-06-04:
I do not know how to check if the final binary contains these functions (symbols).  The `file` command returns “DOS executable (COM), start instruction 0xe9010240 5cf41040”, so `nm` and `readelf` do not recognize the file and do not report its symbols.

How is the linker supposed to be invoked in order to remove all unused symbols, when these are not `static`?

I put the above defines in the tasmota/user_config_override.h file.  I guess it is included before `IRremoteESP8266.h`.

> Have you tested that your resultant binary is smaller or not when you add or subtract protocols?

I have tested, but I do not see any difference, so probably I am not doing the right thing.  I do not understand right now all the details.

I think that disabling the compilation of complete files, based on defined macros, rather than hoping that the linker will remove the unused methods/classes (object files), is the better approach.

NiKiZe, 2023-06-04:
The compile and linker does the cleanup automatically of unused symbols on the output, nothing you have to care about. Just that your example checks for symbols in parts that is the source of linked output.

You could, just to verify try with everything enabled, and then modify the actual definition of _IR_ENABLE_DEFAULT_ and compile to verify if the final binary size changes. But we are not able to help you debug non library code.

The best approach is still to use -D options at compile time.

crankyoldgit, 2023-06-04:
> How is the linker supposed to be invoked in order to remove all unused symbols, when these are not `static`?

The _linker_ is used to turn object code (.o/.a) files into an executable. It's done automatically as part of the build process.
The final files is _static_. There is no dynamic library management or loading on these microcontrollers.

> I put the above defines in the tasmota/user_config_override.h file. I guess it is included before `IRremoteESP8266.h`.

No, it isn't. The library is built separately. Putting in those in `user_config_override.h` will have no effect on how the library is built. None of our library code includes the file `user_config_override.h`, so it is simply not referrenced by our library.
 

> I have tested, but I do not see any difference, so probably I am not doing the right thing. I do not understand right now all the details.

Then commonsense should indicate to you that your change are not disabling/removing any of the code.

> I think that disabling the compilation of complete files, based on defined macros, rather than hoping that the linker will remove the unused methods/classes (object files), is the better approach.

Sadly & sorry, that's wrong thinking. I understand what you're thinking, but that's not how things really & optimally work.
This is exactly what the _linker_ does. It's one of it's main jobs.
e.g. If you use one function from the _math_ library, only the code required for that one _function_ is included in your code. Not every function of the _math_ library. That task is what the linker does. It only links in the required code.

dilyanpalauzov, 2023-06-09:
I tries once again now I can confirm that the size of the binary changes, when I enable or disable protocols.  By accident I tried to include/exclude exactly WHIRLPOOL_AC from the build and it was always reported as supported protocol.  It turned out, that this was not my fault.

In Tasmota, lib/lib_basic/IRremoteESP8266/library.json contains `"build": ["flags": [ "-I$PROJECT_DIR/include", "-includetasmota_options.h" ]`, the latter contains `#include "../tasmota/my_user_config.h"`, and that one contains ` #include "user_config_override.h"`.  So the definitions form user_config_override.h are passed to IRremoteESP8266 when building.

For performance reasons it would be good, if unused code is not complied at first place, rather than relaying on the linker to remove the compiled, unused code.

closed 2023-06-09 by dilyanpalauzov

NiKiZe, 2023-06-09:
Compiling with multithreading properly enabled on modern hardware takes no time at all. Adding complexity to shave of a few ms and causing other issues isn't worth it.