crankyoldgit/IRremoteESP8266 #2001: Compile errors with esp8266/Arduino 3.1.2

igor-d-n, 2023-06-06:
### Version/revision of the library used
IRremote 2.8.5
[Arduino core for ESP8266](https://github.com/esp8266/Arduino) ver 3.1.2 (latest)

### Describe the bug
Code can't compile. Always errors:
ccCS7VxT.s: Assembler messages
ccCS7VxT.s:11346: Error  bad register name  	a9
ccCS7VxT.s:11346: junk at end of line, first unrecognized character valued 0x9
ccCS7VxT.s:11347: Error  bad register name  	sp
ccCS7VxT.s:11347: junk at end of line, first unrecognized character valued 0x9
ccCS7VxT.s:11349: Error  bad register name  	a13
.....
#### To Reproduce
Make project and compile

NiKiZe, 2023-06-06:
Which program are you compiling? And what makes this point to this library?

What do I need to recreate this issue? (Assume that we know nothing of your setup, and that only platformio is installed on a clean computer)

crankyoldgit, 2023-06-06:
The closest thing I can find to this is: https://github.com/esp8266/Arduino/issues/8672
It looks like a tool chain issue. Not so much our library, as such.

crankyoldgit, 2023-06-06:
also: https://stackoverflow.com/questions/73116518/understanding-esp8266-assembly-errors-in-arduino-ide-on-upgrade-from-boards-pack