crankyoldgit/IRremoteESP8266 #2004: Even the examples don't work

ParagDas2315, 2023-06-16:
### Version/revision of the library used
The latest version

### Describe the bug
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp: In member function 'void IRac::argo(IRArgoAC*, bool, stdAc::opmode_t, float, float, stdAc::fanspeed_t, stdAc::swingv_t, bool, bool, int16_t)':
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:492:36: error: 'round' is not a member of 'std'
   ac->setTemp(static_cast<uint8_t>(std::round(degrees)));
                                    ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:492:36: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.h:7,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:7:
C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:494:44: error: 'round' is not a member of 'std'
     ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
                                            ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:494:44: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.h:7,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:7:
C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp: In member function 'void IRac::argoWrem3_ACCommand(IRArgoAC_WREM3*, bool, stdAc::opmode_t, float, float, stdAc::fanspeed_t, stdAc::swingv_t, bool, bool, bool, bool, bool, bool)':
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:540:44: error: 'round' is not a member of 'std'
     ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
                                            ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:540:44: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.h:7,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:7:
C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp: In member function 'void IRac::argoWrem3_iFeelReport(IRArgoAC_WREM3*, float)':
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:566:42: error: 'round' is not a member of 'std'
   ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
                                          ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:566:42: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.h:7,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:7:
C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp: In member function 'void IRac::coolix(IRCoolixAC*, bool, stdAc::opmode_t, float, float, stdAc::fanspeed_t, stdAc::swingv_t, stdAc::swingh_t, bool, bool, bool, bool, int16_t)':
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:741:44: error: 'round' is not a member of 'std'
     ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
                                            ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:741:44: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.h:7,
                 from c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:7:
C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp: In member function 'void IRac::ecoclim(IREcoclimAc*, bool, stdAc::opmode_t, float, float, stdAc::fanspeed_t, int16_t, int16_t)':
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:1131:44: error: 'round' is not a member of 'std'
     ac->setSensorTemp(static_cast<uint8_t>(std::round(sensorTemp)));
                                            ^
c:\Users\parag\OneDrive\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp:1131:44: note: suggested alternative:
In file included from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\parag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
[... 35 lines cut ...]
I just wanted to get the IR remote reciever values, but the library causes error whenever imported

#### Example code used
The IRrecvDemo code included in the library

#### Expected behaviour
To read the values from IR remote 

#### Output of raw data from [IRrecvDumpV2.ino]

### What brand/model IR demodulator are you using?
ESP32

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
=

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
No

### Other useful information

NiKiZe, 2023-06-16:
Looks like a duplicate of #1989
Can you try to pull down master and see if that works for you?

crankyoldgit, 2023-06-17:
> Looks like a duplicate of #1989

Agreed.

crankyoldgit, 2023-08-29:
Technically it hasn't been confirmed to fix his issue, but given the lack of response, lets assume it does.

Closing.

closed 2023-08-29 by crankyoldgit